RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #1: Hoverboard shut down after 1 sec

marcodalco74, 2023-02-18:
Hello,
Thank you very much for your project! It's really cool project for the 2nd gen. hoverboard but I have an issue: after installing the Hoverboard 2.0 master/slave firmware and connect the Arduino Nano everything works fine for about one second, then the overboard shut down. 
Is there a control of battery voltage that shut down the HB? My HB comes from factory with a 25v battery (I don't know why it hasn't a typical.36v battery).
The serial monitor on Arduino show iVolt around 32... The battery voltage during the test was 26v
millis: 101	iSpeedL: 0.00	 iSpeedR: 0.00		iAmpL: 0.00	 iAmpR: 0.20		iVolt: 33.37
millis: 99  	iSpeedL: 0.49	 iSpeedR: 0.00		iAmpL: 0.20	 iAmpR: 0.20		iVolt: 32.86
millis: 100	iSpeedL: 0.49	 iSpeedR: 0.49		iAmpL: 0.00	 iAmpR: 0.40		iVolt: 32.38
I don't know if the issue has been caused from the voltage of the battery but for sure I need help. 
Thanks
Marco

marcodalco74, 2023-02-19:
Update: adding 3 batteries everything's working fine. 
I would keep the original battery in order to use the own charger. Any suggestion to tricky the reading voltage battery? 
May be reducing the R2 value? 
[image]

RoboDurden, 2023-02-21:
Oh great, my first issue :-) Thank you for using my firmware :-) Sorry for late response, did not get an email notification. Thanks to the YouTube comment.

Change the voltage here: 

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Inc/config.h#L76

I should change the code to number of lithium cells like the other firmware's. Bit someone might want to use a lead acid battery..

marcodalco74, 2023-02-21:
The fact is that it isn't a real issue, your firmware works great; I just sent my SOS ;-)
Thank you for the indication of the voltage parameter. To be honest, I have never been able to compile the code in Keil in order to get .bin firmware. I'd intended to follow a tricky hardware solution in order to adapt any kind of battery voltage just rotating a trimmer. For sure I'll try to understand how to compile your great firmware but the risk to make something wrong while I compile the firmware and creating a real issue on a fully functional firmware doesn't give me a good push to follow this way.
About the quick rotation (YouTube), thank you very much for the exhaustive answer. Opening the logbandsaw link I've been a bit scared...I need time to go deep on that topic.
The last thing THANK YOU.

RoboDurden, 2023-02-21:
I will upload 24V 7s liion firmware binaries later the day :-)

RoboDurden, 2023-02-21:
Have uploaded the 7s = 25.2 Volt binaries :-)

https://pionierland.de/hoverhack/gen2/ReadyToFlash/

Here my new BAT_CELLS define in config.h:

```
#define BAT_CELLS         10        // battery number of cells. Normal Hoverboard battery: 10s
#define CELL_LOW_LVL1     3.5       // Gently beeps, show green battery symbol above this Level.
#define CELL_LOW_LVL2     3.3       // Battery almost empty, show orange battery symbol above this Level. Charge now! 
#define CELL_LOW_DEAD     3.1       // Undervoltage lockout, show red battery symbol above this Level.

#define BAT_LOW_LVL1     BAT_CELLS * CELL_LOW_LVL1
#define BAT_LOW_LVL2     BAT_CELLS * CELL_LOW_LVL2
#define BAT_LOW_DEAD     BAT_CELLS * CELL_LOW_DEAD

```

RoboDurden, 2023-02-21:
The iOdomL and iOdomR to turn the motors into servo motors will have to wait until i have implemented the FOC code of the Candas1 fork of the gen 2.0 firmware. My code relies on the hall parameters of the FOC code that Emanuel Feru has introduced.

marcodalco74, 2023-02-21:
**The new code works perfectly! Thank you.**
About the "combo command" (speed =0 and steer = +-1000 for a very short time) doesn't work.
I noted that the wheels are not blocked at speed 0 (a good choice for power saving); there is an option to lock the wheels at speed 0 ?
About the iOdomL and iOdomR, I have no intention to use them in RC, for my purpose the serial connection using steer and speed commands is perfect.

RoboDurden, 2023-02-22:
I do not see special code that frees the motors on speed 0. Which makes me fear that this firmware does not have regenerative braking. Do you see negative amps on your Arduino log when braking or going down hill. But I think this must be so there may be some code in bldc.c to disable pwm when speed is set to zero.
Sometimes in the future I might add a iCmd to the send struct and change iSpeed and iSteer to iParam1 and iParam2. The you could program the battery voltage on startup or disable the "speed 0 freewheeling", switch into servo mode, etc.
But first I want the basis of an object oriented hoverboard OS.....

marcodalco74, 2023-02-23:
Thank you again Robo.

closed 2023-02-23 by marcodalco74

RoboDurden, 2023-03-15:
@marcodalco74 i left a silly bug in the slave binaries that limits the max speed to +-300 instead of +-1000. 
Have uploaded new binaries.
Sorry for this.