RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #2: [Question]

grafvonvolta, 2023-02-22:
Hi Robo, 

thank you for the repo. 

I have a question regarding another layout. 
[image]

It has a GD32F130C8T6.

Do you know by any chance if this layout is compatible to 2.1 or 2.2? If not, I would have to figure out/confirm the pin mapping in defines.h, correct? I guess the most crucial is the mosfet output.

Thank you in advance

Max

RoboDurden, 2023-02-22:
Would be awesome if you test the pin connections of this board. Yes, only a new defines_2-5 would/should be needed.
And yes, the 6 gate driver pins are crucial.
It looks to have three gate driver chips
 That would make testing the pins easier as you should have a direct connect from the gd32 to the two input pins of those half bridge gate drivers.
Would also be great if you could send detailed photos to roland@robo4future.de where every chip description is readable.

Then I would add an official 2.y layout to my repo and we could see how many more people would love a defines.h for that layout.

I think I also have such boards at my other location (solar warehouse).

Thanks for using my repo.

grafvonvolta, 2023-02-22:
OK great. I will send you some pictures soon. Let's see how far I get with the mapping without destroying the mosfets ;). I have two of those boards and one 2.3. So I'm also curious to get that one to work. .

RoboDurden, 2023-02-23:
When you attach a needle to one pole of a multimeter and select the beeping diode test mode, it is quite easy to find the tiny gd32 pins. And when you look at the PDF of your gate driver chips you can quickly target the 6 MOSFET driver pins with the multimeter pole that has no needle.
There should be an op amp chip on the layout which outputs the analog inputs of the current.

The serial pins only have a resistor between header and gd32. But the diode beeper will only work behind that SMD resistor :-/

Digital output is harder to find because a little transistor is driven.

It might be possible to program a binary that sets all possible Io pins to output low and send loops through all pins only setting one pin to pwm.
As the MOSFET half bridges need two MOSFET to open for a short cut, this should be safe.
The st-link dongle has a serial one wire debug interface.. don't know but maybe this could be used to log the output of that  analysing binary.. 
Otherwise the tx pin of one of the serial ports would have to be identified manually.

But I think it will be faster if people like you would have the happiness to use a needle for 60 minutes :-)

Be aware that the orientation of the writing in the gd32 is misleading !! Find the little dot to see pin 1. My many pictures of gen 2.0 should help you.

grafvonvolta, 2023-03-16:
Hi,

here is a picture of my board. So far I could confirm that the mosfet and HAL pins are as in version 2.0 an 2.1. I did not check any leds. I did not manage to get a rotation yet. I plan to do some debugging. 

I'll keep you updated.

[image]

RoboDurden, 2023-03-17:
You might want to follow the progresse in this parallel issue: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/3

So you can debug (step through code) with Keil and st-link dongle ?

When you get the serial communication running you can install my Arduino code and get the log data.
Then you will see if the two ADC inputs are correct.

Is the on/off button working ?