RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #3: [Submission] Another board layout

Batman313v, 2023-03-09:
## My Board:
I have another board with another different layout 😅 

### Layout of Board A (Master):
[image]

### Layout of Board B (Slave):
[image]

## Plan:
I saw [this](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/2#issue-1595662196) submission and will be trying to also document my board using the same process. I just ordered some needle probes for my multimeter as well as a digital microscope for better close up imaging. Those will be here by the end of the week.

### Steps:
**The Steps I plan to follow:**
- Probe the pins of the GD32F130C8T6 to find the routing to the MOSFET drivers.
- Images each IC for documentation
- I also hope to find the pins that controlled the LEDs as well as see if the buzzer pin is different on this board.

I decided to create a new issue because my board is different from the other submission I and figured it would be easier to keep things organized this way. I will update this thread with more info as I go. I am pretty new to imbedded firmware but look forward to learning as much as I can to help get this layout documented as well.

Batman313v, 2023-03-10:
# Update:
I was able to get the routing for the MOSFETs and Drivers and it is as follows:
[image]

#### Key:
- Red for Visible Traces
- Blue for Traces that move between layers

The MOSFETS are `Samwin HF FK2A4 SW076R68E7T`
The MOSFET Drivers are `IDR ID2005 M19AKERG`

This is all I was able to do tonight on this board. Is there anything else I should probe to get the defines_2-4.h file working on this board or should I start on Board A tomorrow? I do eventually want to get the LED pins documented as well but that's more of a secondary thing since I can just control lights from the main Arduino sending the commands until then.

I also have another hoverboard (the same kind) that I plan to document where everything is plugged into (including the LEDs since I forgot to do that the first time) during the disassembly of that one.

RoboDurden, 2023-03-10:
Fine. I added some images to https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/Schematics_2.4

and added a https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Inc/defines_2-4.h

the mosfet pins are the same as layout 2.0 :-)
If you are lucky most of the other pins as well.

But you should check.

At least the hall inputs and the adc inputs. See https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/Schematics_2.4/hoverboard_gen2-4_pins.jpg

Of course the self-hold digital ouput and the on/off button are also important.
The buzeer and the red and green led help to see that the board is working as intended.

In fact it would be good if you try to verify all //TODO lines of defines_2-4.h :-/

Keep on going :-)

Batman313v, 2023-03-11:
## Update
I have more pins figured out with a few minor changes to the defines_2-4.h file but I have a few questions.

First let me say thank you so much for the help. I am learning so much on this project and really appreciate it.
My knowledge of PCB design is still fairly limited so I don't completely understand where to look to find some of the pins. These are the ones I have left over still:
```cpp
// Mosfet output
#define MOSFET_OUT_PIN GPIO_PIN_13	//TODO
#define MOSFET_OUT_PORT GPIOC		//TODO

// Timer BLDC short circuit emergency shutoff define
#define TIMER_BLDC_EMERGENCY_SHUTDOWN_PIN GPIO_PIN_12	//TODO
#define TIMER_BLDC_EMERGENCY_SHUTDOWN_PORT GPIOB		//TODO

// ADC defines
#define VBATT_PIN	GPIO_PIN_4			//TODO
#define VBATT_PORT GPIOA				//TODO
#define VBATT_CHANNEL ADC_CHANNEL_4
#define CURRENT_DC_PIN	GPIO_PIN_6		//TODO
#define CURRENT_DC_PORT GPIOA			//TODO
#define CURRENT_DC_CHANNEL ADC_CHANNEL_6

// Self hold defines
#define SELF_HOLD_PIN GPIO_PIN_2		//TODO
#define SELF_HOLD_PORT GPIOB			//TODO

// Charge state defines
#define CHARGE_STATE_PIN GPIO_PIN_0		//TODO
#define CHARGE_STATE_PORT GPIOF			//TODO
```
I'm gonna do some research on ADCs tonight and see if I can figure out how this works. I tried to probe around a little and I just don't know enough to get any valuable information out of it. Everything I probed just went to vss or vdd pins but no GPIOs.

I think the self hold is a latching circuit? I don't know much about those yet either so I'll be looking into that as well.

I don't even know where to begin for the charge state pins... I thought that was handled in the charger itself.

---

And I had some questions about these ones:
```cpp
#define UPPER_LED_PIN GPIO_PIN_1	//TODO
#define UPPER_LED_PORT GPIOA		//TODO
#define LOWER_LED_PIN GPIO_PIN_0	//TODO
#define LOWER_LED_PORT GPIOA		//TODO
```
Are these only for hoverboards that had under glow lighting?

```cpp
// Debug pin defines
#define DEBUG_PIN GPIO_PIN_4	//TODO
#define DEBUG_PORT GPIOB		//TODO
```
Is this referring to a debug header? Like this one?
[image]

---

I have a lot more photos and better documentation for these boards now and I took apart another one to document where some of the other headers went to. The photos aren't complete yet but I've added the new routes I found today and some labels for the unknown connectors.
[image]
[image]

### Documentation
Here is the documentation I have so far in markdown format:

```md
MCU: https://www.gigadevice.com/product/mcu/arm-cortex-m3/gd32f130c8t6

# A Board
## Buzzer
PB8

# B Board
## MOSFETs
#### Driver A (G)
PA10 -> HIN
PB15 -> LIN

#### Driver B (B)
PA9   -> HIN
PB14 -> LIN

#### Driver C (Y)
PA8   -> HIN
PB13 -> LIN

## BLE Speaker
TODO

## HALL
HALL C (Y) -> PB4
HALL B (B) -> PB5
HALL A (G) -> PB0

## UNPOP 1 (LED 0)
**PINS**
`  0  |  1   |   2  |  3 `
`NULL | NULL | NULL | V+`

**MCU**
0 -> PC13
1 -> PC14
2 -> PC15

## UNPOP 2
LED 0 PINS rotated 90° CC

## LED 1
**PINS**
`  0  |     1     |   2  |  3 `
`NULL | GREEN LED | NULL | V+`

**MCU**
0 -> PF0 (RED)
1 -> PF1 (GREEN)
2 -> PB2 (BLUE)

## TO MASTER
` 0  | 1 | 2 | 3 `
0 -> V+
1 -> PA2
2 -> PA3
3 -> GND

## UNPOP 3
` 0 | 1 `
0 -> V+
1 -> PA7
```

RoboDurden, 2023-03-12:
Good work.

The 8 pin SO-8 seems to be that dual op-amp for adc amplification: https://hmsemi.com/downfile/HM8631_HM8632_HM8634.PDF (simply search the pdf for C32X

[image]

OutA and OutB lead to a SMD resistor.
From behind that resistor you should already find the diode tester beeping when scratching with the needle over all MCU pins.

If not you need to probe the nearby and components to learn where the information of OutX is going.

Do you have Keil installed or should I compile for you ?

In Keil you can search the defines to see where they are used in the code to understand what purpose they i have.

I live outdoors again since yesterday. But I should have some minutes to unpack and open my Toughbook ..

Batman313v, 2023-03-12:
After probing around for a few hours I think I figured it out. I followed all the components but I kept getting further and further away from the MCU eventually stopping elsewhere then I noticed these two stray leads that went directly from the MCU to the Op-Amp and sure enough I think OUT A goes to PA0 and OUT B goes to PA1. These are the only two pins that set the tester off. I'm not 100% confident about this as there doesn't seem to be any resistors when going this route. But this is the only path I could find leading to any pins.

[image]

I don't have it installed yet, I'll work on that next 👍

RoboDurden, 2023-03-13:
That is okay if OUT is directly connected to a ADC input Iof the MCU. That little SMD resistor from OUT to -IN is the feedback resistor of the op amp loop that sets the factor of amplification.

Sorry when my instructions had been misleading and it took you several hours.

Will update the defines_2-4.h today.

But if you plan to install Keil anyway you might want to fork my repo, change the file yourself and make a pull request. I think that is what it is called.
Then I can import your changes with a single click :-)

Batman313v, 2023-03-14:
Gotcha that makes sense. It's all good 👍 I learned a lot poking around and researching different circuits. Once I get Keil installed and setup I'll definitely submit a pull request once I get a working file

RoboDurden, 2023-03-15:
I fear you still need to figure out which of the two adc pins is battery voltage and which is current_dc :-/
The current is measured at the shunt smd resitor to the right of the mosfets and to the left of the capacitor.
You should be able to measure a low resistance (multimeter) from either +INB or +INA to the left side of that shunt smd resistor. As that shunt resistor is very low you should see the same resistance to GND from one of tose +IN.
The other +IN might have some resistance (1k - 10k ?) to the red positive battery cable.

Still todo:

MOSFET_OUT_PIN

RoboDurden, 2023-03-15:
### MOSFET_OUT_PIN
seems to be an ordinary LED output ?
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8fba5e58c2f9179bbe6932fbedecf690698f5e33/HoverBoardGigaDevice/Src/led.c#L91

### TIMER_BLDC_EMERGENCY_SHUTDOWN_PIN
Is initialized here but never used somewhere else in code.
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8fba5e58c2f9179bbe6932fbedecf690698f5e33/HoverBoardGigaDevice/Src/setup.c#L176

### SELF_HOLD_PIN
This important pin keeps the mosfet open after the on/off button got pushed !
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8fba5e58c2f9179bbe6932fbedecf690698f5e33/HoverBoardGigaDevice/Src/main.c#L306
and turns off power on Shutdown:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8fba5e58c2f9179bbe6932fbedecf690698f5e33/HoverBoardGigaDevice/Src/main.c#L513

What is that ?P6504  8 pin smd chip on the master bottem right ? I guess it is a step down chip from battery voltage (63 Volt smd capacitor) to about 14 V (25V smd capacitor) to drive the gate voltages of the mosfets.
These 14V then get regulated down to 5V by the 78M05 which then can regulated down to 3.3V by the tiny A1117.

But i am not sure if 14V are needed because we have gate-driver-chips anyway which might generate their gate voltages themselves.
Anyway, the PWR Btn should somhow lead to a pin of that little 8pin chip.
On to this same pin should "somehow" connect the MCU SELF_HOLD_PIN pin.
That is how you can identify it.

With the Gen1 boards there is a big TIP127 transistor that is activated by the on/off button an hold by the Supply_Enable_MCU pin. See the schematics i added to https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/Schematics_2.0/hoverboard_gen2-0_uart_pins_gd32F130C8.jpg

### BUTTON_PIN 
this the on/off (POW) button. So also a connection (i guess with some smd resistor in between) to MCU pin.
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8fba5e58c2f9179bbe6932fbedecf690698f5e33/HoverBoardGigaDevice/Src/main.c#L457

## CHARGE_STATE_PIN
This seems to be a digital input that hast to be high in order to enable the motors:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8fba5e58c2f9179bbe6932fbedecf690698f5e33/HoverBoardGigaDevice/Src/main.c#L381

If we can not find it it should be okay to simply comment this line because chargeStateLowActive in initialised as set = true

## DEBUG_PIN
seems to be never used in code. Ctrl+Shif+F = Find In Files.

I have updated the defines_2-4.h

Batman313v, 2023-03-30:
## Update
[image]
I did a little studying up on latch circuits and while I can't seem to find a datasheet for this IC I hope I understand the circuit enough to figure this out. 

### SELF_HOLD_PIN
I'm guessing this is PA6. The Header on the right in order from top to bottom is 12V GND SW. I'm guessing SW stands for Switch and is probably used in another variant of this board or for QC testing which I think would reinforce my hope as setting PA6 should accomplish the same thing as a switch.

### BUTTON_PIN
PA5

## TODO
I still need to find the **CHARGE_STATE_PIN** as this would be nice to have as a backup incase the battery voltage gets to low.

## Notes
If the **MOSFET_OUT_PIN** seems to be for an LED output it is probably for some boards that have other external LEDs like the ones with it on the front and back. My boards don't have those so I don't think I need to worry about it.

I'm also not going to worry about the **TIMER_BLDC_EMERGENCY_SHUTDOWN_PIN** as I wouldn't even know what this is referring to. Possibly some older models with a timer IC to prevent overheating? As you pointed out it isn't used again so I don't think I'm gonna worry about this one either. This also goes for the **DEBUG_PIN** as I can just use UART and the LEDs for some basic debugging.

I finally got Keil installed on my machine, I thought I was having driver issues for the longest time but turns out my ST-Link was bad. I ordered a new one and it should be here soon.

RoboDurden, 2023-03-30:
Nice to hear from you again :-)
If you add battery plus (36V), battery minus (gnd) and DC out (the step down regulated voltage, probably something like 14V) to you circuit, we could figure out what that little chip likely is doing.
And you could check if pressing the onoff button will temporary activate the 14V , 5V , 3.3V voltage regulators chain.

I will compile a firmware for you later this day.

RoboDurden, 2023-03-30:
So here the first test binaries: [HoverBoardGigaDevice/BinariesToTest/](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/HoverBoardGigaDevice/BinariesToTest/)

Please check the new defines_2-4.h before flashing.

Here some ideas in yellow for better understanding of the step down regulator:

[image]

Batman313v, 2023-03-30:
Gotcha 👍 I'll go chase more of those traces today and try hooking up a battery. The overlap at the bottom is because they put the traces on top of each other on opposite sides of the board, I'll redraw that later today so it makes more since

cjoe-51, 2024-09-30:
Hi, I have hoverboard with exactly same PCB but another processor. There is MM32SPIN 05PF inside. Is this processor equivalent to GD32F130C8T6 or will the necessary firmware be different?

I-hate-2FA, 2024-09-30:
you should look in mm32 repo

RoboDurden, 2024-09-30:
As your board will be 2.4.5 , yes please start a new issue and post photos from the front side with cables bent outside so as much as possible is visible of the Mainboard.

You can install the pinFinder firmware of AiLife to verify the pins and then switch to his sinusoidal firmware.
Or go with the pin definitions of 2.1.5 and modify MM32 EFeru FOC port Trontin has published recently...
It looks like your board has dual opamp Chip for two phase currents, which are needed for FOC.

AnthonyLia, 2025-10-15:
@RoboDurden  apologies. The previous board I flashed successfully (2.1.9) did not have significant hardware differences or this requirement that slave board needs external 12V on comms in addition to 36V power, so I just assumed all boards are likely not interdependent on the master/slave pairing.

Anyways, I'm having more difficulties. I have tried your autodetect a lot, and I'm not getting good/complete pins. 

[2_1_5_autodetect_round_1.txt](https://github.com/user-attachments/files/22915424/2_1_5_autodetect_round_1.txt)
[2_1_5_autodetect_round_2.txt](https://github.com/user-attachments/files/22915423/2_1_5_autodetect_round_2.txt)
[2_1_5_autodetect_round_3.txt](https://github.com/user-attachments/files/22915422/2_1_5_autodetect_round_3.txt)

BUTTON and SELF_HOLD were probably never going to work since there's no SMD stuff to let them behave with MCU. 
However, autodetect can only get 2 out of 3 phase pins, and I tried lots of times in round 3. Also autodetect thinks 2 of the 3 discovered phases **are** the PA0 and PA1 that the repository has as VBATT and CURRENT_DC. 

With remote_uart and TestSpeed.ino, it's not spinning smoothly like it did on 2.1.9; so I am trying to find better pin definitions. VBATT and CURRENT_DC were also not recording good values (none were around 36V on voltage, and current was never recognized). Why is autodetect not getting all pins correctly? (I did use the precompiled bin)

RoboDurden, 2025-10-15:
The PHASE pins currently have no use in the firmware. And most boards only offer 2 phase inputs anyway. Phase currents are needed for FOC. So if you get two PHASE pins, you could try one of the EFeru FOC ports..
PHASE pins must be analog inputs, which can only be PA0 ... PAx and PB1 and PB2, if i remember correctly. So you getting PA0 and PA1 makes sense.

To detect VBATT, it might be good to use the charger as the power supply, because with my motor on and motor off, the voltage drops quickly from 42V to 36V.
Otherwise put your hand on the motor. But with a full battery, voltage might not drop significantly.
Same goes for CURRENT_DC. Toggle to manual mode as my autodetect code is no really working.
I see a shunt resistor on your board, so on of the analog inputs should give CURRENT_DC.

[image]

There are calibrate defines in defines.h you might need to add to your defines file.

```c++
// ADC value conversion defines
#ifndef MOTOR_AMP_CONV_DC_AMP
	#define MOTOR_AMP_CONV_DC_AMP 0.201465201465  // 3,3V * 1/3 - 0,004Ohm * IL(ampere) = (ADC-Data/4095) *3,3V
#endif
#ifndef ADC_BATTERY_VOLT
	#define ADC_BATTERY_VOLT      0.024169921875  	// V_Batt to V_BattMeasure = factor 30: ( (ADC-Data/4095) *3,3V *30 )
#endif

```

AnthonyLia, 2025-10-17:
Thanks again for the help Robo. I did check 6 output pins versus defines_ad.h and I did notice a typo in your repo. Phase B low should be PB14 but it’s PB4.
This unfortunately didn’t solve my problem though.
I did flash RemoteDummy and still get a misbehaving board:

https://github.com/user-attachments/assets/20695b41-4fe8-4a9e-9654-408dc93f59a1

I triple checked that the 6 phase pins are all the same for defines_ad.h and defines_2-1-5.h; and the motor does at least spin during Autodetect:

https://github.com/user-attachments/assets/0512ec61-a5b8-442d-b06d-324762b7536e

It probably shouldn’t take 80 watts of power to free spin the motor this slowly, but it at least spins and I assume Autodetect causes such high power demand.

I’m at a dead end now. I tried:
- changing TIMER_BLDC_PULLUP to GPIO_PUPD_PULLUP and GPIO_PUPD_PULLDOWN
- commenting out SELF_HOLD and BUTTON
- measuring connection of 6 MCU pins to mosfets, and got about 99ohms (because of SMR) for all 6
- verifying that hall sensors are PB0, PB5, and PB4 on MCU, and they are
- changing BAT_CELLS on Autodetect to 10 instead of 6
- rechecking with 2.1.9 board:

https://github.com/user-attachments/assets/b57662ee-f9b3-4974-891b-10800dbae814

Restating known facts: motor does work perfectly with 2.1.9 board. The 2.1.5 board shorts/misbehaves with Dummy or Remote/TestSpeed. The 2.1.5 board at least spins with Autodetect. 
To me it seems like it’s something related to config (but maybe not directly in defines or config.h) that changes for Autodetect.
Please help me figure out the issue

RoboDurden, 2025-10-17:
Sorry for your trouble. 
80 watts for Remote autodetect sounds too much.
What is the power surge for your 2.1.9 board ? During hall pin detection.

Motor is spinning smoothly after successful hall ordering ?

Use the same Bldc_bc as remoteAutodetect, not bldc_sine.

Set constant speed in RemoteDummy 

I don't understand your power set-up with two power supplies:

[image]

Did you make a new defines file from scratch or modified the existing .h file?