RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #4: Wrong values in CLAMP betwen master and slave communication

matixan, 2023-03-14:
Im preety sure that there should be ifdef debug for that. Actually its clamped to -300 and 300. I spend smth like 5 hours figuring this out haha

```
// Calculate result pwm value -1000 to 1000
	pwmSlave = CLAMP(pData->iPwmSlave,-1000,1000);
```

RoboDurden, 2023-03-14:
Oh thank you very much for opening this issue.
Yes i feel very emberassed because this must have been me when i rewrote the comms files.
It is so much pain when wasting hours on bug that other have made carelessly.

matixan, 2023-03-14:
Thanks for fast response! Im glad someone is actively working on these split boards. It looks like every never hoverboard is using some variants of these split boards. Ill be mind blowed if someone were ablo to run FOC on it.  Big win for 
https://github.com/Kambodia91/Hoverboard-firmware-hack-FOC-MASTER-SLAVE and @candas1 for trying this out.
I belive that in close future it will work on gd32f130 as well.

RoboDurden, 2023-03-14:
I have already tried the Candas1 FOC code but i do not have the PlatformIO toolchain installed and with Keil compiler v6 it gives a ton of errors.
I still dream to first build a object oriented hoverboard OS and then implement this gen2.x code and afterwards add the FOC code.
Unfortunately i can not debug with the Keil IDE. My oo-code is compiling but the main() is not called or some uart libraries are not c++ compatible :-(
And as i strongly dislike linux, all these stm32 enviroments are no pleasure to me.

RoboDurden, 2023-03-14:
@matixan if have uploaded my oo-code. If you have Keil installed, please take a quick look at https://github.com/RoboDurden/Hoverboard-ObjectOriented-OperatingSystem

RoboDurden, 2023-03-14:
Thank you @Candas1  for posting here :-)
I already feared that these split boards have no phase current sensing via MOSFET shunt voltages and that this might be needed for foc.
Still have not found the happiness to understand foc.
So thanks for the info that FOC needs phase currents.

And bad news that the new Artery boards also no longer offer that feature.

It seems that all boards with nice dedicated gate driver chips have no phase current sensing ?

There are new algorithms for sensor less bldc that can calculate the position and start under full load. Maybe the phase currents can somehow be calulated only with the total current and then phases ?

As I said, i never really digged in the theory of FOC :-/

matixan, 2023-03-14:
I have some progress regarding @Candas1 fork. It compiles out of the box with some warnings(I changed mcu to C8) I flashed it anyway and theres no smoke. I got CLI ASCII working on steering uart, buzzer beeps when motors disconnected, standstill is IMO working as motor is resisting when im trying to run it by hand. Also theres some debug info.
[image]
Im gonna dig deeper as it might be working. My platform is TTSD22 with 10 inch motors. Now im figuring out how i can controll the motor

matixan, 2023-03-14:
> I have already tried the Candas1 FOC code but i do not have the PlatformIO toolchain installed and with Keil compiler v6 it gives a ton of errors.
I dont think platformio is problematic, it downloaded everything itself

Candas1, 2023-03-15:
> I have some progress regarding @Candas1 fork. It compiles out of the box with some warnings(I changed mcu to C8) I flashed it anyway and theres no smoke. I got CLI ASCII working on steering uart, buzzer beeps when motors disconnected, standstill is IMO working as motor is resisting when im trying to run it by hand. Turning off dont work at all, also no sound on startup. Actually whole code is very mixed as theres loop() running from main(). Also theres some debug info. [image] Im gonna dig deeper as it might be working. My platform is TTSD22 with 10 inch motors. Now im figuring out how i can controll the motor
> 
> EDIT: after uncomenting more code motor is spinning

Cool
Yeah during my test I was controlling from debug so I probably commenting something out

Candas1, 2023-03-15:
> > I have already tried the Candas1 FOC code but i do not have the PlatformIO toolchain installed and with Keil compiler v6 it gives a ton of errors.
> 
> I dont think platformio is problematic, it downloaded everything itself

Vscode/platformio is nice, you can even get it to run in the cloud (limited time)

https://user-images.githubusercontent.com/20670049/225224072-07f61967-c070-415e-93df-e448801b5544.mp4

Candas1, 2023-03-15:
And just one comment, I have probably left the fork in Sinusoidale, not FOC.

RoboDurden, 2023-03-15:
> I dont think platformio is problematic, it downloaded everything itself

Can you debug a gen 2 board with a simple st-link v2 clone on platformIO ???
Then I might give it a try to make progress with my oo repo.
Can you debug a gen 1 board = stm32f103 ???

matixan, 2023-03-15:
I have only one set of mainboards. Debug works, but i had to change flash id to even fit debug code.
[image]
At this moment my progress will slow down due to burned phase. I tried to stop motor unplugging battery :))

RoboDurden, 2023-03-15:
After unplugging the battery, the capacitors should get drained and the board turning off.
Spinning motors might generator electricity and the energy gets pumped into the capacitors until they blow?

You might want to check my hoverboard repair videos.
https://youtu.be/zo2Att8zO3A

Will install Pio on some sunny spring day. Living outdoor with my solar car again..

QGB, 2023-04-19:
can not get steer data  (always 0)

RoboDurden, 2023-04-20:
One liners don't start me thinking.