RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #5: User Interface on 2.2 hardware

IerwinDE, 2023-03-19:
I'm trying to get 2.2 Boards to work with an UART controller.
Until now I managed to get the Fork of Herleybob to spin the wheels but since that firmware doesn't really seem to be supported I considered switching.
I don't even know is UART is really supported there because the comments in the config.h still suggest hooking UART to the left sideboard port and I don't think that exists on the 2.2 hardware.
Does this version actually support the 2.2 Hardware yet? The precompiled bins don't contain 2.2 yet, so I guess it doesn't.

I am primarily trying to find out if there is a working fork that supports UART for the 2.2 hardware and how the pin layout for UART is.

Any help would be appreciated

RoboDurden, 2023-03-19:
Great if you know a firmware that makes 2.2 spin. Please post the link to his gihub repo.
I might adapt the pin defines from that source to a defines_2-2.h 

I see a nice Rx tx header above the St link flash header to the left of the MCU in my images.

IerwinDE, 2023-03-19:
Check out this comment over in @JRomainG repo:
https://github.com/JRomainG/GD32F130K6-hoverboard-hack/pull/2/commits
I think the changes have never been merged but I checked out the code and it does spin the wheels at 120RPM like it said it would.
No futher testing done by me so far

IerwinDE, 2023-03-19:
And you are probably right about those full sized exposed serial pins. I have no Idea how I could not see those

RoboDurden, 2023-03-19:
Please post a link to the full source that you did compile and which runs at 120 rpm

IerwinDE, 2023-03-19:
I dont have much experience working with GitHub, so I will try my best to get you exactly where I downloaded it from. As I said this is an unmerged commit, So I dont really know how to link to it.
I think this should be the one:
https://github.com/JRomainG/GD32F130K6-hoverboard-hack/tree/2147123518045b09231634e227fe695129171a37

RoboDurden, 2023-03-19:
Nice, his defines definitely look different:
https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/defines.h#L34
```
#define HALL_V_PIN GPIO_PIN_12
#define HALL_W_PIN GPIO_PIN_3
#define HALL_U_PIN GPIO_PIN_8
#define HALL_V_PORT GPIOA
#define HALL_W_PORT GPIOB
#define HALL_U_PORT GPIOB
```

my format:
**https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Inc/defines_2-0.h#L75**
```
// Hall sensor defines
#define HALL_A_PIN GPIO_PIN_11
#define HALL_A_PORT GPIOB
#define HALL_B_PIN GPIO_PIN_1
#define HALL_B_PORT GPIOF
#define HALL_C_PIN GPIO_PIN_14
#define HALL_C_PORT GPIOC
```

porting his pin configs into a defines_2-2.h should not be difficult.
Thanks for pointing me to this code.
Will do it myself in the next week - maybe.
Am living outdoor. So it depends on how much joy you already invest.

IerwinDE, 2023-03-19:
I feel I should start to dig into the pin mapping and hopefully understand a little better what's happening.
I have been poking around with UART Signals on various pins but nothing worked.
Guess the first thing I have to do is get serial debugging to work so I see what the firmware is doing.
At the moment it's just a black box for me

RoboDurden, 2023-03-19:
I don't know what this repo does or if it is complete.
But you might get debug output at that one UART header which used to connect the two twin boards

https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/config.h#L68
https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Src/setup.c#L87

Try listeing with 38400 baud on that Rx pin.

IerwinDE, 2023-03-19:
No luck so far.
But there are to many variables in my setup that might be the reason it doesnt work right now.
Would be good to have firmware that is known to wokr because my wiring usually isnt ;)

RoboDurden, 2023-03-20:
Yes, read issue 3 and learn about using a sewing needle to find/verify pin connections: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/3

With motors spinning i guess you have already 90% of the work done !

Would be great if @Herleybob  could join this discussion.
The comments on his pull request are helpful: https://github.com/JRomainG/GD32F130K6-hoverboard-hack/pull/2
Unfortunately it seems he never published his diagrams :-/

"This is still a work in progress and a port from the original Niklas Fauth firmware. It does not read current (and does not shut off based on current, etc), I believe board temp and voltage is working correctly. I am running this on a 40v battery with no issues. (41.11v fully charged). I need to add a picture of the board and the pinouts, but as it sits when you compile it, it will run the motor up to 120RPM and hold that RPM. ADC conversions are working. The current firmware supports both the main(no buzzer) and aux board(buzzer). Set definition in the config file.

What is needed is either communication between the two boards, or seperate adc inputs per board. But as of right now basic motor control is working."

He also added this section to the readme.md: https://github.com/JRomainG/GD32F130K6-hoverboard-hack/tree/2147123518045b09231634e227fe695129171a37#connection

"TODO: Add picture/diagram of the board and connections. Most are all common sense but the ADC and USART require special connections.

For the ADC to work, you must change the link cable between boards from the 4-pin blue plug to the 4-pin white plug next to it (see diagram). You can then use the pin labeled in the diagram for the ADC input.

For the USART to work, you CANNOT use the labeled tx/rx empty vias on the board due to them being tied to the bldc output. You must use the one labeled in the diagram.

Note: For my arduino to receive correct tx transmission, you must jumper or bridge the resistor on the tx output line, labeled in the diagram."

The smaller GD32G130K6:

[image]

A closeup picture of my board:
[image]

So these Rx1 and Tx1 pins above the flash header connect to RX1=A10 and TX1=A9
These are indeed used by the bldc mosfet pins: 

https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/defines.h#L48

According to his defines.h https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/defines.h#L24
the rx and tx pins (blue header ?) are TX = A15, RX = A7
And it seems there is a smd resistor between A15 and the blue header so his Arduino serial did not work.
You can probe A15 directly beside the MCU as it goes downside to the backside layer there.
With a multuimeter and meauring the resistence from these two mcu pins to the pins of the blue header, you verfiy that.
And you might already receive debug log output on the A15 pin close to the MCU using the sewing needle

He added 6 adc inputs here: https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/defines.h#L81

But only initializes 3 inputs here: https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Src/setup.c#L144

that would be adc_buffer.adc , adc_buffer.temp and adc_buffer.batt

adc_buffer.adc might be the speed input and that pin would be ADC_CHANNEL_2 , but it is not defined in defines.h. Maybe defined by GD32F130 .. Then look in the big datasheet to find the pin.

You would have to uncomment this line https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/config.h#L84

And put "//" before that one: https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Inc/config.h#L94

But most important for you right now i guess is getting debug output.

What chaintool are you using ? There is only a makefile. So no PlatformIO or Keil projekt.

Roland, Germany and :-)

IerwinDE, 2023-03-20:
I didnt even notice the readme that was comitted so that is a great additional source of information for me.
Right now I am using Ubuntu in WSL so I can run make and st-link commands directly.
For the older Hardware I have I used plaformio so Keil would actually be a third variant I would have to dive into.

IerwinDE, 2023-03-20:
I poked a needle to A15 and connected it to a USB2UART Adapter, then pressed the power button.
Did not get any Serial Data this way.
I will double check but so far it doesnt look good

IerwinDE, 2023-03-20:
I removed all my changes from config.h and recompiled the original code with #define CONTROL_MOTOR_TEST and poked A15 again.
This was the output:
´´´
10:51:47.192 -> 1:0 2:700 3:366 4:165 5:1528 6:4462 7:1760 8:15
10:51:47.332 -> 1:0 2:725 3:365 4:165 5:1523 6:4462 7:1760 8:15
10:51:47.473 -> 1:0 2:750 3:366 4:161 5:1524 6:4461 7:1760 8:15
10:51:47.566 -> 1:0 2:775 3:358 4:165 5:1526 6:4462 7:1760 8:15
10:51:47.706 -> 1:0 2:800 3:360 4:165 5:1528 6:4462 7:1760 8:15
10:51:47.846 -> 1:0 2:825 3:360 4:159 5:1528 6:4461 7:1760 8:15
10:51:47.940 -> 1:0 2:850 3:368 4:165 5:1538 6:4462 7:1760 8:15
´´´

I dont unterstand that yet, but I guess it means that A15 is indeed the debug TX

IerwinDE, 2023-03-20:
Yeah I would prefer to get your firmware running. And then it would be good to have your repo linked in the main project for all the Split boards you support.
My project is a 4WD Kart with a human driver.
I only have one old Mainboard and the two 2.2 at the moment and I am not sure if it's even possible (or smart) to build a 4WD application with this mixture. Especially since I am aiming for high speed and I still don't know if the limits of the 2.2 Hardware are similar to the old one (like running 12S).
But it is what I have so until I get another cheap board I will follow this route.

I think it will be best to have a central controller that reads the analog inputs from brake and throttle and sends them via UART to all wheels.
Motor Control should be done in torque mode.

RoboDurden, 2023-03-20:
At some later point of my oo operating system I would like to add i2c control.
Then all your four motors could connect to the same i2c bus. With torque commands it is not neccesary to pair two motors. And driving a curve will need different motor speeds for left and right anyway.
With some gen 2 layouts missing a second UART, i2c seems to be the solution.

IerwinDE, 2023-03-20:
I2C might be a great addition especially for the split mainboards.
You could even build units of one wheel and one board that can be easily swapped between projects and onlöy have to connect i2C and Power

IerwinDE, 2023-03-20:
Aside from that I wish there was some kind of database to guesstimate what kinf of hardware might be inside a particular hoverboard. Sucks to play the lottery with these used boards

IerwinDE, 2023-03-20:
> Yes please send high resolution photos to [roland@robo4future.de](mailto:roland@robo4future.de)
> 
> Use resistance measure with a multimeter to confirm the rx1,tx1,rx2,tx2 MCU pins. I guess that rx1 and tx1 are useless. TX2 = A15, RX2 = A7 blue header ?

A15 is connected to TX2 with a 1K Ohm Resistor.
Cant find a good point to probe A7 yet. At least when I try to probe directly on the chip I only read 19KOhms to RX2 as well as TX2

Herleybob, 2023-03-20:
Hey all, it's been some time since i worked on this firmware and unfortunately i don't have these boards anymore. 

As far as the tx line, i removed the resistor and jumped the pads with solder. I did have debug communication working. 

If i remember correctly, the current firmware was only a test that spun the motors at a preset rpm, no adc control. Performing above modifications should allow adc input. 

sorry i can't add more, it's been a few years and a few projects in between then and now!

RoboDurden, 2023-03-20:
Okay i compiled the first 2.2 binaries.
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/HoverBoardGigaDevice/ReadyToFlash

Take a look at https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Inc/defines_2-2.h what might still be done

don't know if the firmware will work without adc for dc_current.
Maybe without that SELF_HOLD_PIN, the board will chut itself down.

all of course at your risk.
I would try the the test binary as it also simply spins forward and backward.

For the serial connection you will need my arduino code found here.

roland and :-)

IerwinDE, 2023-03-20:
I'll give the binaries a try later.
Which board ist master, which is slave and do they need to be connected or can I start by flashing just one?

RoboDurden, 2023-03-20:
Only flash one and test.
We have no second UART yet

RoboDurden, 2023-03-20:
> Aside from that I wish there was some kind of database to guesstimate what kinf of hardware might be inside a particular hoverboard. Sucks to play the lottery with these used boards

Yes but there are so many Chinese clones..And the old 32f103 get more and more rare.
If we had a 500+ community we could get an old manufacturing line alive..

IerwinDE, 2023-03-21:
So I just tried to flash the binary, but get an error:
```
st-flash --reset write "hoverboard 2.2 master test.bin" 0x8000000
st-flash 1.7.0
2023-03-21T07:57:00 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
file hoverboard 2.2 master test.bin md5 checksum: d4348c96d88fa9a295ea40e5d22df3a, stlink checksum: 0x00fa70ec
2023-03-21T07:57:00 INFO common.c: Attempting to write 212964 (0x33fe4) bytes to stm32 address: 134217728 (0x8000000)
2023-03-21T07:57:00 ERROR common.c: addr too high
stlink_fwrite_flash() == -1
```

RoboDurden, 2023-03-21:
I have also been thinking about a test-pins binary that set all possible pins to input and then cycles over all pins by only setting one pin to digital Out and blinking it's index.
As two MOSFETs need to be active at the same time to cause a shortcut that might be safe.
Would you test such a binary ?
This would make identifying led and UART pins easier. But not that helpful in the whole picture.

RoboDurden, 2023-03-21:
You should also check if the 5v and 3.3v voltage regulators get powered with onoff button pushed.
Against the question if there is a 14-15 volt regulator before that 5V volt regulator.

RoboDurden, 2023-04-08:
> Flashed the other firmware again to verify my wiring and that one still lets the wheel spin. The precompiled does not. Thats all aI can say for now

I am back home now and can also test this 2.2 layout.
Please send me the binary of that other repo that at least does some spinning. It would help me a lot to test my hardware and my firmware. Please than bin file to roland@robo4future.de
Thanks.

IerwinDE, 2023-04-09:
check your spam box, I sent you a binary

RoboDurden, 2023-04-09:
yes i received the bin file. nThank you very much.
Sad to hear that you have continued with a gen1 board.
"Alone.. you always.. lose!"
May still have some happiness left to test my firmware on this 2.2 layout this week.

IerwinDE, 2023-05-03:
Have you had any luck with the firmware yet?
I happen to have a new idea for a robotic project where the 2.2 board and weaker motors I have would be perfectly fine so I might be able to invest some time into getting ot to work.
Main focus for me would be slow driving and controlled by a microcontroller for autonomous navigation.
I think that means PWM needs to be functional

RoboDurden, 2023-05-03:
Sorry, left alone I had no happiness. Maybe I will be back at my train station in a month.

IerwinDE, 2023-05-03:
So I am still trying to get my version to do anything but spin the wheel at a fixed speed.
Even though the Test mode is supposed to cylce from forward to backward motion the oly thing I am getting is constant speed in one direction.
And I dont think that can be related to missing pin mappings or something.
```
#define CONTROL_MOTOR_TEST
#define CONTROL_MOTOR_TEST_MAX_SPEED 600        // sweep slowly from -MAX_SPEED to MAX_SPEED (0 - 1000)
#define CONTROL_MOTOR_TEST_RPM 220
#define CONTROL_MOTOR_TEST_PADDING 10
```

When I set this and turn the board on, the wheel accelerates to a fixed speed and keeps that. No Sweeping at all.

I also tried to get the Serial Interface to work (connected to the RX, TX and GND pins you labeled uart2 in the picture above).
First using an ESP8266 but since that didnt work I tried with an USB2UART connector.
I had some success when using #define DEBUG_SERIAL and was able to receive Debug output.
I guess that at least means that one UART TX is mapped correctly.

Unfortuantely I have not been able to receive anything but the Debug messages yet.
I disabled UART_DEBUG and went with CONTROL_SERIAL_USART3 and FEEDBACK_SERIAL_USART3 instead. Tried the webtool to send commands but no reaction whatsoever.

IerwinDE, 2023-05-03:
Yeah... its been to long sonce I last worked on this and I didnt bother to read this thread before.
It seems the firmware version that i am using does not really support a lot of features yet.
For example in the whole code "CONTROL_SERIAL_USART3" is never used.

So I guess we should get back to the original plan of getting your fork working on the 2.2 board.
I need some guidance for that. I hope its mostly transfering stuff from Herleybobs version over to yours but I dont know where to start and I dont want to break the board wit wrong mappings

RoboDurden, 2023-05-03:
> Flashed the other firmware again to verify my wiring and that one still lets the wheel spin. The precompiled does not. Thats all aI can say for now

So i think it is save to test my binary.
Even if the motors do not spin you might check if the serial communication works.
Then you could get feedback that way..

As my firmware is working for the 2.0 layout i am pretty sure that it should also work with this 2.2 layout.
It needs a complete mapping.

By now i have some other 2.0 repo in my online compiler !
So you might port the defines_2-2.h to that code and see if you get serial feedback with that repo.
But installing keil or platformIO should help you more to do some debugging..

IerwinDE, 2023-05-03:
I flashed the HoverBoardGigaDevice/BinariesToTest/hoverboard 2.2 master test.bin (think you moved the file from ReadyToFlash to BinariesToTest.
I dont get any reaction from the board. No spinning, no serial.
Its hard to tell if the board is even turning on at all

RoboDurden, 2023-05-03:
Now it is up to you if you want to continue with

1. continue with the harleyBob firmware
2. my 2.x repo
3. or replacing the defines.h with my defines_2-2.h here https://pionierland.de/hoverhack/?p=gen2

IerwinDE, 2023-05-03:
I'd preferr to get your 2.x repo working.
I will find a way to compile it so I can play with the settings.
For now I preferr to compile on my local machine so I will try that

IerwinDE, 2023-05-03:
```
Build Time Elapsed:  00:00:01
Load "C:\\temp\\Hoverboard-Firmware-Hack-Gen2.x-main\\HoverBoardGigaDevice\\Objects\\Hoverboard.axf" 
Erase Done.
Programming Done.
Verify OK.
Application running ...
Flash Load finished at 21:45:53
```
looks like compiling and flashing using Keil was way easier than I expected.
Of course at this point I have absolutely no Idea what I am doing and I do not even know if it is sfe to power on the board with what I compiled.
I just uncommented "#define LAYOUT_2_2" in the defines_2-2.h and hit F7 and F8

IerwinDE, 2023-05-03:
So I started to scrape the very top of the iceberg that is understanding what is actually happening in the code.
As far as I understand the main loop should send packages over UART so I don't really understand why I am not getting any output with the wiring that gave me debug messages.
I tried to use the debug feature of Keil hoping I could simply set some breakpoints and step through the code, but even though the execution seems to halt somewhere, it never shows me the line it is at.
My learning curve would be very long if it requiresd to really learn how to master such a powerful tool, so I hope you have some ideas how I should proceed.

RoboDurden, 2023-05-04:
Sorry I also failed with the debug features.
You need to find the correct Keil settings which I failed to find.
Like adding debug to the assambler output...
There's an issue from me here on GitHub somewhere crying for help :-/

platformIO is said to have working step over debug but that tool chain seems to be more complicated...

Please check that the MCU is getting 3.3v when pushing the on button !!!!

In the defines_2-2.h you can see that the hold pinis still not defined.

Try keep pushing the on button to see if 3.3V at least stays on as long as pushing.

Study the first commands of the main() function.

Try to route at least one led output and update the define pin. Then set led on as soon as possible to get this as debug output.

Roland.

RoboDurden, 2023-05-04:
Compare the initialisation lines in main() between my gen2.x and HarleyBob. Both repos are based on the same flo99354 repo ?!

IerwinDE, 2023-05-04:
First of all I tracked down the connection of the Pin labeled "Red" on the outmost socket of the board.
It goes through a transistor and some resistors and ends at PIN 24 (PA14).
What would be a good variable to assign this output to in defines that actually does something when the board turns on?

I couldnt find any information about the root of Herleybobs code so I compared the main() of the two repos (see: https://github.com/JRomainG/GD32F130K6-hoverboard-hack/blob/2147123518045b09231634e227fe695129171a37/Src/main.c).
While there are some similarities all the method names seem to be different so I dont think they are based on the same repo.

Yesterday when I was trying to step debug I noticed that I can actually see when the button is pushed in Keil (Peripherals->GPIO->Something). I still have to mnake sure the chip is getting 3.3v when I turn on the board and not just powered through the ST-Link

RoboDurden, 2023-05-04:
Don't connect 3.3V from stlink dongle to board. Instead you should be able to use the open pin to probe the 3.3V of the MCU :-)

Define the led just as one of the led outputs.
Then for testing move InitGPIO to the beginning of main() and turn the red led on like

GPIO_init();
	
	// Activate self hold direct after GPIO-init
	gpio_bit_write(SELF_HOLD_PORT, SELF_HOLD_PIN, SET);

If then the rer led indeed turns on you could move the code backwards line by line to see where the code hangs.

IerwinDE, 2023-05-04:
Ok, lets go in baby steps, just to be safe.
I flashed Herleybobs firmware again and connected the led. For now I do not have a motor attached.
When I press the power button the led lights up and stays on.

Then I flashed your 2.2 again (no changes to the defines yet) and the LED also lights up when the power button is pressed, but goes dark when the button is released.

Im not 100% sure how the mapping works, so I will wait for your confirmation before I break something.
Is this correct way to map the LED to Pin 24/PA14?
```
#define LED_RED GPIO_PIN_14			
#define LED_RED_PORT GPIOA
```

RoboDurden, 2023-05-04:
That's progress :-)

Now search my code to find where LED_RED is used so you know at which line it is turned on.
Watch my tutorial video on the gen2.x firmware !
When found uncomment that line to verify that you now can indeed use the red led as a marker to test It the code execution has reached a certain point.

As I said you need to find the proper pin for 

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Inc/defines_2-2.h#L112

Otherwise it is no surprise that the board turns off after you release the on button.

You could test if the motors start spinning when you keep pressing.
But for that I think you have to uncomment the while loop-forever in main() that waits for the on button to be released.

RoboDurden, 2023-05-04:
> ```
> #define LED_RED GPIO_PIN_14			
> #define LED_RED_PORT GPIOA
> ```

Yes that would be gpioa14

But if the red led turns already on without updatingthe defines than it might simply be powered directly by 3.3 V and can not be turned on off programmatically.

IerwinDE, 2023-05-04:
meanwhile I probed PIN1 for 3.3v and its only there as long as the button is pressed.
I guess that means the chip is not staying on.

I will route the green led

IerwinDE, 2023-05-04:
geez stupid small stuff!
Upon further inspection I think that i made a mistake before and in reality GREEN is connected to PA14 while RED is connected to PA13.
But as long as the chip is not staying powered on I guess there must be something else going wrong.
I dont yet understand what is keeping it turned on. I assume there should be some GPIO that indicates that it should stay powered and changes when its time to power down? Maybe that is what needs to be mapped correctly first?

RoboDurden, 2023-05-04:
Read my words. I told you to identify the hold pin !!!

Now try to uncomment this line https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Src/main.c#L334

The comment above that line is wrong. It should say wait until button released, not wait until button pressed.

The keep the on button pressed and see if the test firmware makes the motor spin.

IerwinDE, 2023-05-04:
unfortunately no spinning this way.
Sorry that I missed the part about the hold pin.

I tried to find out how its done in Bobs Firmware but it confuses me. This is the PowerDown Method:

```void poweroff() {
	#ifdef AUXBOARD
		for (int i = 0; i < 8; i++) {
			buzzerFreq = i;
			delay_1ms(100);
		}
	#endif

	#ifdef MAINBOARD
		buzzerPattern = 0;
		enable = 0;
		delay_1ms(800);
		gpio_bit_write(BUZZER_PWR_PORT, BUZZER_PWR_PIN, 0);
		while(1) {}
	#endif
}
```
The main also sets this pin to high right after the init.
So it seems a lot like this was the hold pin but it is also the pin controlling the buzzer?
Does that make sense at all?

RoboDurden, 2023-05-04:
Put 

gpio_bit_write(BUZZER_PWR_PORT, BUZZER_PWR_PIN, 0);

Right after while(1) i the HB firmware to see if the board turns off.

As the MCU has less pins it might be possible to use the buzzer pin as the hold pin because buzzing has such high frequency that it would not effect the hold transistor.

Do you get debug output with my firmware after uncommenting the wait_release loop and keep pushing the on buton ?

We need at least the most simple feedback of being able to turn a led on.

GpioInit is the most simple blink code.

IerwinDE, 2023-05-04:
Looks like I just killed one of my boards.
I can no longer connect with ST-Link :(
Im trying to recap what was the last thing I did...
I defined 
```
#define LED_GREEN GPIO_PIN_14
#define LED_GREEN_PORT GPIOA
```
and put GPIO_init and gpio_bit_write(LED_GREEN_PORT, LED_GREEN, SET) in line 285 of the main.

I flashed that version and it lit up the green led for as long as the button was pressed.

Next I looked up Bobs define to find that BUZZER_PWR_PIN is GPIO Pin 11.
So I defined:

```
#define SELF_HOLD_PIN  GPIO_PIN_11		// lerwinDE
#define SELF_HOLD_PORT GPIOA				// lerwinDE
```

and...
well I am not sure if I was even able to flash that version. I think I already got the st-link connection error message when I first tried to flash it.

But I am affraid for now I must assume that one board is broken or in a state that is not flashable

RoboDurden, 2023-05-04:
Don't think that the buzzer pin can do much harm as it was already used as the buzzer output.

And I am not aware that moving the InitGPIO before anything else can kill the board.

And as the green led did light up that pin could not do something else.

Thank you for the progress with the green led.

If you send me your PayPal address to roland@robo4future.de I will donate 5€ to you.

I now  for sure will continue your work. But for the next weeks i am working on a new roof at my solar warehouse.

I think you should continue with the second board :-)
If you succeed with repeating a working step but can not flash the first board then there is indeed something wrong with the first board.

Have you ever uncommented the wait-release loop ?
Then leave the GpioInit where it was but turn the green led on in the while(1) loop.

To test if the code reaches that main loop.

But of course I understand if you now at least want to make a break.

IerwinDE, 2023-05-04:
Absolutely no need to donate money.
I would be really happy though if someone reading this would like to sell me an extra 2.2 board.
Since there is no working firmware for these boards yet, I bet a lot of people have them sitting on the shelf after they pulled them out of boards.

I also have not given up on the board yet.
I read that it is possible that the own code prevents the device from entering flash mode and there apears to be a workaround using the RESET pin. I will have to dig into that.

RoboDurden, 2023-05-04:
Is the 3.3V on button press still working ?
With the gen1 boards you should bridge the on button because the hold-transistor can fall of during flashing.
As i told you to not power the board from the st-link, this might have caused your board to turn off while flashing :-(

Then indeed the reset pin might help ?!

I have added a blinking of the green led in the test mode 

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Src/main.c#L354

and i uploaded new binaries with A14 pin for green led.

That you was able to lit the green led was quit a big progress - in my opinion.

I have a lot of hoverboards with gen2 hardware here at my warehouse. Will have a look at them.
But i do not remember these 2.2 layouts with the nice black housings.

Maybe you want to make a photo of the hoverboards plastic cover and wheels. I think the wheels have more little connectors.
So maybe the wheels also look different from the side. I do not remember.
Would be good if people could identify this gen2.2 hardware from the outside.

Greetings from 07926.

IerwinDE, 2023-05-04:
I think you should not leave this green LED stuff in the repo for now.
I just tried to flash the latest blinking version using my mapping and now bricked the second board.

RoboDurden, 2023-05-04:
thats bad news :,(
**Did you reconnect the 3.3V from stlink dongle to board before continue flashing ?**
Had you first test the HB firmware to verify that the second board is okay ?
And like my uploaded test firmware you only added

`			gpio_bit_write(LED_GREEN_PORT, LED_GREEN, (steerCounter%200) < 100);		`
and

```
#define LED_GREEN GPIO_PIN_14			// lerwinDE
#define LED_GREEN_PORT GPIOA			// lerwinDE
```

If that change indeed made the green led lit then i can not imagine how that active output line could damage the board.
Only if A14 is not connected to the tranistor of the green led.

IerwinDE, 2023-05-04:
Yes, shame on me. I ignored your advice and flashed again with 3.3v from ST-Link.
It worked a hundred times before, what are the chances of it not working 2 times in a row, right?
Welp.... yes, I know that was dumb.
The second board was okay before I flashed.
Only after it bricked I realized that I also have:
```
#define SELF_HOLD_PIN  GPIO_PIN_11		// lerwinDE
#define SELF_HOLD_PORT GPIOA				// lerwinDE
```
in my defines, so that might be the acural reason for the brick and not the LED stuff.

I will try to learn about ways to unbrick

IerwinDE, 2023-05-04:
OMG ITS ALIVE!!!!!
I successfully unbricked one of the boards by following the guide in this video:
https://www.youtube.com/watch?v=20fUiiUJerE&t=556s

So basically I just hit the full erase button in ST-Link Util while poking another GND Line on NRST of the chip.
I can now connect again using ST Link Util so I am confident that I will also be able to flash again

RoboDurden, 2023-05-04:
Hallelujah:-)
And I found a gen2.2 board here in my solar warehouse.
So I will join the party in the next two days :-))

[115 more replies]