RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #6: Individual board

BengkelDepan, 2023-04-04:
great job for your code. but i have idea with this hoverboard, can i copy your code and flashing to singel board instead dual board master and slave? i want to use this board control bldc motor individualy. will it works?

BengkelDepan, 2023-04-05:
do you have tutorial to extracting program file from my hoverboard split board with MCU GD32F130C6T6. i have STLINK V2 dongle and STCubeProg. then i want extracting internal program from this MCU and reprogram for other purpose. im newbie in this programing field but im interesting to learning.

RoboDurden, 2023-04-05:
Extraction may be possible and I had found some links years ago. But forgot were i posted them. Sorry.

BengkelDepan, 2023-04-06:
i found many file on folder hoverboardgigadevice. should i upload all file to my hoverboard?

closed 2023-04-06 by BengkelDepan

BengkelDepan, 2023-04-10:
which code should i delete if i want this code for singel board only, i dont want masterslave communication i want only singel boar control. if i load master only, like you said, the board will read slave data like code below.
void SendSlave(int16_t pwmSlave, FlagStatus enable, FlagStatus shutoff, FlagStatus chargeState, uint8_t identifier, int16_t value);

RoboDurden, 2023-04-10:
First get the master_test binary to work.
Then try to control from Arduino with master binary.
And finally if you wish you can play with removing such no longer needed functions.
But you can leave them as they should do no harm.
Only when you want to reuse that master-slave rx and tx pins for something else would you indeed need to remove that old functionality.