RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #8: GD32F103 on gen 1 main board

Clarepz, 2023-04-21:
The board looks very close to the original one in NiklasFauth first hack but it as a GD32 instead of the STM32. Wich board layout do I need to choose ? Thank you
[image]
[image]

RoboDurden, 2023-04-21:
This is a gen 1 board. The gd32f103 is compatible to the stm32f103.
Choose the eferu foc firmware which is way better then the original Niklas Firmware.

closed 2023-04-21 by RoboDurden

Clarepz, 2023-04-21:
Thankyou I was confused