RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #10: Layout compatibility for a GD32F130C6T6 master/slave board

yaydogan, 2023-04-27:
Hi,

I have a master/slave board with GD32F130C6T6 chipset. I would like to flash the firmware and control it using a remote control receiver.  My board looks different and was wondering how I can confirm if I can flash with this firmware. Attached are pictures of the boards and the chip.

Thanks!
[image]
[image]
[image]

RoboDurden, 2023-04-27:
Please first check the images of the different layouts already known on my repo home page.

RoboDurden, 2026-01-16:
Thanks for the four pictures. Now i will asign a new board number to it and add it to our wiki interactive board finder.
You can already begin with https://youtu.be/VaCOnroVCrQ "Split Hoverboard Gen2.x : NowAutodetect over ST-Link :-)"