RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #12: Other Board

mistljo, 2023-08-22:
Does anyone know this board type?

[image]
[image]
[image]

RoboDurden, 2023-08-23:
If is not one of my layout 2.x then it is probably new to us here.
Nice clean design.
Should ne no big deal to at least trace the three hall pins with a diode tester and a needle.

I can not read the MCU type in the chip.
If it is only the 32kb version of the gd32f130 then our simpleFOC port that Candas1 is close to complete will probably not run on it.
But mostly the 64 kB gd32f130x8 and not x6 is used.
The 6 MOSFET output pins very likely are like all 2.x layouts because of hardware bldc pwm feature..

Greetings from sunny Germany :-)

mistljo, 2023-08-24:
It's a gd32f130 cbt6 MCU.

RoboDurden, 2023-08-25:
it is a 48 pin  mcu on board so it must be gd32f130K6 or K8. Please look more carefully.
I guess the board can be purchased here: https://id.aliexpress.com/item/1005004149419180.html

Have assigned the layout 2.6 to it: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/Schematics_2.6

Please tell more about your project so i can see if i would like to help you - or not.

metalinspired, 2024-03-03:
Hi.

Here are two more images on which part numbers are easily read.
[image]
[image]

Since I'm more interested in using hoverboard motors as wind turbine generators I can send you these boards if you are interested.

RoboDurden, 2024-03-03:
Watch my YouTube video https://youtu.be/D_PfKFIc_kc

and https://youtu.be/33x2NgaMOTA

You can use the control board as active rectifier as well as a step up charger..

AWT or VWT ?
What battery voltage ?

But if postage to Germany is cheap you are welcome to donate them :-)

metalinspired, 2024-03-03:
Well I'll be damn, you already got it all set up :)
Very interesting.
I was thinking of doing HAWT.
I'm also in Germany so postage would be no issue.
However, I'm having second thoughts now that you pointed out that I can reuse these boards :P
But tell you what, since I got another hoverboard lying around which should be identical and if I do not fry these while tinkering I'll send you one pair.
I got a scooter battery which should be 36V which I was thinking of reusing.
Not even sure if that is plausible here or would I be better off with something like car battery.
This whole subject is quite new to me.
I know the principles behind it as I was tinkering with electric motors since early age but I am no engineer.
Now to get my self a st-link and dig a bit through your code :)

RoboDurden, 2024-03-03:
These guys also started with a bldc controller (VESC) and a skateboard bldc hub motor : https://www.kickstarter.com/projects/kitex/wind-catcher-portable-power-generating-wind-turbine

I would be more happy if you continue you wind generator than sending me the hardware.

As hoverboard motor is only 14 rpm/V, it  is possible to generate 36 Volt with a wind generator diameter harvesting 300 Watt at 7 m/s: https://rechneronline.de/windkraft/

If tsr  tup speed ratio would be 5, that would be 50 m/s 
With a 2m diameter wind generator, that would 50/(2 pi) = 8 revs and 480 rpm.
480 / 14 = 34,3 Volt.

RoboDurden, 2024-03-03:
And I would be very excited if you would test this propose of mine: https://youtu.be/fl9UnHpHE5Y