RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #14: Motor overheating

DaviB45, 2023-09-01:
Hi there!

I noticed that the original fw lasted quite a bit longer on my v2.0 board with the same battery and the motors weren't getting so hot. (Before the conversion I got around 10km range, now it's got around 3). What could be the issue and can it be solved? The same thing happened to me before (after flashing the hack fw) with a different kind of board (1 main esc and 2 slave gyro boards type of hoverboard).

RoboDurden, 2023-09-01:
Interesting. So you could drive 10 km with a hovercart conversion like 
[image]
but only 3km with the same setup but the gen2.0 firmware ?

If you have this test setup for a gen1 board, you should test the EFeru FOC  firmware which is more energy efficient and more silent.

@Candas1 might also have completed the SimpleFOC port for the gen2.0 board

https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/6#issuecomment-1694702246

I am eager to test it but still need to drive three days with my solar camper to get home.

Would be great if you have PlatformIO installed and would also like to test it.

I think you have to compile the dev branches of his repos.

But you will not be able to triple your range with a FOC firmware.
So I do not really know what to say to your findings.
I for sure would love to triple my solar camper range. But I do not think that this is possible.
Yest, my motors also get hot.
But 20% heat in the motos is to be expected and that would be about 50 Watt per motor. So motors have to get hot..
Please say more about your conversion hovercart.

And if you successfully flashed my gen 2.x firmware recently, please help here

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/9

Candas1, 2023-09-01:
I wouldn't advice anyone else using what I developed for now.
I just developed the drivers for gd32, and I am trying to use simpleFOC library in the right way for this controllers.
But it's not a firmware a user can just flash and use.

This new development will definitely be more efficient then gen 2.X with the Sinusoidal commutation, extrapolation, FOC, lag compensation.
But it's still hard to compare with the stock firmware which was developped for balancing. 
I think the hacks give you more power/acceleration than stock firmware.

RoboDurden, 2023-09-18:
I still doubt the tripple range with original firmware.

Yet @WizzardDr found a misconfiguration in setup.c:8 

```
// Internal constants
const int16_t pwm_res = 72000000 / 2 / PWM_FREQ; // = 2250
```
which leads to a full pwm duty cycle for +-1125 and not +-1000
Have corrected this

```
//----------------------------------------------------------------------------
// Set pwm -1000 to 1000
//----------------------------------------------------------------------------
void SetPWM(int16_t setPwm)
{
	//	bldc_inputFilterPwm = CLAMP(setPwm, -1000, 1000);	// thanks to WizzardDr, bldc.c: pwm_res = 72000000 / 2 / PWM_FREQ; == 2250 and not 2000
	bldc_inputFilterPwm = CLAMP(1.125*setPwm, -1125, 1125);
}
```

Have not done speed and efficiency tests but the motor is hearingly spinning faster now with  main.c:332

```
		#ifdef TEST_SPEED
			speed = 20 * (ABS((	((int32_t)steerCounter+100) % 400) - 200) - 100);
			speed = CLAMP(speed , -1000, 1000);
```

So a 11.25% speed increase.