RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #16: lLayout 2.7 = GD32E230C8T6

RoboDurden, 2023-09-07:
thanks to @WizzardDr who added this new layout which needs a new Keil compiler target: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/pull/15

As i am not really experienced at all with this Keil platform, help from anybody would be appreciated :-)

I see that this board has three mosfet gate driver chips instead of analog gate drivers like the original 2.0.
Be aware that this firmware here was written for the 2.0 layout where the low side mosfets are turned on with a LOW signal !!
With layouts having gate driver chips however, the on-signal might be HIGH for both high-side and low-side mosfets.
If this is not correctly applied, all three half-bridges will short cut and burn !!

When testing a new layout it is mandatory to use a CC constant current power supply with only 1-2 A max !!!

WizzardDr, 2023-09-07:
I noticed when I was tracing the connections that the Hi and Lo sides of the driver connect to `TIMER0_CH2_ON` `TIMER0_CH2` respectively whereas the other designs had Lo to _ON. I double and triple checked since I expected it to be the same. It may be  that the on and off timing also needs inverting because of this. I will do some testing and look at how the PWM timers are setup to drive the outputs once I get it to compile.

RoboDurden, 2023-09-07:
@Candas1 has begun with a nice spreadsheet of the pin connections for various board layouts: https://docs.google.com/spreadsheets/d/15msbDAIMxC2rIkq8Au8vf82ub1qEaS67Lc1cFb86Jpc/edit#gid=0

RoboDurden, 2023-09-07:
I think that all `#include "gd32f1x0.h"` lines can be removed from the `Inc/*.h` files.
But for the `Src/*.c` file i guess we have to add a `#ifdef TARGETXY #ELSE #ENDIF` to choose between `#include "gd32f1x0.h"` and `#include "gd32e2x0.h"`
I might know how to do this with PlatformIO, but not with Keil.

RoboDurden, 2023-09-07:
Okay, i have replace all `#include "gd32f1x0.h"` with `#include "../Inc/target.h"` in all *.c files and removed all `#include "gd32f1x0.h"` in the *.h files.

`target.h` simply is

```
#ifdef GD32E230
	#include "gd32e230.h"
#else
	#include "gd32f1x0.h"
#endif
```

This compiles successfully for the f130 target.
Still  need to find the correct `#include "gd32e230.h"`

Will publish my code now.

RoboDurden, 2023-09-07:
#include "gd32e23x.h" is the correct wording. Now it compiles but i get 10 compiler errors.
But these should not pose big problems:

```
Src/setup.c(63): error: use of undeclared identifier 'NVIC_PRIGROUP_PRE4_SUB0'
        nvic_priority_group_set(NVIC_PRIGROUP_PRE4_SUB0);
                                ^
Src/setup.c(117): error: too many arguments to function call, expected 2, have 3
        nvic_irq_enable(TIMER13_IRQn, 0, 0);
        ~~~~~~~~~~~~~~~                  ^
C:/Users/PAN CF-LX6/AppData/Local/Arm/Packs/GigaDevice/GD32E23x_DFP/1.0.2/Device/Firmware/Peripherals/inc\gd32e23x_misc.h(69): note: 'nvic_irq_enable' declared here
void nvic_irq_enable(uint8_t nvic_irq, uint8_t nvic_irq_priority);
     ^
Src/setup.c(327): error: too many arguments to function call, expected 2, have 3
        nvic_irq_enable(TIMER0_BRK_UP_TRG_COM_IRQn, 0, 0);
        ~~~~~~~~~~~~~~~                                ^
C:/Users/PAN CF-LX6/AppData/Local/Arm/Packs/GigaDevice/GD32E23x_DFP/1.0.2/Device/Firmware/Peripherals/inc\gd32e23x_misc.h(69): note: 'nvic_irq_enable' declared here
void nvic_irq_enable(uint8_t nvic_irq, uint8_t nvic_irq_priority);
     ^
Src/setup.c(347): error: too many arguments to function call, expected 2, have 3
        nvic_irq_enable(DMA_Channel0_IRQn, 1, 0);
        ~~~~~~~~~~~~~~~                       ^
C:/Users/PAN CF-LX6/AppData/Local/Arm/Packs/GigaDevice/GD32E23x_DFP/1.0.2/Device/Firmware/Peripherals/inc\gd32e23x_misc.h(69): note: 'nvic_irq_enable' declared here
void nvic_irq_enable(uint8_t nvic_irq, uint8_t nvic_irq_priority);
     ^
Src/setup.c(430): error: too many arguments to function call, expected 2, have 3
                nvic_irq_enable(DMA_Channel3_4_IRQn, 2, 0);
                ~~~~~~~~~~~~~~~                         ^
C:/Users/PAN CF-LX6/AppData/Local/Arm/Packs/GigaDevice/GD32E23x_DFP/1.0.2/Device/Firmware/Peripherals/inc\gd32e23x_misc.h(69): note: 'nvic_irq_enable' declared here
void nvic_irq_enable(uint8_t nvic_irq, uint8_t nvic_irq_priority);
     ^
Src/setup.c(489): error: too many arguments to function call, expected 2, have 3
        nvic_irq_enable(DMA_Channel1_2_IRQn, 2, 0);
        ~~~~~~~~~~~~~~~                         ^
C:/Users/PAN CF-LX6/AppData/Local/Arm/Packs/GigaDevice/GD32E23x_DFP/1.0.2/Device/Firmware/Peripherals/inc\gd32e23x_misc.h(69): note: 'nvic_irq_enable' declared here
void nvic_irq_enable(uint8_t nvic_irq, uint8_t nvic_irq_priority);
     ^
6 errors generated.
compiling setup.c...
compiling bldc.c...
compiling main.c...
compiling commsBluetooth.c...
compiling comms.c...
compiling it.c...
compiling led.c...
compiling commsSteering.c...
compiling commsMasterSlave.c...
RTE/Device/GD32E230C8/system_gd32e23x.c(202): error: use of undeclared identifier 'FMC_WS'
    FMC_WS = (FMC_WS & (~FMC_WS_WSCNT)) | WS_WSCNT_2;
    ^
RTE/Device/GD32E230C8/system_gd32e23x.c(202): error: use of undeclared identifier 'FMC_WS_WSCNT'
    FMC_WS = (FMC_WS & (~FMC_WS_WSCNT)) | WS_WSCNT_2;
                         ^
RTE/Device/GD32E230C8/system_gd32e23x.c(202): error: use of undeclared identifier 'FMC_WS'
    FMC_WS = (FMC_WS & (~FMC_WS_WSCNT)) | WS_WSCNT_2;
              ^
RTE/Device/GD32E230C8/system_gd32e23x.c(202): error: use of undeclared identifier 'WS_WSCNT_2'
    FMC_WS = (FMC_WS & (~FMC_WS_WSCNT)) | WS_WSCNT_2;
                                          ^
4 errors generated.
compiling system_gd32e23x.c...
".\Objects\Hoverboard.axf" - 10 Error(s), 0 Warning(s).
```

@WizzardDr please continue as i want to work on something else :-)

RoboDurden, 2023-09-07:
5 compiler errors with 

```
#ifdef GD32E230
	#include "gd32e23x.h"
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b);}
#else
	#include "gd32f1x0.h"
	
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b, c);}
#endif
```
:-)

RoboDurden, 2023-09-07:
Okay if i simple remove that strange line with //FMC_WS... this leads to a successful compilation:

```
#ifdef GD32E230
	#include "gd32e23x.h"
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b);}
	#define TARGET_nvic_priority_group_set(a)	// that function does not exist for this target = not needed ?
	#define TARGET_adc_vbat_disable()
#else
	#include "gd32f1x0.h"
	
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b, c);}
	#define TARGET_nvic_priority_group_set(a){nvic_priority_group_set(a);}
	#define TARGET_adc_vbat_disable(){adc_vbat_disable();}
#endif
```

@WizzardDr , you continue !

WizzardDr, 2023-09-07:
Well this is unfortunate, I didn't see your updates, I have wasted my time doing what you have just done, but I did it worse and slower lol. I looked into the FMC stuff and found "The FMC provides a security protection function to prevent illegal code/data access on theFlash memory." so it probably isn't important. Thanks for getting it to compile.

RoboDurden, 2023-09-07:
Only test the binary with a 1-2A CC power supply !!!
I have cleaned up your Schematics_2.7 a bit. Tell me if i deleted some nice images.

The master and slave boards seem to be different, slave having new room for the buzzer and bluetooth on board !

RoboDurden, 2023-09-07:
I am currently working to add your defines_2-7.h ! 22 errors at the moment..

RoboDurden, 2023-09-07:
Okay i added the defines_2-7.h and removed the compiler errors.
```

#ifdef GD32E230
	#define LAYOUT_2_7	// https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/16
#else
	#define LAYOUT_2_0	// https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2
	//#define LAYOUT_2_1	// https://github.com/krisstakos/Hoverboard-Firmware-Hack-Gen2.1
	//#define LAYOUT_2_2	// 2023/05/11 only MASTER and TEST_SPEED: motor is spinning but needs a push to startup :-/
	//#define LAYOUT_2_4	// NOT READY !!! https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/3
#endif
```

It seems to me that the gate driver chips expect a negated signal for low side mosfets. So maybe the 2.0 code will run.

[image]

But again, be careful !

RoboDurden, 2023-09-07:
If you do not find the CHARGE_STATE_PIN then you need to activate (remove //)

RoboDurden, 2023-09-07:
Okay i added 	`#ifdef CHARGE_STATE_PIN` to the only two references..
`Shift+Strg+F` is your friend !

WizzardDr, 2023-09-10:
@RoboDurden I do not understand how the pulse time is correct with the PWM running in mode 1. In mode 1 "When counting up, OxCPRE is low when the counter is smaller than TIMER0_CHxCV, and high otherwise.".

How does `timer_channel_output_pulse_value_config(TIMER_BLDC, TIMER_BLDC_CHANNEL_G, CLAMP(g + pwm_res / 2, 10, pwm_res-10));` dictate the PWM on time rather than PWM off time?

RoboDurden, 2023-09-11:
What was wrong with UCC1800_dcdc_chip.PDF ?
I did not program this firmware. I only added the different layouts and rewrote the serial communications.
I guess you are the better person to answer this pwm init code.
It will take me longer to understand that then it would take you. Candas1 i think would know the answer instantly, but i am happy that he is busy with the simpleFOC port, so i do not like to ask him.

Have you started testing the binary with a constant current power supply ???
If the defines_2-7.h is close to complete, you could give it a try.
All your pull requests are not really worth anything if you will not test the 2.7 layout at some time.
Before that, further pull requests do not really make the current gen2.x firmware better.

WizzardDr, 2023-09-11:
My problem at the moment is that https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-flash seems to have bricked the mcu... I connected it and it seamed to be unlocked, It read a 4kB image. Then I followed the step and now it just says the chip protection is enabled and it won't let me set the option bits to unlock it. STM32CubeProgrammer returns `  15:36:31 : Error: Expected value for STM32F1 Option Byte "RDP": 0xA5, found: 0x1
  15:36:31 : Error: Option Byte Programming failed`, ST-LINK Utility doesn't work either.

WizzardDr, 2023-09-11:
I lifted the Boot0 pin and used UART and got it to load the firmware. (I selected "Read Unprotect (MCU)" in STM32CubeProgrammer)

WizzardDr, 2023-09-11:
I have tested it with 20v (1a) with the battery cell count set to 5, but the  power doesn't seem to be latching ON.

RoboDurden, 2023-09-11:
add 

```
#define DEBUG_LED_PIN 	UPPER_LED_PIN
#define DEBUG_LED_PORT 	UPPER_LED_PORT
```

to your `defines_2-7.h` and understand how the `DEBUG_LedSet` macro should make the upper_led go on and off.
Then you can test if the code gets executed and were it stops. Or if it reaches the `while(1)` loop..

I never got the debug step by step mode working with this Keil ide.
Porting this firmware to platformIO would make debugging (also using the great StmStudio) much easier..

WizzardDr, 2023-09-11:
Yes, I guess I'll have to trace more of the pcb and find the led pins.

WizzardDr, 2023-09-14:
I've been trying to get it to work and have come to the realization that the main code doesn't seem to be running at all. I've blocked out the main code with `ifndef` and just have a small section to turn a pin on (plus some some bits to stop it complain about speed being an undefined symbol) ```
#define BLINKTEST
#ifdef BLINKTEST
#define BLINKTEST_PORT GPIOA
#define BLINKTEST_PIN GPIO_PIN_14
#include "gd32e23x.h"
#include "../Inc/defines.h"
int main (void){
	gpio_mode_set(BLINKTEST_PORT , GPIO_MODE_OUTPUT, GPIO_PUPD_NONE,BLINKTEST_PIN);	
	gpio_output_options_set(BLINKTEST_PORT, GPIO_OTYPE_PP, GPIO_OSPEED_10MHZ, BLINKTEST_PIN);
	gpio_bit_write(BLINKTEST_PORT, BLINKTEST_PIN, 0);
}
int32_t speed = 0; 												// global variable for speed.    -1000 to 1000
int32_t steer = 0; 												// global variable for steering. -1000 to 1000
FlagStatus beepsBackwards = SET;  			// global variable for beeps backwards
DataSlave oDataSlave;
#endif
#ifndef BLINKTEST

#define ARM_MATH_CM3
........
```
This should be working nonmatter what, right?

RoboDurden, 2023-09-14:
This is the code that works on my gen2.0 test setup:

```
#define BLINKTEST

#ifdef BLINKTEST

#include "../Inc/target.h"
#include "../Inc/config.h"
#include "../Inc/defines.h"

uint32_t iLoop = 0;
int main (void)
{
	rcu_periph_clock_enable(RCU_GPIOA);
	rcu_periph_clock_enable(RCU_GPIOB);
	
	gpio_mode_set(SELF_HOLD_PORT , GPIO_MODE_OUTPUT, GPIO_PUPD_NONE, SELF_HOLD_PIN);	
	gpio_output_options_set(SELF_HOLD_PORT, GPIO_OTYPE_PP, GPIO_OSPEED_10MHZ, SELF_HOLD_PIN);
	gpio_bit_write(SELF_HOLD_PORT, SELF_HOLD_PIN, SET);
	
	gpio_mode_set(LED_GREEN_PORT , GPIO_MODE_OUTPUT, GPIO_PUPD_NONE,LED_GREEN);	
	gpio_output_options_set(LED_GREEN_PORT, GPIO_OTYPE_PP, GPIO_OSPEED_10MHZ, LED_GREEN);
	while(1)
	{
		gpio_bit_write(LED_GREEN_PORT, LED_GREEN, (iLoop++ % 200000) < 100000 );
	}
}

int32_t speed = 0; 												// global variable for speed.    -1000 to 1000
int32_t steer = 0; 												// global variable for steering. -1000 to 1000
FlagStatus beepsBackwards = SET;  			// global variable for beeps backwards
DataSlave oDataSlave;

#else
...
#endif
```

WizzardDr, 2023-09-14:
That code didn't work either. I don't think the mcu is entering the main code. What option bytes are set on yours?

RoboDurden, 2023-09-15:
All Option bytes must be set in the St link utility. But if you can flash successfully then I think your code should run.
Do you have a 36v battery connected ?
Or better a cc power supply set to 26v and 2A.
Do not connect the 3.3v of the flash header.
The led needs the 15V on the led header.
Check this with a multimeter.

From which country are you ?

WizzardDr, 2023-09-15:
I have flashed it successfully through UART and verified the memory after a power cycle.

I was originally testing with 29v power but since trying to just get a blink program to work I have just had the board being powered by the FTDI adapter, on my desk with everything (motor, led, etc.) unplugged. I was using my mustimeter to test if the I/O pin was changing, it was not.

I have just tested your code with it connected to the LED's and powered by my PSU at 29v.
As with all the other 29v tests, when I hold the button one of the LED's lights up (I think this is just due to the way the transistor is held by the pull-up) and as soon as I let go the power turns off (the power MOSFET latch isn't being held by the MCU).

RoboDurden, 2023-09-15:
I was interested in the time zone country that you currently live.

I guess you have not yet verified the SELF_HOLD_PIN so it might be better to have a switch to replace the on off push button.

Maybe you succeed with step by step debugging in the Keil ide.

You are the first to flash via uart. Maybe you write to the wrong address.

Why not use the flash header and use the SWD protocol ?

Install platformIO and write a blink program from scratch for this MCU. Then you should be able to step over debug line by line.

My platformio simpleFOC firmware via i2c might be ready in a few days. So if have a running blink with PIO platformIO, you could continue from that and forget the gen2.x

Search the web for your MCU code and Keil..

RoboDurden, 2023-09-15:
i think to debug with Keil you have to compile with debug info:
[image]
But i never succeeded with that.

WizzardDr, 2023-09-15:
uVision hard crashes (instantly closes) when I click "Options for Target>Debug>Use: ST-Link Debugger>Settings" if I have the ST-Link V2 (china version) is connected to USB, or it crashes if I have that dialogue open then I plug in the ST-Link to USB.
So hardware debug isn't an option via the ST-Link

I tried using the Arduino CommunityGD32Cores board manager and blink sketch, but that uploaded then didn't run too.

RoboDurden, 2023-09-15:
no crashes here with st-link dongle connected (and motor powered)
[image]

WizzardDr, 2023-09-15:
I think is is the ST-LINK firmware. What firmware is your ST-LINK working with?

RoboDurden, 2023-09-15:
Version: V2.J40.S7
Type: STM32+STM8 Debugger

WizzardDr, 2023-09-15:
I don't have a firmware updater (link007) for that version. Do you?

RoboDurden, 2023-09-15:
i am watching a movie..
I think i flashed that firmware with STM32CubeProgrammer.

WizzardDr, 2023-09-15:
I might give up and do that too 😂😭. thanks for the help, I didn't know you could flash directly through cube, ill look into it. I'll leave you to your movie now. Thanks

WizzardDr, 2023-09-15:
[image]
I loaded Keil on my laptop and the st-link debug seems to work. I put a break point in main.c but it stops before there I think. The screenshot shows what happens after clicking start debug.

WizzardDr, 2023-09-16:
[image]
I think I have figured out how the debugger works. I think this is where it is getting stuck. What does it mean?

RoboDurden, 2023-09-16:
GD32E23x_User_Manual_EN_Rev1.5.pdf page 75: **High Speed Crystal Oscillator (HXTAL)**

_...
he HXTALSTB flag in control register 0, RCU_CTL0 indicates if the
high-speed external crystal oscillator is stable. **When the HXTAL is powered up, it will not be
released for use until this HXTALSTB bit is set by the hardware.** This specific delay period is
known as the oscillator “Start-up time”. As the HXTAL becomes stable, an interrupt will be
generated if the related interrupt enable bit HXTALSTBIE in the Interrupt register RCU_INT
is set. At this point the HXTAL clock can be used directly as the system clock source or the
PLL input clock.
..._

In the `do(){}while();` loop the mcu waits for `RCU_CTL0 & RCU_CTL0_HXTALSTB` to become `1`.
After `timeout++` has reached `timeout`, the loop is left. But as `RCU_CTL0 & RCU_CTL0_HXTALSTB` is still `0`, the mcu hangs with `while(1){}` :-(

You might have a hardware fault on your split board and try your second board.

You might try the other oscillator 72 Mhz config:

```
47: #define __SYSTEM_CLOCK_72M_PLL_HXTAL         (uint32_t)(72000000)
48: //#define __SYSTEM_CLOCK_72M_PLL_IRC8M_DIV2    (uint32_t)(72000000)
```

Always hit **F12** on some code you do not understand !

Would be nice if you tell me the steps needed to get debug working on Keil. I never succeded :-(

WizzardDr, 2023-09-16:
Thanks, I will try that.

Regarding how I got debug working:

As you know the st-link crashed Keil on my Desktop. I suspect this is due to having ST stuff on my desktop before installing Keil, I have read online regEdit fixes and incorrect DLLs. But to get it to work I installed it on my laptop.
[image]
This is the OS info. AFAIK there wasn't any ST or Keil stuff on my laptop before.

First I installed Keil, from here: https://www.keil.com/demo/eval/arm.htm 
Install (MDK538a.EXE)
[image]
This is the Keil uVision version info. Drivers were installed during the install.
[image]
This is the ST-Link Driver info.
On my PC I did manage to put the same firmware (Version: V2.J40.S7 Type: STM32+STM8 Debugger) on my ST-LinkV2.

With the ST-LinkV2 plugged in I opened Keil and selected the ST-Link under Options for Target.
[image]
I also enabled "Debug Information" like you suggested.
[image]

Then I cleaned the output and built the firmware.
[image]

Then I loaded the firmware. (after later installing STCubeProgrammer I did have to unplug and replug the usb to get it to let go of the ST-Link so Keil could discover it.) Then I clicked start debugger on the right.
[image]

When the debugger started I think the MCU is running.
[image]
So I click the Stop to see the trace of where it is in the code.
[image]

RoboDurden, 2023-09-16:
Can you set a breakpoint a few lines prior to that while(1){} loop and then step-over debug line by line ?
Great images. When i succeed with your workflow, i will add them to a how-to debug the Gen2.x code :-)

WizzardDr, 2023-09-16:
Yes I could.

But I now I have changed from `__SYSTEM_CLOCK_72M_PLL_HXTAL` to `__SYSTEM_CLOCK_72M_PLL_IRC8M_DIV2` I can't get it to do so, it is just stopping on a memory address rather than a line of code, although I did get it to step one time. The response to my button clicking doesn't seem entirely consistent, though that my just be my lack of knowledge on how to use the debugger.

RoboDurden, 2023-09-16:
This other clock is just try and error.
Did your hoverboard work before you took it appart or was it already broken ? If so, then test the other board.
You should also do an extensive google search for everything connected to the GD32E230.
Maybe watch this: https://www.youtube.com/watch?app=desktop&v=XeYkDJJYmbs (i haven't - again watching a movie).

From which country to currently post ?

WizzardDr, 2023-09-16:
I think it worked, but I don't know for sure because it got it with a dead battery. I will watch that video thanks.
From the UK but that isn't to say the UK's time zone 😂

WizzardDr, 2023-09-16:
I think the reason my debug works on my laptop and not on my desktop is because there are other `STLinkUSBDriver.dll` files on my desktop (in locations referenced by windows environment variables). I did a scan with WizTree and searched `STLinkUSBDriver.dll` on both my computers. Maybe remove all of the software with non Keil instances of `STLinkUSBDriver.dll`.

I think the change to `__SYSTEM_CLOCK_72M_PLL_IRC8M_DIV2` did work, though now it is getting stuck on `nvic_vector_table_set(NVIC_VECTTAB_FLASH,VECT_TAB_OFFSET);` something to do with the memory address the linker assigned for the program flash. I think I am slowly getting there...

RoboDurden, 2023-09-16:
maybe you can learn from these GD32E230 Keil projects:

https://github.com/Quenii/GD32F2x/tree/master/demo/GD32E230_Demo_Suites_V1.0.1/GD32E230C_EVAL_Demo_Suites/Projects/01_GPIO_Runing_Led
https://github.com/RT-Thread/rt-thread/tree/master/bsp/gd32e230k-start

22:40 here in Germany, good night - and good luck.

WizzardDr, 2023-09-16:
I think it is a similar thing. I have downloaded GD32E23x Firmware Library from that video. With the change to `__SYSTEM_CLOCK_72M_PLL_IRC8M_DIV2` it works and the debug gets to main()!!!!!!

Now if I can figure out the target differences I should be able to get the hoverboard firmware to work!!!

WizzardDr, 2023-09-16:
YAY!!!!
I'm now in `main()`

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/16#issuecomment-1709849075 🤔

`nvic_vector_table_set(NVIC_VECTTAB_FLASH,VECT_TAB_OFFSET);` 🤔 It can't access the memory if it is protected... 

[image]

This ain't the solution to undefined `FMC_WS`... This is:
 
[image]

The memory was protected 🤣

The self latch is working.... now to test the main code 🤞

WizzardDr, 2023-09-16:
Motor noise and some holding torque... high current draw without spinning though... getting there!

WizzardDr, 2023-09-16:
YAY It works!!! I had to connect the yellow motor wire to `BH_PIN`, `BL_PIN` and the yellow motor wire to `YH_PIN`, `YL_PIN`

WizzardDr, 2023-09-17:
I've been looking through the code. There is `#ifdef MASTER` in `main()` that covers most of the code. where is the slave's code?

RoboDurden, 2023-09-17:
8:15, Good morning.
It is in commsmasterslave.c.
A slave is a slave :-/

Great job you are doing !!!

WizzardDr, 2023-09-17:
Everything in the `while(1)` in `main()` is in the `#ifdef MASTER`. How does it get called?

WizzardDr, 2023-09-17:
Oh, from commsmasterslave.c I think I have found it. Is it all via hardware interrupt DMA13?

RoboDurden, 2023-09-17:
Yes, `it.c` i think is where all the magic begin. `main()` is called by the when the hardware is ready.

WizzardDr, 2023-09-17:
I think the problem that I have with setting the option bytes via SWD is that the ST-Link programs try to set the lock byte levels 0, 1 and 2 with 0xAA, 0xBB and 0xCC whereas the  GD32E32x needs 0xA5 for no protection.
Due yo any value except 0xA5 or 0xCC: Protection level low and 0xCC: Protection level high the ST-Link Utility always locks it.

There may be a way to set 0xA5 via SWD using a different method, maybe with OpenOCD (I don't have a Linux OS to test with).

For now I am going to do the same as I did with the master to program via UART.

I lifted the Boot0 Pin so it isn't tied to GND, and added a 10k pull-down resistor.

RoboDurden, 2023-09-17:
this looks dangerous. Is that neccessary to uplaod via uart ?

RoboDurden, 2023-09-17:
I do not have any ideas for this byte unlock problem. I also do not fully understand what you want to say. Maybe STM32CubeProgrammer has more options to set the bytes ?

WizzardDr, 2023-09-17:
The "Read Unprotect (MCU)" in the STM32CubeProgrammer is the only way I have managed to reset the lock byte so far, it is just a workaround, the correct solution is to have the SWS programming software set the byte correctly.

I'm sure if I had a GD-Link their software is setup to set the byte correctly, or maybe OpenOCD  can do it via the ST-Link but I don't know.

The problem is the Software thinks the chip is an STM32F1 variant since there isn't a STM32E23. So it tries to set the byte as though tis is an STM32F1.

The required values for unlocked and locked are different.

The GD32E230
[image]

The STM32F1
[image]

So when the value is set to `00` it think it failed to unlock the chip but in fact it has locked the chip since the value isn't `0xA5` (The GD32E23 unlock value).

RoboDurden, 2023-09-17:
Sorry i am to tired now to understand. I think i will not be able to help you.
When you finally have found the best possible ways for other user to flash such a 2.7 board, please make a step by step tutorial.

WizzardDr, 2023-09-17:
ATM the easiest way for other user would be to just buy a GD-Link and use "GD-Link Programmer.exe" which is very similar to "STM32 ST-LINK Utility.exe".

If I find a way to use an ST-Link without first unlocking via UART I will share.

RoboDurden, 2023-09-17:
links not working for me. page not found.
23:53 here, good night.

[71 more replies]