RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #20: version 2.3

mateuszfcg, 2023-09-17:
Hello. Has anyone run the disc version 2.3? At first I had a problem with uploading any batch, but I managed to do it for a while. now I have trouble starting it. when I connect the power supply, the engine only starts slightly and holds it in one position. I cannot establish communication with Arduino. I tried on different GD pins. please help. Does anyone have the file already configured for this disc? Regards

RoboDurden, 2023-09-17:
This is a bad translation engine, surely not Google ?

I guess it was me who did add the 2.3 layout so i might have such a board laying aorund somewhere here.
But no-one yet has begun with a defines-2_3.h file.
If you start tracing the 3 hall sensor and 6 mosfet pins, i will support you. If i find my 2.3 boards i might build just another test setup.
But you should do some work yourself first.

DerPinguin77, 2023-09-18:
Hi im new here but i also have one of these Hoverboards.
I would love to help with finding tracing those hall sensor and mosfet pins. But im fairly inexperienced so i cant really help with programming. I have flashed one Hoverboard before but it had the more common single motherboards. 
I also have some other questions:
Are there plans in integrating something like a PWM control and would it be possible to integrate in the version for this board?
Can u use both boards standalone by just flashing both boards with the Master version?

RoboDurden, 2023-09-18:
If you have a new layout, please post some hi-res photos here.
I might be happy to finally program an pin-auto-detect firmware that creates a new defines_2-xy.h file.
No, i will not add further control methods beside the existing uart. 
But i am currently working on an i2c control for our new simpleFOC firmware: https://github.com/Candas1/Split_Hoverboard_SimpleFOC/issues/11
When this is ready, people might add new Pilot classes..

Yes, slave board can be flashed with the master firmware. But it needs the CHECK_BUTTON define to be disabled whenthe  slave board ist dircetly powered and has no on button:

`	#define CHECK_BUTTON		// disable = add '//' if you use a slave board as master
`

DerPinguin77, 2023-09-18:
Im currently working on the pinout. Is but sadly the firmware woult be pretty useless to me wothout einter ADC or PWM control. I dont know much about Githubs so i will post my results here when im done.

RoboDurden, 2023-09-18:
ADC would not be a big problem. 
If one of the two uart rx/tx pins can work as an anlaog input (like the long cable of the gen1 board), the reading the adc value and turn it into +-1000 will only take a few lines.

PWM would need some few timing calculations to translate the high/low times to a pwm ratio..

If your board is not the layout 2.3 that this issue is dedicated to, please create your own issue and attach high res photos there so i can assign a new layout number.
You can attach high-res fotos to this issue and i will

DerPinguin77, 2023-09-18:
This is the board i have. I used your original photo and drew in the connections I measured. Hope this is enough. If not please tell me what you need and i will try to find the connections. It would be cool if you could add the few lines you need for ADC control because I dont have the knowlege to do that.
[image]

RoboDurden, 2023-09-18:
Would be nice if you could uplad an aligned fornt/back image like [2.7 front-back](https://raw.githubusercontent.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/main/Schematics_2.7/hoverboard_gen2-7_front-back.jpg)
Then some led pins etc can be traced on the photo.

mateuszfcg, 2023-09-18:
`#ifdef SLAVE	// this layout has buzzer on the slave board !
	#define BUZZER
#endif

#define TODO_PORT GPIOA				// this should be a pin that does no harm if input or output
#define TODO_PIN	GPIO_PIN_15	// B15 is not accessibla on the smaller QFN32 32 pin MCU version

#define DEBUG_LED_PIN 	GPIO_PIN_1	// lerwinDE: blue led panel
#define DEBUG_LED_PORT 	GPIOF				// lerwinDE: blue led panel

// LED defines
#define LED_GREEN 			TODO_PIN	// lerwinDE: GPIO_PIN_14 - in conflict with flash pins DIO and CLK !!!
#define LED_GREEN_PORT 	TODO_PORT	// lerwinDE: GPIOA
#define LED_ORANGE 			TODO_PIN	// TODO
#define LED_ORANGE_PORT	TODO_PORT	// TODO
#define LED_RED 				TODO_PIN		// lerwinDE: GPIO_PIN_13 - in conflict with flash pins DIO and CLK !!!
#define LED_RED_PORT 		TODO_PORT		// lerwinDE: GPIOA

#define UPPER_LED_PIN 	TODO_PIN	// TODO
#define UPPER_LED_PORT 	TODO_PORT	// TODO
#define LOWER_LED_PIN 	TODO_PIN	// TODO
#define LOWER_LED_PORT 	TODO_PORT	// TODO

// Mosfet output
// seems to be an ordinary LED output ?
// led.c:91	gpio_bit_write(MOSFET_OUT_PORT, MOSFET_OUT_PIN, counter_Blue >= setValue_Blue ? RESET : SET); 
#define MOSFET_OUT_PIN TODO_PIN		// TODO
#define MOSFET_OUT_PORT TODO_PORT	// TODO

// Brushless Control DC (BLDC) defines
#define TIMER_BLDC_PULLUP	GPIO_PUPD_PULLUP	// robo, based on Herleybob:defines.h
// Channel G
#define RCU_TIMER_BLDC RCU_TIMER0
#define TIMER_BLDC TIMER0
#define TIMER_BLDC_CHANNEL_G TIMER_CH_2
#define TIMER_BLDC_GH_PIN GPIO_PIN_8		// GPIO_PIN_8	robo, based on Herleybob:defines.h
#define TIMER_BLDC_GH_PORT GPIOA				// GPIOA robo, based on Herleybob:defines.h
#define TIMER_BLDC_GL_PIN GPIO_PIN_13		// robo, based on Herleybob:defines.h
#define TIMER_BLDC_GL_PORT GPIOB				// robo, based on Herleybob:defines.h
// Channel B
#define TIMER_BLDC_CHANNEL_B TIMER_CH_1
#define TIMER_BLDC_BH_PIN GPIO_PIN_9		// robo, based on Herleybob:defines.h
#define TIMER_BLDC_BH_PORT GPIOA				// robo, based on Herleybob:defines.h
#define TIMER_BLDC_BL_PIN GPIO_PIN_14		// robo, based on Herleybob:defines.h
#define TIMER_BLDC_BL_PORT GPIOB				// robo, based on Herleybob:defines.h
// Channel Y
#define TIMER_BLDC_CHANNEL_Y TIMER_CH_0
#define TIMER_BLDC_YH_PIN GPIO_PIN_10		// robo, based on Herleybob:defines.h
#define TIMER_BLDC_YH_PORT GPIOA				// robo, based on Herleybob:defines.h
#define TIMER_BLDC_YL_PIN GPIO_PIN_15		// robo, based on Herleybob:defines.h
#define TIMER_BLDC_YL_PORT GPIOB				// robo, based on Herleybob:defines.h

// Timer BLDC short circuit emergency shutoff define
// Is initialized here but never used somewhere else in code.
// setup.c:176	gpio_mode_set(TIMER_BLDC_EMERGENCY_SHUTDOWN_PORT , GPIO_MODE_AF, GPIO_PUPD_NONE, TIMER_BLDC_EMERGENCY_SHUTDOWN_PIN);  
#define TIMER_BLDC_EMERGENCY_SHUTDOWN_PIN TODO_PIN	// TODO
#define TIMER_BLDC_EMERGENCY_SHUTDOWN_PORT TODO_PORT		// TODO

// Hall sensor defines
#define HALL_A_PIN GPIO_PIN_1	// robo, based on Herleybob:defines.h A = U ?
#define HALL_A_PORT GPIOA				// robo, based on Herleybob:defines.h A =  U ?
#define HALL_B_PIN GPIO_PIN_2	// robo, based on Herleybob:defines.h B = V ?
#define HALL_B_PORT GPIOA				// robo, based on Herleybob:defines.h B = V ?
#define HALL_C_PIN GPIO_PIN_0		// robo, based on Herleybob:defines.h C = W ?
#define HALL_C_PORT GPIOA				// robo, based on Herleybob:defines.h C = W ?

// Usart master slave defines
#define USART_MASTERSLAVE USART1	// robo no second uart port yet.
#ifdef USART_MASTERSLAVE
	#define USART_MASTERSLAVE_TX_PIN GPIO_PIN_6		// lerwinDE :-)
[... 42 lines cut ...]
#ifdef MASTER

	// Charge state defines
	// This seems to be a digital input that hast to be high in order to enable the motors. 
	// main.c:381: chargeStateLowActive = gpio_input_bit_get(CHARGE_STATE_PORT, CHARGE_STATE_PIN);
	// If not found it should be okay to simply comment this line because chargeStateLowActive in initialised as set = true
	#define CHARGE_STATE_PIN GPIO_PIN_0		// TODO
	#define CHARGE_STATE_PORT GPIOF				// TODO
#endif

// Debug pin defines - seems to be never used in code.
#define DEBUG_PIN TODO_PIN	// TODO
#define DEBUG_PORT TODO_PORT			// TODO

`

Above is the code of my configuration but not everything is ok
-UART does not work
-Motor does not run stably (sound like skipping steps in a stepper motor but it is rotating) mosfets are heating up 
-Starting the board requires a jumper all the time
-buzzer works as soon as I change to SLAVE in config

any ideas ? 

Sorry for the translation but I have a hard time with languages :)

RoboDurden, 2023-09-18:
**Always use a CC constant current power supply with max 2A when testing new boards !!!!!!!!!**

mateuszfcg, 2023-09-18:
I powered it initially via inverters to limit the current.

DerPinguin77, 2023-09-18:
These are the best images i could get in the dark. I acidentally photographed the Slave board and i forgot to remove the sticker on the back. If its not good enoungh i can try again tomorrow
[image]

RoboDurden, 2023-09-18:
Thank you. I mirrored the backside vertically and added it to the 2.3 folder.
The 5 led pins should be no problem:
[image]

21:15 here in Germany, i go to bed.

RoboDurden, 2023-09-19:
@DerPinguin77 i do not see a seconde uart header on this layout :-/
The primary uart header normally is needed to connect master with slave.
Without at least one additional input pin for speed, you would need an additional Arduino or Esp anyway to calculate speedL and speedR and send them via the only UART to the two boards.

In the bottom left i see a 4 pin header with `GND | DATA |  5V`. But on the photo i see that the two DATA pins are connected.
So at best one input pin.

The 4 pin header to the top right i guess is the charging connector ?
The 2 pin header to the right is the on/off button ? With the slave, this header would not have any function as it is powered directly over the uart cable ? Then the slave board might have that DATA pin and the BUTTON pin to receive speed and steer data.

I stick to my understanding that adding a cheap ESP S2 Mini with bluetooth and wlan is always the better choice.

RoboDurden, 2023-09-19:
@mateuszfcg i have added your `defines_2-3.h` to my repo and updated the `defines.h` and `config.h`.
If you want to make changes to the code (main.c , setup.c , bldc.c , etc.), please fork this repo by clicking on
[image]
Then you can make change to your own copy of this repo and start a "pull request" so i can incooperate your changes with a single click :-)

@DerPinguin77 if you have a constant current power supply it would be great if you start testing the new `defines_2-3.h`

I will check the defines later this day if they confirm with your drawing.

P.S. the two photo sensors (light barriers) on the backside could be used as digital inputs. The last board i tested allowed one input to be used as analog input.

RoboDurden, 2023-09-19:
okay i updated https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Inc/defines_2-3.h

Am not sure if the ordering is right:

```
// Hall sensor defines
// mateuszfcg tested with PA1,PA2,PA0 
#define HALL_A_PIN	GPIO_PIN_1
#define HALL_A_PORT	GPIOA
#define HALL_B_PIN	GPIO_PIN_2
#define HALL_B_PORT	GPIOA
#define HALL_C_PIN	GPIO_PIN_0
#define HALL_C_PORT	GPIOA
```

I added the led pins by looking on the photo:

```
// LED defines, colors probably mismatch !
#define LED_GREEN 			GPIO_PIN_15
#define LED_GREEN_PORT 	GPIOA
#define LED_ORANGE 			GPIO_PIN_14
#define LED_ORANGE_PORT	GPIOA
#define LED_RED 				GPIO_PIN_3
#define LED_RED_PORT 		GPIOB

#define UPPER_LED_PIN 	GPIO_PIN_4
#define UPPER_LED_PORT 	GPIOB
#define LOWER_LED_PIN 	GPIO_PIN_5
#define LOWER_LED_PORT 	GPIOB

#define DEBUG_LED_PIN 	LED_RED
#define DEBUG_LED_PORT 	LED_RED_PORT
```

I disabled both uart ports. 
```
//#define USART_MASTERSLAVE USART1	// robo no second uart port yet.
//#define USART_STEER_COM USART0					
```
Please someone check the rx tx pins of the white uart header below the GD32 MCU :-)

DerPinguin77, 2023-09-19:
Hi, came home from work just now. I think it would not be a problem if each of those boards only had one free input pin. It isnt as nice but you could just use the one pin of each board to control the motor connected to it. I can understand that for robotic projects the esp uart thing is much more practical, but I would prefer to use the board standalone without any extra stuff, cause I just need basic speed control.
I can start testing later this day but it could take a while until i figure out how to flash and compile all of this. Do i need the esp for testing or is there some kind of test mode?
I will continue to test the header you have asked for and i will also test the pins od the infrared gates.
Also i have another question. Will there be a way to calibrate the end points of the adc? On the other firmware i have used, you could press the powerbutton for a long time and you woult enter calibration mode. There you coult mode the pot to the max and min positions.
I am also from Germany btw. :)

RoboDurden, 2023-09-19:
yes i think that is the correct download.
You may also need st-link-utility to unlock the MCU. 
It looks like the red DATA pin is either pulled to 3.3V or GND. Then it is not that useful for adc
Any idea what the yellow pin you added is for ? Might be analog read of battery voltage.
To later also make use of the coming SimpleFOC firmware, you also need the two adc outputs of this dual op-amp:

[image]

RoboDurden, 2023-09-19:
yes, but i am not sure if Keil will be able to unlock the MCU. 
Watch my youtube tutorial for that firmware.
If you only want to control one motor with a potentiometer, one of the uart pins might do.
But please confirm the

```
	#define USART_MASTERSLAVE_TX_PIN	GPIO_PIN_6
	#define USART_MASTERSLAVE_TX_PORT	GPIOB
	#define USART_MASTERSLAVE_RX_PIN	GPIO_PIN_7
	#define USART_MASTERSLAVE_RX_PORT	GPIOB
```

Then i will look in the MCU user manual to see if one of these pins can also do adc.

mateuszfcg, 2023-09-19:
@DerPinguin77 1. unlock the GD32 via ST-LINK Utility (this will remove the preloaded firmware)
2. Start keil and upload the firmware directly from keil.
If you don't want to load (e.g. "no target conection" message pops up) you have to short NRST pin to GND and after approx. 1sec after pressing DOWNLOAD let go. 
The second option instead of manually connecting to GND is to add a transistor to ST-LINK to the RST pin.
Resistor R72 closer to the white connector. 
Good luck

And getting back to the topic of the UART the white connector is:
TX=PB6
RX=PB7

So far I have established a connection to the Arduino with these pins.

I still have a problem with VBat and IAMP. No matter what voltage I put on the power supply it is still Iamp=0.42, IVolt 36.0 in the serial monitor.

Another issue is the hold up when a button is pressed. The button is definitely on PB2

RoboDurden, 2023-09-19:
Thanks, so no adc on these pins.

So you use

```
	#define USART_STEER_COM_TX_PIN	GPIO_PIN_6
	#define USART_STEER_COM_TX_PORT	GPIOB
	#define USART_STEER_COM_RX_PIN	GPIO_PIN_7
	#define USART_STEER_COM_RX_PORT	GPIOB
```

With 	
`#define USART_STEER_RCU RCU_USART0`
or 
`#define USART_STEER_RCU RCU_USART1`
 ?

RoboDurden, 2023-09-19:
0.42 is because of bldc.c :
```
// Calculate current DC
	#ifdef CURRENT_DC_PIN
		currentDC = ABS((adc_buffer.current_dc - offsetdc) * MOTOR_AMP_CONV_DC_AMP);
	#else
		currentDC = 0.42; 	// testing with no CURRENT_DC_PIN yet
	#endif
	
```

RoboDurden, 2023-09-19:
36V also set in bldc.c

```
	// Calculate battery voltage every 100 cycles
	#ifdef VBATT_PIN
		if (buzzerTimer % 100 == 0)
			batteryVoltage = batteryVoltage * 0.999 + ((float)adc_buffer.v_batt * ADC_BATTERY_VOLT) * 0.001;
	#else
		batteryVoltage = BAT_CELLS * 3.6;		// testing with no VBATT_PIN yet
	#endif

```

RoboDurden, 2023-09-19:
If you indeed have serial connection between arduino and hoverboard please post your new values of:

```
// Usart steer defines
//#define USART_STEER_COM USART0					
#ifdef USART_STEER_COM
	#define USART_STEER_RCU RCU_USART0			
	#define USART_STEER_AF	GPIO_AF_0				
	#define USART_STEER_COM_TX_PIN	GPIO_PIN_11
	#define USART_STEER_COM_TX_PORT	GPIOB
	#define USART_STEER_COM_RX_PIN	GPIO_PIN_10
	#define USART_STEER_COM_RX_PORT	GPIOB
#endif

```

RoboDurden, 2023-09-19:
Then you need to find the correct adc pins:

```
// ADC defines
//#define VBATT_PIN	GPIO_PIN_0				// robo, no gpio_mode_set() inHerleybob:setup.c
#define VBATT_PORT GPIOA							// robo, no gpio_mode_set() inHerleybob:setup.c
#define VBATT_CHANNEL ADC_CHANNEL_17	// robo, based on Herleybob:setup.c: adc_regular_channel_config(2,ADC_CHANNEL_17,ADC_SAMPLETIME_13POINT5);
//#define CURRENT_DC_PIN	GPIO_PIN_1	// robo, no gpio_mode_set() inHerleybob:setup.c
#define CURRENT_DC_PORT GPIOA					// robo, no gpio_mode_set() inHerleybob:setup.c
#define CURRENT_DC_CHANNEL ADC_CHANNEL_2	// robo, maybe adc_buf_t:adc is potentiometer input

```

RoboDurden, 2023-09-19:
Why did you uncomment 

`//#define USART_STEER_TX_BYTES 2   // Transmit byte count including start '/' and stop character '\n'`

it is no longer used in my code.

mateuszfcg, 2023-09-19:
`// ADC defines
//#define VBATT_PIN	GPIO_PIN_0				// robo, no gpio_mode_set() inHerleybob:setup.c 17 A
#define VBATT_PORT GPIOA							// robo, no gpio_mode_set() inHerleybob:setup.c
#define VBATT_CHANNEL ADC_CHANNEL_17	// robo, based on Herleybob:setup.c: adc_regular_channel_config(2,ADC_CHANNEL_17,ADC_SAMPLETIME_13POINT5);
//#define CURRENT_DC_PIN	GPIO_PIN_1	// robo, no gpio_mode_set() inHerleybob:setup.c
#define CURRENT_DC_PORT GPIOA					// robo, no gpio_mode_set() inHerleybob:setup.c
#define CURRENT_DC_CHANNEL ADC_CHANNEL_2	// robo, maybe adc_buf_t:adc is potentiometer input
`

I have tried all the ADC pins and there are still no readings

RoboDurden, 2023-09-19:
You need to surround multi line code with ```

mateuszfcg, 2023-09-19:
> Dlaczego odkomentowałeś
> 
> `//#define USART_STEER_TX_BYTES 2 // Transmit byte count including start '/' and stop character '\n'`
> 
> nie jest już używany w moim kodzie.

communication with the arduino without it did not work at all

RoboDurden, 2023-09-19:
please repeat your test and put the `//` again before `#define USART_STEER_TX_BYTES 2`
I guess you have made multiple changes at the same time..

mateuszfcg, 2023-09-19:
ok you are right.

mateuszfcg, 2023-09-19:
> Ok masz rację.

RoboDurden, 2023-09-19:
[image]

```
// photo diodes / light barriers on the backside
#define PHOTO_L_PIN		GPIO_PIN_15
#define PHOTO_L_PORT	GPIOC
#define PHOTO_R_PIN		GPIO_PIN_14
#define PHOTO_R_PORT	GPIOC
```

So the photo sensors on the backside seem to be digital io only.

DerPinguin77, 2023-09-19:
Hi, I got the Firmware flashed and the motor ran pretty well. Sadly my battery wasnt charged up so i couldnt get to full speed until the bms switched it off. My cc powersupply cant supply enough current. I put my battery on the charger and i will test again tomorrow after work. 
The only issue i had was with the power latch. The mainboard would instantly turn off if if released the powerbutton so i had to hold it. Idk if this was intended or if youre supposed to install a switch instead of the original pushbutton.

DerPinguin77, 2023-09-19:
@RoboDurden So all of the pins I tested wont work for adc? Maybe u could use the pins currently used for the uart. Ans some of the digital pins that i tested for the uart.
Or is could be an option in the software you could comment out and choose between uart and ADC
Or are there any other unused analog pins of the chip that could be used? Maybe the led pins on the 3 pin header?

mateuszfcg, 2023-09-19:
Tomorrow I will check what pins can be used and let you know. For now, good night because it's late.

RoboDurden, 2023-09-19:
no, the uart pins are also digital only.
the led pins are driven by a smd transistor, so not available for adc. Beware of the 15V on the led header, they will kill the MCU.

You can find the adc capable pins in GD32F130xxDatasheetRev3.7.pdf . Simply search for adc_
The Hall sensor can do adc :-/

Would be nice if you could check all possible adc pins. 
The two adc inputs of the dual-opamp must be two of them.
Same goes for the yet missing VBAT pin and the DC_CURRENT pin.

If you have identified these 4 adc pins, then maybe you also have found a free adc pin to be used for a potentiometer.

But again i want to suggest that you should think bigger and go for an ESP32 S2 Mini.

RoboDurden, 2023-09-19:
If the power gets off wenn releasing the onoff button, this pin definition is wrong:

```
#define SELF_HOLD_PIN		GPIO_PIN_13
#define SELF_HOLD_PORT	GPIOA		
```

This HOLD pin should go to a big mossfet to the lower right..

RoboDurden, 2023-09-19:
2A should be well enough to spin the motor freely. If more current is drawn (and the mosfet get hot), there might be something wrong with the 6 mosfet pin definition or the 3 hall pins !

RoboDurden, 2023-09-19:
@mateuszfcg if you are from Poland, you are 1 hour ahead of us ?
Here in Bavaria it is 22:38 and i also turn off the lights :-)

mateuszfcg, 2023-09-20:
**@DerPinguin77**  The double operational amplifier is certainly from the current, from the voltage may also be although it is rather done on the voltage divider. By the hour I also sit down further to look for pins.

DerPinguin77, 2023-09-20:
These are the new pins i tested:
[image]
Hope this is enough if i should test morte please tell me

RoboDurden, 2023-09-20:
Today i have published the first running I2c-SimpleFoc firmware and impelemented the new EspressLRS support (CRSF decoder) method that @WizzardDr contributed so i am done for today:

**NOW rc EspressLRS support (CRSF decoder) !!!** https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26

One dual opamp for sure is to measure two low-side currents of two phases. This is needed for the FOC. Not needed for this simple `bldc.c` here.

DerPinguin77, 2023-09-20:
Will the FOC also work with this board? I also continuewd to test arount the hall sensors. Myabe this hepls to fix the poroblem with me motor.
Btw. I didnt habe problems with the motor skipping. Cant say anithing about the mosfet temp because i only hat it on for a few secs.
This is the again updated version.
[image]

RoboDurden, 2023-09-20:
Would be nice if you only update jpg files and not png files.
Yes your board has an GD32F130C8 and not C6 (which only has 32 kb Flash).
You would have to add the `defines_2.3.h` in the arduino format like the others.
But that SimpleFOC firmware is way more complex and you should be sure that the hall/mosfet pins are correct.

DerPinguin77, 2023-09-20:
I think they are correct because the motor ran smoothly in my case. I have tested them again and couldnt find any errors but I also dont 100 percent understand what im doing there so there might be something wrong.
The only poblem i had was that the motor wouldnt start spinning again and the fuse blew after the first time of cycling between min and max speed(I think this happend when it wanted to go in reverse)
 I can also test the connections of the mosfet driver again. Maybe the "error" is there.
The only thing i dind't test for today is the voltage sens pin. I couldnt really locate the voltage divider you talked about so i dont really know where to start. Maybe i acidentally found it with the 3 traces i tested for the power latch pin. 

Would be cool if the FOC firmware would run on the board somtime in the future. Im doing my best to help with finding all the pins.

mateuszfcg, 2023-09-20:
> #define PHOTO_L_PIN		GPIO_PIN_15
> #define PHOTO_L_PORT	GPIOC
> #define PHOTO_R_PIN		GPIO_PIN_14
> #define PHOTO_R_PORT	GPIOC

what are the digital pins used for when reading? 
I have no idea where to look for them

and regarding the ADC, I am starting to see what can be changed to get at least one ADC pin

DerPinguin77, 2023-09-20:
Mofet pins are also correct in the drawing. Tested them again right now

DerPinguin77, 2023-09-20:
@mateuszfcg Thanks for helping with that. Im done for today but tomorrow i could look up some ADC pins in the datasheet and see if they are unused on the board.

mateuszfcg, 2023-09-20:
[image]

I
Hall sensor pins are connected to the ADC. I suggest you solder the resistors 
1.R75,R76
or
2.R74,R77
or
3.R73,R78
and connect one of the Hall sensors e.g. to the DATA plug via resistors. 

II
use of an external ADC transducer via i2C. I2C can be run on the LED pins except that resistors and transistors must be removed from the board.

All LED pins are I/O.

RoboDurden, 2023-09-20:
I would not solder anything on the board. I will never accept code to this repo that needs hardware modifications.
An ESP32 S2 Mini does not really cost more then an I2C ADC module: https://en.aliexpress.com/item/1005004506970676.html

DerPinguin77, 2023-09-20:
Your Link is broken somehow. Can u also use a normal esp32? Would have been nice if it was all self contained in the board but if there arent any free pins the esp will work to. Just need to learn how all of this uart stuff works

mateuszfcg, 2023-09-20:
ESP is the best solution in my opinion.

mateuszfcg, 2023-09-20:
@DerPinguin77  Will you send a video of how your engine works ? Maybe in my case there is something damaged in the driver mosfet because I am concerned about the operation of the BLDC motor

RoboDurden, 2023-09-20:
Simply search aliexpress for `esp32 s2 mini`.
But any ESP32 module works. The double headers of the ESP32 S2 Mini are no problem with the usual `Prototype Board`s

DerPinguin77, 2023-09-20:
@mateuszfcg I can send a video but i cant let it run for long or my fuse will blow. One moment

DerPinguin77, 2023-09-20:
Her is a Short video of the motor running. The sudden stop isa when i released the power button and it turned off. It cant get to full speed without blowing the fuse so the porblem should be something else.

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/145379857/2e0a0dc3-8b78-4b5b-b365-31301689fbab

mateuszfcg, 2023-09-20:
I was more concerned with the low speed. In my case nothing is burning just the performance characteristics are unsatisfactory to me and I have a little concern that something is wrong with the driving of the mosfets though. I'll post my pin configuration tomorrow because I'm already going to bed, but it's strange that you're burning a fuse. 

Regarding the video, those sounds are already vibrations through the high speed. Notice the sound at low speed kind of hard

RoboDurden, 2023-09-20:
This board has a mosfet driver chip. The 2.0 layout has analog mosfet drivers made of three little smd transistors.
It might be that this mosfet driver chip needs a different deadtime when turning off the high side and turning on the low side. Or some other adjustment of https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/2e3e2dbd8994b9281278be360cb46f0302034b13/HoverBoardGigaDevice/Src/setup.c#L255
I have no experience with bldc mosfet driver chips.
Maybe @robcazzaro has an idea.

robcazzaro, 2023-09-20:
@RoboDurden, sorry but I have no experience with high power mosfet drivers. If you have a datasheet, we can check if it needs a different signal polarity for the H and L phase, or a different deadtime.

RoboDurden, 2023-09-21:
Thanks robcazzaro (=Rob ?). I remember that you were the first to notice that the low side mosfets need inverted signals.
From the pdf i think that this also applies to this driver chip:
[image]
I have only found the Chinese datasheet: https://www.fortiortech.com/public/upload/files/20200715/20e40353f0c9715259019ff1d7ce0e54.pdf
Here the google translation: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/Schematics_2.3/FD6287T%20bldc%206x%20gate%20driver.zh-CN.en.pdf

I think the motors are spinning way to fast in the two videos attached/linked here. I only remember too fast rotating when the phase cables have the wrong ordering. But here the motors speed up very smoothly, so i think the phase/hall ordering is correct.
Problem ist that the phase colors at the mosfets is yellow,blue,green. But with the hall cables it is yellow,green,blue.
A bit confusing. But the hall pin definitions of @mateuszfcg i think compensate for that:

[67 more replies]