RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #22: Layout 2.8 = MM32SPIN05PFOP :-(

RoboDurden, 2023-09-19:
https://www.mindmotion.com.cn/en/products/mm32mcu/mm32spin/mm32spin_specific_mcu/mm32spin05/

- 32bit Arm® Cortex®M0 processor as the core
- Maximum operating frequency is up to 72M
- 32K bytes of Flash memory
- 4K bytes of SRAM
- Boot loader supports Chip Flash and ISP (InSystem Programming)
- Embedded factorytuned 48/72MHz high speed oscillator
- One 12bit ADC and 1μS of conversion time (up to 13 channel)
- Supported peripherals: Timer, UART, I2C, SPI and ADC
- One 16bit 4channel advancedcontrol timer with 4channel PWM output, dead-time
generation and emergency stop
- Serial wire debug (SWD)
- Two UARTs
- One I2C
- Two SPIs

https://www.keil.com/dd2/mindmotion/mm32spin05nt/

https://github.com/topics/mm32

Would be nice if someone knows how to add this device to the Keil IDE.

reed-dan, 2023-10-30:
Does the information here help add it to KEIL?

https://github.com/iclite/mm32_startup/tree/master

RoboDurden, 2023-10-30:
Would be nice if @WizzardDr would explain how he added the GD32E230 target to the project:
[image]
If a new target would be added, then your MM mcu can be choosen:
[image]

Then this repo might already compile.
It might need some fixes like fore the E230 mcu in target.h

```
#ifdef GD32E230
	#include "gd32e23x.h"
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b);}
	#define TARGET_nvic_priority_group_set(a)	// that function does not exist for this target = not needed ?
	#define TARGET_adc_vbat_disable()
#else
	#include "gd32f1x0.h"
	
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b, c);}
	#define TARGET_nvic_priority_group_set(a){nvic_priority_group_set(a);}
	#define TARGET_adc_vbat_disable(){adc_vbat_disable();}
#endif
```

But as WizzardDr succeeded with adding a new target, i now am quite optimistic that he and me could add your MM mcu and i/we resolve all compiler errors.

But are you willing to invest some hours to trace the pins ??

RoboDurden, 2023-10-31:
Okay, a target can be added in menu->Project->Manage->ProjectItems..
[image]

But now i am stuck at adding the Device files and i do not know the name of the propper #include file:
[image]

ideas welcome.

reed-dan, 2023-10-31:
When I look at my board it is slightly different than this board, appears to have the same processor though.

[image]
[image]
[image]

Board Source: Jetson Plasma 
Model: Plasma (JPLSM-BLK)
UPC: 811991031504

Slave Board Identifiers
Taotao Technology
2025
pu20200616011
94v-0 y-d
e253642
816117r

RoboDurden, 2023-10-31:
To me this looks identical to the photos of the first board.

But i have no idea from where to get the MM counterplay of the Standard Peripherals:
[image]

I have nearly zero experience with the Keil Ide :-/

reed-dan, 2023-11-02:
Adding some documentation

https://www.mindmotion.com.cn/products/mm32mcu/mm32spin/mm32spin_specific_mcu/mm32spin05/

[AN0012_MM32_Series_Keil_pack_install_SC _translated EN.pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13233795/AN0012_MM32_Series_Keil_pack_install_SC._translated.EN.pdf)
[UserManual_MM32SPIN05x_q_EN.pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13233796/UserManual_MM32SPIN05x_q_EN.pdf)
[DS_MM32SPIN05x_version_F_EN.pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13233797/DS_MM32SPIN05x_version_F_EN.pdf)
[FAQ_MM32SPIN05_EN.pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13233798/FAQ_MM32SPIN05_EN.pdf)
[AN6301_Sensorless_Trapezoidal_Control_of_BLDCM__translatedSC (1).pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13233799/AN6301_Sensorless_Trapezoidal_Control_of_BLDCM__translatedSC.1.pdf)
[DS_MM32SPIN05x_version_E_EN.pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13233800/DS_MM32SPIN05x_version_E_EN.pdf)

RoboDurden, 2023-11-02:
Yes thanks, i have added the pdf documents to the manuals folder.
But i fear we still do not know how to install the Standard Peripherals like ADC, DMA, GPIO

reed-dan, 2023-11-22:
Also found in

Board Source: JETSON ALPA
Model: Alpha (JALPHA-BLK)
FCC: 2AV54ALPHA
https://fcc.report/FCC-ID/2AV54ALPHA

BigVoo, 2024-01-06:
Looking for help on this board.  I've burned through many of the RioRand motor controllers and decided to use one of the 4 different hoverboard motors that I have.  Any help will be appreciated.  I'm a newbie, but I love tinkering and learning new things.

RoboDurden, 2024-01-06:
If you have different boards you better start with a GD32F130 MCU and not this not yet supported MM32SPIN05.

BigVoo, 2024-01-06:
I do have a GD32F130 C8T6 MCU board, but it's a newer board and does not match any of the examples.  I wasn't sure if I could flash or not since it didn't exactly match.  If I can, then that will be perfect since that's the board and motor that I want to use.

RoboDurden, 2024-01-06:
Here is a download link for the STM32-ST-Link-Utility maybe it includes such a driver you asked for:
https://www.robosoft.de/forums/en.stsw-link004.zip

The Keil IDE ( https://www.keil.com/demo/eval/arm.htm ) will have to wait.

RoboDurden, 2024-01-06:
No you need to start a new issue here, upload high res photos of front and backside and start tracing the pins. See the 2.6.1 issue currently being active.
This 2.8 is not yet traced either.

I always like to add new layouts, so feel free to go ahead :-)

AILIFE4798, 2024-01-06:
@RoboDurden heres a toturial of using keil with mm32spin maybe itll help i cannot tell
[mm32spinexample(changeZIPto7Z).zip](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13850089/mm32spinexample.changeZIPto7Z.zip)
[MM32_KEIL_Pack_2.18.zip](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13850090/MM32_KEIL_Pack_2.18.zip)

AILIFE4798, 2024-01-06:
> Okay, a target can be added in menu->Project->Manage->ProjectItems.. ![grafik](https://private-user-images.githubusercontent.com/12238841/279305984-83f28222-e6c2-43d8-bafc-e8dda6bee352.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1NjYxODQsIm5iZiI6MTcwNDU2NTg4NCwicGF0aCI6Ii8xMjIzODg0MS8yNzkzMDU5ODQtODNmMjgyMjItZTZjMi00M2Q4LWJhZmMtZThkZGE2YmVlMzUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDE4MzEyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4MTVjZmJmNzY2YjE5ZDMxYWY5MjlmYThjYWI0ZjQ1NGU0ZmM1MGFkMzJjMDA1Y2RkOWFhZDNlMGMxZmYyNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fWja0vwgkiqWET_oVL8XWL_BgfZD-J6cyn7iFKEzQh8)
> 
> But now i am stuck at adding the Device files and i do not know the name of the propper #include file: ![grafik](https://private-user-images.githubusercontent.com/12238841/279306504-316fb6c9-54dd-4716-9a97-a5f2560a25b1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1NjYxODQsIm5iZiI6MTcwNDU2NTg4NCwicGF0aCI6Ii8xMjIzODg0MS8yNzkzMDY1MDQtMzE2ZmI2YzktNTRkZC00NzE2LTlhOTctYTVmMjU2MGEyNWIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDE4MzEyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhZjQ4M2ZhNzU4Yjc0ZjkyNjg0YTBlNWQ5OGMxNzA2N2MwNmE0ZGE5ODU2MzhkN2RlOGIxYjY5NDI3ZDdhMmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Qxw3GJsblj5VGnGIT7mCydQ-IXOirEhXSr7X3tVJcoU)
> 
> ideas welcome.

i am not sure if this is your problem but i think you can copy it from example
[image]

AILIFE4798, 2024-01-08:
@RoboDurden can you please look into it now i have included the new build target 
it still have 83 error but those are related to constant not defined can you fix them
[image]
#50

RoboDurden, 2024-01-08:
Thanks for you work ! I have accepted the pull request :-)
But i did not like that you changed

```
#ifndef ADC_BATTERY_VOLT
	#define ADC_BATTERY_VOLT      0.024169921875 
```
to

```
#ifndef ADC_BATTERY_VOLT
	#define ADC_BATTERY_VOLT      0.02488682634 	// V_Batt to V_BattMeasure = factor 30: ( (ADC-Data/4095) *3,3V *30 )
```

This would apply to all other layouts and not only the 2.6.
Instead i will add

`	#define ADC_BATTERY_VOLT      0.02488682634 	// V_Batt to V_BattMeasure = factor 30: ( (ADC-Data/4095) *3,3V *30 )
`
to the defines_2-6.h

RoboDurden, 2024-01-08:
i have uploaded my first changes, so please @AILIFE4798 sync your code.

```
#ifdef GD32E230
	#define LAYOUT 7
#elif MM32SPIN05
	#define LAYOUT 8
#else
	#define LAYOUT 6
#endif

//#define MASTER		// uncomment for MASTER firmware. Choose USART0_MASTERSLAVE or USART1_MASTERSLAVE in your defines_2-?.h file
//#define SLAVE			// uncomment for SLAVE firmware. Choose USART0_MASTERSLAVE or USART1_MASTERSLAVE in your defines_2-?.h file
#define SINGLE			// uncomment if firmware is for single board and no master-slave dual board setup
```
Looks like many files in the Inc project folder should not be there. Maybe you can fix this:
[image]

RoboDurden, 2024-01-08:
I am down to 75 errors and 0 warning.
Many errors are to my PB8 -> PORT_B + PIN_8 marcors that i introduced lately to make the defines_2-x.h easier.
But these macros are GD32 specific so i now first have to investigate how MM is coding PORT and PIN.
This might take some hours.

AILIFE4798, 2024-01-08:
yes,the adc calibration should be done on every board because the resistors and mcus are not exactly identical

you can delete those files in inc its not needed 
i am just experimenting with it and those file is included in device already
[image]

RoboDurden, 2024-01-08:
done and repo updated :-)

AILIFE4798, 2024-01-08:
do you have real hardware layout2.8 to test on or should i buy more of them to test
because the real problem will start from how to flash it

RoboDurden, 2024-01-08:
MM is also using port defines like GPIOB but not pin defines like GPIO_PIN_9 :-/ So i must learn the defines to acces a port/pin for MM.
With GD32 the GPIO_PIN_9 is defined in gd32f1x0_gpio.h:
```

/* GPIO pin definitions */
#define GPIO_PIN_0                 BIT(0)                /*!< GPIO pin 0 */
#define GPIO_PIN_1                 BIT(1)                /*!< GPIO pin 1 */
#define GPIO_PIN_2                 BIT(2)                /*!< GPIO pin 2 */
#define GPIO_PIN_3                 BIT(3)                /*!< GPIO pin 3 */
#define GPIO_PIN_4                 BIT(4)                /*!< GPIO pin 4 */
#define GPIO_PIN_5                 BIT(5)                /*!< GPIO pin 5 */
#define GPIO_PIN_6                 BIT(6)                /*!< GPIO pin 6 */
#define GPIO_PIN_7                 BIT(7)                /*!< GPIO pin 7 */
#define GPIO_PIN_8                 BIT(8)                /*!< GPIO pin 8 */
#define GPIO_PIN_9                 BIT(9)                /*!< GPIO pin 9 */
#define GPIO_PIN_10                BIT(10)               /*!< GPIO pin 10 */
#define GPIO_PIN_11                BIT(11)               /*!< GPIO pin 11 */
#define GPIO_PIN_12                BIT(12)               /*!< GPIO pin 12 */
#define GPIO_PIN_13                BIT(13)               /*!< GPIO pin 13 */
#define GPIO_PIN_14                BIT(14)               /*!< GPIO pin 14 */
#define GPIO_PIN_15                BIT(15)               /*!< GPIO pin 15 */
#define GPIO_PIN_ALL               BITS(0,15)            /*!< GPIO pin all */
```

The PORT definitions for MM are in reg_gpio.h :

```
#define GPIOA                           ((GPIO_TypeDef*) GPIOA_BASE)
#define GPIOB                           ((GPIO_TypeDef*) GPIOB_BASE)
#define GPIOC                           ((GPIO_TypeDef*) GPIOC_BASE)
#define GPIOD                           ((GPIO_TypeDef*) GPIOD_BASE)
```

But i can no identify the PIN definitions yet. Maye a MM32 code example would  help

I think i have 2.8 but not here in the train station i think. And honestly i do not have the time=happiness at the moment to build yet another test setup. So it would be great if you had 2.8 hardware. Was $2 a typo or can you really buy fully assembled 2.8 boards for USD 2 ???

RoboDurden, 2024-01-08:
Maybe these are the definitions we need: https://github.com/koendv/MM32SPIN27-Arduino/blob/525f27bd91b756aeabad3b7a706f60cb9fb75275/mm32/cores/arduino/HAL_Lib/hal_gpio.h#L89

But i would like to find these defines in the target files you added.

AILIFE4798, 2024-01-08:
[image]
here is a example of mm32 blynk

$2 is for a pair of master and slave layout2.8.1 board by razor no led or motor 
i think its hard to beat
a set of gen1 board with mainboard+2 side board+led board is $12 still cheap(seller say its full GD32 so compatiable with eferu)
keep in mind its china everything is dirt cheap

RoboDurden, 2024-01-08:
Where can i download this blynk example ?

We have problems to port the EFeru gen1 firmware to the newer AT32 Gen1 boards ! So if there is no GD32 mcu on theser $12 boards i would be caucious.

AILIFE4798, 2024-01-08:
you can download example in pack installer
[image]

AILIFE4798, 2024-01-08:
yes i have also seen the at32 board its selling for even cheaper but i know its not supported so i wont try to buy it
it is very easy to identify by theres no mcu on the side board every single gpio goes to the main board so the main board to side board cable is very wide
idk how they get so much gpio from just one mcu maybe with multiplex idk

AILIFE4798, 2024-01-08:
Incase you cannot open the link I downloaded the image from seller for you to confirm layout 2.8.1
[image]
[image]

RoboDurden, 2024-01-08:
Well this is definitely not a 2.8 layout. See my photos in Schematics_2.8/ folder.

I do not find the gpio_bit_write() definition for the MM target. When i search my project with strg+shift+F i only find the GD definition :-/
This would help me to cast PA6 to the correct types ..

RoboDurden, 2024-01-08:
48 errors. code uploaded.
i have to make a break. sun is finally shining so i can saw some wood to feed my oven ! It is currently -6°C outside and +11°C inside  !

AILIFE4798, 2024-01-08:
the gpio definition is in device/system_mm32spin06xx_s.c/reg_gpio.h i guess
not sure if its what you wanted

the layout is not exactly same as 2.8 but i would think its pin compatible
after all theres only one mm32spin layout
and from the blury picture you can make out the mindmotion and arm logo so thats why its so cheap i guess

RoboDurden, 2024-01-08:
in my keil project i do not have a sub folder, only
[image]
In you screenshot above you have
[image]
 :-/

AILIFE4798, 2024-01-08:
[reg_gpio.h.txt](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13859055/reg_gpio.h.txt)
i have this sub folder after i compiled once

AILIFE4798, 2024-01-08:
[MM32SPIN0x_q_Lib_Samples.zip](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13859162/MM32SPIN0x_q_Lib_Samples.zip)

you should check these example because i didnt understand more about mm32 then you lol
but as a dedicated bldc driver mcu it should have advanced timer

RoboDurden, 2024-01-08:
40 errors and repo updated.

RoboDurden, 2024-01-08:
37 errors and repo updated.
But this is not looking good.
All the timer functions are different. So i can not simple adapt some naming conventions like i may have succeded here in defines_2-8.h:

```
// found in hal_tim.h of MM32SPIN0x_q_Samples
typedef enum {
    TIM_Channel_1 = 0x0000,  ///< TIM Channel 1
    TIM_Channel_2 = 0x0004,  ///< TIM Channel 2
    TIM_Channel_3 = 0x0008,  ///< TIM Channel 3
    TIM_Channel_4 = 0x000C,  ///< TIM Channel 4
    TIM_Channel_5 = 0x0010   ///< TIM Channel 5
} TIMCHx_Typedef;

	#define TIMER_BLDC 		TIM1
	#define RCU_TIMER_BLDC 		RCU_TIM1
	#define TIMER_BLDC_CHANNEL_G 	TIM_Channel_3
	#define TIMER_BLDC_CHANNEL_B 	TIM_Channel_2
	#define TIMER_BLDC_CHANNEL_Y 	TIM_Channel_1
```

next error goes to `timer_interrupt_flag_clear` which is unkown to MM
Thanks for the MM32SPIN0x_q_Samples which i can compile with selecting "using default compiler 6".
But i looks like i or someone else has to dig deeper (in these examples and the user manual) to understand how the timer stuff is coded in MM.

AILIFE4798, 2024-01-08:
yeah the mm is not as fully compatible as GD but all the newer board have mm now so it would be great if it could be supported if you needed any help with chinese datasheet i can probably help otherwise ill wait for you uto adapt the code and also wait for the hardware to arrive

RoboDurden, 2024-01-08:
This $2 board for sure will have to get it's new layout number like 2.19
But maybe i adapt you idea to assign sub numbers to help identify the board layout.
So 2.0.x would be gd32F130, 2.1.x would be gd32E230 and 2.2.x would be MM32SPIN05
The MM32SPIN05 does only have 32kb which would make it incombatible with my SimpleFOC repo https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC that is already working with I2c control...

$12 for a GD32 gen1 might be better. Some years ago you could buy a set for $17. But nowdays all Gen1 sets are $25 :-(
I can buy used hoverboards for $25 including shipping. A lot people would like to have a <$15 source for Gen1 boards. But including VAT and shipping :-/

AILIFE4798, 2024-01-08:
the problem is used hoverboard is starting to contain split board so i have to spend another $12 and a few week to get a set of gd gen1 board
so maybe after all gen2 is better because for more then 2 wheel application its better to use uartbus

RoboDurden, 2024-01-08:
And Gen2 have nice led drivers on board.
A $2 board for sure is something we all want to be added to this Gen2.x.
I just wanted to say that a cheap MM board with 64 kB would be better.
But that SimpleFOC project seems to be dead anyway. This SimpleFOC library to me seems to be too complex.. It did work for a gen2.0 board but then for a gen2.5 it did not work :-(
I will make a break for some hours, maybe the rest of the day.
As you do not have a MM board ready to test, there is no hurry anyway.

P.S. it would be great if you now someone in China close to the Chinese Communist Party who would be allowed to read (and discuss with me) my "masterplan for world peace": https://www.robodurden.de/?iLan=4&iType=11&iFormat=21

RoboDurden, 2024-01-08:
down to 36 errors. But now i really need to study the timer examples.

RoboDurden, 2024-01-08:
The example TIM2_BasicExample makes use of HAL_LIB/hal_tim.h which i also need in the gen2.x project.
Would be nice if you update your code with my repo and try to add such "hardware abstraction layer = hal" library files.
Because i still do only have the two st
[image]
artup files.

AILIFE4798, 2024-01-08:
i see allot of gen2 board that look like other layout but have mm32 now
the problem is they are more expensive then gen1
mm32spin06 would be 64kb so theres still hope for foc just need to replace the mcu which i would have guessed is very cheap
isnt gen2.0 and 2.5 have almost identical hardware just different pinout i guess the phase current sensing just wouldnt work well enough
anyways is it possible to have sine wave drive but with constant speed pid loop at least
so its easier to sync multiple motor

yes the board would take a while to arrive 
and i also find that gen1 side board only which contain a gd32f103c8t6 
it only cost $0.5
so ill buy a bunch of them and selvage the mcu to replace the layout2.6 so i have 64k

my file is a exact duplicate as your repo now and i still have subfolder i think its due to keil settings no need to install seperately
or maybe your device pack is not installed correctly
[image]

i hate politics :)

AILIFE4798, 2024-01-08:
Ok I'll create a folder and add all those little files for mm32 target later

AILIFE4798, 2024-01-08:
#51 i have added the files

RoboDurden, 2024-01-08:
I only see the .c files and not the .h files. Sure they are not needed ??

AILIFE4798, 2024-01-08:
ok i will also add it just in case

RoboDurden, 2024-01-08:
why did you change the [RTE/_GD32F130C8/RTE_Components.h](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/pull/51/files#diff-29232d9f8d45648cfcd117ff108142586f7d35424359a380c66414b44a29ecaa) file, does 2.6 still compile ?

I do not really like the location were you put the mm hal files: [Src/MM32_HAL_lib/dtype.h](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/pull/51/files#diff-914285803ea2faa2cfc3320f273f04ab44c6cf86a305466275ebbcf696383c95)

In the examples they are hidden in Device\MM32SPIN0x_q\

AILIFE4798, 2024-01-08:
oh no it doesnt work any more
ill investigate

AILIFE4798, 2024-01-08:
> In the examples they are hidden in Device\MM32SPIN0x_q\
in my example it does not have Device\MM32SPIN0x_q\
[image]
it looks like this
and the tim2 example does not contain any hal lib file
[image]

RoboDurden, 2024-01-08:
And does your example compiles ?
Mine does and it looks like this:
[image]

I by now also try to import the hal files..

AILIFE4798, 2024-01-08:
actually go ahead and include ```#include "hal_tim.h"```
it will compile just fine without aditional error

RoboDurden, 2024-01-08:
I am lost..
I have added the MM32_HAL as a sub folder to Inc/ 

[image]

and try to copy the complete example code into the gen2 project to see that it works:

```
#elif defined MM32SPIN05	
	
	#include "MM32_HAL/hal_conf.h"
	#include "mm32_device.h"

	#include "MM32_HAL/hal_device.h"
	#include "MM32_HAL/hal_rcc.h"
```
But i get lots of compiler errors like
```

Inc/MM32_HAL/hal_rcc.c(292): error: use of undeclared identifier 'RCC_CFGR_SWS_HSI'
        case RCC_CFGR_SWS_HSI:
             ^
Inc/MM32_HAL/hal_rcc.c(293): error: use of undeclared identifier 'RCC_CR_HSI_72M'
            return (RCC->CR & RCC_CR_HSI_72M) ? HSI_72MHz : HSI_48MHz;
                              ^
Inc/MM32_HAL/hal_rcc.c(293): error: use of undeclared identifier 'HSI_72MHz'
            return (RCC->CR & RCC_CR_HSI_72M) ? HSI_72MHz : HSI_48MHz;
                                                ^
Inc/MM32_HAL/hal_rcc.c(293): error: use of undeclared identifier 'HSI_48MHz'
            return (RCC->CR & RCC_CR_HSI_72M) ? HSI_72MHz : HSI_48MHz;
                                                            ^
Inc/MM32_HAL/hal_rcc.c(296): error: use of undeclared identifier 'RCC_CR_HSI_72M'
            result = (RCC->CR & RCC_CR_HSI_72M) ? HSI_72MHz_DIV6 : HSI_48MHz_DIV6;
                                ^
Inc/MM32_HAL/hal_rcc.c(296): error: use of undeclared identifier 'HSI_72MHz_DIV6'
            result = (RCC->CR & RCC_CR_HSI_72M) ? HSI_72MHz_DIV6 : HSI_48MHz_DIV6;
                                                  ^
Inc/MM32_HAL/hal_rcc.c(296): error: use of undeclared identifier 'HSI_48MHz_DIV6'
            result = (RCC->CR & RCC_CR_HSI_72M) ? HSI_72MHz_DIV6 : HSI_48MHz_DIV6;
                                                                   ^
```

Maybe something missing in
[image]

RoboDurden, 2024-01-08:
Sorry i give up for today.
Your pull request does so many changes which i do not understand :-/
If you succeed with a pull request that does not generate more errors than my online repo with this code added to target.h:

```
#ifdef GD32E230
	#include "gd32e23x.h"
	#define TARGET_nvic_irq_enable(a, b, c){nvic_irq_enable(a, b);}
	#define TARGET_nvic_priority_group_set(a)	// that function does not exist for this target = not needed ?
	#define TARGET_adc_vbat_disable()
#elif defined MM32SPIN05	
	
	#include "MM32_HAL/hal_conf.h"
	void TIM2_IRQHandler(void)	// just for testing
	{
			TIM_ClearITPendingBit(TIM2, TIM_IT_Update);		// fine, now TIM_IT_Update is defined
	}

	#include "mm32_device.h"

	#include "MM32_HAL/hal_device.h"
	#include "MM32_HAL/hal_rcc.h"

	#define timer_interrupt_flag_clear TIM_ClearITPendingBit
```

and 2.6 still compiling successfully
then i will accept your pull request.

good night from Germany :-/

AILIFE4798, 2024-01-08:
yes,it is a mess that i have created
sorry about that
this is my first time using github to clone repo and create pull request
so things do go wrong
this is also my third?day using keil so also not very experienced
im trying my best to make here a better place for everyone
now its working again

AILIFE4798, 2024-01-10:
i have tried to remove everything except hall2led now
but pinmode in setup.c still error can you help fix it
https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x/blob/MM32-TEST

RoboDurden, 2024-01-10:
today i finally have begun with a RemoteAutodetect that might automatically detect all the pins.
Will push my repo online later this day.

Remember that i can not download your fork because it is to big.
Please wait until i have moved the keil project to a seperate repo and then fork from this.

Click F12 on "pinmode" to see where i define it.
THE GPIO_Init will also have to be rewritten based on the mm32 examples.
I do not really see an example for setting a pin to input :-/

RoboDurden, 2024-01-10:
Okay here is the new repo with only the Keil project for GD32: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32
And this is our temporary project for MM32: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-MM32

Ideas welcome !

RoboDurden, 2024-01-10:
I think we really should delete everything from the MM32 code so the the `main()` etc is clean and easy to read.
And then step by step copy the functionality from the GD32 repo and port it to the mm32 mcu..

[236 more replies]