RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #26: NOW rc EspressLRS support (CRSF decoder) !!!

RoboDurden, 2023-09-20:
Thanks to @WizzardDr who implemented the code: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/pull/24

In `config.h` you have to uncomment the "remote" you want to use:

```
#define REMOTE_DUMMY
//#define REMOTE_UART
//#define REMOTE_CRSF
```

In `remote.h` you could include new remotes like RemoteAdc or RemotePwm, etc.

```
	void RemoteCallback(void);		// must be implemented by all remotes
	void RemoteUpdate(void);			// must be implemented by all remotes

	#ifdef REMOTE_UART
		#include "../Inc/remoteUart.h"
	#endif
	#ifdef REMOTE_CRSF
		#include "../Inc/remoteCrsf.h"
	#endif
	#ifdef REMOTE_DUMMY
		#include "../Inc/remoteDummy.h"
	#endif
```

I have tested REMOTE_UART and REMOTE_DUMMY on my 2.0 test setup with only the one board running as the master. Have burned the second board some time ago so i at the moment can not test the master-slave communication between two boards.

Next step for a more complete firmware would be an abstract `pilot.h` _class_  and implementing _classes_ `pilotPwm.c`, `pilotSpeed.c`, `pilotTorque.c`, `pilotDistance.c` , etc. that contain the closed loop for various tasks.
 
But as today i also finished the I2C control of the new SimpleFOC firmware for these split boards, this next step might be for you who first put the closed loop into main.c and then like with WizzardDr i move the code into such a pilot system.

You can also test the new SimpleFOC firmware with an I2C remote (still called pilot there). This is the arduino code: https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC/tree/main/arduino%20examples/ESP32_Mini_S2_I2c_Master

Ideas welcome.

**Here in this issue please test and report the new REMOTE_CRSF.**

For problems with the new SimpleFOC firmware, open issues in that other repo of mine.

Thanks again to WizzardDr for contributing !!!

Andrii067, 2024-12-10:
Hi I tried to make coaxing through crossfire but no changes 
I tried to change places Rx tx did not help 
I took a known working Rx tx coaxed them (they support communication) put in the transmitter output crossfire and input channels I have mood but no reaction to change hoverboard put firmware define REMOTE_DUMMY wheel spins both ways but from crossfire does not work maybe there is some instruction and debugging methods ?

RoboDurden, 2024-12-11:
As you are posting in the EspressLRS issue, maybe your remote control is the cause of the poor serial communication.
So try TestSpeed.ino first: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/tree/main/Arduino%20Examples/TestSpeed

Maybe your hoverboard control board has a faulty dcdc converter which generates the bad noise. Try the other board to see if the crossover talk disappears.

Get yourself an oscilloscope to see the noise: https://www.aliexpress.com/item/1005006019697163.html

Andrii067, 2024-12-11:
Hello, I just saw your answer today, there were quite a lot of tests using UART and esp32, I managed to make a connection and output speed tests, it works and didn't cause any problems, even somehow with huge delays of 10 seconds, I managed to do what I connected to the esp32 crossfire receiver and actually controlled it using the remote control, but the delays were too huge, I don't really understand why it didn't want to connect directly, I was already thinking about making the receiver give ppm at the output, but it doesn't know how, it can only do sbus, mavlink, crsf and some serial rx and tx, if anything, I have an xf nano 50 receiver

eluchhh, 2025-02-06:
Hello, I need help. I want the hoverboard's controller to be managed by the ELRS receiver. How can I do this?

RoboDurden, 2025-02-06:
First rule of project mayhem: you do not ask questions.

eluchhh, 2025-02-06:
"Sorry, but I couldn't figure out your project, that's why I'm asking."

I-hate-2FA, 2025-02-07:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/main/HoverBoardGigaDevice/Inc/config.h

it feels like you are from gen1 though,just sayin

eluchhh, 2025-02-07:
Thank you, I am a beginner in this matter.