RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #27: Help identify board layout (GD32F130C8T6)

miro-ux, 2023-09-25:
Hi, I'm a complete noob. 👋🏻

I can't tell if  any of the layouts are the one I have.

I was able to unlock the board and flash it from Keil but now I'm afraid to power it on due to now knowing which layout to config.

The MCU is a GD32F130C8T6

I'm also not sure if I got the master/slave right.

Master:
[image]

The Slave looks identical except it has the Switch+Switch LED I think and has no charger port.
[image]

MOSFET + MCU view:
[image]

Please let me know if you need anything else to help me identify or what should I do if it's a new layout.

RoboDurden, 2023-09-26:
Good morning, 10:06 here in Germany :-)
Please help with the three gate driver chips, i can not really read the code:
[image]

I have assigned the new layout 2.11 :-)

Do NOT test a new layout without a 2 A cc (constant current) power supply !!

Most layouts share the same 6 mosfet gate driver pin because the GD32F130 only has one advanced timer whose output pins are very limited.

You need to trace the three hall inputs. Very nice double image you made :-)

Also important ist the onOff push button and the HOLD output which keeps the power on when onOff is realeased.

```
// Self hold defines
// important pin keeps the mosfet open after the on/off button got pushed !
// main.c:306: gpio_bit_write(SELF_HOLD_PORT, SELF_HOLD_PIN, SET); 
// and turns off power on Shutdown:
// main.c:513:	 gpio_bit_write(SELF_HOLD_PORT, SELF_HOLD_PIN, RESET); 
#define SELF_HOLD_PIN		TODO_PIN
#define SELF_HOLD_PORT	TODO_PORT

// Button defines
// on/off (POW) push-button. So also a connection (i guess with some smd resistor in between) to a MCU pin.
// main.c:457: if (gpio_input_bit_get(BUTTON_PORT, BUTTON_PIN)) 
#define BUTTON_PIN	TODO_PIN
#define BUTTON_PORT	TODO_PORT
```

I will add a new defines_2-11.h the next hours.

Welcome :-)

RoboDurden, 2023-09-26:
all three hall sensors seem to go underneath the mcu :-(
[image]

If you do not want to probe with a multimeter and a needle, we have to find the mcu pins by exclusion.
Try to identify the three led panel. I have test code to link them to the hall sensors so they blink when rotating the motor manually:

#define TEST_HALL2LED	// led the 3-led panel blink according to the hall sensors

RoboDurden, 2023-09-26:
I guess the led pins map to some of
```

PA14
PA15 = PORTA PIN15
PB3
PB4
PB5

PB8
PB9
or
PF6
PF7
```
[image]

If you have a max 2A CC power supply ( or CC step down dcdc module) you can try and error..

RoboDurden, 2023-09-26:
Okay you have to update your downloaded code and uncomment your layout in config.h :
```
	#define LAYOUT_2_0	// https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2
	//#define LAYOUT_2_1	// https://github.com/krisstakos/Hoverboard-Firmware-Hack-Gen2.1
	//#define LAYOUT_2_2	// motor is spinning but needs a push to startup :-/
	//#define LAYOUT_2_3	// 2023/09/19 motor does not run stable (sound like skipping steps in a stepper motor but it is rotating). mosfets heating up :-(
	//#define LAYOUT_2_4	// NOT READY !!! https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/3
	//#define LAYOUT_2_11	// NOT READY !!! https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/27
```

In remoteDummy.c uncomment line 19:

`		//speed = 0;	// uncomment this to turn the motor manually when TEST_HALL2LED
`

Then turning the motor by hand might already light up some leds.

UART header might also already work.
So you can load the arduino TestSpeed.ino to and ESP32 (or orther) and might receive the log struct data.

You will need a permanent on switch instead of an the push button to keep the bord powered until you have traced the two pins i mentioned earlier.

Of course there is the risk that you will break your board.
With a 2A cc power supply the risk is acceptable - i think.

RoboDurden, 2023-09-26:
The half bridge gate drivers are the IRS2002 and seem to be compatible with the firmware:
[image]

RoboDurden, 2023-09-26:
There are no photo-diodes / light barriers on the backside AND id do not see a header to plugin two connectos with exteran photo-diodes ??

miro-ux, 2023-09-26:
Hi @RoboDurden! Thanks a lot for your quick response. 

I got good news -- bad news:

I flashed the layout following your instructions.
The good news is that i was also able to power it with 2A cc limit (at 12V) and I got a single green light on the LED strip. No magic smoke... 😊 Turning the motor with halls does nothing to leds though... 
[image]

Also, I did trace that all the battery status LEDs and they are all here:
[image]

I also changed it in the layout:

```
// LED defines, colors probably mismatch !
#define LED_GREEN 			GPIO_PIN_15
#define LED_GREEN_PORT 	GPIOA
#define LED_ORANGE 			GPIO_PIN_3
#define LED_ORANGE_PORT	GPIOB
#define LED_RED 				GPIO_PIN_4
#define LED_RED_PORT 		GPIOB

#define UPPER_LED_PIN 	GPIO_PIN_5
#define UPPER_LED_PORT 	GPIOB
#define LOWER_LED_PIN 	GPIO_PIN_6
#define LOWER_LED_PORT 	GPIOB
```

The bad news is that I can no longer connect to the target to flash the changes 
I get `No Target Connected` & `Error: Flash Download failed  -  Target DLL has been cancelled`.
The ST-Link is connected... it's the board that doesn't work anymore. 😟

I tried everything. Restarting, reinstalling St-link USB drive... updated it's firmware... Nothing works!
The green light is still there when i supply 12v and short the switch. Just can't comm with the board.
St-link Utility says to try different options like  `Connect under reset` but nothing works.

Any idea what could it be and how to unbrick it if it's not fried?
Thank you!

RoboDurden, 2023-09-26:
If you are running Windows, make a batch file in the st-flash-exe  folder:

st-flash --connect-under-reset write hoverboard.bin 0x8000000.bat containing the single line
`
st-flash --connect-under-reset write hoverboard.bin 0x8000000`

https://pionierland.de/hoverhack/gen2/ReadyToFlash/st-flash%201.7.0%20Windows%20x86%2064bit.zip

locate your latest hoverboard.bin file in HoverBoardGigaDevice/BinariesToFlash

If this does not work, you need the NREST pin
Or simply turn hoverboard off and on again and then click the enter key to start the bat file..

Unplugging the st-link-v2 of course also helps sometimes..

Try tracing the three hall inputs !

RoboDurden, 2023-09-26:
You can also rework the RST pin of these cheap ST-Link-V2 dongles to work for our SWD protocol (not the UART):
[image]

miro-ux, 2023-09-27:
I see. My cheap clone is a little different so I used [this article](https://research.kudelskisecurity.com/2020/04/15/swd-part-3-swo-and-nrst/) since it looks exactly the same. Surprisingly it uses exactly the same MCU as the board (GD32F130C8T6):
[image]

What I do not get if I should now connect the NRST of the ST-Link to the NRST of the board?

[image]
Otherwise it's just not connected to anything right?

RoboDurden, 2023-09-27:
There mostly is a smd resistor close to that NREST pin that you simply hold a male prototyping cable connector to:
[image]

miro-ux, 2023-09-30:
Thanks for all your help again! I've had no success with connecting the RST wire but it may be due to different issue.

After a lot of trial and error I think the cheap clone may be faulty and bending pushing it up and down in the USB port worked the best.

I wrote this `auto-flash.bat` which basically spams the st-flash command and helped me speed up the flash process and solving `NRST is not connected` error:

```
@echo off

:repeat
REM Remove REM below to add 1 sec delay 
REM timeout /t 1 /nobreak > nul
color 08
cls
st-flash --connect-under-reset write hoverboard.bin 0x8000000 || goto :repeat

color 0A
echo:
echo ================================
echo ^|       FLASH SUCCESSFUL       ^| 
echo ================================
echo:
echo Press Ctrl+C to exit or any other key to flash again.
pause > nul

goto :repeat
```
(It goes into ./HoverBoardGigaDevice/BinariesToFlash/)

That's what it looks like before connecting the ST-Link dongle and right after it's connected it flashes it successfully without the NRST wire

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/7188826/e4e6881d-ab44-4d0a-bc03-0848d438ed7a

I can now see 2 leds flash when I rotate the motor by hand:

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/7188826/ca72d5b7-8165-4c95-afe3-77b00d85cd59

I couldn't get the middle hall/led work and I tried every single pin:
```
#define HALL_A_PIN	GPIO_PIN_1	// confirmed
#define HALL_A_PORT	GPIOA
#define HALL_B_PIN	GPIO_PIN_2	// unknown - ??????
#define HALL_B_PORT	GPIOA
#define HALL_C_PIN	GPIO_PIN_0	// confirmed
#define HALL_C_PORT	GPIOA
```

There is no issue with the led itself because I can change the Red, green, orange and the led works so it's either the wrong hall pin or something else overwriting it?

How do I test the motor if it works? I've tried with `speed=100` in `remoteDummy.c` and commenting out `//#define TEST_HALL2LED` but it doesn't spin. It only makes it a little harder to rotate the motor so it is working I think but can't get it to spin. 

What should I try next?

RoboDurden, 2023-10-01:
It is a bit risky to also connect the 3.3V from st-link to hoverboard. Better only power the board with a 2A CC power supply.
Please mount the motor so it can rotate freely. You will rip some crucial cables if the motor gets a full speed command.
speed can go from -1000 to +1000. So 100 might not be enough to start the motor.
I usually test with 300.

Which led is not working ? And make sure the other two have the correct color naming:

```
// LED defines, colors probably mismatch !
#define LED_GREEN 			GPIO_PIN_15
#define LED_GREEN_PORT 	GPIOA
#define LED_ORANGE 			GPIO_PIN_14
#define LED_ORANGE_PORT	GPIOA
#define LED_RED 				GPIO_PIN_3
#define LED_RED_PORT 		GPIOB

#define UPPER_LED_PIN 	GPIO_PIN_4
#define UPPER_LED_PORT 	GPIOB
#define LOWER_LED_PIN 	GPIO_PIN_5
#define LOWER_LED_PORT 	GPIOB
```

disable the following lines by adding // before each:

#define DEBUG_LED_PIN 	LED_RED
#define DEBUG_LED_PORT 	LED_RED_PORT

Then the DEBUG_LedSet macros  in defines.h is defined to nothing and some code somewhere in the project will not overwrite the LED_RED behavior:

```
#ifdef DEBUG_LED_PIN
  #define DEBUG_LedSet(bSet){gpio_bit_write(DEBUG_LED_PORT, DEBUG_LED_PIN, bSet);}
#else
  #define DEBUG_LedSet(bSet)
#endif
```

If `LED_ORANGE` is not working than hit `strg+shift+f` and search the entire project for LED_ORANGE.
Then you see if the `TEST_HALL2LED` code gets overwritten which causes the `LED_ORANGE` not to blink like the other two.

The motor should spin nicely with `TEST_HALL2LED` being active.

RoboDurden, 2023-10-01:
Nice bat file ! Thank you !

RoboDurden, 2023-10-01:
And please close the motor when sending speed commands. It can not spin freely with the backside bearing missing.

RoboDurden, 2023-10-03:
@miro-ux your bat file is not behaving like in your nice video :-(
It simply repeats to flash forever. I need to remove the REM and increase the time to 5 seconds to pull the plug of the st-link usb dongle.
That is not really of any use. We need to try flashing very fast until it succeds and then get your nice success message. 
Your `|| goto :repeat` does not work for me.

```
@echo off

:repeat
REM Remove REM below to add 1 sec delay 
timeout /t 5 /nobreak > nul
color 08
cls
st-flash --connect-under-reset write firmware.bin 0x8000000 || goto :repeat

color 0A
echo:
echo ================================
echo ^|       FLASH SUCCESSFUL       ^| 
echo ================================
echo:
echo Press Ctrl+C to exit or any other key to flash again.
pause > nul

goto :repeat
```

RoboDurden, 2023-10-03:
Okay, i get flash error -1 :-/

```
st-flash 1.7.0
2023-10-03T12:01:09 WARN common.c: NRST is not connected
2023-10-03T12:01:09 INFO common.c: F1xx High-density: 64 KiB SRAM, 256 KiB flash in at least 2 KiB pages.
file firmware.bin md5 checksum: cd6e8fd7c3d1d8cf72a50efc990c350, stlink checksum: 0x0035c6bf
2023-10-03T12:01:09 INFO common.c: Attempting to write 33340 (0x823c) bytes to stm32 address: 134217728 (0x8000000)
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08000000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08000800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08001000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08001800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08002000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08002800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08003000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08003800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08004000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08004800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08005000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08005800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08006000 erased
2023-10-03T12:01:09 WARN common.c: Failed to unlock flash!
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08006800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08007000 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08007800 erased
2023-10-03T12:01:09 INFO common.c: Flash page at addr: 0x08008000 erased
2023-10-03T12:01:09 INFO common.c: Finished erasing 17 pages of 2048 (0x800) bytes
2023-10-03T12:01:09 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-10-03T12:01:09 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-10-03T12:01:09 INFO flash_loader.c: Clear DFSR
2023-10-03T12:01:09 ERROR flash_loader.c: Write error
2023-10-03T12:01:09 WARN flash_loader.c: Loader state: R2 0x800 R15 0xFFFFFFFE
2023-10-03T12:01:09 WARN flash_loader.c: MCU state: DHCSR 0x3000B DFSR 0x8 CFSR 0x10000 HFSR 0x40000000
2023-10-03T12:01:09 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1

```

RoboDurden, 2023-10-03:
Sorry for the confusion. I think i simply had to reboot my Laptop. Windows :-/

rhodyboland, 2024-05-23:
Hi I have this board also from a razor hovertrax. I worked on this last year and am revisiting now.
I have a motor spinning using these defines:
// Self hold defines
#define SELF_HOLD_PIN GPIO_PIN_2
#define SELF_HOLD_PORT GPIOB
// Button defines
#define BUTTON_PIN GPIO_PIN_14
#define BUTTON_PORT GPIOC
// to shutdown main.c:485 changed to:
if (gpio_input_bit_get(BUTTON_PORT, BUTTON_PIN))
		{
      while (gpio_input_bit_get(BUTTON_PORT, BUTTON_PIN)) {}
			ShutOff();
    }

// Hall sensor defines
#define HALL_A_PIN GPIO_PIN_0
#define HALL_A_PORT GPIOA
#define HALL_C_PIN GPIO_PIN_11
#define HALL_C_PORT GPIOB
#define HALL_B_PIN GPIO_PIN_1
#define HALL_B_PORT GPIOA

// Brushless Control DC (BLDC) defines
// Channel G
#define RCU_TIMER_BLDC RCU_TIMER0
#define TIMER_BLDC TIMER0
#define TIMER_BLDC_CHANNEL_G TIMER_CH_2
#define TIMER_BLDC_GH_PIN GPIO_PIN_10
#define TIMER_BLDC_GH_PORT GPIOA
#define TIMER_BLDC_GL_PIN GPIO_PIN_15
#define TIMER_BLDC_GL_PORT GPIOB
// Channel B
#define TIMER_BLDC_CHANNEL_B TIMER_CH_1
#define TIMER_BLDC_BH_PIN GPIO_PIN_9
#define TIMER_BLDC_BH_PORT GPIOA
#define TIMER_BLDC_BL_PIN GPIO_PIN_14
#define TIMER_BLDC_BL_PORT GPIOB
// Channel Y
#define TIMER_BLDC_CHANNEL_Y TIMER_CH_0
#define TIMER_BLDC_YH_PIN GPIO_PIN_8
#define TIMER_BLDC_YH_PORT GPIOA
#define TIMER_BLDC_YL_PIN GPIO_PIN_13
#define TIMER_BLDC_YL_PORT GPIOB

RoboDurden, 2024-05-23:
Thanks for your defines.

You have an old source code. I have moved to PB14 nstead of GPIOB and GPIO_PIN_14 quite long ago. So the Uart protocols might no longer be compatible!

AnthonyLia, 2025-06-23:
Hello. I have this board and have tested the 2.1.9 master dummy bin in BinariesToTest. Nothing happened except the motor brakes when I "turn it on" or hold the power button. I built my own single dummy bin for 2.1.9 by changing LAYOUT to 9 in config.h; I did leave LAYOUT_SUB as 1 as it didn't look to me like that's going to mess anything up. I also did select the correct F130 environment in PlatformIO.
The board did have trouble connecting, and I thought it was the NRST issue, but it finally flashed my bin by unplugging STLink and replugging a few times with the power button held. Now with the new bin flashed, it's the same behavior, with maybe a  bit more braking/holding resistance. No dummy spinning. I am using a 10A power supply.

Where do I start on this? The motor is known to work. I did try to verify PB0/PB1/PA5 for hall sensor pins on the board, but found no continuity to any of the MCU pins when the board has no power and didn't want to risk shorting pins when it's powered up (not the most precise fingers/dexterity). The board is exactly the same as the one in this issue though.

RoboDurden, 2025-06-23:
set the power supply to max 2A !!!
With RemoteDummy, the three led should blink according to the motor roation.
platformio support is still experimental. 
You might want to install the Keil IDE..
You could compile with debug information and then set lots of iBug=1 , 2,3,4 in the main() function and check with StmStudio if the code starts executing. https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Tips-for-programming,-debugging-and-flashing

With platformio you may be able to debug step by step. Have not tested that yet.

AnthonyLia, 2025-06-23:
Thanks for the quick reply. It was much simpler, the defines in your repo still had PB0/PB1/PA5, but for this board it's definitely PA0/PA1/PB11 like @rhodyboland found

AnthonyLia, 2025-06-25:
How easy is it to find the ADC pins on these boards? I'd like to find the current/voltage ones so the board can have a little more protection if possible

AnthonyLia, 2025-06-27:
Okay, just to save time, can you confirm if you mean these autodetect binaries?

[image]

It says they were last committed last year, so not sure if they have the update that fixed the floating point issue you're talking about in autodetect

rhodyboland, 2025-07-09:
Hi, revisiting this again. I've updated to the latest code. Motor spinning in dummy mode (SINGLE). Power button and latch working, buzzer working. Here is my defines_2-1-9.h:
```
// from https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/27

#if defined(SLAVE) || defined(SINGLE)	// this layout has buzzer on the slave board !
	#define HAS_BUZZER
#endif
/* GD32F130 48pin possible IO pins: 
	C13 C14 C15 F0 F1 A0 A1 A2 
	A3 A4 A5 A6 A7 B0 B1 B2 B10 B11
	B12 B13 B14 B15 A8 A9 A10 A11 A12 A13 F6 F7
	A14 A15 B3 B4 B5 B6 B7 B8 B9 
	
	mostly used for 6 BLDC mosfet pins: B13 B14 B15 A8 A9 A10
	mostly used for USART0: B6 B7
	mostly used for USART1: A2 A3
	ST-Flash pins: A13 A14 (also used as green and red led on 2.2)
	
	so mostly available for other use:	
	C13 C14 C15 F0 F1 A0 A1 A4 A5 A6 A7 B0 B1 B2 B10 B11 B12 A11 F6 F7 A12 A15 B3 B4 B5 B8 B9 
	so available for analog input:
	A0 A1 A2 A3 A4 A5 A6 A7 B0 B1 	
*/

// LED defines, colors probably mismatch !
// -------------------- haven't looked at these - my leds are missing
#define LED_GREEN 			PA5
#define LED_ORANGE 			PA14
#define LED_RED 				PB3

#define UPPER_LED 	PB4
#define LOWER_LED 	PB5
// --------------------

// Mosfet output
// seems to be an ordinary LED output ?
// led.c:91	gpio_bit_write(MOSFET_OUT_PORT, MOSFET_OUT, counter_Blue >= setValue_Blue ? RESET : SET); 
//#define MOSFET_OUT TODO_PIN		// TODO

// Brushless Control DC (BLDC) defines
#define TIMER_BLDC_PULLUP	GPIO_PUPD_PULLUP	// robo, based on Herleybob:defines.h
// Channel G
#define BLDC_GH		PA10		// channels G=green and Y=yellow swopped compared to 2.0
#define BLDC_GL		PB15
// Channel B
#define BLDC_BH		PA9
#define BLDC_BL		PB14
// Channel Y
#define BLDC_YH		PA8
#define BLDC_YL		PB13

// Timer BLDC short circuit emergency shutoff define
// Is initialized here but never used somewhere else in code.
// setup.c:176	gpio_mode_set(TIMER_BLDC_EMERGENCY_SHUTDOWN_PORT , GPIO_MODE_AF, GPIO_PUPD_NONE, TIMER_BLDC_EMERGENCY_SHUTDOWN);  
//#define TIMER_BLDC_EMERGENCY_SHUTDOWN		TODO_PIN	// TODO

// Hall sensor defines
// mateuszfcg tested with PA1,PA2,PA0 
#define HALL_A	PA0	// rhody tested
#define HALL_B	PA1	// rhody tested
#define HALL_C	PB11	// rhody tested

// GD32F130 USART0 TX/RX:	(PA9/PA10)AF1	, (PB6/PB7)AF0 , 	(PA2/PA3)AF1 , (PA14/PA15)AF1 GD32F130x4 only!
// GD32F130 USART1 GD32F130 TX/RX: (PA14/PA15)AF1 , (PA2,PA3)AF1	, (PA8/PB0)AlternateFunction4
#define USART1_TX		PA2 // both usart1 and 0 are these pins
#define USART1_RX		PA3

// ADC defines
// ----- haven't looked at these yet either
#define VBATT	PA4
#define CURRENT_DC	PA6
// -----

// Self hold defines
// important pin keeps the mosfet open after the on/off button got pushed !
// main.c:306: gpio_bit_write(SELF_HOLD_PORT, SELF_HOLD, SET); 
// and turns off power on Shutdown:
// main.c:513:	 gpio_bit_write(SELF_HOLD_PORT, SELF_HOLD, RESET); 
#define SELF_HOLD	PB2	// rhody tested

// Button defines
// on/off (POW) push-button. So also a connection (i guess with some smd resistor in between) to a MCU pin.
// main.c:457: if (gpio_input_bit_get(BUTTON_PORT, BUTTON)) 
#define BUTTON	PC14 // rhody tested

#ifdef HAS_BUZZER
	// Buzzer defines
	#define BUZZER	PB9 // rhody tested
#endif

//#define CHARGE_STATE TODO_PIN

// photo diodes / light barriers on the backside
//#define PHOTO_L		TODO_PIN
//#define PHOTO_R		TODO_PIN
```

The most annoying issue so far is the firmware seems to disable going into reset/boot mode when holding down the power button. The firmware still seems to not run while the power button is held - but you can't connect to the GD32. My work around is as you suggested above, connect the rst pin on my st-link. 

I hold a jumper from the rst on st-link to the resistor that is connected to NRST on the GD32. (As you mentioned above)

[image]

If I use st-link to erase the chip it will program without this. Just holding the button.

I've identified that PF0 Is connected to NRST if this helps.

RoboDurden, 2025-07-09:
Bridging the on off button is not for entering a reset mode but simply to prevent board from turning off during flashing.

This forward stuttering also for BLDC_BC ?

rhodyboland, 2025-07-09:
Ah ok, I see. I will solder a pin to the nrst to make it easier, with keil set to connect under reset and hw reset it programs nicely with the nrst connected.

Yes it works alot better in BLDC_BC. Both directions smooth. The wheel velocity is also a lot higher for the same 1000 command.
I'm new to sine control - does this indicate the mapping is different?

RoboDurden, 2025-07-09:
sorry to hear that my new BLDC_SINE does not work for you.
It might not even be your 2.1.9 board but the bldc motor having the three hall sensors not perfectly aligned..

There is a direction related offset of +- 30° in bldcSine.c that you could play with:

`				iOffset = sectorChange < 0 ? 390 : 330;		// +-30° +360° to prevent negative angle
`

that is 360+30 and 360-30
Would be nice if you test 20, 10, and 0.
If you find a better setup, continue with 5° steps.

Mabe You could also download and install StmStudio an see the stuttering in realtime: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Tips-for-programming,-debugging-and-flashing
You should monitor `sector10x` which shows you the 60° hall steps, scaled by 10 to be visibile in the 0°-360° window of `angle_idx`.

RoboDurden, 2025-07-09:
But these repeating steps in the time from one hall step to the next (`hall_time_step`) indicates that my  hall sensors are not perfectly in place:

[image]

The are 6 hall change events happening when one of the 3 hall sensors go on or off.
So between step 1 to 2 and 4 to 5, my motor hardware as an offset, that could be taken into account.
But the low pass is doing a good job already.

So i wonder, what your StmStudio plot looks like.

RoboDurden, 2025-07-09:
okay, bug resolved, repo is working again.

rhodyboland, 2025-07-10:
This is still previous commit before all changes yesterday.
Clockwise (positive speed, rough direction). Note that at speed 100 it is smooth in both directions. 
This is at 200 where it is pulsing. I'll find the transition point next.

[image]

Slo-mo video:

https://github.com/user-attachments/assets/4e3dcdfb-9116-4beb-a67d-462bf07b2d1c

Anti-clockwise (negative, smooth) -200 speed.

[image]

Let me know if you need more data. I'll change to the latest commit also.

RoboDurden, 2025-07-12:
@rhodyboland you have changed the pin definitions for the three phases :-/
```c++
#define BLDC_GH		PA10		// channels G=green and Y=yellow swopped compared to 2.0
#define BLDC_GL		PB15
// Channel B
#define BLDC_BH		PA9
#define BLDC_BL		PB14
// Channel Y
#define BLDC_YH		PA8
#define BLDC_YL		PB13

// Hall sensor defines
// mateuszfcg tested with PA1,PA2,PA0 
#define HALL_A	PA0	// rhody
#define HALL_B	PA1	// rhody
#define HALL_C	PB11	// rhody
```

But when i look at the photo of 2.1.9 here at the top and compare it to 2.1.0, then the phase colors are NOT changed.
Please upload photos of your boards.

Your hall definitions are also completely different from the current ones:

```c++
// Hall sensor defines
// mateuszfcg tested with PA1,PA2,PA0 
#define HALL_A	PB0	// robo from front+back-photo
#define HALL_B	PB1	// robo from front+back-photo
#define HALL_C	PA5	// robo from front+back-photo
```

If your three motor phase cables are not green-blue-yellow (like on the photos), than swapping HALL_A and HALL_C might do the same job.
The 6 pins for the motor phases are identical for 95% of all boards, so i am hesitant to follow your pin definitions.
Especially when there are 2.1.9 boards that do have the green-blue-yellow order.

AnthonyLia, 2025-10-05:
Two things:
1. I believe the Arduino example is incorrectly reading amperage. I put my firmware’s current limit to 12A and using TestSpeed I slightly gripped the motor while spinning to add resistance and got an Arduino Serial Monitor reading of 13.6A. It seems the values are 1/10th of what the Arduino is printing (meaning I only induced 1.36A) as I verified with my power supply’s digital display.
2. Is there any documentation of exactly how to command over UART? I added some print statements to see what iSpeed, iSteer, and wState are, and I can’t figure out exactly what iSteer is and if it’s needed. I plan to command my motors with a Raspberry Pi so I’m guessing my UART-send code will be similar but different to yours. I want to know what exactly the GD32F130 is expecting and if it needs wState (which I’m assuming is “wheel state”) and if it needs iSteer, or simply a speed can be commanded?

RoboDurden, 2025-10-06:
You did not upload photos of your boards :-(

Please copy src files to ChatGpt and get answers quicker.