RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #28: Yet another Layout?

TheJP, 2023-09-28:
Hello :)

Just found this repo and not sure if my boards resemble any of the already provided layouts.

Could someone maybe point out which of the two white connectors is meant for the on/off button? The other should be an LED port, but confusingly uses the exact same connector. (I'm a novice in hardware but an experienced programmer.)

# Probably Master (Side where the on/off button is located):

Chip is a: GD32F103

[image]

# Probably Slave (Other Side)

[image]

RoboDurden, 2023-09-28:
The onoff button is probably to the right where the dcdc step down converter for the 5V and 3.3V is generated.
The first board with the buzzer is probably the slave and the second one with the dcdc step down components is the master.
So there i guess you should plug in the onoff button.

Please make a closer photo of the 48 pin MCU so we can see if it is a GD23F130 or some other chip.

TheJP, 2023-09-28:
Both boards seem to be nearly identical (I didn't find a difference so far). I guessed the top picture to be the master, because the on/off button is located on that side of the hoverboard.

> The onoff button is probably to the right where the dcdc step down converter for the 5V and 3.3V is generated.

Thanks for the hint. So the top right next to the buzzer would be the on/off plug?
It has to be one of those two-pin plugs (top or bottom of the board, just unsure which one), And the on/off has to be connected to the board on the first image.

> Please make a closer photo of the 48 pin MCU so we can see if it is a GD23F130 or some other chip.

It reads GD32F130 directly on the MCU. I tried creating a better picture, but my phone cannot focus when I go any closer than the following. But I think on that picture it is kind of readable. (Both boards have the same MCU, this is the board from my first picture.)

[image]

RoboDurden, 2023-09-28:
Sorry i rotated the photos: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/Schematics_2.12

This is the dcdc hardware of the master (without the buzzer) and probably the onoff switch:
[image]

RoboDurden, 2023-09-28:
Yes, a GD32F130C8 is compatible with this firmware and the simpleFOC firmware will also run on it.
Now you need to trace the pins.
hall sensor
onoff button
hold button
led buttons

Please make a high resolution photo of the backside and the fronside (with the cables pushed out of the way).

RoboDurden, 2023-09-28:
A sorry, this is no coil but simply the tranistor bent to the left :-(
Both boards have a dcdc step down:
[image]

So the board with the buzzer is the master and the onoff button should be to the left.

TheJP, 2023-09-28:
Both boards have a buzzer and both are directly connected to the power-source.

> So the board with the buzzer is the master and the onoff button should be to the left.

Is left near the buzzer or far from the buzzer?

RoboDurden, 2023-09-28:
when you disassemble everything make new photos.
hall sensor header is down, buzzer left, mosfets top. onoff persumably right.

TheJP, 2023-09-28:
Backside

[image]

RoboDurden, 2023-09-28:
and a nice photo of the frontside with cables pushed out of sight.

TheJP, 2023-09-28:
Wasn't sure which cables from the front were bothering you. So here are pictures where I temporarily disconnected the ports that lead to the other board. (Same board twice, 1. without flash, 2. with flash)

[image]

[image]

RoboDurden, 2023-09-28:
Okay, the mosfet pins mostly are the same for all gd32F130 boards.
And the hall pins seem to be PA0, PA1 and PA2
[image]

The 5 led pins most likely are PA15, PB3, PB4, PB5 and PB6

Now you need a cheap st-link-v2 dongle to unlock the boot loader and install the Keil ide to compile and test a new defines_2-12.h

And a constant current max 2A power supply (or step down converter with CC).

RoboDurden, 2023-09-28:
no i am not sure about the led.
But this enough for me for now.
good night from Germany

TheJP, 2023-09-28:
Thanks for looking into it! :)

I have an st-link-v2 USB dongle. For a power supply I have an original hoverboard li-ion battery pack.
Probably don't have time to test it out until next week. I'll write here for an update.

Good night from Switzerland

Hishamahmed97, 2024-03-08:
@TheJP @RoboDurden Hello guys,
Can you help how I can correctly build the keil project for this GD32F103 ?

RoboDurden, 2024-03-08:
And for sure do not post triple times :-((