RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #29: Support for board YFS-N2?

ltzen77, 2023-09-29:
Any change any of the existing code will support this one? says YFS-N2 on the back.

[image]
[image]

RoboDurden, 2023-09-29:
Very unlikely. 
I do not find any information about a LKS 32bit C6T8 MCU on the internet:
[image]

ltzen77, 2023-09-29:
ah of course - with a magnifier i read:  
LKS
LKS32MC
R11C6T8
2128

RoboDurden, 2023-09-29:
no compiler support i guess.

AILIFE4798, 2024-01-06:
please read datasheet
[Uploading 2210201730_LINKOSEMI-LKS32MC033H6P8_C5209379.pdf…]()

RoboDurden, 2024-01-06:
Thanks AiLife. Unfortunately i am still a beginner with this Keil IDE. So i do  not know how to add a new target.
Someone else did it with the GD32E230
And we also need a new target for the MM32SPIN05 :-/
See https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/22

AILIFE4798, 2024-01-06:
the datasheet i sent is for wrong IC
heres the correct one
[2110081130_LINKOSEMI-LKS32MC088C6T8_C2798416.pdf](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/files/13849969/2110081130_LINKOSEMI-LKS32MC088C6T8_C2798416.pdf)
i also know nothing about keil but heres the device package from manufacture website maybe itll help
[https://www.lksmcu.com/index.php/DevicePack/](url)

AILIFE4798, 2024-01-14:
anyone have this board,do not replace the mcu with gd32,it is pin compatiable but 5v,so dont do it in any case

RoboDurden, 2024-01-14:
You mean there is not 3.3V regulator on that layout but only the 5V regulator which directly powers the MCU ?

AILIFE4798, 2024-01-14:
yes you can see there is no ams1117 which would be on other board

AILIFE4798, 2024-01-14:
the mm32spin is also 5v tolerant ! please for your own good make sure the voltage before you do anything
[image]

RoboDurden, 2024-01-14:
Have you tested the mcu voltage ? If the board does not need 5V, it could go directly from the dcdc converter to 3.3V.

AILIFE4798, 2024-01-14:
ams1117 is only rated for 12v,so it must be regulated to 5v first,and the hall sensor always need 5v,so it have to be generated

RoboDurden, 2025-04-30:
@I-hate-2FA can give you more info on the LKS i think.
I would search the web for a LKS32 blink example.
And for sure ask https://chat.deepseek.com !

I-hate-2FA, 2025-04-30:
Use keil or pyocd

RoboDurden, 2025-04-30:
As I do not offer LKS32 binaries, I already assumed that he tries to compile and upload with Keil Ide. But that the st-link dongle might be the problem.

ltzen77, 2025-04-30:
the ambition with stm 32 cube was to see so that stlink could connect at all. which failet. but ill see if any of the others will work better, thanks

I-hate-2FA, 2025-05-01:
The dongle is not the problem, all core m devices can use it, if you used the correct device pack in keil

I-hate-2FA, 2025-05-01:
Because mm32 and lks is not direct clone, the register location is different, you need to use different device pack
If you still can't connect, the MCU is unlocked, you need to find a way to unlock, maybe can check the mm32 as refrence

RoboDurden, 2025-05-01:
If you want to make an autodetect binary for LKS32, you have to add another #ifdef clause in RemoteAutodetect.c that defines the available pins (digital and analog):

```
	#ifdef GD32F103	// no PF0 and PF1 on 48 pin package
		#define COUNT_PinDigital 29
		PinAD aoPin[COUNT_PinDigital] = {
				{PC13,"C13",0}	,{PC14,"C14",0}	,{PC15,"C15",0}			,
				{PA0,"A0",1}		,{PB11,"B11",0}		,{PA4,"A4",1}		,{PA5,"A5",1}	,{PA6,"A6",1} 	,
				{PA7,"A7",1}		,{PB0,"B0",1}		,{PB1,"B1",1}		,{PB2,"B2",0}	,{PB10,"B10",0}	,
				{PA1,"A1",1}	,{PB12,"B12",0}	,{PA11,"A11",0}	,{PF6,"F6",0}	,{PF7,"F7",0} 	,
				{PA12,"A12",0}	,{PA15,"A15",0}	,{PB3,"B3",0}		,{PB4,"B4",0}	,{PB5,"B5",0}	,
				{PB6,"B6",0}		,{PB7,"B7",0}		,{PB8,"B8",0}		,{PB9,"B9",0}	,{PA2,"A2",1}	,
				{PA3,"A3",1}
				};

	#else
		#define COUNT_PinDigital
		PinAD aoPin[COUNT_PinDigital] = {
				{PC13,"C13",0}	,{PC14,"C14",0}	,{PC15,"C15",0}	,{PF0,"F0",0}	,{PF1,"F1",0}		,
				{PA0,"A0",1}		,{PB11,"B11",0}		,{PA4,"A4",1}		,{PA5,"A5",1}	,{PA6,"A6",1} 	,
				{PA7,"A7",1}		,{PB0,"B0",1}		,{PB1,"B1",1}		,{PB2,"B2",0}	,{PB10,"B10",0}	,
				{PA1,"A1",1}	,{PB12,"B12",0}	,{PA11,"A11",0}	,{PF6,"F6",0}	,{PF7,"F7",0} 	,
				{PA12,"A12",0}	,{PA15,"A15",0}	,{PB3,"B3",0}		,{PB4,"B4",0}	,{PB5,"B5",0}	,
				{PB6,"B6",0}		,{PB7,"B7",0}		,{PB8,"B8",0}		,{PB9,"B9",0}	,{PA2,"A2",1}	,
				{PA3,"A3",1}
				};
	#endif
```

When adding the LKS32 target to Keil, you have to create a new define like `LKS32MCR11`

[image]

@I-hate-2FA , are the 6 bldc mosfet pins an the uart pins the same for LKS32 ?

```
	mostly used for 6 BLDC mosfet pins: B13 B14 B15 A8 A9 A10
	mostly used for USART0: B6 B7
	mostly used for USART1: A2 A3
```

@ltzen77 , for sure, do only test with a 2A constant current power supply !!!

I-hate-2FA, 2025-05-02:
> If you want to make an autodetect binary for LKS32, you have to add another #ifdef clause in RemoteAutodetect.c that defines the available pins (digital and analog):

problem is the timer code may be different, in mm32 autodetect i do not initialize adc timer, only the necessary timer1 to drive motor

> are the 6 bldc mosfet pins an the uart pins the same for LKS32 ?

it is the same place but different wiring

[image]

[image]

p1.4-p1.9(pb4-pb9 if we follow the naming), but it alternate between ch0hi,ch0lo,ch1hi like this, it is way better for pcb routing this way id say

it is for LKS32MC08X , i dont know which one you have