RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #30: Layout 2.0 with GD32F130C8T6 not enough flash memory when attempting to flash binary code in stLINK UTILITY

mrWallyMan, 2023-10-02:
Hello, I successfully compiled a few firmwares (I think at least) however when I attempt to flash them to my split boards using the stLink utility I get an error stating that there is not enough flash memory on the device. The MCU on my boards is the GD32F130C8 which shows up in the stLink utility as a STM32F10xx Medium Density. It also shows up with only 64Kbytes of flash memory. I am trying to flash a few different firmwares onto them to no avail as they are all showing up in my file explorer in windows as 445KB (weiminshen99_gd32-hover-master-slave_master slave.bin AND weiminshen99_gd32-hover-master-slave_mw masterslave slave.bin) and 384KB (vedderb_bldc_hover-vesc_binary_2023-09-30_04_55_26.bin) and 765KB (vedderb_bldc_hover-vesc_binary_2023-09-30_04_55_26 hex.hex).

 

I’m not sure if the vesc firmware is even compatible with my boards but if I can get that to work on there this would be ideal but, at this point if I can get any of them to work on these split boards I’ll be incredibly happy. If you can help me make this work I’ll be more than happy to make a donation for your time. I’ll make an even larger donation if you can help me get the vesc firmware up and running on these boards as well.

 

Thank you in advance for any assistance you may provide and thank you so much for making all your work available open source. If you would like me to post this somewhere else online so the information is available to everyone else I would be more than happy to do so as well, just send me a link to where I should post and I will do so.

That was all copied from my email I had sent you.

I just now stumbled upon a bit of info stating that these particular boards are only compatible with UART input. Does this mean that I can control these with my flipsky VX3 remote by hooking it's receiver up VIA its UART wires? I'll sure hoping to be able to utilize this remote, that would be sweet!

RoboDurden, 2023-10-02:
You need to find the right layout of your board. Otherwise the pins are wrong and you can kill your hoverboard by flashing the wrong layout.

Regarding uart control via RC: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26

mrWallyMan, 2023-10-02:
Awesome, I can handle that I think. Now I just need to get past the whole only having 64k of flash memory so I can upload the compiled binary onto the boards

RoboDurden, 2023-10-02:
Please first state which layout you have.
And please post some photos of the board and of the plastic hoverboard casing.
So other people can see from the outside which layout is inside.
thanks !

mrWallyMan, 2023-10-02:
It's layout 2.0. I no longer have the outer shell, I took the hoverboard apart years ago
[image]
[image]
[image]
[image]

RoboDurden, 2023-10-02:
Yes you have the 2.0 :-)
So the binaries in https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/BinariesToFlash/
Should work for you.
Simply flash the master_test.bin to your master and the master motor should spin forward and backwards.
Then flash the master_slave.bin to the slave and connect the master-slave uart cable. Then both motors should spin forward and backwards.
Then install the arduino test code on some arduino or ESP and read the serial log data from the master board.
In the end you can flash the master.bin to the master and let the ESP control your two boards.

Good night (from Germany) and good luck.

mrWallyMan, 2023-10-03:
while trying to flash the firmware on both the master and the slave I get the following:

19:01:23 : Device ID:0x410 
19:01:23 : Device flash Size : 64KBytes
19:01:23 : Device family :STM32F10xx Medium-density
19:01:31 : Unexpected error
19:01:31 : Error occured during program operation!
19:01:31 : Programming error @ 0x08000000!
19:01:43 : Programmed memory Checksum: 0x001DE200

I wish I had more of the older style mainboards as I have never had any issues flashing firmware onto  those ones after I figured out how to compile custom firmware using platformio extension in visual studio.

RoboDurden, 2023-10-03:
Use st-flash.exe from https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/HoverBoardGigaDevice/BinariesToFlash

And/or rename your bin file into hoverboard.bin and use auto-flash.bat

mrWallyMan, 2023-10-04:
How do I use auto-flash.bat?

RoboDurden, 2023-10-04:
rename your bin file into hoverboard.bin

mrWallyMan, 2023-10-04:
I'll try that here shortly and report back. Thank you for all your help.

Is the vedder/vesc firmware compatible with any of the split main boards? It would be ideal if I can use vesc on.

RoboDurden, 2023-10-05:
Oh i see that by now the VESC rpeo in my online compile does successfully compile. But it only takes less then a second ??? It used to take 16 seconds - far longer than then the hoverboard firmwares.
The split hoverboards only have 64kB, some only 32 kB. The VESC.bin takes a massive 384 kB - which would be too much for the Gen1 boards that have 256 kB (i think).

I am not sure what benefits VESC would bring to hoverboards..

mrWallyMan, 2023-10-07:
If vesc won't fit on the hoverboard controllers what's the use having it as an option? I was hoping to be able to use the vesc firmware because my intentions are to use these hoverboard boards to make an electric skateboard.

RoboDurden, 2023-10-07:
Yet another stupid fun project
 while millions of others starve to death on this planet :-(

sp3swj, 2023-11-24:
Hi, few words about VESC, I have FLipsky VESC 75350 and this wirks with no problem with a lotnof motors. I tried to use motor from hoverboard with VESC an motor not work smooth and properly, even on VESC discord not found help yet

https://youtu.be/1wEMIYY7nd0?si=Gr08XAH1rIolYlbW

I use vizard, and manually dedtwct motor, modify parameter manualy - not success

Motorbis ok - works with DUMMY bin from github
https://youtu.be/O05wrJU0uIU?si=NHVWwcCB4Y4UUitK

VESC is ok too, because imtested in meantim
https://photos.app.goo.gl/dB49a7Hj1FmQ5SrFA

Summaries ;-)  i dont know how to force VESC to work smooth and quiet with such motor

Backemf voltage read from motor looks ok, sensors working ok

https://photos.app.goo.gl/GkbnJkxEygifJ9Xq7

Now continue with learning more abou  hacked firmware ;-) ... time to install Platformio ;-)

Few words about processor, in one board i kill C6 processor- Then I solder C8 - works ok - HEX DUMMY program loaded from "DOS" as described above.

i Noticed that oryginal software spin whell more smooth and silent than hacked ... time to analyse what method is silent.

DrHongos, 2024-02-22:
Hello builders! First of all thanks for all this work and congrats on its recent update! 
I am dealing with this boards and its my first time hacking a hoverboard, have a current situation out of my capabhilities..
Using firmware from https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesReadyToFlash
hoverboard 2.1.1 master Dummy.bin => works perfect
But then the 'interesting' ones:
hoverboard 2.1.1 master Uart.bin & hoverboard 2.1.1 single UartBus id0.bin & id1 => Error
After flashing any of the above, when i try to start the hoverboard just emits the starting sounds that are interrupted by a few beeps, green led turns into red and it shuts down.. if i press again the start button just turns the red led on.
Also, what are idX versions on UartBus?
Any chance you have some idea to continue? Thanks a lot!

RoboDurden, 2024-02-22:
Maybe you don't have 36volt battery and board shutting down.

UartBus needs uncommenting the define in Arduino code.

UartBus and Uart communicate with additional empty  Uart  header , not the master slave header.

AILIFE4798, 2024-02-24:
[image]
how to fix this

RoboDurden, 2024-02-24:
Is this a question ??
Seems to be unrelated to this issue.

AILIFE4798, 2024-02-24:
yes gd32f130 can erase cannot upload