RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #31: Question

Carte24306, 2023-10-23:
I'm working on board v2.0 but there is a lot of different firmware's. I was wondering if this is the newest firmware to use with an Arduino.

hoverboard 2.0 single UartBus id3.bin
hoverboard 2.0 slave.bin

RoboDurden, 2023-10-23:
You either choose a master and a slave binary for the two boards. Or a single UartBus with differed Ids for each board.

Master Server allows speed plus steer. With uartBus you have to calculate steer on the Arduino. See Arduino ide example.

Master Server at the moment for some reason is a bit choppy / stuttering.

Carte24306, 2023-10-23:
And just to make sure i do this right your saying

Board one gets
hoverboard 2.0 single UartBus id1.bin

and board 2 gets
hoverboard 2.0 single UartBus id2.bin

and this is the wiring
[image]

Then after I flash i just put the code on the Arduino and plug it in where I showed before
is this the right code?
[https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/Arduino%20Examples/TestSpeed/TestSpeed.ino](url)

If I missed anything important ples let me know and thank you for your time I doing this for a school project

RoboDurden, 2023-10-23:
Yes yes yes.
Please read the updated readme of this repo.
Thanks.

Always good to first test with a 2A cc constant current power supply or dcdc step down converter.

Carte24306, 2023-10-24:
so I have it wired up I ended up doing master and slave uart both boards are flashed and wired up but the issue I'm having is when I plug in the Arduino Uno and power it on it place a musical beep but that's it I tried hitting the power button and also holding it it did nothing?

and I ran rx to tx and tx to rx I didn't change any of the bin files and i didn't modify the code and I used the Test speed code
  Think you can help?

RoboDurden, 2023-10-25:
First flash the single_dummy on each board to make the motor spin without any Arduino UART control.

RoboDurden, 2023-10-25:
Sorry i had the wrong ESP32 tx pin 35 instead of the usual 37 in the TestSpeed.ino in the last update:     `HoverSetupEsp32(oSerialHover,19200,39,35);      // baud, rx, tx` - when testing my 2.0 test setup this morning the motor did not spin because it did not receive the tx data :-(