RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #32: Stuck with programming [Solved]

pacraf, 2023-10-25:
I will write this for other stuck with GD32F130 programming. 
Generally for me (board v.2.0) worked solution of unlocking chip using st-flash.exe under windows.
after that I sucesfully programmed board many times, until once it after "succesfull" programming, stopped to respond.
somechow uC worked, as it hold itself power (after using button), but it was not responding for programmer.

I tried the way of looped .bat file (where you believe to get luck and after xx powerup tries you should be lucky and programming will start. believe me, I tried hard. 

So thinking about exchanging GD chip, last hope is "NRST way".
And here my observation - it worked from first try.

Programmer I have (Nucleo board programmer) is the one where NRST signal is already available.
[image]
But for clones there is on this github also description how to wire it. This I did not tested.
Anyway, what is important I think that you do not have to desolder NRST pin of GD32F130 to connect this signal. (that was my understanding, that it is neccesary to desolder it from actual board layout, and wire it to NRST signal of programmer)

[image]

 Seeing that on boear this pin no 7 (NRST) is connected to resistor 10k, I soldered to this point (arrow black below). This is very easy to do (imagine desoldering one of this small pins from GD chip...) 

[image]

So here I have to say that:
- I have no idea if clone programmers will also work the same way (but it is worth trying)
- If I properly got generally NRST idea (with desoldering NRST pin from GD chip?), or it wans meant like I did. But as I did not found clear description - here you have tested way... 
most informative for me was this [link ]([JRomainG/GD32F130K6-hoverboard-hack: Hoverboard sideboard hack for GD32 F130K6 boards (github.com)](https://github.com/JRomainG/GD32F130K6-hoverboard-hack))

pacraf, 2023-10-28:
Maybe I have bad luck, maybe I am just noob, whatever the problem is, I could not access both boards. 
for anyone having errating programming experience, like this screen below, please reinstall your toolchain.
Suggested reboot did not helped, but SOMEHOW reinstall - indeed. 

```
> C:\stlink\bin>st-flash --reset write hoverboard.bin 0x8000000
> st-flash 1.7.0
> 2023-10-27T17:56:23 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
> file hoverboard.bin md5 checksum: 78fc13f7c1e91616f9c34d44c484b22f, stlink checksum: 0x002172f1
> 2023-10-27T17:56:23 INFO common.c: Attempting to write 22112 (0x5660) bytes to stm32 address: 134217728 (0x8000000)
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08000000 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08000400 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08000800 erased
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08000c00 erased
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08001000 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08001400 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08001800 erased
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08001c00 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08002000 erased
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08002400 erased
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08002800 erased
> 2023-10-27T17:56:24 WARN common.c: Failed to unlock flash!
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08002c00 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08003000 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08003400 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08003800 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08003c00 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08004000 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08004400 erased
> 2023-10-27T17:56:24 INFO common.c: Flash page at addr: 0x08004800 erased
> 2023-10-27T17:56:25 INFO common.c: Flash page at addr: 0x08004c00 erased
> 2023-10-27T17:56:25 INFO common.c: Flash page at addr: 0x08005000 erased
> 2023-10-27T17:56:25 INFO common.c: Flash page at addr: 0x08005400 erased
> 2023-10-27T17:56:25 INFO common.c: Finished erasing 22 pages of 1024 (0x400) bytes
> 2023-10-27T17:56:25 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
> 2023-10-27T17:56:25 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2023-10-27T17:56:25 INFO flash_loader.c: Clear DFSR
> 2023-10-27T17:56:25 INFO flash_loader.c: Clear CFSR
> 2023-10-27T17:56:25 INFO flash_loader.c: Clear HFSR
> 2023-10-27T17:56:25 INFO common.c: Go to Thumb mode
> 2023-10-27T17:56:25 ERROR flash_loader.c: Flash loader run error
> 2023-10-27T17:56:25 WARN flash_loader.c: Loader state: R2 0x0 R15 0x0
> 2023-10-27T17:56:25 WARN flash_loader.c: MCU state: DHCSR 0x0 DFSR 0x0 CFSR 0x0 HFSR 0x0
> 2023-10-27T17:56:25 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
> stlink_fwrite_flash() == -1
```

After all these problems I got, I can confirm for anyone interested 
- if you power board from 3,3V external source (I was not sure if onboard powersupply is OK/NOK) - you can program chip.
- after this unstable behaviour, I had to unlock again uC. How the hell it could happen it locked during loading .bin... 
I did it by stlink - like suggested. (someone wrote to even lock / unlock it twice, I went this way too)
[image]

after this , again beatufill, full of green - mind relaxing -  colour screen below... 

[image]

pacraf, 2023-10-28:
the only thing I do not understand, is that st-flash reports that NRST is not connected (see above WARN: NRST not connected), but it is. 
St-link reports it correctly (see second above stlink = connect under reset method applied) 
but somehow it works, so OK for me...

RoboDurden, 2023-10-28:
> Suggested reboot did not helped, but SOMEHOW reinstall - indeed.

What did you reinstall ?
ST-Link-Utility ? Keil ? Some driver ?

flashing a GD mcu with STM software can show something like a NRST not being connected..

pacraf, 2023-10-28:
In fact I reinstalled all tools. 
I used windows mechanism to restore to date xxx. then installed ST-Link and st-flash and openocd.
Keil is still not installed back... was not used for me until now.