RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #35: Another Layout?

Terramoto, 2023-11-08:
I got this hoverboard that has a board flagged tt-sd2.1 which appears to be an older version of layout 2.0. I would like to use this issue to also ask for information on how i would go about turning this into a feedback wheel with this firmware.

Master:
[image]

Slave:
[image]

pacraf, 2023-11-09:
if you suspect that it is similar (equal) to 2_0 then check pinout. I think this would be the way... then after seeing that it is like that, take secured (overcurrent protected) power supply and load firmware.

Terramoto, 2023-11-13:
> if you suspect that it is similar (equal) to 2_0 then check pinout. I think this would be the way... then after seeing that it is like that, take secured (overcurrent protected) power supply and load firmware.

Wasn't sure how to do it, got it over a light and started to trace. It does match with the 2.0. I'll flash the firmware and see how it goes. Thanks!

pacraf, 2023-11-14:
for test purpose the easiest is dummy firmware. 
from ready to flash folder take file 
[image]
and if you really have the same pinout - it will rotate wheel... you can load he sam file to both boards.
don't forget to power it safely - via power suppl with overcurrent limit set to 1A for example.

Terramoto, 2023-11-15:
> for test purpose the easiest is dummy firmware. from ready to flash folder take file [image] and if you really have the same pinout - it will rotate wheel... you can load he sam file to both boards. don't forget to power it safely - via power suppl with overcurrent limit set to 1A for example.

Unfortunately i was not able to flash it, i'm not sure if it's because of this but i just notice the chip on my board has half the specs on the 2.0 layout. It flashed part of the firmware, i believe it ran out of memory.
[image]
I have the c6 and the c8 is on the 2.0. Am i wrong?

pacraf, 2023-11-15:
You have to know that I am not expert.  But reading this GitHub project issues I understand that 32kB is ok for this firmware. Not enough for upcoming SimpleFOC. 
how do you load it?

Terramoto, 2023-11-16:
I used stm32cubeprogrammer, connected the stlink to board, read from the chip, opened the bin file and download it.

pacraf, 2023-11-16:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/30

Program like described there.

Terramoto, 2023-11-16:
st-flash.exe seems to have done the job:

```
st-flash.exe --connect-under-reset write hoverboard.bin 0x8000000
st-flash 1.7.0
2023-11-16T09:48:30 WARN common.c: NRST is not connected
2023-11-16T09:48:30 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
file hoverboard.bin md5 checksum: f3d6a2194d5d67fa05e8e6e73841bac, stlink checksum: 0x001e93b7
2023-11-16T09:48:30 INFO common.c: Attempting to write 20632 (0x5098) bytes to stm32 address: 134217728 (0x8000000)
2023-11-16T09:48:30 INFO common.c: Flash page at addr: 0x08000000 erased
2023-11-16T09:48:30 INFO common.c: Flash page at addr: 0x08000400 erased
2023-11-16T09:48:30 INFO common.c: Flash page at addr: 0x08000800 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08001000 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08001400 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08001800 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08002000 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08002400 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08002800 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08002c00 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08003000 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08003400 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08003800 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08003c00 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08004000 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08004400 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08004800 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08004c00 erased
2023-11-16T09:48:31 INFO common.c: Flash page at addr: 0x08005000 erased
2023-11-16T09:48:31 INFO common.c: Finished erasing 21 pages of 1024 (0x400) bytes
2023-11-16T09:48:31 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-11-16T09:48:31 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-11-16T09:48:31 INFO flash_loader.c: Clear DFSR
 21/ 21 pages written
2023-11-16T09:48:32 INFO common.c: Starting verification of write complete
2023-11-16T09:48:33 INFO common.c: Flash written and verified! jolly good!
```

Terramoto, 2023-11-16:
Hey, i'm just reopening this because i'm not sure if you want to do something with it.
The firmware for layout 2.0 works on this board which seems to be an older version of layout 2.0 board.

Terramoto, 2023-11-16:
I'm trying to run the testspeed script with a esp-wroom-32, i know the pins are either 16,17 or 1,3
I've set serial2 in 
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/bfbe2bd157759a93c8a9b3d2b1368cb2b04f3095/Arduino%20Examples/TestSpeed/TestSpeed.ino#L16

which according to 
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/bfbe2bd157759a93c8a9b3d2b1368cb2b04f3095/Arduino%20Examples/TestSpeed/hoverserial.h#L17

is for 16,17, however i'm getting hex values:
[image]

It's been a while since i played with esp32 so i think i might be missing something.

Terramoto, 2023-11-17:
When running the TestSpeed the device only respondes whenever i power it up. How do so that it keeps reading the serial data? At the moment i'm having to keep clicking the power button, it i keep it press it doesn't do anything.

pacraf, 2023-11-17:
I don’t understand your question…

Terramoto, 2023-11-19:
> I don’t understand your question…

Testspeed code runs a loop sending different speed and directions, however my motor only seems to catch one whenever i press the power button. It runs at a speed and direction and then stops and unless i press the power button again, it wont do anything

pacraf, 2023-11-20:
Understood. I wonder what can be the reason. 
in arduino sketch there is possibility to see debug messages. You see communication in terminal ? If you uncomment Rx debug , then you will see if checksum is correct… don’t know what to advice. Dummy firmware rotates motor ?

RoboDurden, 2023-12-01:
Sorry i did not get a notification.
When the motor spins nicely forward and backward when you bridge the on/off button, then pin definiton of HOLD is wrong and this layout is not 100% identical to the 2.0

```
// Self hold defines
#define SELF_HOLD	PB2
```

Then you should install Keil and trace the correct output pin to keep the power on after the on off button is released.