RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #36: New board PPM and PWM RC remote?

dexterbot80, 2023-11-11:
Hello, First of all I want to congratulate you on this project. Very useful for bringing broken hoverboards back to life....
I have two pieces of this model and would like to be able to actuate them simultaneously from an RC remote control to create a 4-motor RC platform.
Have you seen this model before? Can someone guide me?
Thank you
[image]
[image]
[image]
[image]

pacraf, 2023-11-11:
It seems to be different from photos on main page. So could be different layout. Controller looks like ok (gd32f130 c6t8); battery 24V
I would start with identifying all important pins, so mosfets , halls all others. Look into thread with 2_10. Pin by pin identified what is where connected. 
Anyway I can not say that your work will make it happen that Robo will create new layout. 
I barely compile prepared firmware, select options, but newer added anything new… but this info for sure will be necessary to start.
Btw in this thread 2-10 board you will see that I did this 4 wheeler. Rc version.

dexterbot80, 2023-11-12:
> It seems to be different from photos on main page. So could be different layout. Controller looks like ok (gd32f130 c6t6?); battery 24V I would start with identifying all important pins, so mosfets , halls all others. Look into thread with 2_10. Pin by pin identified what is where connected. Anyway I can not say that your work will make it happen that Robo will create new layout. I barely compile prepared firmware, select options, but newer added anything new… but this info for sure will be necessary to start. Btw in this thread 2-10 board you will see that I did such 4 wheeler. Rc version.

Hello @pacraf, Thank you for your very quick reply, I saw your 4-wheel vehicle and I want to do roughly the same thing, only I want to attach a flight controller and a GPS to it so that it can work autonomously.
It would be extremely useful if you could make a step-by-step tutorial and a wiring diagram for your machine.
Thank you 💯

pacraf, 2023-11-15:
Wiring is shown here, as it is lolin s2 connected to 4 boards, with diodes , as you can see on main page of this repo. Additionally you will connect your receiver or FC. I also experimented with pixhawk, but finally I have it only with RC receiver. 
diagram makes no sense, as it is only one pin from FRsky receiver connected to lolin S2…

I am stuck now with hardware making, so proper box , installing motors , making it rain proof… 
But this will take for me long time…

dexterbot80, 2023-11-16:
> Wiring is shown here, as it is lolin s2 connected to 4 boards, with diodes , as you can see on main page of this repo. Additionally you will connect your receiver or FC. I also experimented with [...]

Hello, Thank you for your answer. 
The Lolin code is located [here ?](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/Arduino%20Examples)

pacraf, 2023-11-16:
1. Yes, there are sketches, I play with this 4wheeler where author of this repo did nice example of 4 wheeler control.
Then you have sketch ESP ppm . This one shows example of using ppm library, which is typical RC receiver (all channels on one wire) .
So if you want 4 wheeler with RC, then add this example from ESP PPM to 4Wheeler. 

But if you want other control than PPM, then you need to transfer values from your RC to iSpeed and iSteer… this is generally idea.

2 yes and no. All boards are connected like on diagram. 4 in my case. In this scenario there is no master and slave (remove this cable) , but all boards are connected like this diagram shows. All are “singles” addressable one by one , with unique ID. This ID you have to compile with Keil so each bin file is different. Don’t load everywhere the same file because this will be mess. All will want to respond to lolin on the same time so this will not work. Take your time and read some threads on thin git. You will understand all aspects, there are other options to control 4 wheeler but I did not use them .

3. No. Rc signal goes to lolin. This board need to translate Rc signal to send properly commands to all boards . Read inside sketch esp ppm, there is written as I remember that ppm signal goes to input 3? Not sure. Read it there.
4. Don’t understand your statement. All boards (what I know) have 2 uart. One is used normally to connect between master slave (in original unmodified hardware) this you don’t need (in scenario that each is “single” with unique ID.
Second uart is normally not soldered, this one you have to identify on your board, solder pins, and connect like shown on diagram you have above.

dexterbot80, 2023-11-16:
> 1. Yes, there are sketches, I play with this 4wheeler where author of this repo did nice example of 4 wheeler control.
>    Then you have sketch ESP ppm . This one shows example of using ppm library, [...]

1. I have just one unsoldered UART on the SLAVE board that corespond with pins PB6 and PB7.
2. Maybe i need to identify pins PB6 and PB7 on the MASTER board to - here i do not have connector pinouts for this pins...
3. I have an extra white wire to the MASTER MOTOR that is connected with this red wire on the board
[image]
[image]
[image]

pacraf, 2023-11-17:
Hmm scenario like you proposed on diagram (but one thing different, that in fact master connecting with lolin , and then slave connected to master. You know what I mean? Just change on your diagram words master and slave). So this scenario is possible, however not tested by me.

Regarding pins for uart , I am pretty sure that you have them. These are the small pads not soldered close to uC. Visible on your photo.

dexterbot80, 2023-11-17:
> Hmm scenario like you proposed on diagram (but one thing different, that in fact master connecting with lolin , and then slave connected to master. You know what I mean? Just change on your diagram [...]

Hello professor :)
If you mean these small pads on the MASTER board, I found them, but I expected there to be a dedicated connector for this.
I wouldn't really want to use such thin wires around here - I catch my fingers :)
[image]

Regarding my diagram, I was saying to write the firmware for the SLAVE board on the microcontroller on the MASTER board and vice versa.
In this way, I still have the possibility to mount a connector for UART and I think it would be much more resistant.
[image]
It would be very good if this could be done, but I still have no idea what this white wire that enters the motor from the MASTER board is used for.
[image]
[image]

I have no idea if the operation of the engine would be affected if I installed the firmware dedicated to the SLAVE board on the MASTER board.
Thank you

pacraf, 2023-11-18:
Oh not the small control pads. I mean this connector pads 
[image]
Regarding white wire, I have also no idea. Can you measure voltage there? When board works of course.

regarding scenarios, as I said, my is different than your intended. I have all of them separate ID, and threat them as singles. Lolin makes calculation for turning .

pacraf, 2023-11-18:
Wait… I think that I also mixed things. The pins I marked are for programming… not for uart. Sorry for confusing.

dexterbot80, 2023-11-18:
> Wait… I think that I also mixed things. The pins I marked are for programming… not for uart. Sorry for confusing.

Hello, I loaded the firmware in both boards and now I need the code for ESP32.
I only have the [Olimex ESP32-GATEWAY](https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware) board and I tried to change the UART pins to 36 and 39 but without success - it keeps resetting after i upload the code.
Can you please send me your functional code for ESP32 and explain a little where I connect the PWM pin of the RC receiver?
Also, if you have one, please send a code for the ESP32 that starts the motors without the remote control.
Thank you

pacraf, 2023-11-18:
Yes, better start with the test sketch, and get it working. This sketch works on lolin for me. It is just generator of set values for drive. So no receiver, it just sends commands to drive. 
you have to pick correct communication, for me it is with uncommented UartBus, as this is the firmware I use. 

Firmware is here and it works.
Arduino Examples/TestSpeed/TestSpeed.ino
 I don’t know if it will work on your board, but should work… anyway I am not good in all that stuff. 
 
Regarding pwm - here I can’t help either, I suppose on net you will find something working easy to modify. My receiver works with PPM (this is different than your single channel PWM), and I just found example sketch and it works on lolin. It also - as I already wrote- in this repo as ready to test . But for you it will not help.  

But start with test, first of all you have to see communication alive.

pacraf, 2023-11-18:
For pwm input look for examples like here 
https://projecthub.arduino.cc/kelvineyeone/read-pwm-decode-rc-receiver-input-and-apply-fail-safe-113bac

dexterbot80, 2023-11-18:
> Yes, better start with the test sketch, and get it working. This sketch works on lolin for me. It is just generator of set values for drive. So no receiver, it just sends commands to drive. you have [...]

Something is wrong here, the ESP32 keeps resetting after I upload the code. This is the Serial Monitor output

`ELF file SHA256: e33092a70d7d9f2a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Hello Hoverbaord V2.x :-)
E (15) uart: uart_set_pin(692): tx_io_num error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40087f38
file: "C:\Users\mproz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\cores\esp32\esp32-hal-uart.c" line 215
func: uartBegin
expression: uart_set_pin(uart_nr, txPin, rxPin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)

abort() was called at PC 0x40087f3b on core 1

Backtrace: 0x40083349:0x3ffc5000 0x40087f45:0x3ffc5020 0x4008cd21:0x3ffc5040 0x40087f3b:0x3ffc50c0 0x400d2b9e:0x3ffc50e0 0x400d1cc4:0x3ffc5140 0x400d157a:0x3ffc5190 0x400d239e:0x3ffc51c0

ELF file SHA256: e33092a70d7d9f2a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Hello Hoverbaord V2.x :-)
E (15) uart: uart_set_pin(692): tx_io_num error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40087f38
file: "C:\Users\mproz\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\cores\esp32\esp32-hal-uart.c" line 215
func: uartBegin
expression: uart_set_pin(uart_nr, txPin, rxPin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)

abort() was called at PC 0x40087f3b on core 1

Backtrace: 0x40083349:0x3ffc5000 0x40087f45:0x3ffc5020 0x4008cd21:0x3ffc5040 0x40087f3b:0x3ffc50c0 0x400d2b9e:0x3ffc50e0 0x400d1cc4:0x3ffc5140 0x400d157a:0x3ffc5190 0x400d239e:0x3ffc51c0`
[image]

Please send me the code in the configuration that works for you.
Thank you

pacraf, 2023-11-18:
sure, below is the example code, it waits until you send in console to lolin new values of speed and steer. 
so in console you just enter for example 100,200,0  (and enter to send it) and is should command to board.

But i am pretty sure you will also have the same error, as example from this repo works. 

some serial example from arduino work for your board? you know, some basic serial communication, examples maybe... I don't know what is the problem...

```
//    Tested with Arduino Pro Mini 3.3V and Hoverboard-TX to pin 9 and Hoverboard-RX to pin 8
//
//    PB6 (Hoverboard-TX) and PB7 (Hoverboard-RX) can handle 5V I/O-Level :-)
//
//    please share feedback to https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x

#define _DEBUG      // debug output to first hardware serial port
#define DEBUG_RX    // additional hoverboard-rx debug output
#define REMOTE_UARTBUS

#include "util.h"
#include "hoverserial.h"

#ifdef ESP32
  #define oSerialHover Serial1    // ESP32
#else
  #include <SoftwareSerial.h>    // not compatible with RCReceiver because of interrupt conflicts.
  SoftwareSerial oSerialHover(9,8); // RX, TX 
  #define oSerialHover Serial    // Arduino
#endif
SerialHover2Server oHoverFeedback;
char myData[30] = { 0 };
int iSteer = 0;
int iSpeed = 10;

void setup()
{
  #ifdef _DEBUG
    Serial.begin(115200);
    Serial.println("Hello Hoverbaord V2.x :-)");
  #endif
  
  #ifdef ESP32
    HoverSetupEsp32(oSerialHover,19200,39,37);      // baud, rx, tx
  #else
    HoverSetupArduino(oSerialHover,19200);    //  8 Mhz Arduino Mini too slow for 115200 !!!
  #endif

  pinMode(LED_BUILTIN, OUTPUT);
}

unsigned long iLast = 0;
unsigned long iNext = 0;
void loop()
{
  unsigned long iNow = millis();
  digitalWrite(LED_BUILTIN, (iNow%2000) < 500);

  //int iSpeed = 3 * (ABS( (int)((iNow/20+100) % 400) - 200) - 100);   // repeats from +300 to -300 to +300 :-)
  //int iSteer = 3 * (ABS( (int)((iNow/400+100) % 400) - 200) - 100);   // repeats from +100 to -100 to +100 :-)

  
  byte n = Serial.available();
  if (n != 0) {
    byte m = Serial.readBytesUntil('\n', myData, 30);
    myData[m] = '\0';  //null-byte // (sscanf(myData, "%d,%d,%d", &y1, &y2, &y3) == 3) {
    int y1, y2;  //, y3;
    if (sscanf(myData, "%d,%d", &y1, &y2) == 2) { 
      Serial.println("new iSpeed/iSteer values are:");
      Serial.print("iSpeed = ");
      Serial.println(y1);
      Serial.print("iSteer = ");
      Serial.println(y2);
      //Serial.print("int y3 = ");
      //Serial.println(y3);
      iSpeed = y1;
      iSteer = y2;
    } else Serial.println("error in input!");
  }

  boolean bReceived = Receive(oSerialHover,oHoverFeedback);   
  if (bReceived)
  {
    DEBUGT("millis",iNow-iLast);
    DEBUGT("iSpeed",iSpeed);
    DEBUGT("iSteer",iSteer);
    HoverLog(oHoverFeedback);
    iLast = iNow;
   }

  #ifdef REMOTE_UARTBUS
    if (iNow > iNext)
    {
      iNext = iNow + 100;
      //DEBUGLN("time",iNow)
      int iSpeedL = CLAMP(iSpeed + iSteer,-1000,1000);
      int iSpeedR = -CLAMP(iSpeed - iSteer,-1000,1000);
      
      HoverSend(oSerialHover,0,iSpeedL);  // hoverboard will answer immediatly on having received this message ...
      delay(10);
      HoverSend(oSerialHover,1,iSpeedR);  // but wait 10 ms for the RX line to be clear again
    }
  #else
    if (bReceived)  // Reply only when you receive data
      HoverSend(oSerialHover,iSteer,iSpeed);
  #endif

}
```

dexterbot80, 2023-11-18:
> 39,37

Thanks @pacraf i have the same error / reboot 
Shouldn't it also work with arduino UNO? Do I need to make other settings to use Arduino instead of ESP32?

pacraf, 2023-11-18:
I have no idea. I was also like you playing with different boards, not successful. Finally bought lolin. It is only 4-5 euros… but I know, always better utilize what we have…

pacraf, 2023-11-18:
Don’t give up with your esp32. It’s far better than uno. Try other examples with two serials utilization. Did you choose correctly settings for your board in IDE?

dexterbot80, 2023-11-18:
> Don’t give up with your esp32. It’s far better than uno. Try other examples with two serials utilization. Did you choose correctly settings for your board in IDE?

I can't find Lolin S2 here, but I found a board that has the same ESP32-S2FN4R2 microcontroller.
I will buy it and keep you updated - I never give up :)

dexterbot80, 2023-11-28:
> Don’t give up with your esp32. It’s far better than uno. Try other examples with two serials utilization. Did you choose correctly settings for your board in IDE?

Hello @RoboDurden, Hello @pacraf 
In the end, the S2 mini also arrived, but I couldn't start the robot :)
I think I have a different configuration of the pins on the board because after I upload any of the firmware versions here, the Buzzer is no longer heard. In the videos presented by @RoboDurden  I noticed that the Buzzer is heard after each firmware upload.
I also noticed that my boards are very similar to V2.5 but I didn't find a special dedicated firmware for V2.5...
So we have to discover Schematics_2.14 together - now winter is coming so we need to make a snow robot quickly :)

This is the TX signal from S2 mini 
[image]

And this is the nervous S2 mini
[image]

RoboDurden, 2023-11-28:
Oh sorry, without the @RoboDurden  I would not have noticed this new issue. Sorry for that.
Will read this issue tomorrow.
If this is indeed a new layout, please upload high resolution photos of front and back with cables bent outside. So all components and circuit lines can be seen.

Good night from  Germany. I am living outdoor again and this night temperature will drop to -6°C. No big deal, my coldest night was -22 °C :-)

dexterbot80, 2023-11-28:
> Oh sorry, without the @RoboDurden I would not have noticed this new issue. Sorry for that. Will read this issue tomorrow. If this is indeed a new layout, please upload high resolution photos of [...]

Nice to meet you robo - come on whatsapp too, +40762654488
In the summer we both make a solar car :)

RoboDurden, 2023-11-29:
Yes, this will be 2.14.
But please upload high resolution photos of front and back with cables bent out of view. So all components and circuit lines can be seen.

RoboDurden, 2023-11-29:
Do not upload binaries for a different layout. This can kill your board.
As this is a new layout, there do not exist any binaries yet.
With a new layout 2.14 it will be very very helpful if you install the Keil IDE and can compile für hitting F7 and upload to hoverboard with F8.
It will require many compilations until we have a working binary.
I will not do that for you.

And forget about the esp32 and rc remote for now.

First we need to identify the the hall input pins and make sure that the 6 MOSFET output pins are as usual.

Then we(you) can compile the first single_test.bin that makes the motor spin forward and backward without any esp32.

Yes it would be could if have traced the pb6 and pb7 pins so you can establish a serial communication to the lolin32 S2 Mini. This will help to receive log data for the hover firmware.

Some boards only offer one serial port because these split boards only need a communication between each other.

The you will need single_uartBus binaries an control each Motor individually from Esp32 like @pacraf has successfully done :-)

If you have a second serial port header in your board, you could control only two masters and they in turn control their attached slaves.
In the image you posted above you did this with two master_Uart binaries . This will need two serial ports on the Esp32.
Older Esp have the hardware ports , so after using the first die Arduino debug Output Ther would still be two hardware UART ports left for your sketch.
But Esp32 S2 does only have two UART hardware ports. So you better go with my uartBus protocol.
But this is future talk.

Please upload hires photos of front and back without any cables blocking the view.

Big thanks to Pacraf for being active here :-)

dexterbot80, 2023-11-29:
> Do not upload binaries for a different layout. This can kill your board. As this is a new layout, there do not exist any binaries yet. With a new layout 2.14 it will be very very helpful if you install the Keil IDE and can compile für hitting F7 and upload to hoverboard with F8. It will require many compilations until we have a working binary. I will not do that for you.
> 
> And forget about the esp32 and rc remote for now.
> 
> First we need to identify the 3 hall input pins and make sure that the 6 MOSFET output pins are as usual.
> 
> Then we(you) can compile the first single_test.bin that makes the motor spin forward and backward without any esp32.
> 
> Yes it would be good if you have traced the pb6 and pb7 pins so you can establish a serial communication to the lolin32 S2 Mini. This will help to receive log data from the hover firmware.
> 
> Some boards only offer one serial port because these split boards only need a communication between each other.
> 
> Then you will need single_uartBus binaries an control each Motor individually from Esp32 like @pacraf has successfully done :-)
> 
> If you have a second serial port header on your board, you could control only two masters and they in turn control their attached slaves. In the image you posted above you did this with two master_Uart binaries . This will need two serial ports on the Esp32. Older Esp have three hardware ports , so after using the first for Arduino debug Output Ther would still be two hardware UART ports left for your sketch. But Esp32 S2 does only have two UART hardware ports. So you better go with my uartBus protocol. But this is future talk.
> 
> Please upload hires photos of front and back without any cables blocking the view.
> 
> Big thanks to Pacraf for being active here :-)

Hello @RoboDurden,
Please tell me if you can see the pictures well enough.

[image]
[image]
[image]
[image]
Thank you

RoboDurden, 2023-11-29:
Great, i now will compile them into a front-back image and add the 2.14 layout to my repo :-)

dexterbot80, 2023-11-29:
> Great, i now will compile them into a front-back image and add the 2.14 layout to my repo :-)

I powered the hoverboard from a laboratory source and limited the current to 0.7Ah - I did not notice an increase in consumption after loading several firmware versions, but I noticed from the first loaded firmware that the buzzer stopped working and that makes me think I have a different configuration of the pins.
Thank you @RoboDurden

RoboDurden, 2023-11-29:
Yes the pins are very likely different from other layouts.
As @pacraf suggested, read some other layout issues here to learn how to trace the pins.

RoboDurden, 2023-11-29:
okay i have added 2.14.

the 5 led pins should be no problem as the 5 transistors with its 5 gate resistors are clearly to be seen and it should be easy to trace them on the photos:
[image]

RoboDurden, 2023-11-29:
uart0 mostly is
```
	#define USART0_TX	PB6
	#define USART0_RX	PB7
```

uart1 mostly
```
	#define USART1_TX		PA2
	#define USART1_RX		PA3
```
I guess, PA2/PA3 will be the master-slave serial.

Only the slave has a second uart header:

[image]
That could be PB6 and PB7

Most important for now are the 3 hall pins. It seems they trace only on the front as i do not see vertical traces on the backside:
[image]

As you have a CC constant current power supply we can risk it to assume that the 6 mosfet output pins are as usual.

So if we have a clue on the 3 hall pins, i can create a new defines_2-14.h and you can start compiling a first single_test.bin

RoboDurden, 2023-11-29:
@dexterbot80 , try to trace the 3 hall pins with a multimeter (and a needle). You might already succed with this:

[image]

RoboDurden, 2023-11-29:
i can not trace the fifth led by photo because it vanished underneath the mcu.
So @dexterbot80 you need to test all the mcu pins with the diode-tester of a multimeter (and a needle):
[image]

RoboDurden, 2023-11-29:
okay i have uploaded a first single_dummy.bin to `BinariesToTest/`.
You will need `auto-flash.bat` from `BinariesToFlash/` and rename the bin file to hoverboard.bin
But as i said, you should please install Keil ide and compile yourself.

The test binary might show some led blinking when the motor is spinning.
if motor is not spinning then rotate by hand and report which led is blinking.
You should  plug in the 3led module and not the 2led cross-like module to be able to get feedback on all three hall pins.

Closely read the new Inc/defines_4-14.h

I did only guess here:
```
// Hall sensor defines
#define HALL_A	PA0		// robo just guessing !
#define HALL_B	PA1		// robo just guessing !
#define HALL_C	PC14	// robo just guessing !
```

Onoff button very likely will not work yet.
You will need to bridge the button so the board is constantly under power.

That's it for me for today - i guess.

dexterbot80, 2023-11-29:
> okay i have uploaded a first single_dummy.bin to `BinariesToTest/`. You will need `auto-flash.bat` from `BinariesToFlash/` and rename the bin file to hoverboard.bin But as i said, you should please [...]

Hello @RoboDurden 
I had guests here and I couldn't keep up with you. I can't upload firmware using STM32CubeProgrammer?
Thank you

RoboDurden, 2023-11-29:
Usually, st-link-utility or STM cube programmer do not work with gd32 split boards.
Please install Keil IDE. 
Or try the auto flash bat.

And get yourself a multimeter and start tracing the 3 hall pins.

dexterbot80, 2023-11-29:
> Usually, st-link-utility or STM cube programmer do not work with gd32 split boards. Please install Keil IDE. Or try the auto flash bat.
> 
> And get yourself a multimeter and start tracing the 3 hall pins.
I think i have to learn how to use Keil :)

[image]

RoboDurden, 2023-11-29:
st-flash.exe in same folder ?
Windows 10/11 ?

RoboDurden, 2023-11-29:
Downloads st-flash.exe from my folder..

dexterbot80, 2023-11-29:
> Downloads st-flash.exe from my folder..
I tested st-flash.exe but it doesn't work...
I ran in compatibility mode with windows 7, 8 and vista but it doesn't work...
I receive this black screen
​
[image]

RoboDurden, 2023-11-29:
??? Of course you need to put the stlinkexe in the correct folder.

dexterbot80, 2023-11-29:
> ??? Of course you need to put the stlinkexe in the correct folder.
Files are in the same folder 
[image]

RoboDurden, 2023-11-29:
Yes. Of course

dexterbot80, 2023-11-29:
> Yes. Of course

> Yes. Of course. But now autoflash.bat is missing.
I tried with both files in the same folder first

RoboDurden, 2023-11-29:
I do not see your problem.
You run the bat file with both the bin and exe I'm the same folder.

And you should install Keil IDE anyway. So no real need to use auto flash.

And it seems you somehow succeeded with stmcube anyway.

dexterbot80, 2023-11-29:
> I do not see your problem. You run the bat file with both the bin and exe I'm the same folder.
> 
> And you should install Keil IDE anyway. So no real need to use auto flash.
> 
> And it seems you somehow succeeded with stmcube anyway.
In STM32Cube i receive 2 red lines when i press Start Programming button
[image]

RoboDurden, 2023-11-29:
Okay, Error.

dexterbot80, 2023-11-29:
> Okay, Error.

You say it's ok but the wheels don't turn :)

RoboDurden, 2023-11-29:
First you to successfully upload the binary.

dexterbot80, 2023-11-29:
> First you to successfully upload the binary.

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/47325794/1d98fa53-dfad-4954-bdf6-996233386a4f

RoboDurden, 2023-11-29:
As i said, STM cube mostly does not work with gen2.x boards.

Install Keil IDE.

dexterbot80, 2023-11-29:
> As i said, STM cube mostly does not work with gen2.x boards.
> 
> Install Keil IDE.
I installed already- i need to open SRC files like this ?
[image]

RoboDurden, 2023-11-29:
Open the Keil project file. I think it is
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Hoverboard.uvprojx

Then F7 
Then F8

dexterbot80, 2023-11-29:
> Open the Keil project file. I think it is https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/HoverBoardGigaDevice/Hoverboard.uvprojx
> 
> Then F7 Then F8
[image]

RoboDurden, 2023-11-29:
Looking good, to guess.
Make sure your layout 14 is set in config.h

Now trace the three hall pins with a diode tester.

If possible, flash the slave board.
Then you can also select remoteUart and check the Esp32 serial connection to that left four pin header. Only connect tx rx and gnd.
Also check that you have 3.3v on the MCU.
You can measure that at the flash header...

dexterbot80, 2023-11-29:
> Looking good, to guess. Make sure your layout 14 is set in config.h
>
> Now trace the three hall pins with a diode tester.
>
> If possible, flash the slave board. Then you can also select remoteUart and [...]

I need to douwnload your updated project and uncomment layout 2_14 here ?
[image]

RoboDurden, 2023-11-29:
#define LAYOUT 14

Of course you need my latest changes :-(

dexterbot80, 2023-11-29:
> #define LAYOUT 14
> 
> Of course you need my latest changes :-(
okkkk, now the battery led flashes when i rotate the wheel

RoboDurden, 2023-11-29:
All three led show up ?
And always two Led overlap ?
Then my guess for the hall pins was correct :-)

But then the motor should do something.

RemoteDummy selected in config.h ?

dexterbot80, 2023-11-29:
> All three led show up ? And always two Led overlap ? Then my guess for the hall pins was correct :-)
> 
> But then the motor should do something.
> 
> RemoteDummy selected in config.h ?

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/47325794/15f57328-bfa3-4054-8ec0-1e83bf5babf6

RoboDurden, 2023-11-29:
The video made für Show only Red and Green Led.
Sure you attached the led module with three led.
Ah yes only the slave board has all three transistors for three led.

So continue with the slave board = no buzzer.

I still wonder why no motor movement at all...

[75 more replies]