RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #39: Think i found an uncovered chip

GeeElectronics, 2023-12-05:
The layout looks a bit like the 2.1 version but the chip is  a different gd32
[image]

RoboDurden, 2023-12-11:
Sorry I do not get notifications :-(
We already have an E230 MCU Layout, so if you start tracing the hall pins and led pins and MOSFET pins, I can assist you .

GeeElectronics, 2023-12-11:
Do you have the pin defines already set so i can Compile? Or should we trace the Pins?

RoboDurden, 2023-12-11:
You need to upload high resolution photos of your layout..
The other E230 layout is 2.7.
If yours is not 2.7. then you need to trace at least the 3 hall pins and the 6 mosfet pins. At least a few of the 6 as the possible configuration is very limited.
3 led output pins would be good to get some visible feedback.
Have a look at some recent issues here to see how tracing pins is done..

RoboDurden, 2023-12-11:
Your layout is new, please upload high res photos from front and backside. As slave probably has different layout, also the two photos of the slave.
Your MCU is the 32 pin version:
[image]
Which means you do not have the standard 6 mosfet pins
```
// Brushless Control DC (BLDC) defines
#define BLDC_GH PA10		// green	, Tommyboi2001 all bldc pins same as 2.0
#define BLDC_GL PB15		
#define BLDC_BH PA9			// blue
#define BLDC_BL PB14		
#define BLDC_YH PA8			// yellow
#define BLDC_YL PB13		
```

But i guess you will have mosfet driver chips, so pin tracing might not be a big deal. Opposed to the old 2.0 layouts with its many little smd transistors and resistors.

RoboDurden, 2023-12-12:
These seem to be the only possible pins for Timer0 which is used for the 6 mosfet bldc 
```
TIMER0_CH0	PA8
TIMER0_CH1	PA9
TIMER0_CH2	PA10

TIMER0_CH0_ON	PA7 PB13
TIMER0_CH1_ON	PB0 PB14
TIMER0_CH2_ON	PB1 PB15
```

So i guess your defines will need 
```
#define BLDC_GH PA10		// green
#define BLDC_GL PB1		
#define BLDC_BH PA9		// blue
#define BLDC_BL PB0		
#define BLDC_YH PA8		// yellow
#define BLDC_YL PA7		
```

But this might only work with a different "Alternate Function", and an update of target.h :

`	#define AF_TIMER0_BLDC(pin)	GPIO_AF_2	// GD32F130: all TIMER0 AF are AF2
`

Johannes09, 2024-06-13:
Hi,

I have a hoverboard with the same layout using the GD32E230 chip, and I'm experiencing difficulties connecting to it via ST-Link. Here's what I've tried so far:

    Pulling the BOOT pin both UP and DOWN.
    Measuring the breakout points on the board, confirming they are connected to PA13 (SWDIO) and PA14 (SWCLK).
    Switching SWDIO and SWCLK.
    Confirming my ST-Link programmer works; I am able to connect to a GD32F120 using STM32 ST-LINK Utility. 
    Ensuring the voltage is steady at 3.3V.

Any suggestions on what I might try next?

Thanks in advance for your help!

I-hate-2FA, 2024-06-13:
im not sure if the gd32 can be connected with swd in bootloader(boot0 high)
the mm32 can be unbricked that way
because im not sure if theres any software that can flash gd32e230 with serial, the best option may be to use nrest, which will force the board to enable swd (assuming the board is not bricked)