RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #40: new board / need help on UARTBUS

JochenKr, 2023-12-08:
Hi,
Thanks for this great project and your work to collect all this different boards.
I want to build some vehical and in the end it needs to run 4 wheels.

So I started to get one hoverboard and disassembled it. I found that it has a GD32F130C8, so I found your repository.
The boards I have look quite similar to your 2.0 Board. I did some trace checking, PWM signals match, UART matches on the 4 Pin connector (PA2 and PA3). Only the HALL sensors are a bit mixed up. I changed to below and the wheel is spinning if REMOTE_DUMMY is set. 
#define HALL_A	PC14
#define HALL_B	PF1
#define HALL_C	PB11

Picture of the slave Board:
[image]

in a next step I tried to communicate with one Board using your Arduino example (Testspeed) and a ESP32.
unluckily I do not get a response and the wheel also doesn't start to spin.
On GD32 side I've configured 
#define SINGLE 
#define REMOTE_UARTBUS 
#define SLAVE_ID	1
#define HAS_USART1
	#define USART1_REMOTE
the other options (HAS_USART0 as well) are commented

To watch the signals I'm calling HoverSend(oSerialHover,1,150,1); on the ESP32
The probes in the plots below are connected on the ESP32 side as this is easier. I did probe directly on PA3 of the GD32 one time to check that the 1kR inbetween is not doing something nasty to the signal, but the signal looks identical measuring on that Pin.
Ch1 is showing the TX of ESP32 (connected to PA3 of GD32), Ch2 is showing RX of ESP32. I put a trigger on Ch2, but it keeps high.
Anything wrong in that frame? Anything I missed to configure on G32 side? I'm compiling with Keil 5.38, don't know if this might be of intrest.
[image]

hope you can help me on this.

RoboDurden, 2023-12-11:
Sorry I do not get notifications on new issues :-(

Pa2 pa3 normally is the master slave UART.
Remote_xy usually uses pb6 pb7.
Check your defines.h

I have added a layout very much like 2.0 but different hall pin.
Check the list.

This is a quick answer..

RoboDurden, 2023-12-11:
With uartBus, the gd32 is only responding after a message for its I'd has been received.
Try RemoteUart first. Then the gd32 should send log data to esp32 every 100ms.

RoboDurden, 2023-12-12:
So it might be a bug that uart1 is not working for Remote :-/

I do not have a test setup nearby, so I can do no testing for the next weeks.

With uartBus you could activate the little debug feature in

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/ac9ecedd64bec843c7a2c5758bc6158515a76aba/HoverBoardGigaDevice/Inc/config.h#L43C18-L43C18

Then also uncomment this line to start receiving led feedback when a certain code line is reached:

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/ac9ecedd64bec843c7a2c5758bc6158515a76aba/HoverBoardGigaDevice/Src/remoteUartBus.c#L130

Sorry for the trouble.

RoboDurden, 2023-12-13:
Some time I have but I am living outdoors for the next weeks. You need to do the debugging :-/
[image]

JochenKr, 2023-12-22:
Sorry for my late reply.
looks pretty cold. Hope you have it somehow cozy!

I used the LED to trace where I get in the code.
Seems that it does not reach this line:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8b9c54ced31c95bce2d6e00405d615c3173f98cb/HoverBoardGigaDevice/Src/remoteUartBus.c#L167
Problem with CRC?

Before that I figured out that iRxDataType is 1 as the LED turns on in:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/8b9c54ced31c95bce2d6e00405d615c3173f98cb/HoverBoardGigaDevice/Src/remoteUartBus.c#L151

JochenKr, 2023-12-23:
I checked in the ESP32 code on this line:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/a773edf341c0aa1a9d472ba934bbecfc663b3b2d/Arduino%20Examples/TestSpeed/hoverserial.h#L80
the content of oData using Serial.printf. The result is 0x1, 0x96, 0x1 0xC141, which makes somehow sense to me as I'm calling HoverSend(oSerialHover,1,150,1).

Then I measured TX with the oscilloscope and tried my very best to encode it as below.
Does not really look correct, does it?
[image]

RoboDurden, 2023-12-24:
Oh was a mess :-(
It seems that the TestSpeed still has the outdated hovererial.h file.
Please copy that file from the Test_4Wheeler (or ESP32_PPM ) arduino example.

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/a773edf341c0aa1a9d472ba934bbecfc663b3b2d/Arduino%20Examples/Test_4Wheeler/hoverserial.h

RoboDurden, 2023-12-24:
Sorry i have not yet found a way in the Arduino IDE to have a commone hoverserial.h within the github file structure for the different examples. Ideas welcome.

JochenKr, 2023-12-24:
Thanks for looking into it. Right, the rootcause was in the hoverserial.h. After using the one from the Test_4Wheeler example the UARTBUS is working fine!
Happy Holidays!

closed 2023-12-24 by JochenKr

RoboDurden, 2023-12-24:
@JochenKr , sorry for all the trouble. 
I would like to assign a new layout number to this board as it is new. Is this the only change you made ?

```
#define HALL_A PC14
#define HALL_B PF1
#define HALL_C PB11
```

JochenKr, 2023-12-24:
Hi,
That's correct! Do you want some additional photos?

RoboDurden, 2023-12-24:
not really needed as the tracing of the pins is already complete.
But of course they would be nice.
More important would be photos of the old hoverboard so people can see from the housing what layout is inside.

JochenKr, 2023-12-24:
Ok, I'll provide later

JochenKr, 2023-12-27:
Here a half disassembled picture. The model is nilox DOC 6.5 green
[image]