RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #42: Layout 2.16 (Razor)

RoboDurden, 2023-12-14:
Thanks to @andyshmandy

[image]
[image]

This board seems to have only one dual opAmp so no overall current dc. And only 2 low-side phase currents (not needed for this gen2.x firmware but needed for the SimpleFOC firmware).

todo:

```
// ADC defines
#define VBATT	PA4					// todo
#define CURRENT_DC	PA6		// todo	robo: as this layout only has one dual opamp (for 2 low side phase currents) this board might not have a overall dc current adc

// Self hold defines
#define SELF_HOLD	PB2		// todo

// Button defines
#define BUTTON	PC15		// todo
```

but already compiling successfully.

andyshmandy, 2023-12-16:
@RoboDurden  thank you for writing up this issue.  Since this board is mostly mapped out I decided to flash the MCU with the 2-16 code changes.  I was digging around and it turns out that I can't find the motors for this board.  To be honest, I think I salvaged the bearings from the motors for another project.  Anyway, the motors should be somewhat universal so I tried hooking up a different motor but I was not able to find a hall sensor combination that would drive the motor.  The motor I tested happened to be 300 watts and this 2.16 board layout was originally paired with a 25v battery.  I had my Constant Current power supply set to 25v and 1.0 Amps.  I increated the voltage to 32 and the amps to 2.0 and still didn't have any success when going through the 6 hall sensor combinations.  I may try a lower wattage motor tomorrow.  By the way, the Red, Green, and Orange LEDs blink and change if I move the motor manually.  I checked the MOSFETS for shorts and found none.  Any Ideas?

I was not able to get the arduino serial monitor to display any output from the board.  It is just showing garbage.  I am not sure what I am doing wrong.

On another note, I have another board layout that I have started to trace that has the same MCU as this one.  I will open another issue and include some photos at sometime this weekend.

Thank you so much for your help!  I am really learning a lot!

andyshmandy, 2023-12-16:
By the way, the photo above with the traces, the part that is in shadow is GND on the other side of the board.  I forgot to turn off that layer when I exported the image from gimp.  I would like to upload the Gimp file to the Schematics folder so that the work I did tracing might be helpful to someone else and they can take advantage of the layers.  The file of over 40MB though :-(

RoboDurden, 2023-12-16:
I must keep the repo as small as possible so downloading the complete repo is possible for everyone..

Is the motor at least vibrating and you only do not find a permutation that works smoothly ?

Garbage on UART usually means wrong baud rate.
Maybe that board runs on 48 Mhz or 72 MHz.
If You have an idea oscilloscope you can measure the pulses to very the 19200 baud.

All motors are compatible I think. Not the sockets.
This PA5 was the inside led ?