RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #43: Layout 2.17?? HP-IBC-V9.7

marshmeloproject, 2023-12-16:
Hi, i'm new to this and was redirected here from EFeru github for split board main controller. I've taken pictures of the both boards front and back but am unsure which is slave and which is master. I'm also confused which version firmware is most suitable for my case. If there is any need for me to markup more details on the pictures below or could help point me in the right direction, please let me know!
Chip is GD32F096C6T8

**Board A**  (links to power switch, bluetooth speaker module, charging port, 2 LED strip and 1 on/off led strip)
[image]
[image]

**Board B**  (links to battery, 2 LED strip and 1 battery led strip)
[image]
[image]

RoboDurden, 2023-12-16:
I have no idea what that RD 32 F 096 C8 MCU is :-(

It might be pin compatible with a GD32F130C8 that you can buy for $1 on AliExpress https://a.aliexpress.com/_EGsx6aH

If you have a smd soldering heat gun..

If the MCU is not supported by the Keil IDE (or PlatdormIO) chances are near zero to program that MCU .

marshmeloproject, 2023-12-16:
> I have no idea what that RD 32 F 096 C6 MCU is :-( ![Screenshot_20231216-140932](https://private-user-images.githubusercontent.com/12238841/290994340-f47a3e7a-086e-4244-9bb3-b2b0cddf52f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3MzcyMDMsIm5iZiI6MTcwMjczNjkwMywicGF0aCI6Ii8xMjIzODg0MS8yOTA5OTQzNDAtZjQ3YTNlN2EtMDg2ZS00MjQ0LTliYjMtYjJiMGNkZGY1MmYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDE0MjgyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkZjZmOTQ1MWFmM2NhMzA0NjYwYjZlNTAyZjk0N2ZjYjZiZWRhZGRhY2ZjYzczYzcyMTQ1MGU3NDJjNTQ0ZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._fXSTUbgKfUEK59r4NRoN3B1sy97YeEetAGfDMrA1fo)
> 
> It might be pin compatible with a GD32F130C8 that you can buy for $1 on AliExpress https://a.aliexpress.com/_EGsx6aH
> 
> If you have a smd soldering heat gun..
> 
> If the MCU is not supported by the Keil IDE (or PlatdormIO) chances are near zero to program that MCU .

Chip is actually GD32F096C6T8, the last two characters hidden by red paint

RoboDurden, 2023-12-16:
C6 means 32 kB instead of 64kB which would be too less for the simpleFOC firmware anyway.
T8 means 48pin package.

Replacing the mcu with a smd heat gun is not such a big deal
[image]
But I have never done it. Watch some YouTube videos :-)
It might even work without new reflow low temperature soldering paste because there stays enough tin solder on the 48 pins.

Take a look at the gd32f130 pin out and the Inc/defines_2-0.h file to locate

Pa2 pa3
These pins should go to the masterSlave UART header .

Then see if the 6 MOSFET output pins match your layout.

Then check NREST and gnd and 3.3V and the flash pins to match your MCU.

Chances are good that a gd32F130 will fit.

Of course, for the community it would be better if find the manufacturer of that new MCU. But it simply may be a custom labeled GD or MM...

marshmeloproject, 2023-12-16:
> C6 means 32 kB instead of 64kB which would be too less for the simpleFOC firmware anyway. For this simple gen2.x the 32bit are enough.
> 
> T8 means 48pin package.
> 
> Replacing the mcu with a smd heat gun is not such a big deal ![Screenshot_20231216-154021](https://private-user-images.githubusercontent.com/12238841/290998670-331c0f16-f5f2-4c67-8412-6d625e61a45c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI3Mzk1OTcsIm5iZiI6MTcwMjczOTI5NywicGF0aCI6Ii8xMjIzODg0MS8yOTA5OTg2NzAtMzMxYzBmMTYtZjVmMi00YzY3LTg0MTItNmQ2MjVlNjFhNDVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE2VDE1MDgxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzYzEzMDNkZmE0NjFiNzA0MzEzMGNmZTNhMzBjNDQxM2MwMmI5ZDViMmU3NDQ5YzgyYjM1NzUzYTI5ZDgzZDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7xzxu_bf7OVHQT2PxCV1P3sp2ky_dTH5kuU9KsmXfNc) But I have never done it. Watch some YouTube videos :-) It might even work without new reflow low temperature soldering paste because there stays enough tin solder on the 48 pins.
> 
> Take a look at the gd32f130 pin out and the Inc/defines_2-0.h file to locate
> 
> Pa2 pa3 These pins should go to the masterSlave UART header .
> 
> Then see if the 6 MOSFET output pins match your layout.
> 
> Then check NREST and gnd and 3.3V and the flash pins to match your MCU.
> 
> Chances are good that a gd32F130 will fit.
> 
> Of course, for the community it would be better if find the manufacturer of that new MCU. But it simply may be a custom labeled GD or MM...

Thanks a lot for the additional info! Unfortunately replacing the chip is beyond my skills at the moment, I will definitely revisit this suggestion in the future when time permits!

closed 2023-12-16 by marshmeloproject