RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #49: Layout 2.6.1

AILIFE4798, 2024-01-03:
I have a board very close to layout 2.6 I'm not sure if I can use that or to open another one
If it is needed I can help trace wire pinout I really want support asap
Here is some as clear as I can get pictures because phone camera quality is bad
[image]
[image]
[image]
Master board is with power button and Bluetooth and slave board is with buzzer

Mcu:gd32f130c6t6(I heard that the rom is too small so it cannot support this firmware?)
Imu:854lc1(no info on Google)
Gate driver: eg2131

The board is got from the cheapest most common brand hoverboard in China called AERLANG(please don't buy it the quality of the wheel is very bad it only have 3 parallel wire but I got mine for free)

I have some follow up question to ask because I want to make a robot with 6 wheel and I have 2 gen 1 hoverboard and this cheapo one is there any way to control these together because it's very hard to buy gen 1 hoverboard

AILIFE4798, 2024-01-03:
@RoboDurden and how can I use your server to build gen 2 firmware I can only find option to use gen 1 and I've successfully flashed gen1 with uart mode

RoboDurden, 2024-01-04:
For this firmware you need to install Keil ide if you need to compile a new binary.

C6 only has 32 kB but this is enough for this simple gen2.x firmware :-)

Please trace the three hall pins and make sure that the 6 monsfet pins are the standard configuration. See defines_2-0.h

2.6 has not been traced yet.

But if you have the three hall sensors and the six mosfet and maybe the green and red led I will add a defines_2-19.h for you and you could start testing.

But you should by all means use a constant current power supply with 1-2 A max !
Simple dcdc step down converter could also do.

RoboDurden, 2024-01-04:
Please add photos of the housing / outside so people can see from the outside what is inside.

What do you mean with 'the wheel is very bad it only have 3 parallel wire " ?
Please describe with a photo :-)

RoboDurden, 2024-01-04:
Yes this is likely compatible with the 2.6 board which is labeled as v1.5 whereas yours is labeled as v1.5.1.
I have uploaded your photos to the 2.6 folder.
If you trace the following pins, i will add a defines_2-6-h file for you:
[image]

The upper left 4 pin header might be PB6 and PB7,
the lower right 4 pin header might go to PA2 and PA3
the three resistors driver the three transistors which might drive the three color led module.
The six mosfet outpins seem to be the default ones.
With the three hall pins traced, you might already get the motor spinning.

Keil is easy to install.

AILIFE4798, 2024-01-04:
@RoboDurden 
this is the reverse enginered schematic
[image]
i think i have missed quite a few connections but its all i can find

AILIFE4798, 2024-01-04:
[image]
This is the wheel from the hoverboard and as you can see it's all plastic and there is only 3 parallel copper wire for the winding so it's probably only rated 150w or something like that way less then the normal aluminum wheels

AILIFE4798, 2024-01-04:
i am hardware guy thats why im here looking for code from more intelegent people like you can you please compile firmware for me i am afraid to use keil to compile my self if not for your server i will not attempt this project so thanks allot

i have got a stlink v2 adapter but i have taken it apart and it is fake with apm32 insted of stm32 is there any way to repair it or where can i get a proper one
i have used it to flash gen 1 board but its stm board insted of gd32 so idk if it will still work with fake stlink

 i have a lab bench power supply so i am ready to test firmware and btw the battery is also of low quality its only 1 cell in parallel one of the worst ive seen so i will use lab bench power supply
im intrested why to use constant current because with dedicate gate driver it should not be able to create a short circuit even if the mcu tries to because of wrong pinout

AILIFE4798, 2024-01-04:
Some additional closeup picture increase you needed it as the full picture is very blurry

[image]

[image]
[image]

[image]
[image]

AILIFE4798, 2024-01-04:
updated schematic
[image]

RoboDurden, 2024-01-05:
Good job .
I will compile a binary the next two days.
But I might need 10+ compilations until every pin is correct.

Keil is really very simple. 
Install
 Open the project file
Hit f7 to compile 
Hit f8 to upload to gd32

You might have more problems installing the bin file..
That plastic wheel looks awful indeed.
Please upload.photos of the housing / box / outside so others won't buy it
Unless it is less the $60 new ;-)

Yes, gate driver chips should prevent a short cut.
But not a continuous current from mosfet a-hi to b-lo.

With new binaries it is really recommended to start with 1-2 Ampere max.

RoboDurden, 2024-01-05:
You have PB5 for two different led:
[image]
[image]

You might be able to solder a tx cable to the lower side of the PB7 RX smd resistor and a rx cable to the  lower side (the MCU side) of the PB6 TX smd resistor:
[image]

I have set the three hall sensors to

```
// Hall sensor defines
#define HALL_A	PB11
#define HALL_B	PA1
#define HALL_C	PC14
```

But it might need to test all six possible permutation until the vibration of the motor turns into a smooth spinning.
I am not really willing to do all these many compilations for you. Download Keil IDE.

The led will also need many permutations until all colors are correct:

```
// LED defines
#define LED_GREEN PA15
#define LED_ORANGE PB3
#define LED_RED PB4

#define UPPER_LED	PB5	
#define LOWER_LED	TODO_PIN
```

The adc input pins will also need some pin testing until voltage and current is logged correctly to ESP32:

AILIFE4798, 2024-01-05:
I hope I didn't got too many pin wrong so it'll work in few try
Where do I connect the stlink to flash it
Can keil work with fake st link and how to unlock gd32 without nrst
Pb5 really is connected to 2 led in parallel(the led have it's own MOSFET but the gate series resistor is connected together
I am not 100% sure the power button and latch pin is correct but we'll see when I test the firmware later

RoboDurden, 2024-01-05:
PA14 = SWCLK and PA13 = SWDIO 
+ GND. Better not contact 3.3V but bridge the onoff button to let the board generate the mcu 3.3V.

cheap clones flash just as nicely as stLink.
Try the autoflash.bat in the readytoflash/ folder !

And run windows St-Link-Utility  to unlock the bootloader and enable all option bits..

plastic wheel might be good when weight is a problem. Like robot arm..

RoboDurden, 2024-01-05:
what is this sot-8 chip to the very left ?
[image]

AILIFE4798, 2024-01-05:
I will try to flash when I'm home
Plastic wheel indeed is very light weight but I'm not sure how much power it can handle is it really worth it
I cannot find smd led can you circle it out in the picture I only find in gen1 board
The 8 pin ic is likely audio amplifier for Bluetooth speaker

RoboDurden, 2024-01-05:
no you test a board that you have not yet flashed to see if some little red led shows up. If not then we do not have such a little led. no problem.

There are applications like lawn mover, where 350+ Watt are not needed anyway.
Only thre copper wires might mean more windings and better efficiency at low power :-) Or the plastic stator needs more space so less copper available for windings :-(
Please share photos of how such a plastic hoverboard looks like ! What is the price to buy new ?

AILIFE4798, 2024-01-05:
i have tested board with original fw no led is found
i have alr taken the hoverboard apart and its not going back together so look elsewhere for the picture 
it cost abt $80 new you may be able to find other ones for even cheaper

AILIFE4798, 2024-01-05:
i have just tested the remote dummy firmware
when i push the button the turn signal lighting continuely
when i release the button the power latch works and the board keep on
the motor is locked up and when i force it by hand the 3led board start spinning with 1~2 led on at same time
when i reverse 2 of the motor phase wire the motor starts spinning back and forth
so i think my pinout is mostly correct 
the slave board have no sign of life at all because it does not have 15v buck regulator so i think i will need to modify it

RoboDurden, 2024-01-05:
Which two phases did you need to swap ?
The slave board is powered via the master slave uart cable if no dcdc is on the slave board.

Try connecting the esp32 to PB6 and pb7

AILIFE4798, 2024-01-05:
i swapped yellow and green
the pinout of rotating led is 15v,orange(master only),green,red
why not use pa2 pa3 as serial as its ment to do
i can solder the pin to smd resistor no problem just curious

RoboDurden, 2024-01-05:
With only one uart header you can use my uartBus protocol to control each single motor by esp32.
Normally you use RemoteUart and send steer + speed to the master which then controls it's slave.
Will be easier as you need the master slave cable of pa2 pa3 to power the slave.

AILIFE4798, 2024-01-05:
I have connected to the uart and when I connect USB to serial and baud 19200 it send something that is not humanly readable
I tried to connect to esp(which I can't believe took forever since esp32 is known territory for me) and it kinda works now but not really the checksum keep failing for some reason idk
[https://media.discordapp.net/attachments/1148619676297986150/1192882399298727986/image.png?ex=65aab14e&is=65983c4e&hm=fabc14e1066a40484bbf7238feab07a60a530f08a01f7870bfe5afda74bd10ec&](url)

AILIFE4798, 2024-01-05:
[image]
Image failed to attach

RoboDurden, 2024-01-05:
no my protocol is not compatible to gen1
I have updated the binaries with this hall configuration:

```
// Hall sensor defines
#define HALL_A	PC14
#define HALL_B	PA1
#define HALL_C	PB11
```

your esp32 log output does not really makes sense to me.
The hover message should begin with CD AB ( or AB CD) and not end with it.

And it looks as if the checksum at the end of the transmitted message is always 0 (tmpFeedback.checksum).

But as i do see CD AB, you indeed do seem to receive the hoverboard data.

Sorry i will not be able to help you with this in the next days.
I am living outdoors and -8°C coming..

AILIFE4798, 2024-01-05:
ok i will test the new hall mapping tomorrow

i am using the testspeed example but modify the serial to serial2 because serial1 does not exist on esp32... the one you have is esp32s2 not the original esp32
but i dont think it matters anyways as it can receive the data
the CDAB at the end actually can cchange to something else when i turn the wheel and it is reproduceablebut its only these two combination no other random chracter

have you change the led pin because it seems to be wrong
it should be
```
#define LED_GREEN PB3
#define LED_ORANGE PA15
```

is there any more pins that seems to be missing i need to check to make it fully support or sth else
if no ill wait patiently when you are free to help me

RoboDurden, 2024-01-05:
when log data is correct you can verify battery voltage and current DC.

Sorry, no more compilations from me today.
As i said, you better install Keil :-)

Good night from Germany.

AILIFE4798, 2024-01-05:
where can i download keil without log in
i downloaded st link utilities from third party
i will try to mess around with the serial to see whats wrong
hopefully ill figure out soon enough
not on a hurry any more now

goodnight from China

RoboDurden, 2024-01-06:
Oh it is an honor for me to have someone from China here on my GitHub project !!

Yes it needs an email to receive the official download link for the Keil IDE..

Then of course I will happily compile as many binaries for your as you need :-)

Just please remember that I currently live outdoors with my solar camper
[image]
[image]

I will try to upload the Windows 64 bit Keil Installation file to my server at the next free WLAN hotspot.

Does the latest binary with swapped hall a and c work for you ?

Otherwise I will upload all six binaries with the six possible permutations.

Good morning :-)

AILIFE4798, 2024-01-06:
morning

id expect allot more people from China here because all the latest and cheapest tech is made and first sold here
so well often need to create custom layout 
anyways if you need support about chinese datasheet etc i can probably help
add me on discord as ```ailife```

i will try keil if you can provide installation file
because after all you do not learn if you dont try new thing
and can you also provide stlink driver
because my desktop cannot install stlink it says the driver is not signed but i have disabled force signing
i have to use my laptop to flash
maybe linux is better after all

the new hall mapping can make the motor spin but not smooth and only in one direction i need to help it to spin other direction
swap phase wire does not help
i think we are closer

and the serial issue have been resolved now
i just made stupid mistake of not commenting out remote uartbus
it certainly didnt help to have no documentation
but well im a hardware guy  

the battery voltage is close enough (24.3 vs 23.6)
the current is random value
i have tried to unplug motor phase and it didnt change anything 
[image]
[image]

AILIFE4798, 2024-01-06:
oh and i have seen your solar camper project on youtube
its really cool imo wish i would do sth like this in the future

AILIFE4798, 2024-01-06:
This is how to solder pin header to access serial for those that don't know
[image]

RoboDurden, 2024-01-06:
nice header! the left pin ist GND ?
I have uploaded all six permutaions as Master_Dummy to https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/HoverBoardGigaDevice/BinariesToTest

One of the hoverboard_ABC.bin should make the motor spin forward and backwards nicely.

All three led are blinking ? You already mention the overlap of two led while the motor is rotating which is as it should be.
The led blinking is activated with this define in config.h

#define TEST_HALL2LED	// led the 3-led panel blink according to the hall sensors

Still no free wlan hotspot nearby. And still 48 km to drive home to my train station. So i guess it will be tomorrow afternoon to have unrestriced internet connection.

AILIFE4798, 2024-01-06:
ok i will test hall sensor layout first and report back
yes the led is blinking in circle and overlapping
yes the left serial header is GND
take your time ill wait for ur support

RoboDurden, 2024-01-06:
Here is a download link for the STM32-ST-Link-Utility maybe it includes such a driver you asked for:
https://www.robosoft.de/forums/en.stsw-link004.zip

The Keil IDE ( https://www.keil.com/demo/eval/arm.htm ) will have to wait.
The Demo is limited to 32 kB but this is all this simple firmware needs anyway.

AILIFE4798, 2024-01-06:
the layout is BCA

AILIFE4798, 2024-01-06:
thanks the st link utilities i downloaded is too old
yours worked just fine

RoboDurden, 2024-01-06:
Okay i have uploaded new binaries with BCA.
So you can now control with ESP and als test conecting the slave to master.
Only current adc pin is still wrong i guess.
When putting a hand on the rotating motor you should see an increasse of 1-2 Amps.
Will also need some try and error binaries.
But that can wait until i am home :-)

Battery chared to full at free eBike charging station so i continue my ride/drive :-)
Roland and off.

AILIFE4798, 2024-01-06:
i have tested the master uart with correct motor wiring and its working now
the slave also working only theres no power off melody like on gen1
the (indicated) current of slave is 1~-255 and current of master is 0~2
the actual current on 24v is only 200ma and i can get to 500 ma by trying to stop the motor with hand but even at 24v its very very strong i cannot do it
at 40v the accelration at 1000 max speed is making me almost lost grip of the motor
its probably good enough for my use case

RoboDurden, 2024-01-06:
So the esp32 current reading corresponds to the load you put on the motor with your hand?

AILIFE4798, 2024-01-06:
no the reading is random and not 0 when motor not connected

RoboDurden, 2024-01-06:
Here is the 913 MB download for you: https://www.robosoft.de/forums/mdk539.exe.tmp
Please tell me when you have downloaded the file so i can delete it from my server.
Of course you have to remove the .tmp to install ;-)

AILIFE4798, 2024-01-06:
here is comparator area pin tracing but i do not understand anything
[image]

RoboDurden, 2024-01-06:
Don't understand what you mean with comperator area :-/
I have uploaded four binaries to BinariesToTest with the sill available adc pins. So one of these binaries might output correct current.
They are master uart binaries.
So you can set eps32 to constant speed of 300 and put your hand on the wheel :-)

AILIFE4798, 2024-01-06:
the area of ok06 comparator ic where the adc is connected to
i guess its pb0

AILIFE4798, 2024-01-06:
i have installed keil where can i download the 2.6.1 source to try out and where to edit

RoboDurden, 2024-01-06:
[image]

Currently, the config.h is already configurated for 2.6 :-)
pin definitions are in defines_2-6.h

RoboDurden, 2024-01-06:
Simply open Hoverboard.uvprojx with the Keil ide (or doubleclick).
Then check config.h to match your board number etc.
then hit F7 to compile
and then hit F8 to upload binary :-)

AILIFE4798, 2024-01-06:
current sens pin is a6
which is the pin that directly go to current sens resistor who would have thought lol
i tried this pin last because i thought its to do with the comparator/opamp but turns out no its direct connected to adc

RoboDurden, 2024-01-06:
In `defines.h` there are two float defines to scale the voltage and currentDC:
```
// ADC value conversion defines
#ifndef MOTOR_AMP_CONV_DC_AMP
	#define MOTOR_AMP_CONV_DC_AMP 0.201465201465  // 3,3V * 1/3 - 0,004Ohm * IL(ampere) = (ADC-Data/4095) *3,3V
#endif
#ifndef ADC_BATTERY_VOLT
	#define ADC_BATTERY_VOLT      0.024169921875 	// V_Batt to V_BattMeasure = factor 30: ( (ADC-Data/4095) *3,3V *30 )
#endif
```

You can defines those two already in defines_2-6.h to get correct readings for this layout..

I assume that you now can compile yourself :-)

AILIFE4798, 2024-01-06:
i think i have sucessfully build my self
i will try to flash it now lets hope for the best

AILIFE4798, 2024-01-06:
yes it really worked!
now the pin is all correct and motor can spin smoothly and voltage/current reading is also accurate

RoboDurden, 2024-01-06:
As you have more insight into the pin tracing of your 2.6.1, it would be nice if you compare the pcb with the photos of the 2.6 layout in the Schematics_2.6 folder.
If you do not see some differences for the pins defines in defines_2-6.h
then we can be quite sure that the 2.6.1. board is compatible with the 2.6 layout.

AILIFE4798, 2024-01-06:
to me it looked the same from the picture
i actually traced mine by using the physical pcb i only draw the image for you and others that have the same board to see
so i cannot be sure but i try to trace some random ones thats all on single layer and its the same
im not quite sure about the battery led but i think it should be the same

AILIFE4798, 2024-01-06:
[image]
and the final pin tracing image update

AILIFE4798, 2024-01-06:
> As you have more insight into the pin tracing of your 2.6.1, it would be nice if you compare the pcb with the photos of the 2.6 layout in the Schematics_2.6 folder. If you do not see some [...]

the buzzer is on master for 2.6 but on slave for 2.6.1

AILIFE4798, 2024-01-07:
> here is comparator area pin tracing but i do not understand anything ![hoverV2 6 1_comparator](https://private-user-images.githubusercontent.com/142502122/294699242-61aa6a4c-1427-45b7-ae98-ef5a2270c2fb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NTQ5MzAsIm5iZiI6MTcwNDY1NDYzMCwicGF0aCI6Ii8xNDI1MDIxMjIvMjk0Njk5MjQyLTYxYWE2YTRjLTE0MjctNDViNy1hZTk4LWVmNWEyMjcwYzJmYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwN1QxOTEwMzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMDYyY2Q5NjM4NjM4N2Y0N2RkNjlkMjgzNTczNjQzMzAyOTg3NGMzNzE3MjAwOGE1NTRkMWRmNWE3MmJhNDIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.t6knlBgllt6r8Hry0MGvVGoBKrA1YZmKZWDYF_1VJDg)

i can understand now it is sensing the voltage from phase wire to ground i think it is used to sens the back EMF or phase voltage

RoboDurden, 2024-01-07:
For the advanced FOC control (which needs more than 32kB MCU), at least two phase currents are needed.
The empty footprint of the SOT-8 would be another dual op-amp.
On most layouts, there are two sot-8 dual opamp chips: one for two phase currents and the other for the currentDC amplification (and the third phase current).

I have reached my train station and will make a few days break from my outdoor life.
But going to bed now.

AILIFE4798, 2024-01-07:
it have the dual opamp fitted i just removed it to find the current dc pin but infact current dc is connected to the adc directly and the opamp is for 2 phase current
maybe i should replace the mcu i have the ability to do it
but the layout 2.8.1 is selling for $2 i cannot resist to buy some 
i will buy some gd32f130c8t6 and if its not needed to replace the mm32spin i will use it on layout2.6.1

AILIFE4798, 2024-01-07:
Maybe I should compare this gen2 board with gen1 board which is using eferu foc to see if it really did make Any difference

AILIFE4798, 2024-01-08:
Can this layout be added to readme as ready now
Or do you need video of actual test to confirm

[9 more replies]