RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #52: 2.18 - (chip tbd) - FLUXX3

reed-dan, 2024-01-10:
Brand: Fluxx (GoTrax)
Model: Fluxx3
URL: https://gotrax.com/products/fluxx-fx3-self-balancing-hoverboard
Manufacturer Date: 02/05/2020
Max: Speed 6mph Weight 220lb 
Battery: 25.2 V 2.6Ah
Charger: Barrel Jack 24V 0.6A

Body: Aluminium
Wheels: Plastic 6.5in
Board Configuration: 2 boards
Chip: TBD (130)
Notes: Battery connector is XT30

Similar to [2.17](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/44 )

AILIFE4798, 2024-01-10:
2.18 is used so it would be 2.19
it should be pin compatiable with 2.17 test with constant current supply

RoboDurden, 2024-01-10:
Yes this will be 2.19
Layout is quite different from 2.17 (also GoTrax) so it will need some pin tracing and testing with constant current supply (cheapest way: LM2596HVS 60V 3A Buck Converter = https://www.aliexpress.com/item/4001360810712.html )

@reed-dan please upload nice top view photos from master and slave and master-backside and slave-backside, this will make pin tracing easier.
A description of the pin headers would also be nice.
I guess the one with the buzzer in the middle is the master as it has the onOff button and the charger socket.

RoboDurden, 2024-01-14:
Okay i have added this as 2.19 to the repo.
@reed-dan the 6 mosfet are compatible to my new autodetect firmware:
[image]

But i can not really see where the two SWD pins for flashing with st-link dongle goes:

PA14: SWCLK
PA13: SWDIO

And you need a uart communicaton for cheap usb-uart-dongle (or esp32 with my autodetect.ino)

PA2/PA3 is usually the  master/slave header:
[image]

PB6 and PB7 could be on the empty header to the left:
[image]
But then PB6 and PB7 seem to be bridged. So you would have to access the still seperate PB6 and PB6 before the final smd resistors.

Best start with my hoverboard autodetect tx=PA2, rx=PA3.bin at https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesToTest

AILIFE4798, 2024-01-14:
there really is no swd connector
thats gonna be a bit hard to solder wire to pin of mcu but it can be done

AILIFE4798, 2024-01-14:
i guess the swd is disabled anyways just use serial to flash

RoboDurden, 2024-01-14:
please upload some screentshots on how to flash via serail and not swd.
To me it looks like PA13 and pA14 at least a connected to some smd resistors/capacitors. So soldering thin cable to those should not be such a big problem.

AILIFE4798, 2024-01-14:
but the problem is if the pin used for something else then swd have to be disabled anyways

AILIFE4798, 2024-01-14:
well the bootloader recover uart pin is a9 a10 which is used by motor so it cannot be used also
thats gonna be a bit tricky

AILIFE4798, 2024-01-14:
[image]

AILIFE4798, 2024-01-14:
pattern1=boot0 pullup boot1 pulldown

RoboDurden, 2024-01-14:
@AILIFE4798  please wait for @reed-dan to start trying. We already have a board that also uses a SWD pin for something LED if i remember correctly. In this case we only have to take care to not use this pin as something like LED_RED because after that it needs the NREST pin to be able to flash again.

reed-dan, 2024-01-18:
> But i can not really see where the two SWD pins for flashing with st-link dongle goes:
> 
> PA14: SWCLK PA13: SWDIO
> 
The swclk swdio pins are filled in with solder but to the left of the mcu

[image]

I will get some better pictures soon. Sorry been wokring  on 2.8.

reed-dan, 2024-01-19:
[image]
[image]
[image]
[image]
[image]

reed-dan, 2024-01-19:
[image]
[image]
[image]
[image]

reed-dan, 2024-01-19:
How do you determine which board is the Slave & Master?

reed-dan, 2024-01-19:
Tracing Work in Progress
MCU
Pin 30-PA9->R83->EG2133 Pin 2
Pin 28-PB15->R59->EG2133 Pin 4
Pin 26-PB13->R82->EG2133 Pin 6
Pin 31-PA10->R60->EG2133 Pin 1
Pin 29-PA8->R85->EG2133 Pin 3
Pin 27-PB14->R84->EG2133 Pin 5

Pin 37-PA14: SWCLK
Pin 34-PA13: SWDIO

Mosfet
Q2: D:R38
Q8: D:R39

reed-dan, 2024-01-19:
[image]

AILIFE4798, 2024-01-19:
The board with power button is master
Most of the time they write it on the board with Chinese but apparently this time not
You should not trace the pin should wait for auto detect
Only trace tx rx and the swd pins

RoboDurden, 2024-01-19:
Yes . Pb6 and pb7 would be good if you plan to use the master slave cable.

RoboDurden, 2024-01-19:
I am still in bed.
If you have Keil installed you can compile the latest autodetect yourself.
Your mosfet pins are compatible.
And the hall sensors detection worked for ailife.

AILIFE4798, 2024-01-19:
is the serial fixed now

reed-dan, 2024-01-19:
It will be a couple of days, I am waiting on desoldering wick so I can take the solder out of the swd holes. It looks like PB6 & PB7 are both unused on the board. Would this something I would need to have connection to permanently or just temp? If it is just temp I ordered some micro probe clips and may be able to hook onto the pins, otherwise I am not sure I can solder to those pins at this time.

Good news is I have a 2.9 board with an gd32f103c8t6 (instead of  gd32f103c6t6) chip that has an "external interface" appears to be connected to PB6 & PB7!

AILIFE4798, 2024-01-19:
i dont think any micro probe can hold onto a qfn48 without shorting out lol
soldering copper wire that is used in motor winding is the easiest way if it is really the case that is unused
but pb6 and pb7 is the only hardware i2c pin so if they want to use hardware i2c for IMU it would be much easier thrn soldering to gd32's leg

reed-dan, 2024-01-19:
I probed them to PB5 pin 41 (Alternate: SPI0_MOSI, I2C0_SMBA, TIMER15_BRKIN,
TIMER2_CH1) does that seem correct?

[image]

RoboDurden, 2024-01-19:
PB5 can not be used as serial0.

Why do i not see the back connections on your transparent pcb photo ?
[image]
[image]

Am i mixing wrong layouts ?

reed-dan, 2024-01-19:
I didn't upload the back side transparent, but you are spot on. I tested those pads previously with my fine point meter probes, but they seemed to no be connected to anything. When I got out the needles we are making progress, and there are nice pads to solder to.

[image]
[image]

RoboDurden, 2024-01-19:
But you can see the footprint of a bluetooth module there :-)
[image]

RoboDurden, 2024-01-19:
[image]

RoboDurden, 2024-01-19:
[image]

AILIFE4798, 2024-01-20:
to easily solder pin header without removing the solder in the hole first you can first cut the pin header into a single pin and solder it 1 by 1(melt the solder from one side of board and push the pin header in)

reed-dan, 2024-02-11:
I desoldered the swd header pin holes today, but the holes are smaller than standard header pins.

RoboDurden, 2024-02-11:
https://youtu.be/qVJqOmmUm2c

danil212211, 2025-09-22:
I have this layout. I resoldered those pins and connected to it. Unfortunately I couldn't use autodetect binary from this repo, cause it's 33kb and GD32F130C6T6 has only 32kb of flash memory. 

I built autodetect from sources, just changed GCC optimization flag to -Oz, and it flashed alright. But sadly, when I tried to detect hall sensors, it didn't find anything. For power I used my lab supply with 24v 1a.

RoboDurden, 2025-09-22:
> GCC optimization flag to -Oz, 

With platformio ?
Please post screenshot.

User interface of my autodetect running ?
Can you detect led pins ?

danil212211, 2025-09-22:
> With platformio ?

I used Keil uVision, set Autodetect macros on in config.h

>  Please post screenshot.
I will post screenshot in an hour or two, cant do it now.

> User interface of my autodetect running ? Can you detect led pins ?

Yes, I used your autodetect interface, I haven't tried detecting led pins, but successfully detected buzzer pin on C13

RoboDurden, 2025-09-22:
You could make bHall a global variable and monitor it (together with iTest) with StmStudio (or MCUViewer):

danil212211, 2025-09-22:
> You could make bHall a global variable and monitor it (together with iTest) with StmStudio (or MCUViewer):
> 
> ` uint8_t bHall = digitalRead(aoPin[iTest].i); `
> 
> If you already see the pin currently tested in my user interface (via uart), only visualizing bHall might help you to identify the hall sensors. Of course, the motor should be spinning.
> 
> my hall identifier was tuned manually:
> 
> ` if ( (iTime > 5) && (iTime < 50) ) // the hall on-time should match the rotation speed `
> 
> Make `uint16_t iTime = msTicks-msTicksOld;` also a global variable to see why my tuning does not work for your motor.
> 
> Read here in the wiki for StmStudio.

Ok, thank you! Maybe threre is some problems with my motors, cause they haven't rotated a bit while test was running, Ill try debugging with your approach and will post results if Ill find anything

danil212211, 2025-09-24:
Yeah, I traced the pins to EG2133 and looked at the datasheet. It seems that I have a faulty board, and I haven't checked if it works before starting disassembling it.

RoboDurden, 2025-09-24:
# Tell me, why didn't you post a screenshot with this `GCC optimization flag to -Oz,` ?

RoboDurden, 2025-09-24:
Awesome, now even RTT_REMOTE is below 32kB:

```
REMOTE_AUTODETECT with Optimization "O1"

REMOTE_USART
Program Size: Code=32028 RO-data=1584 RW-data=648 ZI-data=2112  

RTT_REMOTE
Program Size: Code=35948 RO-data=1628 RW-data=648 ZI-data=3280  

REMOTE_AUTODETECT with Optimization "Oz Image Size"

RTT_REMOTE + WINDOWS_RN
Program Size: Code=24604 RO-data=1596 RW-data=644 ZI-data=3268  

REMOTE_USART
Program Size: Code=23912 RO-data=1572 RW-data=644 ZI-data=2100  
```

> Yeah, I traced the pins to EG2133 and looked at the datasheet and there is no enable pins, and I haven't checked if it works before starting disassembling it, so It seems that I have a faulty board
UPD: The second board has a short circuit, so It's 100% faulty=(

Without analog mosfet driver, it should be possible to repair such a board.
Short cut is probably because of two mofets (lo and hi side of one phase) have burned.
Remove them and see with an oscilloscope if the EG2133 is still outputting the gate voltages for the broken mosfests.
If not, you would have to replace the EG2133 as well.

danil212211, 2025-09-24:
> Awesome, now even RTT_REMOTE is below 32kB:

Glad I could help=)

> Without analog mosfet driver, it should be possible to repair such a board. Short cut is probably because of two mofets (lo and hi side of one phase) have burned. Remove them and see with an [...]

Got it. I think it is too much trouble just for making motor spin=) I think I will use bldc driver for my project instead, would be a lot easier