RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #53: !!! Now with auto-detect to trace the pins !!!

RoboDurden, 2024-01-10:
I have finally begun with a REMOTE_AUTODETECT :-)

For now it will assume the six bldc pins to be 

```
#define BLDC_GH PA10		// green
#define BLDC_GL PB15		
#define BLDC_BH PA9		// blue
#define BLDC_BL PB14		
#define BLDC_YH PA8		// yellow
#define BLDC_YL PB13		
```

Only the advanced TIMER0 can do bldc motor control and the pins are very limited. So most boards have this combination :-)
It is easy to check some of these pins by looking at the pcb.

My autodect simply simulates the hall sensors to make the motor spin. Quite nosiy but 0.5A at 25V = max 10 Watt is okay.
Then it scans all 27 possible io pins to find the three that are in sync with the rotor.

This stage `AUTODETECT_Stage_Hall` already works for my 2.13 test setup:

```
0 try 1
hall 0 = PC14 : 4 ms
1 try 3
1 try 4
hall 1 = PF1 : 4 ms
2 try 6
2 try 7
2 try 8
2 try 9
2 try 10
2 try 11
2 try 12
2 try 13
2 try 14
2 try 15
hall 2 = PB11 : 4 ms
```

Next stage will be the six possible permutations of the newly found three hall sensors :-)

@AILIFE4798 , Ideas and testers welcome !

AILIFE4798, 2024-01-10:
all my board and wheel is at school rn i will test when i have hardware
i will create some more code to trace the 6 possible hall sequence and the button
where did you store the detected pin or do i have to get serial acess first

RoboDurden, 2024-01-10:
At the moment i simply output to PB6. Should change this to PA2 as the master-slave serial ist mostly PA2/PA3
I am already working on the 6 permutations.
But it would be nice if you would test if my method also works on another layout/motor setup.

AILIFE4798, 2024-01-10:
you should output to both

AILIFE4798, 2024-01-10:
if the hall sequence cannot be detected automaticly let user select with button and store it in flash so there is no need for them to compile firmware

AILIFE4798, 2024-01-10:
I have a (at least I think) very good idea that is also very simple to make
First the user should unplug the hall cable and push button
The controller will scan for the only active line that is the switch and remember it
And the controller will guess a pin randomly as the latch
And it should store the tried latch pin in flash
And user just keep pressing and release button until it can hold power
Then afterwards press button will cycle through all the output pin to select the LED in a list that is defined in manual
And long click to confirm it's correct and store to flash
Then after all led and buzzer is selected manually the left you can use your code to detect hall sequence and the adc pins
I'm not sure if it'll work but it sounds promising

RoboDurden, 2024-01-10:
It should be possible to detect the hall order all by the autodetect.
I have uploaded my progress so far:
1. detect the correct three pins.
2. find the correct permutation :-)

serial output:
```
0 try 1 = PC14
hall 0 = PC14 : 4 ms
1 try 3 = PF0
1 try 4 = PF1
hall 1 = PF1 : 4 ms
2 try 6 = PA1
2 try 7 = PA4
2 try 8 = PA5
2 try 9 = PA6
2 try 10 = PA7
2 try 11 = PB0
2 try 12 = PB1
2 try 13 = PB2
2 try 14 = PB10
2 try 15 = PB11
hall 2 = PB11 : 5 ms
wrong oder: 3 2
wrong oder: 4 5
wrong oder: 3 4
wrong oder: 2 1
hall oder 4 :
#define HALL_A = PB11
#define HALL_B = PC14
#define HALL_C = PF1
```

If someone has a layout ready for testing with the usual 6 mosfet pins and can access PB6 = tx, it would be nice if he tests my REMOTE_AUTODETECT (with a 2A constant current power supply) :-)

RoboDurden, 2024-01-10:
uploaded more stable autodetect.
Good night from Germany :-)

AILIFE4798, 2024-01-11:
I still don't have the hardware I'll try when I got
I've looked into the code it's a bit more complicated then I originally intisipated 
It'll take some time to understand
But driving a bldc with commutation overall is not that complicated so I think it'll be possible

AILIFE4798, 2024-01-11:
The goal is I want to remove the need to compile any firmware our self so can you look into store the detectedPinout in eeprom and next startup if stored pin is detected it will automatically switch to remote uart instead of remote autodetect

RoboDurden, 2024-01-11:
Sure that will be nice.
But until I am finished I need serial output anyway.
But eeprom might help to autodetect the hold pin :-)

AILIFE4798, 2024-01-11:
I think the hold pin should be the first to detect
And the button pin
Then the led and buzzer cannot be auto detected user should select manually with short press button to cycle through all the output pin and when it land on the correct led long press to save
After that then detect the hall pin that you have wrote

RoboDurden, 2024-01-11:
I am happy that you stay engaged here :-)
But your autodetec ideas are not that good i think ;-)

I am not planing to add REMOTE_PWM or REMOTE_ADC because ESP32 is so cheap that everybody should control via uart.
And as the every board has a master-slaver uart header, 
it is no problem to read and control autodetect via serial terminal.
Does not really matter if the user has to flash a binary once or twice. And he will need serial communication anyway.
So flash one: run autodetect and report results to me.
Then flash two with the new ReadyToFlash binary.

Next code update will be led detect. User will hive simply hit r=red or o=orange or g=green or u=up or d=down when then led-stage cycles through all the remaining io ports..

Users will need a serial commuication anyway !

AILIFE4798, 2024-01-11:
I agree with the control should be handled by external microcontroller just like a odrive and talk to hoverboard via serial
So yes auto detect the led via serial is better
But If the same binary have a secret code sent via serial will do the calibration and after reboot it'll read the pin from eeprom and work normally
So every thing that can be set with config.h should be able to be set via serial and store in eeprom
So user will never need to compile any binary one will fit all

RoboDurden, 2024-01-11:
Yes with so many layouts people have problems finding their already supported binary. 
As i said, EEPROM might be nice when i am finished with everything.

AILIFE4798, 2024-01-11:
i think with hal_gpio library itll be easier

RoboDurden, 2024-01-11:
yes that seem to have been a bug. But as all people use +-1000, i scale the final pwm to +-1125 which seems to be 100% pwm.

I have optimized the code and added the feature to detect the 5 led:

```
0 try 1 = PC14
hall 0 = PC14 : 4 ms
1 try 3 = PF0
1 try 4 = PF1
hall 1 = PF1 : 4 ms
2 try 6 = PB11
hall 2 = PB11 : 4 ms
wrong oder 0
wrong oder 1
wrong oder 2
wrong oder 3
hall oder 4 :
#define HALL_A = PB11
#define HALL_B = PC14
#define HALL_C = PF1
send 'r'=red or 'o'=orange or 'g'=green or 'u'=up or 'd'=down when led lights up :-)
'c' to reset , '-' to toggle direction , 'l' to list
and 'x' when finished.
try pin 1 = PC15
try pin 2 = PF0
try pin 3 = PA0
try pin 4 = PA4
try pin 5 = PA5
try pin 6 = PA6
try pin 7 = PA7
try pin 8 = PB0
try pin 9 = PB1
try pin 10 = PB2
try pin 11 = PB10
try pin 12 = PA1
try pin 13 = PB12
try pin 14 = PA11
try pin 15 = PF6
try pin 16 = PF7
try pin 17 = PA12
led orange = PA12
try pin 18 = PA15
led green = PA15
try pin 19 = PB3
led red = PB3
try pin 20 = PB4
try pin 21 = PB5
try pin 22 = PB8
try pin 23 = PB9
try pin 0 = PC13
#define LED_RED	PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	TODO_PIN
#define LOWER_LED	TODO_PIN
try pin 1 = PC15
try pin 2 = PF0
try pin 3 = PA0
direction: -1
try pin 2 = PF0
try pin 1 = PC15
try pin 0 = PC13
try pin 23 = PB9
try pin 22 = PB8
#define LED_RED	PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	TODO_PIN
#define LOWER_LED	TODO_PIN
```

Next will be to make a youtube video on this new feature :-)

AILIFE4798, 2024-01-11:
and buzzer can also detect in same way assume its on master ofc
i think you should make video when its more finished and is ready to detect everything like self hold which lets be real is the whole point because self hold is very hard to find manually
and current dc can also be sensed i guess
voltage sens need user interaction to change voltage on their power supply

RoboDurden, 2024-01-11:
next will be scaning the remaining adc pins an user has to hit button when a reading matches the battery voltage, etc.
HOLD bin will be setting all reamining pins to output and high and then step by step set the to 0 until boards turns off. 

No i want testers. So the youtube video might get me users who already test the autodetect.bin

AILIFE4798, 2024-01-11:
i agree,i will test it when i have hardware.
i have a 3 phase control module in aluminium box maybe it can also support this?

RoboDurden, 2024-01-11:
What has happened with your 2.6.1 board ?

RoboDurden, 2024-01-11:
Neither 2.9 nor 2.19 are yet traced and my autodetect does not really need any guide.
Simply choose the serial port you want to use in config.h

```
	#define HAS_USART0	// tx=PB6,rx=PB7	uncomment to connect via 19200 baud serial
	//#define HAS_USART1	// tx=PA2,rx=PA3	uncomment to connect via 19200 baud serial
```

And connect usb-uart dongle to the two hoverboard pins.
The Autodetect.ino Arduino example is simply a path-through so you can read and send from the strg+M arduino terminal having the final ESP32 already conected to the hoverboad.
But any cheap usb-uart dongle and windows-serial-terminal should also do.
Every comand needs a 'return' to be sent :-/

Then bridge the on/off button and turn the cc power supply on:

```
0 try 1 = PC14
hall 0 = PC14 : 4 ms
1 try 3 = PF0
1 try 4 = PF1
hall 1 = PF1 : 4 ms
2 try 6 = PB11
hall 2 = PB11 : 4 ms
wrong oder 0
wrong oder 1
wrong oder 2
wrong oder 3
hall oder 4 :
#define HALL_A = PB11
#define HALL_B = PC14
#define HALL_C = PF1
send 'r'=red,	'o'=orange,	'g'=green,	'u'=up,	'd'=down,	'b'=buzzer
'c' to reset , '-' to toggle direction , 'l' to list
and 'x' when finished.
try pin 1 = PC15
try pin 2 = PF0
try pin 3 = PA0
down = PA0
try pin 4 = PA4
try pin 5 = PA5
try pin 6 = PA6
try pin 7 = PA7
try pin 8 = PB0
try pin 9 = PB1
try pin 10 = PB2
try pin 11 = PB10
buzzer = PB10
try pin 12 = PA1
up = PA1
try pin 13 = PB12
try pin 14 = PA11
try pin 15 = PF6
try pin 16 = PF7
try pin 17 = PA12
orange = PA12
try pin 18 = PA15
green = PA15
try pin 19 = PB3
red = PB3
try pin 20 = PB4
try pin 21 = PB5
try pin 22 = PB8
#define LED_RED		PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	PA1
#define LOWER_LED	PA0
#define BUZZER	PB10
```

AILIFE4798, 2024-01-12:
Well because you say it's 99% ready I thought the 2.6.1 board is not needed any more so I left it at school
I have ordered 4 set of the cheap new layout but it'll take a while to arrive

AILIFE4798, 2024-01-12:
@reed-dan do you want to test blinking example on mm32
You will need to manually trace one led pin and tell me I'll make that blink to see if the environment is working and if it's even possible to unlock a mm32 which I'm very concerned it's not

RoboDurden, 2024-01-12:
No video but now VBatt and CurrentDC detection:

```
0 try 1 = PC14
hall 0 = PC14 : 4 ms
1 try 3 = PF0
1 try 4 = PF1
hall 1 = PF1 : 4 ms
2 try 6 = PB11
hall 2 = PB11 : 5 ms
wrong oder 0
hall oder 1 :
#define HALL_A = PB11
#define HALL_B = PC14
#define HALL_C = PF1
'r'=red,	'o'=orange,	'g'=green,	'u'=up,	'd'=down,	'b'=buzzer
RET to pause,	'-' to toggle direction,	'l' to list
'c' to reset, 'x' to delete and 's' when finished.
try PC15
try PF0
try PA0
pause
try PA0
try PA0
try PA0
try PA0
LOWER_LED = PA0
try PA0
continue
try PA4
try PA5
try PA6
try PA7
try PB0
try PB1
try PB2
try PB10
BUZZER = PB10
try PA1
pause
try PA1
try PA1
UPPER_LED = PA1
try PA1
continue
try PB12
try PA11
try PF6
try PF7
try PA12
pause
LED_ORANGE = PA12
try PA12
continue
try PA15
pause
try PA15
LED_RED = PA15
continue
try PB3
pause
try PB3
LED_GREEN = PB3
try PB3
continue
try PB4
try PB5
#define LED_RED		PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	PA1
#define LOWER_LED	PA0
#define BUZZER	PB10
try PB8
try PB9
#define LED_RED		PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	PA1
#define LOWER_LED	PA0
#define BUZZER	PB10
'f'=battery voltage
RET to pause,	'-' to toggle direction,	'l' to list
'c' to reset, 'x' to delete and 's' when finished.
PA4: VBATT ?= 98.47
PA4: VBATT ?= 98.47
PA4: VBATT ?= 98.47
PA4: VBATT ?= 98.47
PA4: VBATT ?= 98.46
PA4: VBATT ?= 98.47
PA4: VBATT ?= 98.47
try PA5
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
try PA6
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
try PA7
PA7: VBATT ?= 4.43
PA7: VBATT ?= 4.35
PA7: VBATT ?= 4.44
PA7: VBATT ?= 4.38
PA7: VBATT ?= 4.43
PA7: VBATT ?= 4.34
PA7: VBATT ?= 4.43
try PB0
PB0: VBATT ?= 32.97
PB0: VBATT ?= 32.88
PB0: VBATT ?= 32.98
PB0: VBATT ?= 32.88
PB0: VBATT ?= 32.89
PB0: VBATT ?= 32.98
PB0: VBATT ?= 32.89
try PB1
PB1: VBATT ?= 16.30
PB1: VBATT ?= 16.33
PB1: VBATT ?= 16.30
PB1: VBATT ?= 16.33
PB1: VBATT ?= 16.33
PB1: VBATT ?= 16.30
PB1: VBATT ?= 16.33
try PA4
PA4: VBATT ?= 23.54
PA4: VBATT ?= 23.54
PA4: VBATT ?= 23.53
PA4: VBATT ?= 23.53
PA4: VBATT ?= 23.54
PA4: VBATT ?= 23.53
PA4: VBATT ?= 23.53
try PA5
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
PA5: VBATT ?= 98.47
try PA6
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02

#define VBATT		PA4
#define CURRENT_DC	TODO_PIN
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.02
PA6: VBATT ?= 33.01
PA6: VBATT ?= 33.01
PA6: VBATT ?= 33.02

#define VBATT		PA4
#define CURRENT_DC	TODO_PIN
'f'=current DC
RET to pause,	'-' to toggle direction,	'l' to list
'c' to reset, 'x' to delete and 's' when finished.
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
PA5: CURRENT_DC ?= 0.00	
PA5: CURRENT_DC ?= 0.20	
try PA6
PA6: CURRENT_DC ?= 0.60	
PA6: CURRENT_DC ?= 0.81	
PA6: CURRENT_DC ?= 0.60	
PA6: CURRENT_DC ?= 0.60	
PA6: CURRENT_DC ?= 0.20	
PA6: CURRENT_DC ?= 0.81	
PA6: CURRENT_DC ?= 0.81	
PA6: CURRENT_DC ?= 0.81	
PA6: CURRENT_DC ?= 1.01	
PA6: CURRENT_DC ?= 1.41	
PA6: CURRENT_DC ?= 0.40	
PA6: CURRENT_DC ?= 0.20	
PA6: CURRENT_DC ?= 0.40	
PA6: CURRENT_DC ?= 0.40	

#define VBATT		PA4
#define CURRENT_DC	PA6
```

current identification is not easy. I speed up the motor to pwm 700 and it needs 2 seconds no motor to read the offset in order to display a correct reading the following 6 seconds.
Then putting you hand on the motor to increase the current consumption..

ideas welcome

AILIFE4798, 2024-01-12:
try to display all pin result in same time so user only need to do once calibration and choose from them
zero them all for 2second>display average current of each pin 10 times>let user decide which one is it
and to increase the current draw you should acelerate the motor back and forth

RoboDurden, 2024-01-12:
Yes that is a good idea :-) spinning forward and backward while scanning the possible adc pins for some sine function..
That would not need any user interaction.
But not easy to implement :-/

AILIFE4798, 2024-01-12:
it should look like this
```
select the pin that is responding by pressing number key
1->PB0: CURRENT_DC ?= -0.50	
2->PB1: CURRENT_DC ?= 0.20	
3->PA4: CURRENT_DC ?= 0.00	
4->PA5: CURRENT_DC ?= 2.10	
----------------------------------------
1->PB0: CURRENT_DC ?= -0.50	
2->PB1: CURRENT_DC ?= 0.20	
3->PA4: CURRENT_DC ?= 0.20	
4->PA5: CURRENT_DC ?= 2.10	
```

AILIFE4798, 2024-01-12:
because the adc is not accurate enough it will be very very hard to sense automaticly
so i think manual interaction is still required
but accelerate the mass of wheel is easy to create load on motor which otherwise is not easy to do by user

RoboDurden, 2024-01-12:
It's not that easy. Every adc pin has an offset. That is why the first 2 seconds i have to halt the motor and do the callibration.
But then every adc pin is showing a current of 0 A. And as the max A must be something like 20A, you only see 1.4 A with you hand heavy on the motor.
No good to let the user choose.. But if switch von speed +700 to -700, that might generate some 5 A.
But the cc power supply should be set to 2A.

Best would be a fourier transformation to detect the frequency in the current log.
But i am already very close to the 32 kB limit. Already need to optimize code :-(

AILIFE4798, 2024-01-12:
if the current dc had amplifier it would be possible to sense automaticly
but to save cost the current resistor is connected to adc directly

AILIFE4798, 2024-01-12:
> It's not that easy. Every adc pin has an offset. That is why the first 2 seconds i have to halt the motor and do the callibration. But then every adc pin is showing a current of 0 A. And as the max [...]

yes at start stop the motor and zero every possible pin at once
and keep displaying value of every pin to let user see which one is responding when you put hand on motor
it should be possible to read multiple adc at once right

RoboDurden, 2024-01-12:
You better wait for the video :-)
User can already see the adc value chaning while putting hand on motor.
But you can not watch 5 values at the same time.
So if the autodetect of forward/backward does not work (or is to complicated for me now), then my current solution already works quite nicely.
But i need different boards to test, because i had to tune my autodetection quite a bit to get the results i want..

AILIFE4798, 2024-01-12:
ok
i dont know you cant watch multiple values at the same time
then yes for devices that does not have currentdc amplifier it would be better to detect phase current but again that would be too much work
its not too hard to trace manually anyways so your current solution is maybe good enough
ill help test it when i can

reed-dan, 2024-01-12:
> @reed-dan do you want to test blinking example on mm32 You will need to manually trace one led pin and tell me I'll make that blink to see if the environment is working and if it's even possible to unlock a mm32 which I'm very concerned it's not

I should have my buck convert in a few days so I can test somethings out. I found this repository [https://github.com/koendv/t12-958](https://github.com/koendv/t12-958) which references unlocking a MM32L072XX with "When using JLink, the first time firmware is downloaded, type mon flash erase to unlock the device."

AILIFE4798, 2024-01-13:
i have compiled a [firmware](https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32/blob/main/HoverBoardMindMotion/Objects/HoverboardOutput.hex) for mm32 that (should) blink pa5
if you need to change pin i can compile another one
acording to multiple source stlink will not work with mm32
a daplink or jlink should do which just require flashing the stlink but im not sure if the stlink clone is possible to be flashed
as of the software for flashing,stlink utilities 100% wont work
someone mentioned openocd or pyocd to work
but actually it seems like the best solution to upload is using keil directly and with daplink firmware
i have no idea if any of the above is true you have to try it ur self to know

RoboDurden, 2024-01-13:
MM32 topics please in the 2.8 issue, thanks.

Yes i now am able to detect the small back-currents when changing rotation speed.
First autodetect binaries online: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BinariesToTest

```
'f'=current DC
RET to pause,	'-' to toggle direction,	'l' to list
'c' to reset, 'x' to delete and 's' when finished.
now PC13 : -16453
now PA4 : -15050
now PA5 : -16392
now PA6 : 11438
now PA7 : -11938
now PB0 : 1103
now PB1 : -2703
#define CURRENT_DC = PA6 //11438
now PA0 : -14597
now PA4 : -15278
now PA5 : -16417
now PA6 : 11485
now PA7 : -10987
now PB0 : 3828
now PB1 : -4758
#define CURRENT_DC = PA6 //11485
now PA0 : -13445
now PA4 : -15393
now PA5 : -16702
now PA6 : 11434
now PA7 : 7572
now PB0 : -1155
now PB1 : -15467
#define CURRENT_DC = PA6 //11434

```

AILIFE4798, 2024-01-13:
now only button pin and latch pin left
latch pin can be done without eeprom if not possible to use,just ask user to use last printed line in serial monitor
and button pin is easy to detect after latch pin is found

short circuit pin(timer_bldc_emergency_shutdown) is not possible to auto detect

phase current, infrared sensor for pedel, charge detect is possible to auto detect,but is not used in firmware so theres no point to try

RoboDurden, 2024-01-13:
# https://youtu.be/PenYna7ktT0 :-)

AILIFE4798, 2024-01-13:
i have watched the video
i think the way of autodetect led and buzzer need to be changed
it should keep outpot to same pin untill user either press 
g for green, r for red......etc , or +,- for nothing and after any of these input is detected then switch to next pin(or previous id its -)
because keep changing every second will be too fast to react and 3 second it will take forever to detect

RoboDurden, 2024-01-13:
Give it a try ! 1 second for one pin is fast and simply hit the enter key to pause/continue :-)

AILIFE4798, 2024-01-13:
i will when i can :) and real untraced board on the way too

AILIFE4798, 2024-01-13:
Wouldn't it be possible to run board from charger so it's automatically limited to 2a for those people that don't have a lab bench charger for some reason
(Bench power supply is a really good tool everyone into electronic should have one although it's also the most dangerous tool in the lab)

RoboDurden, 2024-01-14:
@AILIFE4798 yes i will test if the 2A Charger will break down on a short cut or if it generally survies.
@Candas1 , @robcazzaro , i finally have written a Gen2.x autodetect firmware: https://youtu.be/PenYna7ktT0 . The SELF_HOLD is next to be done. The GD32F130 has a VBAT pin (not supported by MM32) which mostly seem to be connected to 3.3V:

```
GD32F1x0 User Manual 10.4.12 Battery voltage monitoring
The VBAT channel can be used to measure the backup battery voltage on the VBAT pin. When
the VBATEN bit in ADC_CTL1 register is set, VBAT channel (ADC_IN18) is enabled and a
bridge divider by 2 integrated on the VBAT pin is also enabled automatically with it. As VBAT
may be higher than VDDA, this bridge is used to ensure the ADC correct operation. And it
connects VBAT/2 to the ADC_IN18 input channel. So, the converted digital value is VBAT/2. In
order to prevent unnecessary battery energy consumption, it is recommended that the bridge
will be enabled only when it is required.

```

I am running my autodetect function from within the 16 kHz CalculateBLDC(). So the idea is to set all remaining io pins to HIGH and then step by step set one io pin to LOW for x/16k seconds.
VBAT can also trigger a watchdog i think.

```
WDCHSEL [4:0] : 
Analog watchdog channel select
00000: ADC channel0
00001: ADC channel1
00010: ADC channel2
……
01111: ADC channel15
10000: ADC channel16
10001: ADC channel17
10010: ADC channel18
Other values are reserved.
Note: ADC analog inputs Channel16, Channel17 and Channel 18 are internally
connected to the temperature sensor, to VREFINT and to VBAT analog inputs
```

Maybe you like to share some of your ideas. You might remember that I do not like that low level hardware programming ...

AILIFE4798, 2024-01-14:
the charger have to limit the current
because there is no charging circuit either in the battery or on the hoverboard main board
i didnt have any of the charger to reverse engineer since everyone lost there before selling their hoverboard
but i think it should have current limiting 
only concern is the capacitor smoothing the output can create some damage to the mosfet 

you cannot switch output low that fast you must wait at least 500ms because there is smoothing capacitor on board 
the vbat pin is used to connect a cr2032 battery to power the clock and eeprom to store data when power is off
there is no battery on hoverboard so store the pins in flash
flash have limited lifetime but setting pin is only one time job it is ok to use the flash

RoboDurden, 2024-01-14:
please linke to an eeprom code example :-)

AILIFE4798, 2024-01-14:
there is no battery,eeprom does not work,use flash insted(stored with the code)
https://github.com/lamik/EEPROM-emulation-STM32F1-HAL/tree/master

AILIFE4798, 2024-01-14:
@RoboDurden is the 32k flash enough for remote_allinone (autodetect+uart+uartbus)
or will it for c8t6 only

AILIFE4798, 2024-01-14:
32k is really not enough comming back to arduino level from esp32 which have 4m at least

RoboDurden, 2024-01-14:
I strongly doubt that my autodetect.c and the normale main.c + RemoteUart will fit into the 32 kB.
But i did not really like that eeprom storage anyway.

AILIFE4798, 2024-01-14:
it seems quite complex but i think its manageable
the storage would probably not corporate
i really want to know how the original firmware got foc and bluetooth communication and balancing pid in 32k

AILIFE4798, 2024-01-14:
may be intresting to glitch one of the ic to dump original firmware
i did ordered some extra for experments

RoboDurden, 2024-01-14:
Okay i succeedd with the SELF_HOLD without the board shutting off :-)
It did not even need the watchdog to detect the failure of the 3.3 Volt :-))
But again i had to tune it to my 2.13 test setup.

```
	case AUTODETECT_Stage_Hold:
		SetPWM(0);
		if (adc_buffer.v_batt < 1350)
		{
			aiPinScan[8] = aoPin[iTest].i;
			sprintf(sMessage,"#define SELF_HOLD %s //%i\n",aoPin[iTest].s,adc_buffer.v_batt);
		}
		digitalWrite(aoPin[iTest].i,(buzzerTimer%8000) < 700 ? 0 : 1);	// relase SELF_HOLD for a short time
		break;
```

I have disabled the "auto-next-pin". So user simply hits RETURN to test next pin. That is faster to cycle through the pins and then send 'd' for down or 'g' for green, etc., which then will also move on to next pin.
With 'i' (+RET), the "auto-next-pin" mode can be toggled.

```
0 try 1 = PC14
hall 0 = PC14 : 4 ms
1 try 3 = PF0
1 try 4 = PF1
hall 1 = PF1 : 4 ms
2 try 6 = PB11
hall 2 = PB11 : 4 ms
wrong oder 0
hall oder 1 :
#define HALL_A = PB11
#define HALL_B = PC14
#define HALL_C = PF1
'r'=red,	'o'=orange,	'g'=green,	'u'=up,	'd'=down,	'b'=buzzer
'i'=toggle auto-next-pin,	'-'=toggle direction,	'x'=delete pin
'l'=list,	'c'=reset and 's'=next stage
RET for next pin
now PC13try PC15
try PF0
try PA0
try PA4
try PA5
try PA6
try PA7
try PB0
try PB1
try PB2
try PB10
try PA1
try PB12
try PA11
try PF6
try PF7
try PA12
try PA15
try PB3
try PB4
try PB5
#define LED_RED		PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	PA1
#define LOWER_LED	PA0
#define BUZZER	PB10
#define LED_RED		PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	PA1
#define LOWER_LED	PA0
#define BUZZER	PB10
'v'=battery voltage
'i'=toggle auto-next-pin,	'-'=toggle direction,	'x'=delete pin
'l'=list,	'c'=reset and 's'=next stage
RET for next pin
now PA4PA4: VBATT ?= 98.50
PA4: VBATT ?= 98.50
PA4: VBATT ?= 98.51
PA4: VBATT ?= 98.52
PA4: VBATT ?= 98.50
PA4: VBATT ?= 98.50
PA4: VBATT ?= 98.51
PA4: VBATT ?= 98.50
PA4: VBATT ?= 98.51
try PA5
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51
try PA6
PA6: VBATT ?= 33.04
PA6: VBATT ?= 33.05
PA6: VBATT ?= 33.05
PA6: VBATT ?= 33.06
try PA7
PA7: VBATT ?= 4.45
PA7: VBATT ?= 4.45
PA7: VBATT ?= 4.48
try PB0
PB0: VBATT ?= 33.04
PB0: VBATT ?= 33.03
try PB1
PB1: VBATT ?= 16.40
PB1: VBATT ?= 16.39
PB1: VBATT ?= 16.41
PA2: VBATT ?= 32.62
PA2: VBATT ?= 32.62
PA2: VBATT ?= 32.54
PA2: VBATT ?= 32.20
PA2: VBATT ?= 32.51
try PA3
PA3: VBATT ?= 32.45
PA3: VBATT ?= 32.10
PA3: VBATT ?= 32.10
PA3: VBATT ?= 32.47
PA3: VBATT ?= 32.33
try PA4
PA4: VBATT ?= 23.53
PA4: VBATT ?= 23.54
PA4: VBATT ?= 23.54
PA4: VBATT ?= 23.56
PA4: VBATT ?= 23.53
PA4: VBATT ?= 23.55
PA4: VBATT ?= 23.55
try PA5
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.50

#define VBATT		PA4
#define CURRENT_DC	TODO_PIN
PA5: VBATT ?= 98.50
PA5: VBATT ?= 98.50
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51
PA5: VBATT ?= 98.51

#define VBATT		PA4
#define CURRENT_DC	TODO_PIN
current_DC: wait one cycle :-)
'i'=toggle auto-next-pin,	'-'=toggle direction,	'x'=delete pin
'l'=list,	'c'=reset and 's'=next stage
RET for next pin
now PA5 : -21652
now PA6 : 9124
now PA7 : -22047
now PB0 : -22350
now PB1 : -18842
now PA2 : -22351
now PA3 : -22348
#define CURRENT_DC = PA6 //9124
now PA5 : -21631
now PA6 : 9836
now PA7 : -22102
now PB0 : -21522
now PB1 : -2144
now PA2 : -22352
now PA3 : -22353
#define CURRENT_DC = PA6 //9836
now PA5
#define VBATT		PA4
#define CURRENT_DC	PA6
now detecting SELF_HOLD pin,
release the on/off button !!!
try PC15
try PF0
try PA5
try PA7
try PB0
try PB1
try PB2
#define SELF_HOLD PB2 //1346
#define SELF_HOLD PB2 //1349
#define SELF_HOLD PB2 //1347
#define SELF_HOLD PB2 //1346
#define SELF_HOLD PB2 //1348
#define SELF_HOLD PB2 //1346
#define SELF_HOLD PB2 //1349
try PB12
try PA11
try PF6
try PF7
try PB4
try PB5
try PB8
try PB9
try PA2
try PA3
try PC13
try PC15
try PF0
try PA5
try PA7
try PB0
try PB1
try PB2
#define SELF_HOLD PB2 //1346
#define SELF_HOLD PB2 //1347
#define SELF_HOLD PB2 //1347
#define SELF_HOLD PB2 //1349
#define SELF_HOLD PB2 //1346
#define SELF_HOLD PB2 //1348
try PB12
try PA11
try PF6
try PF7
try PB4
try PB5
try PB8
```

AILIFE4798, 2024-01-14:
how is it possible
then self hold should be tested on first bootup directly so keep pressing the button is not needed

RoboDurden, 2024-01-14:
when the SELF_HOLD is released, the dcdc converter for the 5V regulator is no longe powered.
when the 5V regulator fails, the 3.3V regulator will also fail.
This will start the 3.3V at the mcu to drop.
As i scan the VBAT pin (which the MM32 does not have) at 16 kHz, i am able to detect the failing 3.3 quick enough to set the SELF_HOLD back again to high and the dcdc converter is preventing the mcu crash.

AILIFE4798, 2024-01-14:
ok thats amazing

AILIFE4798, 2024-01-14:
what about detecting the actual vbat voltage of the main battery 24v
it should drop first before 3.3 start to drop
you just need to detect it dropping 5v lower then before

RoboDurden, 2024-01-14:
There is no adc pin connected to after the HOLD mosfet that powers the dcdc converter.

I have added the BUTTON detection and autodetec is now complete :

here the lower half of my gen2.13 log:

```
#define VBATT		PA4
#define CURRENT_DC	TODO_PIN
#define SELF_HOLD	TODO_PIN
#define BUTTON	TODO_PIN
current_DC: wait one cycle :-)
'i'=toggle auto-next-pin,	'-'=toggle direction,	'x'=delete pin
'l'=list,	'c'=reset and 's'=next stage
RET for next pin
now PA5 : -22074
now PA6 : 6503
now PA7 : -22247
now PB0 : -22349
now PB1 : -9999
now PA2 : -22351
now PA3 : -22352
#define CURRENT_DC = PA6 //6503
now PA5
#define VBATT		PA4
#define CURRENT_DC	PA6
#define SELF_HOLD	TODO_PIN
#define BUTTON	TODO_PIN
now detecting SELF_HOLD pin,
release the on/off button !!!
try PC15
try PF0
try PA5
try PA7
try PB0
try PB1
try PB2
#define SELF_HOLD PB2 //1318
#define SELF_HOLD PB2 //1339
#define SELF_HOLD PB2 //1339
#define SELF_HOLD PB2 //1344
#define SELF_HOLD PB2 //1338
#define SELF_HOLD PB2 //1338
try PB12
try PA11
try PF6
try PF7

#define VBATT		PA4
#define CURRENT_DC	PA6
#define SELF_HOLD	PB2
#define BUTTON	TODO_PIN
now 'b'=BUTTON pin,
release the on/off button and hit ENTER
push button and hit ENTER
#define BUTTON	PC15
release button and hit ENTER

#define VBATT		PA4
#define CURRENT_DC	PA6
#define SELF_HOLD	PB2
#define BUTTON	PC15
#define LED_RED		PB3
#define LED_ORANGE	PA12
#define LED_GREEN	PA15
#define UPPER_LED	PA1
#define LOWER_LED	PA0
#define BUZZER	PB10
#define HALL_A = PC14
#define HALL_B = PB11
#define HALL_C = PF1
```

i need testers :-/

RoboDurden, 2024-01-15:
okay i will stop working on this until a few users have tried the binaires.
I have tested from 24V to 29V.
For higher voltages, the hall detection has to be adapted because motor is spinning faster.
Also the HOLD pin detection fails because the voltage drop on HOLD release can no longer be detected.

Both problems can be solved. But i for now will stop here.

AILIFE4798, 2024-01-15:
had to do some maintenance on ESPhome devices today a bit busy
the self hold can be adapted to just keep looking for voltage and if it dropped more then 50 for example then last checked then it know it is the correct pin
dont use absolute value and preprogrammed time use the change and theres only a timeout time that is impossible to reach

RoboDurden, 2024-01-15:
I have already moved from absolute value to percentage: 

```
// init while-case
case AUTODETECT_Stage_Hold: 
  ..
  iVBatMin -=  (2* iVBatMin) / 100;		// 98% of lowest so far measured 3.3V VBat

..

// scan while-case
case AUTODETECT_Stage_VBatt:
  ..
  if (iVBatMin > adc_buffer.current_dc)	// dma_init_struct_adc.memory_addr->[1] is used to read mcu 3.3V
    iVBatMin = adc_buffer.current_dc;	// needed for later AUTODETECT_Stage_Hold

..

case AUTODETECT_Stage_Hold:
  SetPWM(0);
  if (adc_buffer.v_batt < iVBatMin)
  //if (adc_buffer.v_batt < 1350)
  {
    aiPinScan[8] = aoPin[iTest].i;
    sprintf(sMessage,"#define SELF_HOLD %s //%i,\tmin=%i\n",aoPin[iTest].s,adc_buffer.v_batt,iVBatMin);
  }

```

As i said, i will stop answering here until some user has tested the code.

[154 more replies]