RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #54: new board type

RebellRebecca, 2024-01-20:
hey i think i have a board type wich i have not found anywhere else on the internet.
it is allmost the same as the 2.12 board but instead of red it is yellow , has an inductor on the botom right corner and most importantly has a different mcu ( STM32F103 ). this chip is also used in the two in one type hoverboard controllers.  do you know if there exists a firmware for this board?

[image]

RoboDurden, 2024-01-20:
Well porting this repo from gd32f130 to stm32f103 would be much easier than the mm32 of the more modern hoverboards.
But your board with an stm32 mcu must be very old, so no other people would find it again.

And with an stm32f103 just like the Gen1 board, it would not really need much modifications to get the EFeru FOC firmware running on it.
There is already the option to only use left motor or right motor.
And of course all the pins would be needed to be traced again.
My autodetect firmware is not running on a stm32.
But many people here have succeeded with tracing by hand.

I guess it will be easier for you to buy some more $20 used hoverboards until you find a board that there is already a firmware for.

AILIFE4798, 2024-01-20:
I think stm32 and gd32 is compatible you can just flash the existing binary to it and try with constant current supply

AILIFE4798, 2024-01-20:
All the clones are designed to be pin and code compatible with stm32, like cks32,gd32,apm32,ch32  including mm32f103, but the one found on hoverboard is mm32spin, so that's why it is not compatible (or is it? Noone have tried!)

AILIFE4798, 2024-01-20:
This layout have no currentDC and emergency stop
The pa6,7 is connected to imu as always
Pb2,3 is comms masterslave
MOSFET pin is also standardized on stm32 apparently

AILIFE4798, 2024-01-20:
Should be named 2.12.1 it's really quite close

RoboDurden, 2024-01-20:
If i remember correctly, the Gen1 boards originally had an stm32f103 but then mostly a gd32f103 which is compatible to the stm32f103.

So yes , if we add a GD32F103 target to this firmware, the binary should also run on a stm32f103.
Adding a target for stm32f103 will probably not work for all the adc/timer/dma functions, which are from the GD library.

RoboDurden, 2024-01-20:
I do not see a difference between this board and gen2.12 :-) 
Only the mcu.
So after i added a gd32f103 target, the autodetect might already work for this board

AILIFE4798, 2024-01-20:
Isn't the original 2.12 gd32f103c6t6?

RoboDurden, 2024-01-20:
You are right :-)
[image]

AILIFE4798, 2024-01-20:
So just try the original 2.12 code with cc supply and hope for the best

AILIFE4798, 2024-01-20:
If it is not traced then test the autodetect I think other then vcc it'll work
Idk why bother submitting the board if you are not tracing it

RoboDurden, 2024-01-20:
@TheJP are you still interested to get your 2.12 board running ? We have a new users with such a board and i have written an autodetect firmware :-)

RoboDurden, 2024-01-20:
Yes, i added the gd32f103 target and successfully compiled the autodetect firmware:
[image]

AILIFE4798, 2024-01-20:
I think it is stm32f103c8 or c6 anyways so no need to compile another one

Hishamahmed97, 2024-03-08:
@RoboDurden Hi Robo!
I am trying to build on this target with the keil project but I got some errors. Would you show me some directions how to correctly configure it to run on GD32F103

RoboDurden, 2024-03-08:
Do not post multiple times :-(

Hishamahmed97, 2024-03-08:
sorry for that. es tut mir leid 😄

Hishamahmed97, 2024-03-16:
Hello guys. I have been trying to run the autodetect firmware for this board but I got nothing in serial monitor. I've tried the Tx Rx PA2,PA3 and PB6,PB7 both bot didn't get it right. Can you help me configure the keil project correctly for auto_detect because when I build it i got alot of errors.
Thanks for your marvelous efforts.

RoboDurden, 2024-03-17:
Select the gd32f103 target in the drop down list to the top right of the Keil IDE.
Uncomment #define Remote_Autodetect I'm config.h

Hit f7
Hit f8

I think that no one ever had the happiness to make a gd32f103 / stm32f103 board running.

Maybe a gd32f103 is compatible with stm32f103 Keil library.
But stm32f103 is not compatible with gd32f103 Keil Code library.

Erik-96, 2024-09-04:
Hi,
I seem to have same board layout (2015.6.25)) but with a GD32F103C8T mcu. I added some info onto the existing picture
[image]

Here are the Pins I managed to trace.
I reworked the  "Hoverboard-JC2015.7.31.V2.0-Hack-main" from EFERU and got most of them working.
The motor is spinning but I'm having issues with the UART communication.
/ LED defines
#define LED_RED GPIO_PIN_15 //JMA 6.29 ok
#define LED_RED_PORT GPIOA //JMA 6.29 ok
#define LED_GREEN GPIO_PIN_3 //JMA 6.29 ok
#define LED_GREEN_PORT GPIOB //JMA 6.29 ok
#define LED_BLUE GPIO_PIN_12 //* 
#define LED_BLUE_PORT GPIOA //was GPIOA

// Brushless Control DC (BLDC) defines
// Channel G
#define RCU_TIMER_BLDC RCU_TIMER0
#define TIMER_BLDC TIMER0
#define TIMER_BLDC_CHANNEL_Y TIMER_CH_2
#define TIMER_BLDC_YH_PIN GPIO_PIN_10
#define TIMER_BLDC_YH_PORT GPIOA
#define TIMER_BLDC_YL_PIN GPIO_PIN_15
#define TIMER_BLDC_YL_PORT GPIOB
// Channel B
#define TIMER_BLDC_CHANNEL_G TIMER_CH_1
#define TIMER_BLDC_GH_PIN GPIO_PIN_9
#define TIMER_BLDC_GH_PORT GPIOA
#define TIMER_BLDC_GL_PIN GPIO_PIN_14
#define TIMER_BLDC_GL_PORT GPIOB
// Channel Y
#define TIMER_BLDC_CHANNEL_B TIMER_CH_0
#define TIMER_BLDC_BH_PIN GPIO_PIN_8
#define TIMER_BLDC_BH_PORT GPIOA
#define TIMER_BLDC_BL_PIN GPIO_PIN_13
#define TIMER_BLDC_BL_PORT GPIOB

// Hall sensor defines JMA OK
#define HALL_A_PIN GPIO_PIN_0 //was GPIO_PIN_11
#define HALL_A_PORT GPIOA //was GPIOB
#define HALL_B_PIN GPIO_PIN_1 //green //was GPIO_PIN_1
#define HALL_B_PORT GPIOA //was GPIOF
#define HALL_C_PIN GPIO_PIN_2 //was GPIO_PIN_14
#define HALL_C_PORT GPIOA //was GPIOC

// Usart master slave defines JMA OK
#define USART_MASTERSLAVE USART2 //was USART1
#define USART_MASTERSLAVE_TX_PIN GPIO_PIN_10 //was GPIO_PIN_2
#define USART_MASTERSLAVE_TX_PORT GPIOB //was GPIOA
#define USART_MASTERSLAVE_RX_PIN GPIO_PIN_11 //was GPIO_PIN_3
#define USART_MASTERSLAVE_RX_PORT GPIOB //was GPIOA

// ADC defines JMA OK
//#define VBATT_PIN	GPIO_PIN_5//was GPIO_PIN_4
#define VBATT_PIN	5
#define VBATT_PORT GPIOA //was GPIOA
#define VBATT_CHANNEL ADC_CHANNEL_5 //was ADC_CHANNEL_5

#define CURRENT_DC_PIN	7 // temp dummy
#define CURRENT_DC_PORT VBATT_PORT //was GPIOA
#define CURRENT_DC_CHANNEL ADC_CHANNEL_7 //was ADC_CHANNEL_6

// *added pins
#define LIGHTSENSEA_PIN GPIO_PIN_6 //* 
#define LIGHTSENSEA_PORT GPIOA //was GPIOA
#define LIGHTSENSEB_PIN GPIO_PIN_15 //*
#define LIGHTSENSEB_PORT GPIOC //was GPIOA
#define LED_BAR_PIN GPIO_PIN_13 //* 
#define LED_BAR_PORT GPIOC //was GPIOA

#define I2C0_SDA_PIN GPIO_PIN_9 //* need remap
#define I2C0_SDA_PORT GPIOB //
#define I2C0_SCL_PIN GPIO_PIN_8 //* need remap MCU
#define I2C0_SCL_PORT GPIOB //need remap MCU

// Self hold defines JMA OK
#define SELF_HOLD_PIN GPIO_PIN_4 //was GPIO_PIN_2
#define SELF_HOLD_PORT GPIOA //was GPIOB

// Button defines JMA OK
#define BUTTON_PIN GPIO_PIN_3//was GPIO_PIN_15
#define BUTTON_PORT GPIOA //was GPIOC

// Usart steer defines
#define USART_STEER_COM USART0
#define USART_STEER_COM_TX_PIN GPIO_PIN_6
#define USART_STEER_COM_TX_PORT GPIOB
#define USART_STEER_COM_RX_PIN GPIO_PIN_7
#define USART_STEER_COM_RX_PORT GPIOB

#ifdef MASTER
// Buzzer defines JMA OK
#define BUZZER_PIN GPIO_PIN_11 //was GPIO_PIN_10
#define BUZZER_PORT GPIOA //mwas GPIOB

//untraced-Remaining GPIO Pins: PB1, PB2,  PC14, PD0, PD1
//Missing functions: Charger connected,...?

How to go about and use one of your firmware version. 
If I uncomment the line "#define Remote_Autodetect I'm config.h" as stated above in firmware "Hoverboard-Firmware-Hack-Gen2.x-GD32-main" from your repo and build I get over 30 errors.
I must be missing something.

What are the layouts referred to in your firmware, if the targets are the MCU's. What is the correlation to the defines files. 

Your help would be much appreciated.

I-hate-2FA, 2024-09-04:
this repo does not work on f103, only f130 and e230
you need to use gen1 none foc version

RoboDurden, 2024-09-04:
@Erik-96 , did you get the motors rotating with this firmware ? https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack

Is it a Keil project ?
Then you should be able to resolve the 30 erros of this repo.
I lost the happiness to work on the stm32f103 target..

@AILIFE4798 , stay positive :-)

I-hate-2FA, 2024-09-04:
no really it doesnt work on f103, ive tested, theres no f103 microcontroller pack in your project file, and its way too hard to adapt to it(because f103 have no pinmodeAF)

Erik-96, 2024-09-04:
Thank you for your quick replies.

It is indeed a Keil project, however I  had to modiffie many of the defined pins and registers as my board is a different version as the one used in the 15.7.31 v2.

RoboDurden, 2024-09-04:
You should get a working uart keil project and copy and paste code from there to your keil project.
My Gen2.x here will not help you. But if have succeeded, please share a link to your fork, then i might be able to get my 103 target running.

I-hate-2FA, 2024-09-04:
it is def possible to port but without too much modification of the code its very hard ive tried it

I-hate-2FA, 2024-09-04:
the structure need to be changed
or just start a new repo for f103

RoboDurden, 2024-09-04:
the bldc.c and main.c of that gd32f103 repo looks very similar so i do not see a reason why my Gen2.x should not support such a mcu.

Erik-96, 2024-09-05:
Thanks again for your comments.
Small update,
I don't think its relevant for your fork here but I managed to get the "Hoverboard-JC2015.7.31.V2.0-Hack-main" working on my board version with GD32F103C8T.
( ๑‾̀◡‾́)σ"

It needed some more pin remapping  and declaration of alternate pin functions for the UART2 to get it to work. The UART communication was not yet implemented in the above firmware, perhaps I just had an old version.
Also the DMA IRQ handlers did not match with the onces stated in "startup_gd32f10x_hd.s" and the  uart_rx_address used was wrong causing the DMA to read junk.
Pretty obvious and stupid things but unexperienced as I am it did take me a while to figure it out. 
I'm glad I learned something,... still alot of room to improve my skills :-)

I-hate-2FA, 2024-09-05:
i have gd32f103 board id like to know how its possible

RoboDurden, 2024-09-05:
@Erik-96 please put your working fork online to your repos.
Maybe you even are able to fork my https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32 and make the neccessary changes for that target :-)