RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #55: New chip?

StomperQc, 2024-01-25:
I bought some refirbushed hoverboard for cheap and it's a split board with a new chip i guess, i have a lot of those board so if you want one ill send a pair to you and ill pay for shipping, so you can test it.

Brand: Gyrocopter
36v battery

The chip brand is LKS and the first number on it is lks32mc

[image]
[image]
[image]
[image]

AILIFE4798, 2024-01-25:
lks is back...

RoboDurden, 2024-01-25:
Yes, i already search the other issue ...

AILIFE4798, 2024-01-25:
if not for the mm32spin then i may be able to try support it

AILIFE4798, 2024-01-25:
but lets be real the hoverboard looks very shtty the mosfet is tiny and plastic wheel really not worth support this
and no opamp on board (but lks32 have one built in the chip)

RoboDurden, 2024-01-25:
This is the other issue: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/29 "Support for board YFS-N2? "

RoboDurden, 2024-01-25:
@AILIFE4798 ,  maybe you again can find the dataship for this MCU. The other mcu seems to not only have a dual opamp but also a 6ch gate driver inside: LKS32MC03x with built-in 6N Gate Driver

I have so many good old hoverboard motors, that i do not care so much for these newer plastic motors. And as i said, there are lots of applications like lawn mover, where the build quality of the motors is not cruicial.

And smd mosefts does not need to be worse than TO220 ones. And again, there are many use cases where 350+ Watt are not needed.

I fear that these LKS mcu are not pin compatible to the GD32.

So i think we may try LKS only after we have mastered the MM32 mcu..

AILIFE4798, 2024-01-25:
can u give me full marking on chip i cannot read it

AILIFE4798, 2024-01-25:
https://datasheet.lcsc.com/lcsc/2110081130_LINKOSEMI-LKS32MC088C6T8_C2798416.pdf

AILIFE4798, 2024-01-25:
phone is a very good tool to trace a board manually. most phone have a macro lens now,with flashlight on it can deliver a very good quality photo,and also good if you need to solder 2 wire to the mcu pin...

AILIFE4798, 2024-01-25:
https://www.lksmcu.com/index.php/DevicePack-en/
keil pack

AILIFE4798, 2024-01-25:
[image]
pinout is not compatiable

RoboDurden, 2024-01-25:
With dedicated hall and mosfet pins, this will make tracing such a board easier..
@StomperQc , for now it is up to you to install Keil and import the device pack that @AILIFE4798 has linked to.
When you then succeed to successfully compile a blink example,
i will be happy to add a LKS32 target to this firmware.

Thanks for the offer to ship such a hoverboard to me (Germany), but i still have many other Gen2 boards here i first want to trace with my autodetect this year.
Where are you from ? May shipping to @AILIFE4798 would also be an option :-)
He by now is more experienced in adding new targets to the firmware !

RoboDurden, 2024-01-25:
Again, 32kB are absolutly okay for this Gen2.x firmware.
And when you have succeeded with MM32, i might very well like to port the FOC example code that will fit into the 32kB.

AILIFE4798, 2024-01-25:
lks32 is also capable of foc and it have dsp to calculate the foc math so the mcu dont have to
the comparator/opamp is also in the mcu and so itll make the board even cheaper but also harder to code
it is really dedicated so it will not be easy if the example code did not work like the mm32

AILIFE4798, 2024-01-25:
you can send one pair to me but i may have to work on mm32 so itll get delayed a bit

StomperQc, 2024-01-25:
Im from Canada, i dont mind sending a pair to @AILIFE4798
As i said i have many of them, i wont be able to do it by myself i dont have the tech knowledge so i dont mind shipping a pair if i and others can reuse them later instead of trowing them out.

AILIFE4798, 2024-01-25:
you can send a pair i will try to support it when i have time
im from china so it wouldnt be very easy nor cheap to send a full hoverboard so i think ill have to find another one to selvage the wheel which is a bit annoying but at least i can tell you how to upload the code to it
this mcu is very new so there is limited resources for programming it but there is example
i will send you my address if you want to
or you try it your self and ill provide all the resources available from random chinese website

StomperQc, 2024-01-26:
Yes im willing to send a pair of board to china, it would cost around 20-30$ so it's not that bad.

You can send me your adress at klempstock@hotmail.com and ill let you know when it shipped.

I dont mind if it takes time i understand everybody is doing it on their own free time.

AILIFE4798, 2024-02-02:
one of the board is 5v and the other is 3.3...

[image]

AILIFE4798, 2024-02-11:
lks32 also have the example foc program(17kb) and compile without error after minor modification
so it wouldnt be too hard to support just like the mm32
but i think the mm32 example is simpler to implement so ill use that

I-hate-2FA, 2025-04-16:
There is keil device pack for it on the official website 
If you want to make auto detect you can port gd32or mm32 auto detect, but if all of your board is same tracing manually can be done in an hour