RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #56: Wiki: Topic to discuss the Wiki

reed-dan, 2024-01-25:
@RoboDurden  @AILIFE4798

Just wanted to create an area were we can discuss progress wiki.

reed-dan, 2024-01-25:
I am trying to organize things by MCU type and then board layouts. 

I personally don't really like the github wiki markup, and would much rather it be basic html or something. I will give it some time though.

AILIFE4798, 2024-01-25:
move all the board away from readme to wiki because its getting too long

AILIFE4798, 2024-01-25:
i will deal with the esp32 and esp8266 later
the serial have to be added later because again no hardware on me the 2.6.1 is not around

reed-dan, 2024-01-25:
@RoboDurden What MCU is on?
2.1
2.6 GD32F130C6T6?
2.9 Can you confirm GD32F103C6T6 or GD32F103C8T6, I believe it is labeled wrong?

reed-dan, 2024-01-26:
Just added information about setting up Keil and the Blink sketch for 2.8 to its wiki. It needs some formatting help, but it outlines the blink issue I am experiencing.

RoboDurden, 2024-01-26:
My Gen2.x is a fork from Gen2.1 = https://github.com/krisstakos/Hoverboard-Firmware-Hack-Gen2.1
[image]
= GD32F130C6T6 = 32kB , 48 pin.

2.6.0 = GD32F130 ????
[image]

2.6.1 =GD32F130C6T6
[image]

2.9 = GD32F103C6T6
[image]

But i also only looked at the images in the Schematics... folders.

Yes i also would like an interactive search where you select gateDriverChips/analogGateDrivers and then 32pin/48pin and then GD/MM/STM/LKS and then C6/C8 while the list of images decreases..

Would be easy for me to do in javascript. But i am not sure if html is possible on github.

And this is not high up on my todo list. People will have to spend several days on their hoverboard projects. It is okay for them to spend two minutes on our impressive list of supported boards :-)

AILIFE4798, 2024-01-26:
Serial flashing toturial is not coming anytime soon because no board with me again I think I'll have to wait for my other c8t6 boards
And the serial method is only for gd32 and mm32
lks32 does not have bootloader so if SWD is disabled that chip is trashed you cannot reprogram it under any circumstances

RoboDurden, 2024-01-26:
@AILIFE4798  Don't you have online second hand sites in your country ?
Here in the Western so called democracies which in truth are nothing but Postcolonial Fascist Dictatorships of Decadency ( PFDD, www.RoboDurden.de ) you can easily buy used hoverboards for $30 incl. shipping.

Maybe try Ruten 露天, Shopee 蝦皮 , Yahoo! Auctions or Facebook marketplace. If you don't have paypal, a seller might accept Paypal :-)

AILIFE4798, 2024-01-26:
hoverboard is too long to be shiped if the seller didnt take it apart and split it in half first but trust me ive tried didnt find any good ones thats close by so i can pick it up my self
actually needed 2 more for my project so i def wanted some more

AILIFE4798, 2024-01-26:
and ill go make esp8266 flash wiki now

AILIFE4798, 2024-01-26:
> I am trying to organize things by MCU type and then board layouts.
> 
> I personally don't really like the github wiki markup, and would much rather it be basic html or something. I will give it some time though.

idk how to use markup either so i will not have any formatting

AILIFE4798, 2024-01-26:
@reed-dan the esp8266 daplink wiki is complete please read it and try it your self

AILIFE4798, 2024-01-26:
> I personally don't really like the github wiki markup, and would much rather it be basic html or something. I will give it some time though.

if you want to make better wiki with html i can host it on my server and other people will not be able to edit it

AILIFE4798, 2024-01-27:
@RoboDurden please move all board layout from readme to wiki with this [example](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Example-page-for-creating-new-board-layout)

AILIFE4798, 2024-01-27:
@RoboDurden autodetect guide finished please edit it because only you know what it exactly does

RoboDurden, 2024-01-27:
> @RoboDurden please move all board layout from readme to wiki with this [example](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Example-page-for-creating-new-board-layout)

As i said, I do want to keep the large list of known boards on the readme to give hope to new users that their board is one of them.

I think, the wiki only needs special information not the full pin definitions..

But adding all the photos from the schematics folders might be a good idea.

I now live outdoors again. -5°C this night.

AILIFE4798, 2024-01-27:
Because readme is only one page at current rate the known board will grow to none senseable level very quickly
It is better to move everything in readme to wiki
And in first line of readme put a link to wiki
And yes when you have a page for each board then you can include all photo in the end of page

RoboDurden, 2024-01-27:
@AILIFE4798 , all this here will not save the world.
And every 3.6 seconds a human dies from hunger.
The Western so called democracy are responsible for over 1,100,000,000 deaths in Africa since 1955.

If you waste your time on unnecessary things, you are an evil man !

reed-dan, 2024-01-28:
I made some improvements to the side menu to make the MCUs [collapsible](https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab). I think this collapse feature may be nice on some of the other pages as well.

AILIFE4798, 2024-01-28:
Home page if wiki looks nice

RoboDurden, 2024-01-28:
yes, home page of wiki does look nice.
I guess i will remove the many images from readme and put a big link to the new wiki instead.

AILIFE4798, 2024-01-28:
Idk why 2.1,2.6.0,2.17,2.18 is not supported tho

RoboDurden, 2024-01-28:
2.1 has never been tested, that why I only offer a test binary. It should work as the gen2.x is a fork of 2.1. but no one ever tested it.
Same for 2.6.0. there might be more differences than the buzzer.

reed-dan, 2024-01-28:
When I was making the list, I knew I would wont to make a category for mcu that should work pending autodetect.

@RoboDurden I thought you had said somewhere that you wanted to keep track of one other piece of information along with the mcu for each layout, what was that? It was one of the other chips I believe.

AILIFE4798, 2024-01-28:
there is only 4 chips on the board
the mcu, the imu(not used so not this),the gate driver ic, the opamp and maybe some buck regulator ic and bluetooth ic
dont think theres any worth documenting

AILIFE4798, 2024-01-28:
is there anything need to be added to wiki other then serial which ill need to test first

RoboDurden, 2024-01-28:
Well I did propose the wiki so that you could summarize the progress you have made concerning the mm32.
And then start a fresh 2.8 issue that only builds on knowledge from the wiki.
If you like to build a complete wiki then you might want to copy everything from the readme to wiki pages.

AILIFE4798, 2024-01-28:
idk why a new issue is needed but i created it anyways
most part of readme is in wiki now except the board list is still not complete but i think @reed-dan is working on it

RoboDurden, 2024-01-29:
i had a quick search on how to add javascript to github pages, and it seems this can only be done on github.io: https://robodurden.github.io/
But i do not want to spread information on yet another url.

The interactive search could be done with multiple drop down lists :-/
1 First dropdown: analog gate drirvers or gate driver chips.
2. dropdown 32pin/48pin 
3. GD/MM/STM/LKS 
4.  C6/C8

And each dropdown would need little images to show the user where to look on the pcb.

But then the user would only see board images after he has choosen all the four drop down lists.

I would prefer that new users see the full (impressive) list like in read.me and then can click on the four filters to reduce the number of images.

But this requires javascript.

And again, i think the Issues should be the place where we communicate and work.
And the wiki only the place where found knowledge is stored so new users do not read lengthy issues.

AILIFE4798, 2024-01-29:
i can tell you that i suceeded with flashing by serial wiki will be in few hours

AILIFE4798, 2024-01-29:
got distracted by esp32

AILIFE4798, 2024-01-29:
@RoboDurden serial flash guide is complete you can check it out but maybe not try it out

RoboDurden, 2024-01-29:
Yes thank you for this serial flashing tutorial. But I hope that I will never have to flash that way. I would not have had the courage to solder so many thin wires to these unusual MCU pins.

AILIFE4798, 2024-01-29:
Only soldering to one tiny pin is necessary(the boot0)
Tx and rx can be soldered to the date driver cup which is much bigger
And boot1 is pb2 which is a normal gpio and will be connected elsewhere
But still it's not easy to do and so far only the weird gd32e? Series need to do this all the others just need nrst

reed-dan, 2024-01-29:
I was thinking it would be nice to have a link for each specific mcu and include any information specific to the mcu and then any information specific to a board would be on each boards page. I am playing with this with MM32spin05pf in the side menu ( i have it linked to the example page right now). My concern is mobile users because this makes the collapsed menu only expandable by clicking on the carrot on the right of the mcu instead of clicking the whole mcu name. Does that make sense? What are everyones thoughts?

On the mcu page I was thinking like picture of the pins, information specific to setting up said mcu in Keil etc, information about unlocking specific mcu.

reed-dan, 2024-01-29:
@AILIFE4798  I made some formatting grammatical corrections to [CH552 ](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/CH552-as-CMSIS%E2%80%90DAP-link)

AILIFE4798, 2024-01-30:
Some other page may also need to be edited like the esp8266 ones

AILIFE4798, 2024-01-30:
I highly doubt anyone will flash on mobile
Yes it is actually possible I tried just that but no I do not recommend

Maybe this would be better?
<summary><b>
<a href="https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Example-page-for-creating-new-board-layout">MM32SPIN05PF</a>
</b>--(expand)</summary>

reed-dan, 2024-01-30:
> I highly doubt anyone will flash on mobile Yes it is actually possible I tried just that but no I do not recommend
> 
> Maybe this would be better?
> 

Might be nice for people with chromebooks though, depending on the process since a lot of apps are not avialable on chromebooks.

reed-dan, 2024-01-31:
@RoboDurden I think it would be useful to change the board layout numbering. I know this seem like a crazy idea, but it is in an effort to simplify and organize the layouts. It is crazy to see how many layouts have been recently added. I would like the numbering to follow this convention 2.numberidentfyingmcu.layoutnumber

2.1 GD32F130C8
2.1.1 Layout 2.0
2.1.2 Layout 2.3
2.1.3 Layout 2.4
2.1.4 Layout 2.10
2.1.5 Layout 2.11
2.1.6 Layout 2.12
2.1.7 Layout 2.13 (C8T6)
2.1.8 Layout 2.15 (C8T6)
2.1.9 Layout 2.16 (C8T6)
2.1.10 Layout 2.18

2.2 GD32F130C6
2.2.1 Layout 2.5
2.2.2 Layout 2.6.1 (C6T6)
2.2.3 Layout 2.14 (C6T6)
2.2.4 Layout 2.17 (C6T6)
2.2.5 Layout 2.19 (C6T6)

2.3 GD32F130K8
2.3.1 Layout 2.2

2.4 GD32E230C8T6
2.4.1 Layout 2.7

2.5 GD32F103C6T6
2.5.1 Layout 2.9

2.6 MM32SPIN05PF
2.6.1 Layout 2.8

2.7 MM32SPIN06PF
2.8 LKS32MC088

-------------------
or
-------------------
2.6 = MindMotion
2.6.1 MM32SPIN05PF
2.6.1.1 Layout 2.8
2.6.2 MM32SPIN06PF

We could possibly do something along the lines of 

2.(Number for GD/MM/STM/LKS).(number for specific mcu).(Number for gate driver).(pin count).(layout number)

AILIFE4798, 2024-01-31:
It only require changing the layout file name which is not so bad I think it is doable and we are remaking the wiki anyways it would be a good chance to turn back

RoboDurden, 2024-01-31:
Yes I also already have thought about more dimensions for board identification.
And why only numbers ?

Gen2.GD.F130.64.A.48.1 = gdf130 with 64 kB and 48pin and analog drivers version 1

But again ! This will not really help beginners.
It would rather be only for perfection :-(

Gen2.x(.y) where y indicates only a minor PCB difference is simple.

Yes I fear that there are over 100 different layouts out there ...

AILIFE4798, 2024-01-31:
With autodetect we don't need to fear about how much layout there is
There's not even a point to log known layout everyone just use the autodetect for every board it is done in 30 minutes
That's why I really wanted remote allinone because then we only need to provide one file and it's all they need
I know due to flash size it is not possible but it would be the dream(btw c8t6 actually have 128k of flash you can use it)

I think you should go edit the autodetect readme that so complete noob can use this great tool you made them self and not rely on our database because the Chinese is always faster in updating the board design and make it cheaper

AILIFE4798, 2024-01-31:
The benefits of known layout is only not needing to install keil there is pre compiled firmware

reed-dan, 2024-02-04:
I  made some updates o he autodetect page primarily organizational and formatting.

I also creating a page so we can outline submitting a new layout plan to make a template for submitting a new layout.
What information do you think would be useful? [Submit new layout](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/How-to-submit-a-new-layout)

AILIFE4798, 2024-02-04:
I think my new 2.20 and 2.8.1 issue is pretty good

RoboDurden, 2024-02-04:
Okay i have begun with an "interactive" board search for the wiki, so i can reamove the overlong read.me.
But i really think it is not helpful. New people come and go so quickly, all the hard wiki work seems a bit wasted to me:

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/find-your-layout-(incomplete,-only-test)

RoboDurden, 2024-02-04:
Okay, the "initeractive" search  seems to be complete. 
@reed-dan , @AILIFE4798 , do you agree to copy the code into the wiki main page ?

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/find-your-layout

reed-dan, 2024-02-06:
> Okay, the "initeractive" search seems to be complete. @reed-dan , @AILIFE4798 , do you agree to copy the code into the wiki main page ?
> 
> https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/find-your-layout

Sure where were you thinking the top of the home page?

RoboDurden, 2024-02-06:
Thanks @reed-dan for adoptin my dropdown section.
I have removed the board list from my read.me

I do not find that "All The Documented Boards" with only 50% width images very useful...

AILIFE4798, 2024-02-06:
can the control method and flashing guide use one in wiki also
i have made ok ones and if you are not satisfied you can edit it
so all thats left in readme is a short description of whats gen2.x hack and what it can do
and a big link to wiki home page

RoboDurden, 2024-02-10:
Maybe I will change the board numbering this weekend:

Gen2.1.4xf = split board . target 1 . version 4 x=64kb, f=FOC capable.

The s/x flag and the f flag are for information only.

Targets are the Keil targets:

1 gd32f130
2 gd/stm32f103
3 gd32e230
4 mm32spin0x
5 lks..

Ideas welcome.

reed-dan, 2024-02-15:
> Maybe I will change the board numbering this weekend:
> 
> Gen2.1.4xf = split board . target 1 . version 4 x=64kb, f=FOC capable.
> 
> The s/x flag and the f flag are for information only.
> 
> Targets are the Keil targets:
> 
> 1 gd32f130 2 gd/stm32f103 3 gd32e230 4 mm32spin0x 5 lks..
> 
> Ideas welcome.

I think it looks good @AILIFE4798 @RoboDurden  are the mm32spin0x boards close enough that they can be grouped together, or should they be seperate?

RoboDurden, 2024-02-15:
the new board numbering is grouped by the Keil target:

firmware from @AILIFE4798 :
[image]

my Gen2.x firmware:
[image]

MM32SPIN05 and MM32SPIN07 mainliy differ in the 32kB vs 64kB, which is not important here as the free Keil version is limited to 32 kB anyway.

Great to see that with @GRROWL911  we have another coder with a MM32SPIN0x board in https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/59 here :-)

RoboDurden, 2024-02-18:
I have update the board numbering in this repo here.
With the new LKS target we by now have identified 26 Gen2 boards.

Still need to update the wiki main page and the read.me.
I can also rename all the issue titles to resemble the new board numbering.

@AILIFE4798 you might want to adopt the Gen2.4.1 - Gen2.4.4 for your MM32 firmware.
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/target_4%3DMM32SPIN0X

Ideas welcome !

RoboDurden, 2024-02-18:
I have deleted the "all documented boards" list from the wiki main page.
Say so if i should restore it.
If would also like to delete the mcu list from the wiki index:
[image]

I do not think it has much value. The "interactive board identifier" will lead users to the issue of their board.

Multiple places/ways for the same information can quickly confuse people.

AILIFE4798, 2024-02-18:
Since I already made autodetect, I will now not keep track of existing layout

RoboDurden, 2024-02-18:
I am eager to see your all-in-one firmware in a youtube demonstration :-)

Okay, then i will assign new board numbers for mm32 as well, thanks.

AILIFE4798, 2024-02-18:
i will not make all in one firmware but something even better
self programming firmware!
this is to save storage so i can have beautiful UI

AILIFE4798, 2024-02-18:
you will thus need to flash twice but its not gonna be that bad i imagine

[60 more replies]