RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #57: ALL about MM32SPIN0X (layout 2.8 and more)

AILIFE4798, 2024-01-28:
if you had a mm32spin0x board please create a new layout and hopefully help us make the blink code run on it

you can read the [current progress](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/MM32SPIN05PF%E2%80%90Layout-2.8) and [old issue ](#22 )

reed-dan, 2024-01-30:
@AILIFE4798 the link to the old issues just takes me back here.
[Link to issue 22](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/22)

AILIFE4798, 2024-02-02:
@RoboDurden blink code worked

RoboDurden, 2024-02-02:
Good News. I won't drive for the next days, so I could do some programming.
The autodetect as it currently is would need a 16 khz timer (that triggers ADC conversion)
But with the timer working the led stage should already work.
And the hall detection.
The bldc driver would need the advanced timer with three own channels..

Did you make any modifications to the blink example in the Keil package ?

If so, where can I download your project ?

AILIFE4798, 2024-02-02:
I will update my code later
Because there is no crystal hardware on the board you cannot use clock and it was the problem stopping program from running
Hers a proof of it running
I don't have motor so I'm not going to test it you will have to

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/142502122/6524f378-afb3-491e-a699-2050c66896ce

AILIFE4798, 2024-02-02:
https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32
use this project and the code in hoverboardmindmotion

AILIFE4798, 2024-02-02:
@reed-dan also try it
the led pin you traced is wrong i have corrected it now

AILIFE4798, 2024-02-02:
Recently I will work on driving motor and uart
I'm pretty sure it'll be possible
But I can already say foc is not because the lack of comparator on my board

AILIFE4798, 2024-02-02:
Better spend the time rewriting all the gd32 code and then add foc to it

RoboDurden, 2024-02-03:
@AILIFE4798 , i have downloaded your HoverBoardMindMotion/ but when i doubleclick on `HoverBoardMindMotion/Hoverboard.uvprojx` , i still have to change the compiler version from missing-5 to 6. And when i then hit F7 i get 

`Src/main.c(88): error: call to undeclared function '__nop'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
`
It would be nice to have a Keil project that i can download and that does compile successfully like on your computer :-/

RoboDurden, 2024-02-03:
I do not really find Timer code and dma-adc code in `MM32SPIN0x_q_Lib_Samples`
But it looks like `Motor-DK_Low_Power_DEV_Kit_MM32SPIN0x_V1.7\4-Software\03.Hall FOC\SPIN05\MM32SPIN05_HALL_FOC_3R_V_1.3_20230314\KEIL_PRJ` might contain everything we need:

`mm32_it.c` = `Gen2.x:it.c`

```
void ADC1_COMP_IRQHandler(void) 
{
...
		//»ô¶û½Ç¶È¼°¹ÊÕÏ¼ì²â
		HALLModuleCalc(&HALL1);
```

`hallhandle.c ` = `gen2.x:bldc.c`

```
void HALLModuleCalc(HALLType *u)
{
...
	u->RunHallValue = HALL_JudgeState();

...

uint8_t HALL_JudgeState(void)
{
...
	bReadValue1 = HALL_ReadHallPorts();

...

uint8_t HALL_ReadHallPorts(void)
{
	uint8_t HallA;
	uint8_t HallB;
	uint8_t HallC;
	uint8_t HallValue;
	
	HallA = GPIO_ReadInputDataBit(HALL_PORT,HALLU_PIN);
	HallB = GPIO_ReadInputDataBit(HALL_PORT,HALLV_PIN);
	HallC = GPIO_ReadInputDataBit(HALL_PORT,HALLW_PIN);
	
	if(HALL1.CMDDIR == 1)
	{
		HallValue = HallC*4 + HallB*2 + HallA;
	}
	else
	{
		HallValue = 7-(HallC*4 + HallB*2 + HallA);
	}
	
	return HallValue;
}
```

So if we get all this runing we should have the adc for VBATT and currentDC, the hall position and then instead of FOC the simple block commutation of gen2 to drive the motor :-)

AILIFE4798, 2024-02-03:
code updated with compiler v6 
because i was using compiler v5 on my computer

AILIFE4798, 2024-02-03:
Until now noone have tell me if the code really can be run on other boards but at least on mine it can now self hold and working hall2led
Now trying to get timer1 maybe later will get uart
But uart is 5v it may destroy the esp but it dies have series resistor so we'll see
Adc is not so important so it'll be at the end

AILIFE4798, 2024-02-03:
now i have tried to generate some pwm but so far only all 3 channel low side is conducting

RoboDurden, 2024-02-03:
Yes now i can compile :-)
Your new files initialize.c etc. are not in the src folder. Not really important.

The advanced timer TIM1 can be configured to drive the three low side mosfet and the three high side mosfet at the same time.
So when pwm channel 1 is low, low side is high and when channel 1 is high then low side is low and high side is high.
The deadtime when switching can also be configured.
Also if the low/high-side on should be on boolean 0 or 1 = negate or not.

I guess you will find the init procedure in the hall bldc example i mentioned.

RoboDurden, 2024-02-03:
The hall foc example as well as my gen2.x both trigger the bldcClalculate when adc has finished.
That is why we get adc for VBATT and currentDC and/or phase currents for free.

RoboDurden, 2024-02-03:
There is this ever ongoing discussion if ESP is 5VT or not..
Maybe put normal diodes in between anyway. Then the setup is uartBus ready an the 5V get reduced by 0.7V.

AILIFE4798, 2024-02-03:
video
https://youtu.be/g6V8mLyiMQo

RoboDurden, 2024-02-03:
I am happy to see that you are already doing all the work, so i wait until tomorrow to see if you already succeeded :-)

Please study the `0.3.Hall. FOC` the (dma) adc comes for free:

```
int main(void)
{
...
    Peripheral_Init();
    Interrupt_Init();

...

void Peripheral_Init(void)
{
    Board_ADC_Init();

...

void Board_ADC_Init(void)
{
...
    GPIO_InitStructure.GPIO_Pin     = VBUS_PIN;
    GPIO_Init(VBUS_PORT, &GPIO_InitStructure);

    GPIO_InitStructure.GPIO_Pin     = ISUM_PIN;
    GPIO_Init(ISUM_PORT, &GPIO_InitStructure);    
...
		ADC_RegularChannelConfig(ADC1, ADC_Channel_8, 4, ADC_SampleTime_7_5Cycles);
		ADC_RegularChannelConfig(ADC1, ADC_Channel_3, 5, ADC_SampleTime_7_5Cycles);

...

void Interrupt_Init(void)
{
...
    /** ADC EOF interrupt enabled */
    ADC_ITConfig(ADC1, ADC_IT_EOC, ENABLE);
    ADC_ClearITPendingBit(ADC1, ADC_IT_EOC);
	
    /** Initialization TIM interrupt */
    NVIC_Configure(TIM1_BRK_UP_TRG_COM_IRQn, TIM1_UPDATE_INTERRUPT);
	
...

void ADC1_COMP_IRQHandler(void) 
{
...
    if(ADC_GetFlagStatus(ADC1, ADC_FLAG_EOC))
    {

...

		//»ô¶û½Ç¶È¼°¹ÊÕÏ¼ì²â
		HALLModuleCalc(&HALL1);
				

```

RoboDurden, 2024-02-03:
@AILIFE4798 , where do you copy and paste your code from ?
I see that you do the 6 step block commutation directly in `interrupt.c:TIM1_BRK_UP_TRG_COM_IRQHandler()`
This of course will also work. But it is not really compatible to the Gen2 code.

But hey good to see that you make so much progress so quickly :-)

AILIFE4798, 2024-02-03:
i copied from MM32SPIN0x_q_Lib_Samples
i think i will just rewrite everything while maintaining the uart protocol to be the same so you can actually use them together

AILIFE4798, 2024-02-03:
it is easier to rewrite then to understand the code other people write

AILIFE4798, 2024-02-03:
aparently in debug mode you can only see int variable but not uint32_t
i really needed debug because no serial yet

AILIFE4798, 2024-02-03:
maybe i will be able to get foc if the board you ordered for me have comparators
the $2 board is too cheap to afford a comparator not even 1

AILIFE4798, 2024-02-03:
serial with dma and interrupt now working

RoboDurden, 2024-02-03:
I do not understand

AILIFE4798, 2024-02-03:
[image]
board.c line93

because the timer2 is needed for foc i guess to calculate motor speed so it have to use these 3 pin
[image]

AILIFE4798, 2024-02-03:
[image]
adc is kinda working now with interrupt 
this is all i have time for today
tomorrow will print it with serial

AILIFE4798, 2024-02-03:
[image]
maybe a bit more
this is raw value without connecting actual power supply but it is changing so i think it is correct

AILIFE4798, 2024-02-03:
for some reason it is more then 65536 though maybe i am doing sth wrong

AILIFE4798, 2024-02-03:
firmware already at 6.2k without doing anything useful thats no good
i still think i can fit autodetect+uartbus in flash without foc obviously
really wanted remote_allinone

AILIFE4798, 2024-02-03:
well the spin06 that can do foc is 64k anyways itll do

AILIFE4798, 2024-02-04:
i have made code support real hall sensor commutation now if you want you can try
or you can use simulated hall sensor it is also implemented in code
maybe someone with cc supply can test it on real hardware

AILIFE4798, 2024-02-04:
@RoboDurden can u help me why adc value is too big

RoboDurden, 2024-02-04:
`sprintf(buffer, "VBAT: %d V\n\r", vbat);`
Try %i instead of %d:

AILIFE4798, 2024-02-04:
https://youtu.be/d0uP_z5f4XU
motor running senserless because i do not have hall cable
the code is capable to drive it sensered already

RoboDurden, 2024-02-04:
Good job. Don't you have some Dupont male/female prototyping cables ?

AILIFE4798, 2024-02-04:
do you buy your self one of this mm32 layout also
if so you should also give it a try now ive tested it did not create short circuit

RoboDurden, 2024-02-04:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/22 was created by me so i think i have such a board ! But it must be at my solar warehouse and i will not be there before spring time..

AILIFE4798, 2024-02-04:
now i would really benefit from a wireless dap to debug the board when motor is connected but my board is 5v and there is no series resistor on swd line is would defnately destroy the esp

RoboDurden, 2024-02-04:
https://de.aliexpress.com/item/1005006139041941.html

AILIFE4798, 2024-02-04:
im not sure if its fast enough
those are not suteable for ws2812

AILIFE4798, 2024-02-04:
finally motor running smoothly and code updated
@RoboDurden @reed-dan please try it
https://youtu.be/sEuGHXHj0ak
dont ask me how much it took to get the permutation right lol its not easy
now i believe its right and motor running reasonably smooth
but i can still easily stop it with my hand

AILIFE4798, 2024-02-04:
i think the permutation is still wrong it defnately shouldnt be possible to stop the motor with one finger

RoboDurden, 2024-02-04:
I guess you do the same hall simulation as my autodetect.
For compatibility reasons it might be good to stick to the same 1,2,3,4,5,6,1,2, hall positions and swap the three pairs of mosfet until you have a strong rotation like my autodetect.

You should not need to give the motor a push.
And once spinning, the motor can not easily be stopped.

You already tested my autodetect :-)

RoboDurden, 2024-02-04:
16.8V is a bit low.
My autodetect did not get the motor spinning at 25V and below 15% = 150 own ratio ! 20% is needed. 
So yes, 25% at 17V might work.

AILIFE4798, 2024-02-04:
it is normal to need to push the motor because only a change in hall would trigger next stage
im using 24v as shown in video

AILIFE4798, 2024-02-04:
i am using real hall sensor now i sacrificed one of the 2.20 board for the hall cable

RoboDurden, 2024-02-04:
So you have connected hall socket :-)

RoboDurden, 2024-02-04:
But each of the 6 hall positions should lead to a final rotation position that triggers a new hall position.
So no need to give a push.

AILIFE4798, 2024-02-04:
but on boot all fet is at low so it would not start
so simulated hall sensor should sequence through pwm position 1,2,3,4,5,6?
it is easier to debug with simulated hall so i can turn it slowly

RoboDurden, 2024-02-04:
Yes, 1-6: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/baea91b61afc442094e400651ade305a14643c60/HoverBoardGigaDevice/Src/RemoteAutodetect.c#L829

After startup some function like SetPwm(250) should be calles then the motor should begin to rotate because the hall position is somewhere between 1 and 6 which will lead to a rotation of up to 4° that will trigger a new hall position +-1

RoboDurden, 2024-02-04:
Correction: for 1,2,3,4,5,6,1,2,... It needed a negative pwm = -200

So a positive pwm should go with 6,5,4,3,2,1,6,5, ...

AILIFE4798, 2024-02-04:
ok im using TIM_GenerateEvent(TIM1, TIM_EventSource_COM); incorrectly that cause the commutation to constantly lag behind

AILIFE4798, 2024-02-04:
now the motor vibrate way less and although i can still barely stop it with one finger its much much harder

AILIFE4798, 2024-02-04:
there is no such thing as negative pwm here
the pwm is always positive you just change how the hall position is calculated

AILIFE4798, 2024-02-04:
the motor is only using 5w so i think i should be able to stop it no problem
because the motor is mounted if its not theres no way stopping it by hand

RoboDurden, 2024-02-04:
Gne2.x has negative pwm. And speed from -1000 to +1000.

RoboDurden, 2024-02-04:
5W , now you get a feeling how powerful a six wheeler with 1200 Watt would be ;-)

AILIFE4798, 2024-02-04:
[image]

vbat still do not work though

RoboDurden, 2024-02-04:
Then send 42 instead of vbat...

[51 more replies]