RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #58: Layout 2.20

AILIFE4798, 2024-02-01:
[image]
[image]

RoboDurden, 2024-02-01:
This is the 1 USD slave board I guess.
Maybe It will also run with 5V on master slave.

AILIFE4798, 2024-02-01:
yes it is $1 board 
i will trace it later
i imagine it be much easier then the 2.8.1 which use all 0402 tiny resistor it took quite a while

RoboDurden, 2024-02-01:
Why not autodetect ??

AILIFE4798, 2024-02-01:
no motor on hand

AILIFE4798, 2024-02-02:
[image]

AILIFE4798, 2024-02-02:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/pull/1

AILIFE4798, 2024-02-02:
on this layout master and slave share the same pcb
but because im going to use gen1 this board is useless to me now
if anyone want to buy this $1 board them self and modify to master you can install the missing components your self
if you needed guide on how to do it you can ask me

RoboDurden, 2024-02-02:
Maybe It will also run with 5V on master slave. The 15V might only be needed for the Led modules when the gate driver chip generates it's own gate voltages ...

AILIFE4798, 2024-02-02:
gate driver is using 15v 
and the vbat sensing only have 1 resistor in the voltage divider it also will not work

AILIFE4798, 2024-02-02:
despite there is a coil on the pcb this board is designed for linear 15v regulator it would not be hard to install

RoboDurden, 2024-02-04:
@AILIFE4798 , i have added the 2.20 to repo and added a test binary.
Why do you think this board has 128 kB ? I only see the standard C8 mcu
[image]

And did you test the uart0 with PA11 and PA12 ??? According to the datasheet, no hardware serial possible on these pins:
[image]
So i doubt that people will succeed with attaching an ESP32 to that empty header.
There is no alternateFunction AF that i could assign to these pins in target.h :

```
	#define AF_USART0_TX(pin)	(pin==PB6 ? GPIO_AF_0 : GPIO_AF_1)			// GD32F130: AF0 = PB6 , AF1 = PA2 or PA9 or PA14
	#define AF_USART0_RX(pin)	(pin==PB7 ? GPIO_AF_0 : GPIO_AF_1)			// GD32F130: AF0 = PB7 , AF1 = PA3 or PA15 
	#define AF_USART1_TX(pin)	(pin==PA8 ? GPIO_AF_4 : GPIO_AF_1)		// GD32F130: AF4 = PA8 , AF1 = PA2 or PA14
	#define AF_USART1_RX(pin)	(pin==PB0 ? GPIO_AF_4 : GPIO_AF_1)		// GD32F130: AF4 = PB0 , AF1 = PA3 or PA15
```

AILIFE4798, 2024-02-04:
mcu labeled as 64k always have 128k you can use just not tested

i have confirmed the pin i trace is correct but im ot sure if itll work ive not even flashed a firmware on layout2.20 yet

either implement softwareserial or use the i2c pin because these layout2.x imu is always not useable anyways
[image]

AILIFE4798, 2024-02-04:
101 resistor to bottom right is connected to i2c/hardware serial

RoboDurden, 2024-02-04:
okay then i set uart0 back to pb6 and pb6 in your defines2-20 and update the test binary:
[image]

AILIFE4798, 2024-03-16:
for some reason this board must be flashed with serial(core reset fail)
using swd in bootloader also does not work
luckily it is very easy to flash with serial on this board no soldering to mcu pin required

using master dummy hall2led works all mosfet heat up allot and motor does not spin

RoboDurden, 2024-03-16:
How do you power this slave board ?

Now with motor you could try my autodetect?

AILIFE4798, 2024-03-16:
the board is drawing 100ma in idle at 3.3v and 150ma at 24v i think it is faulty
i have acidently connect 5v to the mcu before but it seems to survived i think sth else on the board is burned
i installed the missing 7815 regulator to turn it into master but i have bypassed the p channel mosfet for power switch for now
the motor pin is standard one

AILIFE4798, 2024-03-16:
turns out the gyroscope/imu is shorted after removing it(which is def easier said then done without a hot air gun) it draw 30ma and the stlink does not want to coroprate with st link utilities so i have to use pyocd
auto detect seems to work but i forgot how to use it by now ive used the mm32 one too much i lost the ability to use this over complicated one

AILIFE4798, 2024-03-16:
ok kinda got it to the hall detect but motor just vibrate allot and cannot detect hall sensor

AILIFE4798, 2024-03-16:
[image]

RoboDurden, 2024-03-16:
> but i forgot how to use it by now ive used the mm32 one too much i lost the ability to use this over complicated one

I am still waiting for your YouTube tutorial.

The motor should not vibrate but spin. Not smoothly but happily rotating.

If autodetect only makes motor vibtate, then the six mosfet pins are not correct.

AILIFE4798, 2024-03-16:
I am making changes too fast it's not worth my time making video now
Or I should say not worth my time making a high quality video now
Foc support is so close I'm just a functional board away really coz the spin05 one is not foc capable and the spin06 is broken
After that I'll have to make a new one
You can only appreciate how good/ bad it is by trying your self! It may be only the one who designed it likes it

The motor does spin kinda but the vibration is way more then I remembered from the 2.6.1 board test
I'll re trace the motor just in case

AILIFE4798, 2024-03-16:
I already ran foc test open loop without current sensor and without hall sensor it worked so well at very low speed it is completely smooth it's very impressive

AILIFE4798, 2024-03-17:
[image]
The motor wires is very easy to trace it is definitely the standard ones

AILIFE4798, 2024-03-17:
it is technically possible to not use the standard pin for the lowside you can use normal io for that by only pwm one side but it will complicate the code so i do not think anyone will do it

RoboDurden, 2024-03-17:
32 pin package does not have all 6 standard pins. I also think that only on such boards will there be a problem for autodetect.

With your board I guess you need to check the output of the six mosfet with a dso.

AILIFE4798, 2024-03-17:
32 pin one can use 
PA8,PA12,PA10(ch1,2,3)
PA9,PA7,PA5(ch1n,2n,3n)

AILIFE4798, 2024-03-17:
the gate driver i have is lowside inverted idk if your auto detect can support that i think it may be the problem
mm32 autodetect can support both

RoboDurden, 2024-03-17:
I guess the invert has to be set here: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/main/HoverBoardGigaDevice/Src/setup.c#L274

AILIFE4798, 2024-03-17:
i think so, but it is set to OC_POLARITY_HIGH and OCN_POLARITY_LOW which is correct for hin normal and lin inverted

https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32/blob/6e1bc5a735994ebd8eac7504538dc7eaebbd0037/HoverBoardMindMotion/Src/initialize.c#L151

Eksol, 2024-04-25:
I came across this version of the board. Is it possible to make the second board the main one and use two boards in two different projects?

Is it possible to work with the throttle trigger?

I-hate-2FA, 2024-04-25:
to use adc as input, you can check out [this repo](https://github.com/reed-dan/hoverboard_hack_esp32_manualspeed/tree/beta)

I-hate-2FA, 2024-04-25:
i am the one posted this thread originally, and a gen2 dev

RoboDurden, 2024-04-25:
> board does not work at all with this firmware, not even autodetect, i have tried it for you so you dont need to waste your time

Sorry @AILIFE4798 = @I-hate-2FA i could not help you make autodetect work. did you check the six mosfet pins with an oscilloscope ? 
I guess the timer0 bldc config is wrong ?

RoboDurden, 2024-04-25:
These gen2 boards mostly do not offer adc pins on the headers. Only lots of digital outputs for the leds.
ESP32 S2 Mini is currently below $2 at AliExpress..

I-hate-2FA, 2024-04-25:
> If I understand correctly, this repository is for controlling a motor controller via a uart. I was thinking of using an ADC in the GD32F130 microcontroller to work with the analog throttle trigger.

the code i provided translate adc into uart
the uartbus is a standard protocol used by all gen2 product

I-hate-2FA, 2024-04-25:
> These gen2 boards mostly do not offer adc pins on the headers. 
technically all gen2 have adc available as pa2 and pa3 always have adc, but as i said we should offload as much things as possible from the 32kb micro controller to esp32

Eksol, 2024-04-26:
It's very sad since you can't connect the throttle trigger directly to the motor board. I thought it was possible to use an output for controlling the throttle triggers instead of the UART. Since I don’t need anything more from the motor board. Only the gas trigger.

Eksol, 2024-04-26:
Judging by the datasheet, the gd32f130c8t6 processor has an ADC. I wonder if they are free?

I-hate-2FA, 2024-04-26:
i already said it would be possible but we wont consider adding it, but the code is available you can always modify it to your need

RoboDurden, 2024-04-26:
@AILIFE4798 is correct (and i was wrong), when you do not want uart control, the PA2 and PA3 (which mostly are used for master-slave serial connection), then these two pins could be used in a RemoteAdc.c then reads the adc value to +-1000 speed value.
But then no connection to slave.
The potentiometer could feed both PA2 pins on the two seperate boards.

I-hate-2FA, 2024-04-26:
he want to use 2 master and no slave

RoboDurden, 2024-04-26:
@AILIFE4798 also dislikes my autodetect because his mm32 autodetect is better (which it is.). 
You could still give it a try.. Watch my youtube video: https://youtu.be/uhACc6_LUSQ

Eksol, 2025-03-11:
This board does not have a short circuit detector on pin PB12 ?

I-hate-2FA, 2025-03-11:
pa6

Eksol, 2025-03-12:
> pa6
But PA6 used for measure current DC

Eksol, 2025-03-12:
PA11 connectet with 1 pin FD6287T

AILIFE-4798, 2025-03-13:
It's possible for me to make mistake, but Those 2 pin is un connected though, because on stm32 it's not connected

AILIFE-4798, 2025-03-13:
> PA11 connectet with 1 pin FD6287T

The MOSFET driver pin is not possible to change because it's limited by hardware

RoboDurden, 2025-03-13:
TX port ? I do not understand this empty header of a potential third serial port.
GD32F130 has no hardware uart on pins like PA11/PA12 or PF6/PA12:

```
	//GD32F130xx Datasheet	2.6.7. GD32F130xx pin alternate function	
	#define AF_TIMER0_BLDC(pin)	GPIO_AF_2	// GD32F130: all TIMER0 AF are AF2
	#define AF_TIMER0_BRKIN(pin)	(pin==PA6 ? GPIO_AF_2 : GPIO_AF_2)	// GD32F130: AF2 = PA6 or PB12
	#define AF_USART0_TX(pin)	(pin==PB6 ? GPIO_AF_0 : GPIO_AF_1)			// GD32F130: AF0 = PB6 , AF1 = PA2 or PA9 or PA14
	#define AF_USART0_RX(pin)	(pin==PB7 ? GPIO_AF_0 : GPIO_AF_1)			// GD32F130: AF0 = PB7 , AF1 = PA3 or PA15 
	#define AF_USART1_TX(pin)	(pin==PA8 ? GPIO_AF_4 : GPIO_AF_1)		// GD32F130: AF4 = PA8 , AF1 = PA2 or PA14
	#define AF_USART1_RX(pin)	(pin==PB0 ? GPIO_AF_4 : GPIO_AF_1)		// GD32F130: AF4 = PB0 , AF1 = PA3 or PA15
```

[image]

So i would have to be software serial. But good to have two digital io pins available.

RoboDurden, 2025-06-13:
@Eksol please continue here.
If you have a configSelect.h file, your code is up to date.

Eksol, 2025-06-13:
Why here? How can the board configuration not be working?