RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #59: Layout 2.8.1

AILIFE4798, 2024-02-01:
[image]
[image]
MCU:MM32SPIN05PFOP(32K)
Gate driver: uni u3116s
Voltage:3.3V

No optical sensors is on this board
It is pin compatible with Layout 2.8

AILIFE4798, 2024-02-01:
Turns out mm32spin have internal comparator so it doesnt need one

AILIFE4798, 2024-02-01:
[image]
@reed_dan i have traced the 2.8.1 can you verify it is same as 2.8.0

RoboDurden, 2024-02-04:
@AILIFE4798  , This board really looks very different to 2.8. So i would like to assign a new number 2.21 to it.

It will not help new people to identify their board when 2.21 is hidden as 2.8.1.

Yes, i am okay if you  want to do a standalone MM32 firmware. Maybe you would like to copy my PAxy pin/port mapping macros and the remoteXY.c structure.

Then i indeed would like to change the board numbering to Gen2.GD.x and Gen2.MM.y

AILIFE4798, 2024-02-06:
@reed-dan do you think this board is close enough to yours
it is manufactured by same company but the infamous lks32mc088 is also manufactured by them so it does not mean anything
the pins are not 100% compatible like led and button are not
tracing a mm32board took allot more effort then a gd board even without autodetect
so it is best to minimize the layouts
or seperate the mm layouts from gd ones

reed-dan, 2024-02-07:
> @reed-dan do you think this board is close enough to yours it is manufactured by same company but the infamous lks32mc088 is also manufactured by them so it does not mean anything the pins are not [...]

I think there are significant differences in the 2 boards. I think mine should be 2.8.0 and yours 2.8.1, but that is my opinion.

RoboDurden, 2024-02-07:
A sub number like 2.6.1 should only be used if the boards are really like version x and x.1 of the same manufacturer and only a minor change in the defines2_6.h is needed like buzzer on slave instead of master.

I would like to assign 2.21 to the 2.8.1

GRROWL911, 2024-02-09:
Hello all, I am new here and have a TaoTao 2130 series board with the MM32SPIN05pf. I have already traced most of the required pins, and just need to summarize the drawing. The board has a master and slave, which I have both, yet not traced the slave board yet. I am working on the testing of the pins based on the MM32SPIN05pf code you have already developed. I also have a new hoverboard with the lks32mc088 chips, master and slaves, which I hope to look at next. How do I create a new board definition? And, I am new to GitHub, so I do not know how to contribute my changes either. Sorry for being a newbie here.
[image]
[image]
[image]
[image]

RoboDurden, 2024-02-09:
Great to have another motivated owner of some mm32 boards.

The mm32 firmware is developed actively by @AILIFE4798 .

If you succeed to unlock your board and are ready to upload a firmware, I am confident that he will make a firmware ready for your board.

Please take a look at our wiki here to connect a mm32 board.

This will be the most difficult part.

GRROWL911, 2024-02-09:
RoboDurden, already unlocked the board, already tested the RGB and Headlight LEDs, working on the latching. Have the Hall and Gate pins sorted, all UART pins sorted. Testing bit by bit.

RoboDurden, 2024-02-09:
@GRROWL911 could you please make a YouTube tutorial where you simply walk through all the tools that were needed to successfully upload a new firmware to the mm32 board !!!!

AILIFE4798, 2024-02-09:
@GRROWL911 you can disable latch and button in code to use a jumper to replace it
test the hall and motor first
if your hall and uart pin is not standard you have to wait im workng on support timer3 (already workng but code not updated) and uart2(tx working rx dma not workng)

AILIFE4798, 2024-02-09:
btw unlock mm32 just require keil and stlink v2 no aditional tool needed
some said its easier then the gd32

AILIFE4798, 2024-02-09:
@GRROWL911 your layout is closer to @StomperQc 's lks32 
do not have a comparator so not foc capable
but should work with current feature 
please make a image like i did at my github repo with all the pins i will see
and PLEASE DO NOT CHANGE PINMODEAF IF YOU DONT KNOW WHAT YOU ARE DOING!!! 
i already bricked my board with that

GRROWL911, 2024-02-09:
> @GRROWL911 could you please make a YouTube tutorial where you simply walk through all the tools that were needed to successfully upload a new firmware to the mm32 board !!!!

Once I get the board working and move to the slave, I'll try post some instructions. Basically, just the same as the Wiki posted using ST-Link and pyocd.

GRROWL911, 2024-02-09:
@AILIFE4798 > btw unlock mm32 just require keil and stlink v2 no aditional tool needed some said its easier then the gd32
I was not able to unlock using Keil alone, would erase and fail to write. Using pyocd to do a chip erase worked, then just Keil.

AILIFE4798, 2024-02-09:
you need to choose erase full chip not erase sector

GRROWL911, 2024-02-09:
> @GRROWL911 your layout is closer to @StomperQc 's lks32 do not have a comparator so not foc capable but should work with current feature please make a image like i did at my github repo with all the [...]

@AILIFE4798 the LKS32 boards do look very, very similar, however I have not pulled them out of the HV yet, so no tracing. Once I have all the pins sorted and verified, I'll post the image.

AILIFE4798, 2024-02-09:
[image]

GRROWL911, 2024-02-09:
> you need to choose erase full chip not erase sector

@AILIFE4798 I am not sure how to set Keil up for that. I am using the VS Code edition and Keil for the first time, so I have no idea what I am doing. I've a lot more experience with the NRF52 world. I'd like to know how for the slave boards. I long ago bricked my other master board.

AILIFE4798, 2024-02-09:
i do not have lks32 yet so you are on your own unless you want to send one
not seen any for sell so far

GRROWL911, 2024-02-09:
@AILIFE4798 any hints on the latch/button code, I have the bins traced, however the latch does not hold, and the button does not seem to be reading correctly either. I am working slowly and methodically since this is all new to me. If I cannot get the latch/button parts working, I will just move to the motors. I also need to figure out how to startup the slave boards as they do not have a switch.

GRROWL911, 2024-02-09:
> i do not have lks32 yet so you are on your own unless you want to send one not seen any for sell so far

@AILIFE4798 I'll look at them after I have these ones working.

AILIFE4798, 2024-02-09:
well since when im good at keil ?lol
the hardest part of keil is to download it
if you have done it everything after is easy
[image]

AILIFE4798, 2024-02-09:
idk both of my board latch an button is working fine
make sure it is not commented out because then it will get bypassed by my code
i think you just traced it wrong
latch and button is not easy to trace

GRROWL911, 2024-02-09:
> well since when im good at keil ?lol the hardest part of keil is to download it if you have done it everything after is easy 

@AILIFE4798 I have not yet found any settings pages in the VS Code edition. I already had VS Code from my NRF52 boards, so adding the Keil extension was very easy. That was the easiest part.

AILIFE4798, 2024-02-09:
if you need to download full keil ide download from filecr
using pyocd is also ok

the reason board is bricked is the pinmodeAF does not exist
when you change the pin you must change pinmodeAF to correct one

AILIFE4798, 2024-02-09:
using slave board is very very hard my layout2.20 is still not working
easier to unbrick the master boar

GRROWL911, 2024-02-09:
@AILIFE4798 I suspected the latch pin, however I have verified it to be A7 like this. S- from the switch is traced to B9. I'll keep trying. 
[image]

RoboDurden, 2024-02-09:
Slave board normally gets on when it receives +5V (or +15V) on the master slave uart cable.

Good night from Germany

AILIFE4798, 2024-02-09:
i tried with 15v and 3.3v both not work

AILIFE4798, 2024-02-09:
@GRROWL911 picture is too blurry cannot reverse for you

AILIFE4798, 2024-02-09:
[image]
this is from layout2.8.1

AILIFE4798, 2024-02-09:
the switch detect is always before the diode, and the latch pin is always 1 resistor away from the smd transistor gate

AILIFE4798, 2024-02-09:
@GRROWL911 
What serial pin is your board
Need someone to test if uart2 is working cuz on my board is not but I don't see any reason why it won't

And if u can please contact me on dc

AILIFE4798, 2024-02-09:
And how much power can this board handle?
This and the lks32 board have the tiny MOSFET
It don't have to be bad, but with minimal heatsinking just like the layout 2.21 I still don't think it's any good
Maybe the rds is significantly lower on these so they can get away but idk

GRROWL911, 2024-02-09:
> And how much power can this board handle? This and the lks32 board have the tiny MOSFET It don't have to be bad, but with minimal heatsinking just like the layout 2.21 I still don't think it's any good Maybe the rds is significantly lower on these so they can get away but idk

@AILIFE4798 These are from off-road hoverboard, 8.5" tires, rated for 250W per motor. Also, the heat sink in the picture was removed for tracing.

GRROWL911, 2024-02-09:
Tx is B6, Rx is B4. Only one UART on this board as far as I can tell. The slaves have two.

GRROWL911, 2024-02-09:
Hall sensors are working...

AILIFE4798, 2024-02-10:
Ok nice
The serial is uart1 so it would already works

AILIFE4798, 2024-02-10:
You should already get the motor working now as the hall sensor is working
Motor pin is standard across all board you just may need invert low side which is added in latest update

RoboDurden, 2024-02-10:
> i tried with 15v and 3.3v both not work

But the master slave cable is the only signal connection. Somehow that cable must be able to wakeup the slave.

RoboDurden, 2024-02-10:
If the slave has a complete dcdc converter on board, the Rx and Tx lines might only be pulled to 3.3V by the master but drive a transistor on slave (latch) to activate the dcdc conversion.

AILIFE4798, 2024-02-10:
yeah i think the 15v is supplied by the master but when i used a layout2.6.1 master to provide the 15v it does not work
i dont have the lm2596 around ill use that when i can
and my other bench power supply also not around

RoboDurden, 2024-02-10:
@AILIFE4798 you always speak of a comparator as a replacement of the external dual opAmp chips for the phase currents.
But I thought the phase currents need to be analog values to calculate FOC.
A comparator can only trigger an analog voltage threshold.
In combination of a DAC that could be used to sample an analog value. But surely not at  16 kHz for at least two phase currents  who are not constant over time at all ?

AILIFE4798, 2024-02-10:
ok uart2 really does not work i gave up and used the i2c lines 
gd32 do that as well why cant i

comparator and opamp is same thing hardware side
when used as opamp the output is assigned to a gpio that connected to a resistor to the inverting input as the amplifing factor

the comparator can also use the internal 1.2v refrence to for example use as emergency stop trigger an interrupt

[image]

AILIFE4798, 2024-02-10:
now you also know why taotao technology boards use pb4 pb6 as serial
because pa2 pa3 is for comparator

AILIFE4798, 2024-02-10:
maybe you can install a external comparator your self
it does not need negative supply rail so should be easy
ill work on foc after autodetect that now layout2.21 is fully workng minus the uart2

GRROWL911, 2024-02-10:
Sort of, while the motor does turn, there is a high pitch, loud, whine/buzz when attempting to move. How do you know if you need to invert the low side?

> You should already get the motor working now as the hall sensor is working
> Motor pin is standard across all board you just may need invert low side which is added in latest update
>

GRROWL911, 2024-02-10:
The back of my board shows 12V on the UART line, not sure if that is what yours says.

> i tried with 15v and 3.3v both not work
> 
> But the master slave cable is the only signal connection. Somehow that cable must be able to wakeup the slave.
>

RoboDurden, 2024-02-10:
Yes either test with 12V from master.
Or if the slave has complete dcdc converter, try 3.3V on the Rx or tx line.

And do not test with the battery. Only 1-2 A constant current power supply. The original hoverboard charger also works.

AILIFE4798, 2024-02-10:
> Sort of, while the motor does turn, there is a high pitch, loud, whine/buzz when attempting to move. How do you know if you need to invert the low side?

if the motor move at all you do not want to invert low side
if you set it wrong it cause short circuit turning on low and high side mosfet

it is totaly normal to have high pitch noise
that is from the pwm
even with foc it will persist
use the constspeed to make motor start up easier

AILIFE4798, 2024-02-10:
my board also have 12 v marked on the serial line, but its actually 13.8v acording to the buck converter datasheet
the tx and rx line is 5v, but have a series resistor so you can connect it to esp directly

AILIFE4798, 2024-02-10:
> Yes either test with 12V from master. Or if the slave has complete dcdc converter, try 3.3V on the Rx or tx line.
> 
> And do not test with the battery. Only 1-2 A constant current power supply. The original hoverboard charger also works.

mm32 board usualy come with 0.5a charger those wont work

RoboDurden, 2024-02-10:
I do not remember a high frequency noise with the gs32f130 hoverboards.

AILIFE4798, 2024-02-10:
because gd pwm frequency is above audible range

RoboDurden, 2024-02-10:
```
setup.c
timerBldc_paramter_struct.period						= 72000000 / 2 / PWM_FREQ;
	
Config.h:

#define PWM_FREQ         		16000     // PWM frequency in Hz
```

AILIFE4798, 2024-02-10:
mine is half (4096)

GRROWL911, 2024-02-10:
@AILIFE4798 Just curious, millis++ is in void ADC1_COMP_IRQHandler(void) and in the void SysTick_Handler(void), was this on purpose? The reason I ask is if you set #define SYSCLK_FREQ_XXMHz  96000000, the and use a Delay_MS( 2000 ) it takes about 26 seconds, and I am trying to sort that out.

AILIFE4798, 2024-02-10:
yes idk why the systick is not accurate
thats why millis++ is in adc handler
because even with delay 1 my pid loop cannot react fast enough

AILIFE4798, 2024-02-10:
@GRROWL911 have you got serial working
because someone else also have the layout2.8 cannot get serial to work

[57 more replies]