RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #60: SendLR

BaktiyarBr, 2024-02-06:
hello thank you for your project it saved our lifes 
i have a quiestion im using HoverSendLR function and passing -1000 and 1000 to spin the wheel to oposite directions but it doesnt work how spining to different sides can be done 

2.10 test speed example

RoboDurden, 2024-02-06:
```
  template <typename O,typename I> void HoverSendLR(O& oSerial, I iSpeedLeft, I iSpeedRight) // -1000 .. +1000
  {
    // speed coeff in config.h must be 1.0 : (DEFAULT_)SPEED_COEFFICIENT   16384
    // steer coeff in config.h must be 0.5 : (DEFAULT_)STEER_COEFFICIENT   8192 
    HoverSend(oSerial,iSpeedRight - iSpeedLeft,(iSpeedLeft + iSpeedRight)/2);
  }

```

Please post the serial feedback log sent from hoverboard to ESP32 (or alike).

AILIFE4798, 2024-02-06:
gen2.x does not support tank steering afaik
use uartbus

RoboDurden, 2024-02-06:
`HoverSendLR` should just do that. send a speed for left motor and a speed for right motor.
But have not tested it for a long time now..

AILIFE4798, 2024-02-06:
ok i dk that it would be very helpful

BaktiyarBr, 2024-02-06:
so i have this HoverSendLR(oSerialHover, 1000, -1000); and log says iSpeed0 iSteer 1000 and wheels doesnt move at all
how to use uartbus for this?

AILIFE4798, 2024-02-06:
uartbus master and slave will receive own speed and you need to change the slaveID of them so you can set them to different speed

RoboDurden, 2024-02-07:
Well, @AILIFE4798 might be right and the hoverSendLR is old code from my EFeru FOC hovercar firmware.

For some reason, This gen2 firmware uses a large float array to map steer and speed to speedL and speedR in main.c

So maybe hoverSendLR does not work as expected.

@BaktiyarBr , you could replace the angle mapping with the original code from Niklas fauth 

speedR = CLAMP(speed * SPEED_COEFFICIENT -  steer * STEER_COEFFICIENT, -1000, 1000);
    speedL = CLAMP(speed * SPEED_COEFFICIENT +  steer * STEER_COEFFICIENT, -1000, 1000);

and report back if we can get rid of that huge angle array main.c

Or try uartBus as suggested by @AILIFE4798