RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #62: A32AA019 is this familiar or similar?

saikumaru, 2024-02-16:
Hello!

I would like to use this board through serial control. But haven't been able to identify the IC to start experimenting.

This is what the big chip name reads
XT3011PT
A32AA019

Would appreciate any direction. Thank you!!

Pictures of the board attached 
[image]
[image]
[image]

AILIFE4798, 2024-02-16:
before posting an issue you should remove all the connectors and rf module blocking the pcb
and use the master board not the slave
but no there seems to be no information about this chip
it could been rebranded chip but the pin number is also wrong
they have really cutten corners on this board theres not much on it
if you are desprate ypu can try to read the chip id anyways with swd programmer and search for that if its an rebranded chip you will be able to find it
if it do not have a programming headder its an asic and not reprogramable
now the mm32spin05 board is supported you should just buy one of those

saikumaru, 2024-02-16:
Sorry here is a cleaner picture.
[image]

Am assuming it's the master board because it has a RF receiver on it? Or else the other board too has the same IC on it

RoboDurden, 2024-02-16:
I am also highly pesimistic about this mcu.
Could you please link the exact module you purchased so we have photos from the outside look/housing and other people will not buy this hoverboard.

The 2.2 layout has a 32 pin version of the gd32f130. If you are extremely lucky, your mcu is pin compatible:
[image]
PA14 is SWD CLK and PA13 is SWD_IO. You could test with the multimeter in beeping diode-tester mode to see if these two pins go the the 4 pin female dupont header.

saikumaru, 2024-02-16:
Thank you so much for your reply guys!

This is the exact piece I purchased.
https://hoverpro.in/product/h6-eco-military-blue-hoverboard-with-bluetooth-and-remote/

If there's a way I can identify hoverboard with GD32F130 on this website it would be great. Or if there are links to the products on Amazon, somewhere on the repo/wiki please direct me there.

AILIFE4798, 2024-02-16:
the empity 4 pin connector may be swd but i cannot tell without enough information
the master board have a power button and 15v regulator, the rf module is on slave to spread the pins to use cheaper micro controllers
a mm32 board is $2 for a set or gd32c8 board is $1 for one piece
i developed the mm32 firmware and robo developed the gd32 firmware so you can decide which one you want
i will admit the gd 32 board have 2 opamp and may support foc in the future while your current board do not even have comparator in the chip

saikumaru, 2024-02-16:
> P.P.S. as this 4 pin header is for rf module, it will rather connect to a serial uart and not the SWD header. 

No. I meant there's a seperate unused 4pin header(green circled), nothing soldered. And one black with female headers soldered(red circled) which has the RF for serial.

AILIFE4798, 2024-02-16:
need back of pcb picture to trace

RoboDurden, 2024-02-16:
I have added a "BEWARE of these hoverboards" folder :

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/tree/main/BEWARE%20of%20these%20hoverboards

As we have autodetect at least for gd32 by now, it might be worth some further investigation on how to flash these custom mcu - if possible at all.

But i for sure will have not time for that.

@AILIFE4798 please post the $1 / $2 links again. But i thought the gd32 board was only the slave board ?

RoboDurden, 2024-02-16:
@saikumaru does your hoverboard have plastic wheels or good old aluminium rims ?

Today i fear you can only get complete hoverboards with old gd32f130 as used and no longer brand new :-/

AILIFE4798, 2024-02-16:
yeah gd32 is only slave and i have not able to get it working yet and i cannot debug gd32f130 for some reason to find whats wrong
i will post link later if it still exist

AILIFE4798, 2024-02-16:
MM32 aliexpress: de,aliexpress,com/item/1005005959910034.html
MM32 taobao: item,taobao,com/item.htm?id=743458922569
GD32 taobao: item,taobao,com/item.htm?id=722577319258

RoboDurden, 2024-02-16:
=
MM32 aliexpress: https://de.aliexpress.com/item/1005005959910034.html (does not ship to Germany)
MM32 taobao: https://item.taobao.com/item.htm?id=743458922569
GD32 taobao: https://item.taobao.com/item.htm?id=722577319258

saikumaru, 2024-02-16:
Ok here is the other board. This has the power switch and also a Bluetooth device powered through it.
So this must be the master board? I hope it gives a better idea now

[image]

saikumaru, 2024-02-16:
> @saikumaru does your hoverboard have plastic wheels or good old aluminium rims ?
> 
> Today i fear you can only get complete hoverboards with old gd32f130 as used and no longer brand new :-/

Mine is plastic wheel with rubber tyres. Like the QC sticker says, this one was manufactured in November '23. So definitely new :( but new IC as well

AILIFE4798, 2024-02-16:
yes this is the master but theres also not a whole lot on it
didnt even see self hold circuitry
i think its using asic unfortunately
not available on chinese website even 
or just rebranded chip

the layout2.21 mm32 board have manufacture date of 20240113

AILIFE4798, 2024-02-16:
to try prevent people like us flashing it the manufacture will laser off the original marking and write new one
but if the empty header is swd you can try to read the chip ID to find

saikumaru, 2024-02-16:
> The 2.2 layout has a 32 pin version of the gd32f130. If you are extremely lucky, your mcu is pin compatible: ![grafik](https://private-user-images.githubusercontent.com/12238841/305358252-50fbea8f-b6a9-40fa-bd66-eec8839d4f08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwNzY3MjgsIm5iZiI6MTcwODA3NjQyOCwicGF0aCI6Ii8xMjIzODg0MS8zMDUzNTgyNTItNTBmYmVhOGYtYjZhOS00MGZhLWJkNjYtZWVjODgzOWQ0ZjA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE2VDA5NDAyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlZmYyZmFjNjZiNWQyZmE4MjJkYzIxODE5MzBhYzA1Y2E2Y2Q3NjRkMGM3Y2M4MmEwZmJmM2QxMzBiMGY3MWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dt2N_JDz1CIC5DeoWUQ2_C4kcZIYUVayQ0qn-9yUZsg) PA14 is SWD CLK and PA13 is SWD_IO. You could test with the multimeter in beeping diode-tester mode to see if these two pins go the the 4 pin female dupont header.

This does match! I am able to beep the pins 23, 24 as you pointed out to the two pins of the 4pin dupont header. Should I just got ahead and solder a SWD connector and follow some suggested steps?

AILIFE4798, 2024-02-16:
you should read chip ID with pyocd
and the board is 5v so be careful

saikumaru, 2024-02-16:
> you should read chip ID with pyocd and the board is 5v so be careful

Are these steps documented anywhere please, especially the 5V care? I had played with ESP32/82666/Atmegas in the past, but not something that has an SWD

AILIFE4798, 2024-02-16:
please read wiki...

RoboDurden, 2024-02-16:
I think the "Flash via PYocd" wiki page is still missing.

AILIFE4798, 2024-02-16:
stlink v2 clone can all handle 5v
if you want to use esp32 you need a series resistor 
150R should be enough as the current does not exeed the 20ma rated for protection diode

saikumaru, 2024-02-16:
> I think the "Flash via PYocd" wiki page is still missing.

Yes, I see the link doesn't exist on wiki. Would you suggest something else please? "Flash via Keil"?

AILIFE4798, 2024-02-16:
on my stlink it cannot detect target so i assume it wont work

your board actually is like the mm32 board using chip internal comparator for phase current i can see the circuit in your better picture
maybe you can replace it with a mm32 because mm32spin06 is abailable in 32pin
[image]
unfortunately they not only erased the number, but also the dimple in the chip so i cannot tell which way is up and if the pinout is correct

AILIFE4798, 2024-02-16:
check the ground and VDD connection, the gd32f130 32pin do not have ground???

RoboDurden, 2024-02-16:
I think the 32 pin MCU does not have the 32 pins to the side but underneath. Gnd are 9 pads underneath in the middle of the backside.

AILIFE4798, 2024-02-16:
it is going to be very hard to solder if its the case

RoboDurden, 2024-02-16:
I think replacing the MCU is not worth the trouble anyway. But if doing it the a little heat gun from the backside is the easiest way anyway I guess. So it does not really makes.a difference if the pins are underneath.

But here the pins are sideways, so maybe your Mm32 32 pin version fits.

Or maybe India has stepped up and can now also produce such simple MCU :-)

AILIFE4798, 2024-02-16:
yes true the mcu cost more then $1 probably so just get a new board to replace

saikumaru, 2024-02-16:
I see the pin 32 is connected to ground of 78L05, and also is one of the pins of dupont header. But what should the 4th pin be please?

 I see the 4th pin on header is connected to a sensor on the back, after a resistor. But can't trace it back to MCU anywhere. Should this be a 5V connection?

[image]

saikumaru, 2024-02-16:
>dimple in the chip so i cannot tell which way is up and if the pinout is correct

The dimple is here, and matched the SWCL and SWIO on pins 24,23 with that as reference.
[image]

With name
[image]

AILIFE4798, 2024-02-16:
the 5v is available on another 7805 pin
pin32 gnd match spin05 32pin package
can you check where is 5v connected to chip
the pin is very thick it should be easy compared to 48pin ones

the swd should not be connected to sensor directly
the remaining one should be 5v
can you post back of board picture like i did at layout2.21 and 2.8 issue
and incase you dont know you can disconnect the motor wire by just pull it out with brute force
so its more convinient to reverse engineer and you can replace with other board

saikumaru, 2024-02-16:
Here is the back of PCB.

I don't have option to replace with another board, or I dint understand your suggestion, sorry
[image]

AILIFE4798, 2024-02-16:
[image]
[image]
maybe it really is compatiable

AILIFE4798, 2024-02-16:
counting from top to bottom
1 and 2 is dio and clk you can try both
3 is gnd
4 is probably 5v, but blocked by sticker on the back of pcb,

RoboDurden, 2024-02-16:
Would  be nice to see you successfully flash an unkown mcu.
I also found a 48pin mcu on the EcgSolax 60A mppt:
[image]
[image]

Might be nice to have an open source 60A mppt hack as well.

saikumaru, 2024-02-16:
Pin5- 5V(not sure if it's 5, but it comes from a square shape regulator)

Pin 32- ground

Pin 23,24- SWCL, SWIO

So this is a possible clone of mm32spin06?

AILIFE4798, 2024-02-16:
i have not suceeded with that before but why would you
unless you want to reverse engineer how do they written the foc
because you need to also bypass read protect its not trivial

AILIFE4798, 2024-02-16:
@RoboDurden you can maybe try this
https://www.eevblog.com/forum/microcontrollers/detect-cpu-via-swd-(segger)/

AILIFE4798, 2024-02-16:
it is certainly a stm32f103cbtx clone so maybe it is just code compatiable with the original
you should try to run blink on it first

RoboDurden, 2024-02-16:
@saikumaru , please tell more about your project and what you want to build.
If you are committed to make your new hoverboard never be a hoverboard a gain, then already order some cheap motherboards from the links above for plan B.
And then i guess that simply flashing a blink Keil (free download) example for the different targets GD103 / GD130 / MM32SPIN05 might be the best way to proceed. Maybe you are lucky and you disable read protection by the way and succeed with flashing something that works.

AILIFE4798, 2024-02-16:
the rom table is different in different mcu 
the debugger do not know how to erase this chip to remove read protect
you can only try different ones

btw @RoboDurden a board can self hold by output to button pin

AILIFE4798, 2024-02-16:
not all boards can, but my layout2.8.1 can
can your autodetect tell it apart?

reed-dan, 2024-02-17:
> I think the "Flash via PYocd" wiki page is still missing.

I just posted the [PYocd](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Flash-via-PYocd) page. It is based on the MM32SPIN05PF mcu but the concept should apply to most mcus.

saikumaru, 2024-02-17:
Hi, I have just started using Keil, and this is the error I seem to get with MM32 pack installation.
I know its not a repo specific error, but it would be a great help if you can help me fix it.

[image]

saikumaru, 2024-02-17:
This is where I am now. I tried to use the STlink utility to connect but it doesnt respond.
The connections are only the 4 wires from STlink to the board. I havent connected any other ports that I physically unplugged. Also I havent switched on anything on the hoverboard side. Am I doing it correct?

[image]

reed-dan, 2024-02-17:
Can you take off the cover of your st link v2 and see what mcu it has? I had really bad luck with my st link v2 until I converted it to jlink. https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Converting-St-Link-V2-to-J%E2%80%90LInk

RoboDurden, 2024-02-17:
Make sure the MCU has the 3.3 or 5 volt it needs.
The pcOcd tools needs the NRST = no reset = inverter reset signal to connect. According to the new wiki page.
You can modify st link dongle to support NRST pin. See readme of this repo here.
I would also install the st link utility and give it a try.

But @AILIFE4798  and @reed-dan  are the experts for connecting and unlocking new boards.

RoboDurden, 2024-02-17:
The 32 pin only MCU might use the two swd pins for other use like led output on the 2.2 board.
Therefore the NRST pin might be necessary to slow flashing..

saikumaru, 2024-02-17:
> Can you take off the cover of your st link v2 and see what mcu it has? I had really bad luck with my st link v2 until I converted it to jlink. https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Converting-St-Link-V2-to-J%E2%80%90LInk

[image]

> Make sure the MCU has the 3.3 or 5 volt it needs. 

Should I be adding a new 5V supply for just this purpose or can the 5V from USB be sufficient? Just to confirm I haven't connected the battery/switch plugs back. It's just plain SWD headers now.
[image]

>The pcOcd tools needs the NRST = no reset = inverter reset signal to connect. According to the new wiki page. You can modify st link dongle to support NRST pin. See readme of this repo here. I would also install the st link utility and give it a try.

Yes, I have tried with the stlink utility and it didnt work.

Alright, I will try modifying the stlink and see. But just wanted to double check am good till here

AILIFE4798, 2024-02-17:
get IDCODE error=board bricked :(

AILIFE4798, 2024-02-17:
because the mcu is only 32pin, they might wanna get all outa it, so swd is disabled
some mcu does not have a bootloader like lks32mc, in that case the mcu is single use and you have to replace it

RoboDurden, 2024-02-17:
But why then the extra header with the swdclk and swdio pins ?

AILIFE4798, 2024-02-17:
because you can program it once at the factory when code is not using swd, once they loaded their code that disable swd at startup you cannot flash again

saikumaru, 2024-02-17:
> get IDCODE error=board bricked :(

Oh no!! Or is it just disabled for reprogramming?
I swapped SWCL SWIO for once as suggested and nothing more odd, hope that dint brick it

AILIFE4798, 2024-02-17:
it is ok to reverse dio and clk
it will say no target connected when you do so

saikumaru, 2024-02-17:
Ok what else can I try before I finally give up?

RoboDurden, 2024-02-17:
No swapping the two swd pins will not brick.
@AILIFE4798 you suggested that the swd is disabled because of only 32 pin. I do not see that as a reason for disabling swd after flashing.

AILIFE4798, 2024-02-17:
if there is not enough pins the 2 swd pins can be used as normal gpio so swd is disabled

RoboDurden, 2024-02-17:
And NRST can reactivate it.

[33 more replies]