RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #64: disable PWM does not work

JochenKr, 2024-02-18:
I tried to turn off the PWM outputs but it seems that it doesn't have an effect.

My approach was to force _timer_automatic_output_disable(TIMER_BLDC);_ in CalculateBLDC(void) as below
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/839d1ab7ff600f73b3853bb75097d5406cfda1bf/HoverBoardGigaDevice/Src/bldc.c#L184
```

	if (1)//(currentDC > DC_CUR_LIMIT || bldc_enable == RESET || timedOut == SET)
	{
		timer_automatic_output_disable(TIMER_BLDC);		
		//DEBUG_LedSet(SET,0)
  }
	else
	{
		timer_automatic_output_enable(TIMER_BLDC);
		//DEBUG_LedSet(hall_c == 0,0)
  }
```
The PWM is still present and the wheels can spin as normally.

RoboDurden, 2024-02-18:
Maybe SetEnable() is what you want:

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/839d1ab7ff600f73b3853bb75097d5406cfda1bf/HoverBoardGigaDevice/Src/main.c#L513

JochenKr, 2024-02-18:
I tried to use SetEnable(RESET); with the same method as above.
but still no impact on PWM outputs visible :-(

in main.c I've commented SetEnable(enable); out, to make sure that this one doesn't turn on the PWM again.

RoboDurden, 2024-02-18:
I still do not know what you want to achieve.
I think it is normal that you have full pwm running with speed at zero. The hi-side and low-side mosfets are inverse so no current flowing.

If you want pure silence you may have to study `void PWM_init(void)` in setup.c

JochenKr, 2024-02-18:
Ok, I'll check.
My target is to save energy during idle. With 4 motors the 24 Mosfets have significant switching losses also if no current is involved. These I would like to eliminate

AILIFE4798, 2024-02-18:
mosfet do not need current to stay on
gate driver need less then 1w

JochenKr, 2024-02-18:
Correct, Mosfets do not need current to stay on.
Charging and discharging of capacitors e.g. gate charge Cg mainly generate losses. My setup is far beyond 1W

RoboDurden, 2024-02-18:
I also do not like this ever ongoing pwm. I have lots of signal noise on my i2c bus ...

AILIFE4798, 2024-02-18:
actually i do not understand what the problem is