RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #65: Manual Speed

reed-dan, 2024-02-18:
@RoboDurden  just wanted to let you know I started creating a manual speed esp32 script based off of your speedtest

https://github.com/reed-dan/hoverboard_hack_esp32_manualspeed

Currently it allows for control via serial with commands like
hover|all|500|1
hover|right|500|1
hover|left|500|1
stop

RoboDurden, 2024-02-19:
Nice to see that you want to build a more useful Arduino software.
Maybe you want to add a wiki page that lists all Arduino examples with your new repo added.

But I do understand the purpose of this script language.
Waypoints to drive to ? Or runtime configuration of input method ??

Thank you anyway for staying committed to gen2 hoverboards :-)

RoboDurden, 2024-02-20:
(sorry, forgot to hit the send button)

Nice to see that you want to build a more useful Arduino software.
Maybe you want to add a wiki page that lists all Arduino examples with your new repo added.

But I do understand the purpose of this script language.
Waypoints to drive to ? Or runtime configuration of input method ??

Thank you anyway for staying committed to gen2 hoverboards :-)

reed-dan, 2024-02-20:
@RoboDurden  I think I implement the delays that were in the autospeed incorrectly. Seems I am transmitting the packets to quickly and it is not giving time for the responses from the mcu to be received/processed. Can you explain the logic of how the delays in autospeed work?

AILIFE4798, 2024-02-21:
as you get better you will understand you should not use delay
nodelay aka millis should be used

actually on esp32 you can use delay because there is rtos on esp it can do multitask but it complicates things

as of now you are still sending commands to 2 slave  without delay in between so they reply together and croupt the signal

RoboDurden, 2024-02-21:
Autospeed ?
@AILIFE4798 has a constant speed feature in his firmware.

There are no delay() in my testspeed.ino

If you have problems with 
`iNext = iNow + SEND_MILLIS;`
this delay must be great enough to ensure complete receipt of the answer when uart is is used.

@AILIFE4798 see that remoteUart is simpler to use for most scenarios. If board has a second uart :-/

AILIFE4798, 2024-02-21:
in latest update v129 constant speed work much better now

AILIFE4798, 2024-02-21:
> If you have problems with `iNext = iNow + SEND_MILLIS;` this delay must be great enough to ensure complete receipt of the answer when uartBus is is used.
>

he send to all slaves without delay, only delay between sends, so slave reply together

> @AILIFE4798 see that remoteUart is simpler to use for most scenarios. If board has a second uart :-/

no remote uart cannot spin 2 wheel at different speed as of now so totaly useless for most project but i know it can be easily implemented
but the thing is you can only use 1 uart at a time anyways as grow said so if you used uart2 you still need uartbus

RoboDurden, 2024-02-25:
Only with UartBus do you need to wait for response..
Of course you can 
if (bAnswer) iTimeNoResonse = millis() + x;

and do whatever you want 

if millis() > iTimeNoResonse

reed-dan, 2024-02-25:
@RoboDurden  Are you say that uartbus is not necessary? Is uart bus only necessary when multiple masters are desired, but because there is only 1 esp32 we don't need uartbus?

RoboDurden, 2024-02-25:
RemoteUart needs a dedicated Uart for the one master  (who controls a slave) or one  single.
With RemoteUartBus there can be many masters and singles on the same Uart "bus".
But then you  should wait for the answer of each master/single so the answers do not overwrite each other.

reed-dan, 2024-02-25:
> RemoteUart needs a dedicated Uart for the one master (who controls a slave) or one single. With RemoteUartBus there can be many masters and singles on the same Uart "bus". But then you should wait for the answer of each master/single so the answers do not overwrite each other.

What is the advantage to  controller (esp32)->master->slave in comparison to just controlling everything with the controller?

RoboDurden, 2024-02-25:
UartBus is my "invention" and came longer after the Masters-Slave code.
Maste-Slave makes use of the original cable.
UartBus is more complex.