RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #66: MM32 pin autodetect firmware :-)

RoboDurden, 2024-02-19:
@AILIFE4798 has written a complete new autodetect for his MM32 firmware that saves the found pins to EEPROM that his hoverboard firmware reads on startup. So only one hoverboard firmware needed :-)

The autodetect firmware currently is the pin-finder branch of his MM32 firmware: https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32/tree/pin-finder

This autodetect firmware offeres a uart serial user interface in which you can also set the UartBus id for controlling multiple boards on one uartBus !

Would be great if people with a MM32 board start testing it :-)

RoboDurden, 2024-02-19:
two friendly suggestions to make your autodetect.c code more readable:

i do not like

```
extern uint32_t pins[33][3];
extern uint32_t adcs[10][3];
```

this suggests that the two arrays share the same type of information.
furthermore, pins do not contain a uint32 at all:

```
const uint32_t pins[33][3] = {   // port,pin,pinsrc,ADC,ADC2,HALL AF,HALL TIM,UART AF
{IOA,GPIO_Pin_0,GPIO_PinSource0},

...

#define IOA 1
#define GPIO_Pin_0                  (0x0001U)                                   ///< Pin 0 selected
#define GPIO_PinSource0             (0x00U)
```

using structs will save memory and improves readability:

```
typedef struct {
   uint8_t port;
   uint16_t pin;
   uint8_t pinBit;
} PinMM32;

const PinMM32 pins[33] = {   // port,pin,pinsrc,ADC,ADC2,HALL AF,HALL TIM,UART AF
{IOA,GPIO_Pin_0,GPIO_PinSource0},
...
```

`pinMode(pins[i][0],pins[i][1],GPIO_Mode_FLOATING);
`
could be rewritten to `pinMode(pins[i],GPIO_Mode_FLOATING);
`

with something like 

```
void pinMode(PinMM32& p,uint8_t state){
	switch (p.port){
		case IOA:
			GPIO_InitStruct.GPIO_Speed = GPIO_Speed_50MHz;
			GPIO_InitStruct.GPIO_Mode = state;
			GPIO_InitStruct.GPIO_Pin = p.pin;
			GPIO_Init(GPIOA, &GPIO_InitStruct);
		break;
  ...
```

And better use enum or defines instead of plain integers. I used bits insead of 1,2,3 to easily test for multiple cases:

```
	#define AUTODETECT_Stage_Startup 	1
	#define AUTODETECT_Stage_VBatt 		2
	#define AUTODETECT_Stage_Hold 		4
	#define AUTODETECT_Stage_Button 	8
	#define AUTODETECT_Stage_Led 			16
	#define AUTODETECT_Stage_Hall 		32
	#define AUTODETECT_Stage_HallOrder 64
	#define AUTODETECT_Stage_CurrentDC 128
	#define AUTODETECT_Stage_Results 	256
	#define AUTODETECT_Stage_Finished	512
```

```
	if (!(wStage & (AUTODETECT_Stage_CurrentDC|AUTODETECT_Stage_Hold|AUTODETECT_Stage_Button))	)
		msTicksTest = msTicks + 50;	// never 
```

peace :-)

RoboDurden, 2024-02-19:
p.s. i only see 20kB Code size for your pin-finder firmware. 32kB limit for the free Keil IDE.

Program Size: Code=20028 RO-data=4192 RW-data=748 ZI-data=2388  

Activating the MicroLIB (memory optimized libraries) 
[image]
yield two errors:
```
.\Objects\HoverboardOutput.axf: Error: L6218E: Undefined symbol __use_two_region_memory (referred from startup_mm32spin06xx_s_keil.o).
.\Objects\HoverboardOutput.axf: Error: L6218E: Undefined symbol __initial_sp (referred from entry2.o).
```

My autodetect without this MicroLIB:

AILIFE4798, 2024-02-19:
i will replace the switch case magic number with enum or define
multiple state at same time is not possible in my firmware so it is not needed
for the pins ill look into it not sure if i should change it
i forgot to remove pa13 pa14 and only realizes after everything else is coded and i did not use num_pin define so i cannot remove those
i may also change that

i thought the storage is calculated by program+ro+rw data i have searched for it last time
i have also tried the microlib before and got same error
those are in startup which is in assembly so i cannot do anything about it
but anyways i think its very unlikely i will run out if only 20k is used
i can use max 31k for my firmware cuz the last 1k is left for eeprom to prevent it being erased when programming other firmware

AILIFE4798, 2024-02-19:
this is my first time writing code for other people to use insted of for my self so its not very readable but itll get better

RoboDurden, 2024-02-19:
Yes you are right, for the difference between spin05 and spin06 the 32kb inlcudes program+ro+rw data  I think.

But for the Keil free 32 kB I think I got the error message when only program data exceed 32 kB.

But I should test my autodetect binary with a new gd32f130c6 target to very that.

novval, 2024-02-19:
use gcc or platformio+gcc to unlimited

RoboDurden, 2024-02-19:
Oh yes, please build me a makefile, then I could add these firmwares to my free online compiler: https://pionierland.de/hoverhack

I strongly dislike Linux :-/

AILIFE4798, 2024-02-19:
i have unlimited compile size and compiler v5 in keil dont need to worry about that
and my goal is to not require other people to compile so it is not an issue
only hardware limit is relevent

linux is great

AILIFE4798, 2024-02-19:
@RoboDurden i have edited the code a bit and hopefully its more understandable now
i have added allot of comments also
incase you want to reverse engineer my firmware to make your better
i have written everything my self didnt copy too much from yours
i will now work on the normal firmware side

novval, 2024-02-19:
> Oh yes, please build me a makefile, then I could add these firmwares to my free online compiler: https://pionierland.de/hoverhack
Why not, I'll try to do it when I have free time... so far, there are problems with this ((

>I strongly dislike Linux :-/
gcc is also available for Windows

ps. A good option is Visual Studio Code + Platformio plugin, it can download and install gcc itself.
An example can be found at the EFeru known to us

AILIFE4798, 2024-02-19:
i think @GRROWL911 is using the platformio plugin

AILIFE4798, 2024-02-19:
but can it support mm32 because eferu is gd32

RoboDurden, 2024-02-19:
As many hoverboards only have 32kB anyway, the 32kB free Keil limit is no problem for us.

AILIFE4798, 2024-02-19:
yeah the mm32spin06 is really quite rare and not needed in most case anyways 32k is enough

AILIFE4798, 2024-02-19:
i may have forgot to git push but its done now

GRROWL911, 2024-02-19:
Yes indeed, the platformio plugin for VS Code works, and I can flash and debug with an ST-Link, I have not found where to set a complete chip erase yet. So I use pyocd for that step.

AILIFE4798, 2024-02-19:
hows your progress now
have you got adcs and uart at least

GRROWL911, 2024-02-19:
I have adc and uart1 working, I am struggling with getting UART2 working. I have TX2 working, however when I enable RX2 it does not work and stops RX1 too. The slave boards have two pin outs for UART. I'm pretty sure about the pins, and I think the AF if my problem. Otherwise, I was distracted with the LKS board for a bit. Finished tracing those pins, and erased the chip, and programmed the LEDs ok.

AILIFE4798, 2024-02-19:
same as mine uart2 rx not working
i think bug in the library
with debug it can work i see data in RDR register but interrupt and dma not work
but my firmware do not need 2 uart anyways 1 is enough

nice to see you making progress with lks just as fast as i did with mm
i think it will not be hard as all example and hal library is available

GRROWL911, 2024-02-19:
Was hoping to use the UART2 as a passthrough to the master, perhaps it is not possible.

AILIFE4798, 2024-02-19:
it should be but its not idk why either
because one of my board annoyingly used uart2 for masterslave and im still unable to get it to work and i also bricked it with the clock fix
but to control master and slave just physically connect 2 board tx and rx together then command will be sent to both board
and you have successfully used slave board??
why everyone can do that i cant

GRROWL911, 2024-02-19:
I can work around the issue with my controller which has two UARTs and I can connect to each independently. I was just trying to get this to work.

As a side, I did get UART2 working, if UART1 DMA is not enabled. So comment out this     
      if(UARTEN==UART1){
        DMA_NVIC_Config(DMA1_Channel3, UARTEN, (u32)&UART1->RDR, (u32)sRxBuffer, 1);
    }else{
        DMA_NVIC_Config(DMA1_Channel3, UARTEN, (u32)&UART2->RDR, (u32)sRxBuffer, 1);
    }

leave in this:

    DMA_NVIC_Config(DMA1_Channel5, UART2EN, (u32)&UART2->RDR, (u32)sRx2Buffer, 1);

and the interrupt:

    void DMA1_Channel4_5_IRQHandler(void)

will be executed. Also note that the Channel for UART2 is 4/5, so

        DMA_NVIC_Config(DMA1_Channel3, UARTEN, (u32)&UART2->RDR, (u32)sRxBuffer, 1);

is not correct anyway. Also note the extra parameter, since in 

   void DMA_NVIC_Config(DMA_Channel_TypeDef* dam_chx, UART_TypeDef* uart, u32 cpar, u32 cmar, u16 cndtr)

UARTEN was hard coded, and would not allow for both UARTs to be setup.

    //UART_DMACmd(UARTEN, UART_DMAReq_EN, ENABLE);
    UART_DMACmd( uart, UART_DMAReq_EN, ENABLE); 

This does not fix the problem with getting both working, however at least give the option of one or the other.

GRROWL911, 2024-02-19:
And, yes I did get the slave board working, either connected to the master, or by itself. When by itself, I needed a 13.7V power input on the 12V UART connector, and it works as expected. Note there are no hold, nor switch, and the UART2 for my board reuses these pins, so make sure when compiling for the slave you disable the LATCHPIN and BTNPIN. Also, for my slave, there is a buzzer, however no buzzer on the master.

AILIFE4798, 2024-02-20:
because my firmware is made for just 1 uart i never designed for 2 to be enabled at same time so i reuse UARTEN
i just want to allow board like layout2.21 that use uart2 as masterslave to use uart2
i did not know uart2 is dma channel 4/5 thats why it doesnt work
i can easily fix it now but because to support autodetect i will choose to ignore uart2 anyways as uart2 have too many pin
uart1 only have 6 and 2 of them is used by motor so only 4 wire and 4 combinations
automaticly matching pinmodeAF is very very hard so i choose not support uart2

AILIFE4798, 2024-02-20:
and if i support uart2 i will need to compile tens of hundreds of binary every time which is def not possible
4 firmware every time is ok
only the outdated board from gd32 era use uart2 anyways ,and those board always use uart1 pin for i2c so you can reuse it for serial

AILIFE4798, 2024-02-20:
i think i will get some 7815 regulators to try out the slave board to see if it work

GRROWL911, 2024-02-20:
No need to support UART2, I was just trying to get it to work for my board and figure out why it did not work. I will also be abandoning using UART2 since it does not work.

AILIFE4798, 2024-02-20:
ok if you have any new layout you can help try out the autodetect firmware
it can trace every pin other then the serial in 30 seconds if you are fast

GRROWL911, 2024-02-20:
A note about the ADC. While working on the LKS board, I noticed that the current reading is actually coming from two pins and they use some sort of comparison between the two to get the current. I am not yet sure about the details, and not sure if it improves the resolution of the readings, just thought I would pass it along.

AILIFE4798, 2024-02-20:
no,there is 2 pins going to the current sens resistor,one going to adc and other one go to a comparator
the one going to adc have a hardware rc filter to average the pwm noise, to get good current reading
the one going to comparator is directly connected because it is used to stop the motor in case a short circuit occured it need to be very fast before the mosfet burn out

AILIFE4798, 2024-02-20:
my firmware already support the comparator tim1 break function

GRROWL911, 2024-02-20:
Thank you for the details, I was not sure what that was being used for.

AILIFE4798, 2024-02-20:
if you look closely ,there may be a pin connected to a resistive divider connected to vcc, that is the side it will compare with
the mcu have built in dac and voltage refrence for the comparator but they connected current sensor to inverting input so it cannot be used they used external one

AILIFE4798, 2024-02-20:
@RoboDurden i have started to adapt the main firmware to read pinout from eeprom
i have used typedef like you requested and now it does not compile
you have to fix it
https://github.com/AILIFE4798/Hoverboard-Firmware-Hack-Gen2.x-MM32/tree/main_eeprom

RoboDurden, 2024-02-20:
I am already in bed :-)
Will look into your code tomorrow.

AILIFE4798, 2024-02-20:
ok i have some things to change still

AILIFE4798, 2024-02-20:
i have bypassed the error now

AILIFE4798, 2024-02-21:
minimum function working while booting from eeprom

AILIFE4798, 2024-02-21:
tomorrow i will work on improving autodetect a bit
now main firmware is completely rewritten to adapt using eeprom and digitalWrite pinMode etc
i will make video in 1 month
i still hope someone can test it

RoboDurden, 2024-02-21:
Yes, the human being by nature is 100% egoistic and can not find joy to help someone else with his work which is not his own.

RoboDurden, 2024-02-21:
@AILIFE4798 i know you said that you hate OO, but you should learn to use structs in C.

```
		for(uint8_t i=0;i<64;i++){
			pinstorage[i]=tmp[i];
		}
```

is aweful.

I usually define a struct for eeprom (Arduino code following):

```
#define EEPROM_VERSION 2
typedef struct {
    uint32_t iVersion = EEPROM_VERSION;
    uint16_t  wSetting = 0; 
    char sSSID[31] = "";
    char sPwd[31] = "";
    char sApiKey[20] = "";
    float fV = 27.0;
    float fI = 5.0;
    float fImin = 2.0;
  } Data;
Data oData;
```

Your 
```
uint8_t restorecfg(){
	uint16_t tmp[64];
	EEPROM_Read((u8*)tmp, 2 * 64);
	if(tmp[MAGICNUM_POS]==MAGIC_NUMBER){    //verify config is valid
		for(uint8_t i=0;i<64;i++){
			pinstorage[i]=tmp[i];
		}
```

then would read somehow like

```
uint8_t restorecfg(){
	Data oDataTemp;
	EEPROM_Read((u8*)&oDataTemp, sizeof(Data));
	if(oDataTemp.iVersion==EEPROM_VERSION{    //verify config is valid
		oData = oDataTemp;
```

In Keil C, a memcpy might be neccessary for the last line like

`memcpy(&oData, &oDataTemp, sizeof(Data);
`

peace :-)

RoboDurden, 2024-02-21:
Okay it is a nice idea to remap PA2 and PB1, etc. to one array

```
const MM32GPIO pins[33] = {   // port,pin,pinsrc,ADC,ADC2,HALL AF,HALL TIM,UART AF
{IOA,GPIO_Pin_0,GPIO_PinSource0},
{IOA,GPIO_Pin_1,GPIO_PinSource1},
{IOA,GPIO_Pin_2,GPIO_PinSource2},
{IOA,GPIO_Pin_3,GPIO_PinSource3},
```

You could (or already have) add

```
enum 
{
	PA0 = 0, PA1, PA2, PA3, PA4, PA5, PA6, PA7, PA11, PA12, PA13, PA14, PA16, 
	PB0, PB1 //, etc.
};

Then you had
`pins[PA2].port` , etc.
```

I myself would prefer

`digitalWrite(pinstorage.latch, 1);` instead of  `digitalWrite(LATCHPIN, 1);` with `pinstorage` being such a struct i mentioned above.

You approach
```

#define HALLAPIN pinstorage[0]
#define HALLBPIN pinstorage[1]
#define HALLCPIN pinstorage[2]
#define LEDRPIN pinstorage[3]
#define LEDGPIN pinstorage[4]
#define LEDBPIN pinstorage[5]
#define LEDUPIN pinstorage[6]
#define LEDDPIN pinstorage[7]
#define BUZZERPIN pinstorage[8]
#define BUTTONPIN pinstorage[9]
#define LATCHPIN pinstorage[10]
```

however is more close to the defines.h where you replace pinstorage[10] with PA2 or PB6, etc :-)

AILIFE4798, 2024-02-21:
i am already learning oop,  just afraid of new things as always
i gotta say its not as bad as i thought
the struct will not help, as the eeprom must have all uint16_t 
i think the define is enough to make code readable
the for loop is very obvious just copy everything why would you want to use struct

yes i am aware of the enum pf PA0 PA1 and are going to use it in the autodetect firmware
but in normal firmware i never needed to mention a pin so i didnt added it

i am just not going to change the `pinstorage[64]` under any curcumstances because it will complicates the eeprom code

linux guy will not use autodetect, they like to compile them self!

RoboDurden, 2024-02-21:
You can cast &Data to (uint16_t*)

AILIFE4798, 2024-02-21:
ok thats good to know

AILIFE4798, 2024-02-21:
because you cannot set 115200 baud rate with uint16_t
idk why your uartbus call for 19200 because 115200 is less load on the hover board to wait for message to send as sending is not dma
but anyways it fit in 65535

AILIFE4798, 2024-02-21:
or just remove 2 zeros and add it back in code

AILIFE4798, 2024-02-21:
i think only need to support esp cuz esp cheaper then arduino now
and esp can also do software serial just did not work with your code idk why it worked with eferus

RoboDurden, 2024-02-21:
I do not understand what is not working at 19200 baud !

AILIFE4798, 2024-02-21:
no, i mean when using software serial on esp32 it did not work with your testspeed

RoboDurden, 2024-02-21:
You mean that software serial on esp32 did not work at 19200 but at 115200 it was able to control hoverboard ??

AILIFE4798, 2024-02-21:
at both baud it cannot control hoverboard with gd32, but at 19200 it can control gen1 hoverboard

RoboDurden, 2024-02-21:
So you mean that software serial on esp32 is not working at all with the gen2 firmware and gd32F130 mcu but with EFeru Gen1 and Gd32F103 it does.

I never meant people to use software serial on esp32 anyway.
I don't think that I will have the happiness to test and fix that.

AILIFE4798, 2024-02-21:
yes ok i hope reed-dan will fix it in his code so i can use it

AILIFE4798, 2024-02-21:
cuz then i can use any pins
the og esp32 does not have gpio multiplexer so only those 2 pin can be used

RoboDurden, 2024-02-21:
What is the "og esp32" ?
Of course you can use hardware serial on any/many pins. With esp32 and esp32 s2.

RoboDurden, 2024-02-21:
I am pretty sure that I used different pins on the Lolin Lite.
Will have a look at my Arduino folder tomorrow.

Good night :-)

AILIFE4798, 2024-02-21:
[image]
gpio 9 10 is connected to flash to boot so cannot be used

AILIFE4798, 2024-02-22:
pinfinder command line now support human readable address and pin number
[image]

RoboDurden, 2024-02-22:
@AILIFE4798 this is arduino code from my logband saw https://www.youtube.com/watch?v=sveUO8Rd1TQ running on a LOLIN32:
[image]

```
#define oSerialFringe Serial1
#define oSerialMove Serial2
...
  SetupHover(oSerialFringe,115200,19,23);     //0,4);
  SetupHover(oSerialMove,115200,0,4);          //16,17);
...
template <typename O,typename I> void SetupHover(O& oSerial, I iBaud, I pin_RX, I pin_TX)
{
  // Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  oSerial.begin(iBaud, SERIAL_8N1, pin_RX, pin_TX);
}
```

[240 more replies]