RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #68: Gen2.1.17 (ex2.20) = ??????? (GD32F130)

TheBestJohn, 2024-02-24:
I'd like to add another to the list, This one looks like YK201C instead of the A version of [Issue 44](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/44#issue-2047540382)'s board. I'll be working on it shortly, I just need access to a power supply.
[image]

I can't identify the brand because the exterior has been abused significantly. It does have a bunch of addressable LEDs, a Bluetooth module hot glued to the case and instead of onboard xt60, a spade and barrel connector.

AILIFE4798, 2024-02-24:
you do not need to identify the brand those does not matter, only the mcu model number is important, the rest of work can be done by autodetect

AILIFE4798, 2024-02-24:
only known board that have addressable led is 2.4.1 and 2.4.2

TheBestJohn, 2024-02-24:
I think I may be mistaken on the MCU. It's likely a MM32SPIN like in [Issue 38](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/38#issue-2021027978). They put a metallic sticker on this thing with some form of adhesive that is impossible to get off. I've used acetone etc. However, MM32SPIN seems to have the same blue streak and 3 fiducials and this is an earlier rev off that board anyway

TheBestJohn, 2024-02-24:
> only known board that have addressable led is 2.4.1 and 2.4.2

Ah it seems I was mistaken in that it's not addressable, just an rgb

AILIFE4798, 2024-02-24:
you will need to make sure your board is mm32spin05 or mm32spin06
i will supply you with the firmware needed
and you need trace the serial pin

your board have 5 led driver, so likely no addressable led is required
your board have 1 dual comparator, so i cannot tell it is mm32 or gd32 by that
i only see one linear regulator, so maybe 5v?

TheBestJohn, 2024-02-24:
So I'm pretty sure those stickers had some sort of etchant mixed into the adhesive. 
[image]
there 2 header pinouts on board but one says 3.3v the other says 5v

[This board is V2.0](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/1250510/02e0c6c9-5971-46ba-947d-1a95e67cb23b) of this so I'd be willing to bet the MCU wasn't what changed. I could be terribly wrong though. At the end of the day whether I brick it or don't do anything with it I have the same outcome. I suppose once I get a power supply I could check some voltages

AILIFE4798, 2024-02-24:
the marking is lasered off, i can still see 3 locating dimple in the chip, which match mm32spin, but it could be again a unknown chip
the 5v header is for rf module so seems like the chip is operating at 3.3v the 3.3v one is swd
the board you linked is mm32spin25, which i do not have and are not sure if it is compatiable with spin05, so you will have to try it

AILIFE4798, 2024-02-24:
[image]
this is a leaked unsubmitted board and it also look similar to your
i will delete this comment after you red it
it is not compatiable with spin05 code

TheBestJohn, 2024-02-24:
ah interesting. Thank you for finding that!

AILIFE4798, 2024-02-24:
i will compile blink code for mm32spin25 for you
i already know your led is at pd2 pd3

AILIFE4798, 2024-02-24:
if it worked ill add spin25 target to main firmware and send you autodetect+normal firmware

AILIFE4798, 2024-02-24:
to flash mm32spin25, the step is confirmed to be the same as spin05
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/MM32SPIN05PF%E2%80%90Layout-2.8

incase the firmware i supply is wrong, you may need to unbrick your board at some point because ive certainly done that a couple time and now you will have to test the unstable code directly from me

AILIFE4798, 2024-02-25:
if you ready tell me i have firmware now

AILIFE4798, 2024-02-25:
blink have been sucessfully tested on mm32spin25 now, but all the main firmware have to be completly rewritten, so you will have to wait a long time

AILIFE4798, 2024-02-25:
do you have discord

TheBestJohn, 2024-02-25:
> do you have discord

yessir, genghis.john

RoboDurden, 2024-02-25:
If this is indeed a new MCU target MM32SPIN25 then this board should be Gen2.6.1 I guess. 2.5 is LKS32 ..

Big thanks to @AILIFE4798 for covering the quickly growing MM32 family

RoboDurden, 2024-02-25:
Have added this board as Gen2_6.1 to the "interactive board identifier": https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki

reed-dan, 2024-02-25:
@TheBestJohn  if you barrow a power supply be very careful, because the hoverboard I acquired with a very similar board to this one YK190A the pins on the power supply are opposite of the other boards/power supplied I already had. The positive and negative pins should be labeled on the bottom of the board.

TheBestJohn, 2024-02-25:
[image]
for posterity