RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #69: Should we do an open source hardware Gen2 partner project ???

RoboDurden, 2024-02-25:
@reed-dan ,  @AILIFE4798 , i feel greatly honored that you two helped to improve the Gen2 software.

But most people quickly drop out - because this is the world of the "brain degenerated consumer" and people only come up with "stupid fun projects" that by definition can not bring joy but only fun. Fun is short lived

So in similar way like www.reprap.org it might be a good idea to partner our Gen2 firmwares/wiki with an open source hardware project that would bring true joy to people.

Gen2 boards with their split boards might be better suited for robotics. And the new plastic motors with less weight even better for that case scenario.

Maybe we three would like to pioneer something like that:

- huge 5-6 axis robot arm that can chop wood or dig holes..
-  huge walking dog
- huge 3d fiberglas printer that can print 200cm diameter elements of a complete house or bike frames or bicicle rims
- solar powered tractor for small farms (2000 m²) in Africa

a hoverboard motor has 90 steps, so 4°. 
I have an easy to print 30:1 differential gear in mind that would bring this to 4/30 = 0.133°. With a radius of 1 m (= 2m diameter) that would be a spatial resolution of 2.3 mm !

Or a (3d printed) toothed rack that is driven by a 20 mm gear wheel. That would have a step size of only 20 * pi / 90 = 0.7 mm !!

I we would decide on one such a project, it might be way more easy to build a commited community to the Gen2 software.

Roland, physicist, Ideas welcome

AILIFE4798, 2024-02-25:
to talk about this i think we will have to be great scott for 1 day for the diy or buy video

cost : buy is $2 you cannot possibly beat that lol the mosfet cost more then that
time : buy need 1 month of waiting, diy will likely need more, and it will take at least 5 hour to assemble
easy to use : diy require soldering lqfp48 which is not easy for beginers
flexibility : buy is fully hackable just like diy, but if you need extra feature diy would be easier to implement, but buy also have extra gpio if you need

at the end of the day, there are already open source bldc driver like vesc and odrive if it is really needed
so it is just purely dependent on do your self want to use it
i will opt out because i will not use it my self, even the original project i want to use this with is abandoned now(i know i must mention it it have nothing to do with me, just working with people is allot harder then working with computer, at least when debugging you can see what they are thinking), so working on this is not beneficial for me only for the ones that will use it, which is fine i will continue supporting mm32 series, but i will not do extra things that only very few would benefit from and will take allot of time to do

if you want to try make some robotics from it, it is a good idea to use a 3d printed cyclodial or plenetry gearbox for the motor, someone have already done it successfully, i would want to make a robot dog from these plastic motors but for number of reasons this is not feaseable irl

RoboDurden, 2024-02-25:
No we would not make another bldc driver. The Gen2 boards are out bldc controllers.
With hardware project I mean robotArm/3dPrinter/Walking dog

AILIFE4798, 2024-02-25:
yes it would be good idea to have example project

RoboDurden, 2024-02-25:
But you want to do software and not a huge hardware :-/

AILIFE4798, 2024-02-25:
the gen1 have a bunch of example project, and made a bunch of promotional video, so they are much more popular, gen2 basicly only get attention because eferu put your link

AILIFE4798, 2024-02-25:
> But you want to do software and not a huge hardware :-/

im a hardware guy after all, i would love to do things i like, but irl it wouldnt be possible, a pc is all i have

RoboDurden, 2024-02-25:
I am not talking about example projects.
And i only offered different example for us to choose from.
Then this open source XY would become the main focus of the new Gen2 community and the gen2 firmwares only the support.

I do not see that our firmware can attract a devoted communiy of developers like SimpleFOC.
But a 2 meters x 1 meters fiber glass 3d printer built from a $30 used or $130 new hoverboard that can print bicycle frames or entire tiny houses.. that would be something worthing investing time.

Yes i remember my google streetview drive through your town. Not much room for a huge hardware project.
But the first rule of Robo Durden is: You always speak yes.
You could take a tour through your town in real life and find an old warehouse with some room to spare... advertise the project at your univeristy and found a group of young inventors..

Second rule: you always! speak!! yes!!!

reed-dan, 2024-02-25:
My goal with the project is to first simplify it, make it so anyone can easily build a project. I think the wiki we have all been working on is the first step to that. You both have made amazing progress with autodetect. At least for me once I got a board flashed it was difficult to find information and understand how to control it. Once @AILIFE4798  helped me to understand the control methods I started to work to make autospeed, to make it easier to control the motors from an esp32, with the idea that their are so many existing projects that provide input methods to the esp32 that can be easily adapted. I think once we develop a well documented base the possibilities are endless.

RoboDurden, 2024-02-25:
Okay no hardware partner project.
I still appreciate your worke here :-)

closed 2024-02-26 by RoboDurden