RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #70: Supported board (Yuehengda) Analog gate Drivers+STM32+different Interface?

sjdaliyan, 2024-02-27:
This is a board from one of my hoverboards. 2 mainboards inplace of sideboards.
I read this address searching to be sure support of my board.
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki
My board looks exactly like the ones with analog gate drives but with stm32f103 instead of GD32. and interfaces looks also different. 

Help me.

[image]

 
[image]

[image]

[image]

RoboDurden, 2024-02-27:
Your board is neither 2.1 nor 2.2 so I will become Gen2_2.3 - if you upload high resolution photos of front and backside of master and of slave board please.

One of these autodetect binaries could work for you 

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD%2CSTM_32F103%20tx%3DPB6%2C%20rx%3DPB7.bin

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesToTest/hoverboard%20autodetect%20GD%2CSTM_32F103%20tx%3DPA2%2C%20rx%3DPA3.bin

Watch my YouTube video on how to use autodetect !!

sjdaliyan, 2024-02-27:
Thanks for quick reply.
2 boards are identical except slave board doesn't have some of the components not assembled. like radio control module and 36v output regulator.
I will watch autodetect video.

here are master board's pictures:

[image]

[image]

[image]

[image]

[image]

[image]

[image]

This is 36v regulator on the master board but missing in the slave board

[image]

One of the mosfets from the back of the board

[image]

AILIFE4798, 2024-02-28:
quality of the picture is too bad
cannot even see this
[image]
you need to retake them in a brighter place i guess

RoboDurden, 2024-02-28:
If autodetect works we no longer need sharp pictures :-)

Where could be the pb6 and pb7 pins ?

RoboDurden, 2024-02-28:
Looks like this PT2272 is used in cheap remote controls:
[image]

sjdaliyan, 2024-02-28:
Thank you all for contributing. I tried my best to take best pictures i can with my cell phone.
But its unclear where the pa5 and pa6 go bcz track is under soldermask i have to use a cutter and scratch the solder mask. but i guess they are going to the remote chip. 

I watched auto detect video i think i have to afully assembly the board right? 

[image]

RoboDurden, 2024-02-28:
Do not scratch on your PCB .
Simply test some pins until you see smileys..
But first check that these pins are not 5V or higher !

What is the master slave cable socket ?

sjdaliyan, 2024-02-28:
I will using multimeter.

In this picture ,Connector marked in blue in the middle is the Master-slave Interconnect. 
Are there any schematics available from the boards your firmware support?

[image]

RoboDurden, 2024-02-28:
Flash the pa2 pa3 binary first and test those master slave pins that are below 5V .

Then try pa6 pa7 binary and check the RC chip pins .

Gd32103 of course.

Autodetect so far has only been tested with gd32f130 ...

And of course: 1-2 A current limiting power supply.
No battery !

AILIFE4798, 2024-02-28:
[image]

RoboDurden, 2024-02-28:
Any Arduino tutorial how to use this  PT2272 as a bidirectional Rx tx Uart bridge ???

AILIFE4798, 2024-02-28:
pt2272 is only decode to 4 output i doubt it is fast enough for serial
and def not bidirectional because the rf module is only rx

AILIFE4798, 2024-02-28:
you have a better chance of sucess just directly connect the rf module one to tx and other to rx i think it might work no decoder is needed
in fact i dont know why its used in this case as if decoding is needed the stm32 also can do it with external decoder its even using more gpio

sjdaliyan, 2024-03-02:
> Any Arduino tutorial how to use this PT2272 as a bidirectional Rx tx Uart bridge ???

This is datasheet of the part.i Think its not possible. bcz its made for another purpose. an external uart transceiver must be soldered in place.I know its complicated and i know these boards are very hard to finds so they are rare. so it may not be worth spending this much time

Only one of these boards is working well during the explosion of battery :) I want to use functional one to control my sewing machine like a servo. i dont care about these Chinese boards much just send me a schematic that works well with your firmware that fine for me.

RoboDurden, 2024-03-02:
You should be able tons older two thin wires to the two and resistors. 
Either to the MCU sides or the receiver chip side.
Both sides might work.

If you only want to use one board anyway than try the master slave header with the pa2 pa3 autodetect binary. You won't need a second uart port for you sewing machine :-)

Beware of +5V or even +15V on this header.

sjdaliyan, 2024-03-02:
What about schematic? are there any available to download?

RoboDurden, 2024-03-02:
Not for your board.

sjdaliyan, 2024-03-02:
> Not for your board.

Any supported single motor driver board will be ok.i told you earlier i dont care about this board. i will desolder the parts and use them on another board with better support.

RoboDurden, 2024-03-02:
Sorry, I will not support that.
We gave our best to help you but if you won't even give my autodetect a try then good bye :-(

sjdaliyan, 2024-03-02:
life means collaboration and cooperation. some sort of give and take. you are not the only one with that information. if you forbid things you know from other in fact you forbid yourself their knowledge. sorry if you dont know any common sense. surely thats expecting too much from some one bragging of being a loser. i will dint my solution with or without your help if you think you have any advantage over me because you know something i dont know you are very wrong. and this is surely spirit of a cave man. living in his cell threatens others. sorry if its normal in your country mate.

closed 2024-03-02 by sjdaliyan