RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #71: A GD32F103C8 - B GD32F130C8 New Board?

Hirzi4, 2024-02-27:
Anyone now this Board?

Board A GD32F103C8
Board B GD32F130C8

Is there a Firmware for that?

Board A
[image] 

Board B
[image]

RoboDurden, 2024-02-27:
Please first check 

https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki

Then post high resolution photos with the cables pushed out of the way of master and slave.

Thanks.

Hirzi4, 2024-02-27:
I have checked the wiki but cant find this board.

[image]

[image]

RoboDurden, 2024-02-27:
Then watch my autodetect YouTube video.
You find autodetect binaries for both boards im BinariesToTest/ folder.

Good night from Germany.