RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #72: Unknown mcu ;( someone please help me I’ve spent all my money buying this

suzy311278, 2024-02-29:
[image]
[image]
[image] 

I have no idea how to flash this mcu, I had to buy a whole hoverboard to finish my robot but to my contrary this happened;(

suzy311278, 2024-02-29:
So now there's no way of flashing this ?

RoboDurden, 2024-02-29:
@AILIFE4798 might give you some hope.
But it would be good for his health if he is a sleep right now :-)

Try to buy some of the $2 Gd32 split boards linked in the other issues.

No second hand platform in your country ? Buying an old hoverboard highly increases the chances to get a GD32 MCU controller.

Please post photos of the outside of your hoverboard so other will not buy it.

A shopping link would also do.

AILIFE4798, 2024-02-29:
I already told him to get a replacement board and why this one cannot be used(the SWD pin is used as gpio for led driver, so SWD is disabled)

AILIFE4798, 2024-02-29:
And why is the mcu model number mistyped again
It's xk3011pt

saikumaru, 2024-03-25:
@suzy311278 wow! You're from same city too!!

Can you please update what did you do next?
Let me know if you're ordering one of those gd32 boards.

Even I need it working soon for a similar project. Happy to work together

AILIFE4798, 2024-03-26:
you should either order a mm32 board for $3.5 allot of people has already tested or a gen1 board like he did if you can find one
the mm32 has officialy surpassed gd32 with the new sine wave mode

saikumaru, 2024-03-26:
Hi @AILIFE4798 
AliExpress is kinda banned here right now, can you suggest an eBay link if you don't mind.

Also where does he mention about finding GEN1 board please?

AILIFE4798, 2024-03-26:
i do not know where did he get gen1...most gen1 i can find is with new mm32 chip also(which is not yet supported but i guess would be possible to support so if someone wanted then you can send me one)

any board with mm32spin0x or mm32spin2x is supported you can easily find one for sure but its just very expensive and low quality some dont even have a heat sink the aliexpress one is best
if aliexpress cannot ship to you you will need to use proxy which will make shipping cost even more but it may still be better

saikumaru, 2024-03-26:
>  or a gen1 board like he did if you can find one 

I mean, how did you find out he has GEN1 board? Can you please guide me to that issue. Or conversation. Thank you.

Ok, let me try AliExpress again

AILIFE4798, 2024-03-26:
go to my mm32 repo readme and at the bottom support section