RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #73: GD32F103 build error

Hishamahmed97, 2024-03-08:
[image]

Hello there,
Appreciate your huge efforts doing all these documentation and coding stuff.
I am trying to use the keil project to build the code for target 2 (2-2-1) layout. After choosing the target in the keil target drop-down and build the target, it shows the attached errors (which are mainly because the defines_2-2-1.h include file is not found. I will be much grateful if you give me a little bit of directions on how to proceed in this regard. 

Thanks for your time!

RoboDurden, 2024-03-08:
You first have to run autodetect.
Post the resulta here and I will make the missing defines file.

Watch my YouTube video on autodetect.

Hishamahmed97, 2024-03-08:
I watched it. So you mean to flash first the binary file for auto detect on the repo for GD32F103 and connect the board with arduino through uart pins see the results?

RoboDurden, 2024-03-09:
Yes