RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #74: Esp32 DevKit v1 - Autodetect.ino and hoverboard 2.1.1 master Uart.bin

Dim4ik1992, 2024-03-12:
I'm using Esp32 DevKit v1 -
Autodetect.ino

and hoverboard 2.1.1 master Uart.bin

but when connected to UART I learn a bunch

z pͫz pͫz  Jͫz (z pͫz pͫz  Jͫz (z pͫz pͫz  Jͫz (

Is it possible to connect PWM with these boards? I'm interested in the possibility of controlling forward backward directly from the RC remote control
[image]
[image]
[image]

RoboDurden, 2024-03-12:
If you have a 2.1.1 board you do not need autodetect.
First flash 2.1.1.dummy and see the motors spinning.
Then remoteUart and the testSpeed.ino

If you have an EspressLRS compatible rc receiver check this https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26

Otherwise you will need an Arduino/Esp to receive pwm or ppm input.

AILIFE4798, 2024-03-12:
your baud rate is probably wrong
you will need to figure it out sooner or later no matter if you trace the board manually or use auto detect
or you would not be able to control the board

it WOULD be possible to support pwm input directly, if the mcu have a pin with timer input, but we will NOT do it, you can add it your self all the code is open source

RoboDurden, 2024-03-12:
He indeed go a 2.1.1 board.
He is simply using the wrong .ino

AILIFE4798, 2024-03-12:
oh he using master uart with autodetect on esp

RoboDurden, 2024-03-12:
@Dim4ik1992 this .ino should be what you want: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/tree/main/Arduino%20Examples/Esp32_PPM

AILIFE4798, 2024-03-12:
you can ask @reed-dan to add pwm/ppm support to his newer and better esp control software

Dim4ik1992, 2024-03-12:
this is what I got but the engine is standing still
https://youtu.be/DsELgWZ_OAg

RoboDurden, 2024-03-12:
Hall sensors are okay, I see iOdom change value when you rotate motor.

Please test  the remoteDummy binary.
This will not output serial data but motor should spin.

Maybe the 2.1.1 binaries are too old.
It might help if you have Keil installed and compile 2.1.1 yourself.

Dim4ik1992, 2024-03-12:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/21707765/e73bf289-598b-4216-8fb3-65f97a50e389

RoboDurden, 2024-03-12:
Sorry I will answer tomorrow.

Dim4ik1992, 2024-03-12:
I installed Keil but I don't know how to compile 2.1.1 myself.

the default is

// LAYOUT_2_X is used in defines.h
#ifdef GD32F130 // TARGET = 1
#define LAYOUT 13
#define LAYOUT_SUB 0 // Layout 6 exits as 2.6.0 and 2.6.1
#elif GD32F103 // TARGET = 2
#define LAYOUT 1
#elif GD32E230 // TARGET = 3
#define LAYOUT 1
#elif MM32SPIN05 // TARGET = 4
#define LAYOUT 1
#endif

Dim4ik1992, 2024-03-12:
there are a bunch of boards on the service, different on different chips, both new and old, but no one knows how to flash them, this is a real breakthrough thanks to you)

reed-dan, 2024-03-12:
> you can ask @reed-dan to add pwm/ppm support to his newer and better esp control software

I currently support serial commands and adc potentiometer, it is designed so you can add any input method you would like though. I am working on wifi ap/tac with a webbased controller at this time hopefully it will be complete in the next few days. PWM is definately on my list just haven't gotten there yet.

Please proceed with caution as I haven't tested everything out.

https://github.com/reed-dan/hoverboard_hack_esp32_manualspeed/blob/beta/ManualSpeed_Improved.ino

RoboDurden, 2024-03-12:
Select gd32f130 as target in the top left  drop down list of Keil ide.

And change #define LAYOUT 13
to #define LAYOUT 1

Then f7 
Then F8

RoboDurden, 2024-03-12:
@Dim4ik1992 if you support the Russian war against Ukraine, please do NOT ask me to support you.

Dim4ik1992, 2024-03-13:
I don't support war!!! let's not talk about politics

> @Dim4ik1992Если вы поддерживаете войну России против Украины, пожалуйста, НЕ просите меня поддержать вас.

Dim4ik1992, 2024-03-13:
> @Dim4ik1992 if you support the Russian war against Ukraine, please do NOT ask me to support you.

I don't support war!!! let's not talk about politics

Dim4ik1992, 2024-03-13:
https://youtu.be/CCCWToCRJsM
TestSpeed.ino and ManualSpeed_Improved.ino
The result is the same

RoboDurden, 2024-03-13:
Now the three led flash according to motor rotation.
I don't know why motor is not spinning.

Activate remoteDummy here https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x-GD32/blob/0cfd2716866bae62987bb81a4427e7b3fb6dde70/HoverBoardGigaDevice/Inc/config.h#L43

You will not have serial output anymore but motor should spin without esp32.

Dim4ik1992, 2024-03-13:
https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/assets/21707765/27adc96b-752b-4a1b-8573-30e0f6d03268

RoboDurden, 2024-03-13:
Please post screenshots and code and a detailed description of what you have done or changed.
I can not watch all your videos.

Dim4ik1992, 2024-03-13:
Now in this mode the wheel is spinning
------------------------------------------------------
	// LAYOUT_2_X is used in defines.h
	#ifdef GD32F130		// TARGET = 1
		#define LAYOUT 1
		#define LAYOUT_SUB 0	// Layout 6 exisits as 2.6.0 and 2.6.1

-------------------------------------------------		
		#define REMOTE_DUMMY
		//#define REMOTE_UART
		//#define REMOTE_UARTBUS	// ESP32 as master and multiple boards as multiple slaves ESP.tx-Hovers.rx and ESP.rx-Hovers.tx
		//#define REMOTE_CRSF		// https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26

Dim4ik1992, 2024-03-13:
when it was like this it displayed debugging in the UART but the wheel stood still

//#define REMOTE_DUMMY
		#define REMOTE_UART
		//#define REMOTE_UARTBUS	// ESP32 as master and multiple boards as multiple slaves ESP.tx-Hovers.rx and ESP.rx-Hovers.tx
		//#define REMOTE_CRSF		// https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26

RoboDurden, 2024-03-13:
Okay, so the esp32 communication is the problem.
Try RemoteUartBus instead of RemoteUart.
You will also have to uncomment/activate RemoteUartBus in the Arduino code.

If you then still get log data from hoverboard this is proof that hoverboard has successfully received commands from esp32.

Dim4ik1992, 2024-03-13:
I don't understand why when I'm in .\Inc\config.h
#define REMOTE_UARTBUS
in the file .\Src\remoteUartBus.c
lines are not active
#ifdef REMOTE_UARTBUS

[image]
[image]

hiliving, 2024-03-19:
> I don't understand why when I'm in .\Inc\config.h #define REMOTE_UARTBUS in the file .\Src\remoteUartBus.c lines are not active #ifdef REMOTE_UARTBUS
> 
> ![12](https://private-user-images.githubusercontent.com/21707765/312607384-2cefe934-f8cc-4d88-b645-1a6571126984.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA4MTI1MjgsIm5iZiI6MTcxMDgxMjIyOCwicGF0aCI6Ii8yMTcwNzc2NS8zMTI2MDczODQtMmNlZmU5MzQtZjhjYy00ZDg4LWI2NDUtMWE2NTcxMTI2OTg0LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE5VDAxMzcwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxMDM4Nzg3MmZhMjY0NTI3YmQ1OTNjMDZhYzAzNjYyYmE5ZTg4YmNjZGM2YTBlZTA5OWE2NGY1MGY3MTM4NDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4sfk_Mntt3orieQvaPPgMPFoTPdcJ4QiDagx1frDoSY) ![123](https://private-user-images.githubusercontent.com/21707765/312607389-2fdf4f79-e87a-4ea7-8922-586a932b199f.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA4MTI1MjgsIm5iZiI6MTcxMDgxMjIyOCwicGF0aCI6Ii8yMTcwNzc2NS8zMTI2MDczODktMmZkZjRmNzktZTg3YS00ZWE3LTg5MjItNTg2YTkzMmIxOTlmLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE5VDAxMzcwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc2YjA5YjgzNzY2YjY0OTYzZTFhMzM4NjlmMjY2NzM3NGZiMWYxZTEzMjUzM2U3Y2FjMmM1ODljZjEyMWUyMGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wLj-wjbQHEH8yNGylDNSI-EP1ygXqbt3xkDcpiQ06WI)
> 
> and now he shows a cross on every library
> 
> ![1234](https://private-user-images.githubusercontent.com/21707765/312609696-7f2c9493-8060-485d-a9af-c8eea5630455.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA4MTI1MjgsIm5iZiI6MTcxMDgxMjIyOCwicGF0aCI6Ii8yMTcwNzc2NS8zMTI2MDk2OTYtN2YyYzk0OTMtODA2MC00ODVkLWE5YWYtYzhlZWE1NjMwNDU1LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE5VDAxMzcwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyMWQ4MGRmY2U4NGEzYTM0MTE1MzcxNzkwODBjNGM4NDM0ZjE4YjkzMjdmOGM1Y2NmYTAzMGU2MTU3NzY1YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nPTf7c4UNmAPMFDKmrfehqt7tV22pLcAWTtv3OAyV-Y)
> 
> maybe there was an error in this, due to incorrect characters in the file path, moved the project to C:\Hoverboard-Firmware-Hack-Gen2.x-GD32-\HoverBoardGigaDevice and everything became normal) I’ll try now

My board is the same as yours. Have you finished it?  Mine has no response now, and there are errors in flashing the firmware.

hiliving, 2024-03-19:
@RoboDurden Can you provide some clues on how I should flash this firmware? Following the prompts in readme.md, I am now using st-linkv2 to connect to the debugging port, using STM32 ST LINK Utility, and STM32CubeProgrammer. Attempting all firmware has failed
[image]
[image]

hiliving, 2024-03-19:
@RoboDurden This board is almost the same as @Dim4ik1992 
[image]

AILIFE4798, 2024-03-19:
please check wiki pages first

Dim4ik1992, 2024-03-19:
I think that on my version of the board the bluetooth module is interfering... it most likely sits on TX, because when I uploaded the Kickstarter version, I also could not connect to the uart remote, but I was very surprised when I uploaded the firmware to the additional board, which is without  bluetooth and I was able to connect to it via uart (TTL)
[image]

AILIFE4798, 2024-03-19:
yes bluetooth module need to me removed, as ive already clearly stated in the wiki page, which nobody reads before asking stupid questions
[image]

AILIFE4798, 2024-03-19:
if you do not want to remove the bluetooth module then use the comms masterslave connector like i sugested

RoboDurden, 2024-03-19:
This firmware here is a fork of a fork!
The original firmware did contain Bluetooth communication.
But I was never able to test that.
So I removed the code when I was restricting the communication code to RemoteXy.

If you have a 2.1.1 board which is the original 2.0 in the original fork, you might be able to flash the original firmware and get the Bluetooth module working .
You might then be able to port that code to a RemoteBluetooth.c here.

But not many of the new layouts have a Bluetooth module and they might connect and work differently.

If the Bluetooth module indeed also connects to PB6 pb7, and interferes with esp32 communication,
You might need to use the RemoteUartBus protocol which connects booths board via pa2 pa3 master slave serial port.

RoboDurden, 2024-03-19:
@hiliving , try autoflash.bat

And read the wiki pages.

AILIFE4798, 2024-03-19:
on new board the bluetooth module is another microcontroller designed for bluetooth speaker and communicate through 1wire or softwsre serial, but you need the special app on android to generate the serial command because it identify as a speaker so it is not very easy to use, better use esp32 as a bluetooth module which would soon be supported by reed

RoboDurden, 2024-03-19:
Sorry @AILIFE4798 , I should empty the read.me to force people go to wiki.

Will do that when i am back from my current outdoor trip.

hiliving, 2024-03-19:
> please check wiki pages first

thx very

hiliving, 2024-03-19:
> @hiliving , try autoflash.bat
> 
> And read the wiki pages.

Solved it, so Excellent project ,thx very much.

MathMyne, 2024-05-16:
Hello, we are currently working on a project using hoverboard hardware, and we encountered some difficulties following your tutorial.
We flashed the 2.1.1 hoverboard card with “hoverboard 2.1.1 slave.bin” and hoverboard “2.1.1 master Dummy.bin” for the two cards and it works well.
We want to use an ESP32 WROOM to handle the hoverboard card. We flashed the master with “hoverboard 2.1.1 master Uart.bin” and connected the ESP32 with RX and TX according to the schematics and loaded the testSpeed.ino code.
It shows this error and we are blocked:
[image]
Pictures of the wiring: 
[image]
[image]
Thanks in advance for your help.

I-hate-2FA, 2024-05-16:
You need to use uartbus.bin

RoboDurden, 2024-05-16:
No you can use Master_RemoteUart

But you have to comment `//#define REMOTE_UARTBUS  // one serial bus to control them all :-)`

MathMyne, 2024-05-17:
Hello, thanks for your response, we tried to put in comment the line “#define REMOTE_UARTBUS  // one serial bus to control them all :-)” in your code TestSpeed, but there are some issues.
[image]
Also when we uncomment the precedent line and look at the serial port, the iDom changes when I turn the wheel. And it seems that the CD, AB and FFFF are :
[image]

I-hate-2FA, 2024-05-17:
just use the uartbus firmware on hoverboard and leave esp code untouched
i have had same problem before

MathMyne, 2024-05-17:
Thank you, it works well !