RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #75: Revert to original firmware

lucor123, 2024-03-17:
Is it possible to install the original firmware to make the hoverboard work in the classic way? Has anyone done it?

Thanks!

RoboDurden, 2024-03-17:
Not for these Gen2 split boards.

AILIFE4798, 2024-03-18:
ofcourse you can if you have dumped the firmware before hand
but because gen2 designs is too different and in already compiled firmware it is very hard to edit the pin number and even harder to change the imu code which is different on each board
so universal firmware is not possible itll be better to write it from scratch

RoboDurden, 2024-03-18:
And that has only been done for gen1 boards.

RoboDurden, 2024-03-18:
But if you how to dump please add a wiki page. many people already have asked this question.

AILIFE4798, 2024-03-18:
if i have another gd32 board ill try to dump it
idk how to dump mm32 yet