RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #76: Gen 2.1.1

Freshwater123, 2024-03-23:
I got 2x 2.1.1 hoverboards. I have unlocked the boards but getting errors when i try to write to them. Tried both STM Util and STM Cube prog with same failures leading me to believe it's not a software problem

[image]

AILIFE4798, 2024-03-24:
use pyocd

RoboDurden, 2024-03-24:
Try sutoflash.bat

Freshwater123, 2024-03-24:
> Try autoflash.bat
> 
> https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/wiki/Flash-via-st%E2%80%90flash.exe

Cool, that seemed to do it. Thanks!

Freshwater123, 2024-04-17:
[image]

Was gonna connect an RE receiver and went with your ESP32_PPM Arduino code for my Arduino mini D1 (Clone) and got this message. Also could we run the Arduino mini from the hoverboard somehow so we dont need a powerbank ?

I-hate-2FA, 2024-04-17:
esp8266(d1 mini) only have one serial port, but that is enough for ppm to serial 
im not sure if hardware timer on the esp is used to decode pwm/ppm if not it will work on esp8266
you can power the esp from the 5v supply or the 3.3v supply on the hoverboard, both is rated for 1a and only less then 100ma is used, but esp8266 is power hungry and excessive heat will be generated

RoboDurden, 2024-04-17:
You should not call HoverSetupEsp32 when you  have an ESP8266. HoverSetupArduino might work with SoftwareSerial.

```
#ifdef ESP32
  #define oSerialHover Serial1    // ESP32
#else
  #include <SoftwareSerial.h>    // not compatible with RCReceiver because of interrupt conflicts.
  SoftwareSerial oSerialHover(9,8); // RX, TX 
  //#define oSerialHover Serial    // Arduino
#endif

```

I-hate-2FA, 2024-04-17:
only 1 serial is used so why not just use the hardware serial

RoboDurden, 2024-04-17:
normally used for debug log output to Arduino console..

I-hate-2FA, 2024-04-17:
A esp32s2 and esp8266 is very different

Freshwater123, 2024-04-17:
Yes, my lack of knowledge on the matter

pacraf, 2024-05-03:
I haven't been here for few months and so many changes :)
Anyone can help with issue that when I compile for this GEN2.1.1 (former 2_0) the simplest dummy firmware in keil, then motor shakes (I was not able to reconnect colors of motor to get it spin) 
BUT when I download ready to test ]([hoverboard 2.1.1 master Dummy.bin](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/blob/main/BinariesReadyToFlash/hoverboard%202.1.1%20master%20Dummy.bin)) 
it works normally

my settings in keil should be fine (as I am able to generate for board version 2.1.8 without problem)

the only change is here 
#define LAYOUT 1 //8 = 2.1.8   1=2.1.1

and I have this shaking behavior

RoboDurden, 2024-05-03:
As i am currently working with my 2.1.4 testsetup i quickly tested RemoteDummy and motor is spinning nicely as expected:

```
//#define REMOTE_AUTODETECT		// !! highly experimental 
				// ONLY test with 1-2A constant current power supply !!!!
				// will drive the motor without hall input to detect the hall pins..
				// outputs ascii to uart 19200 baud to TX=PB6 , RX=PB7

#ifdef REMOTE_AUTODETECT
	
	#define HAS_USART0	// tx=PB6,rx=PB7	uncomment to connect via 19200 baud serial
	//#define HAS_USART1	// tx=PA2,rx=PA3	uncomment to connect via 19200 baud serial

	#define SINGLE
	#define MASTER_OR_SINGLE
	#define BAT_CELLS         	6        // battery number of cells. Normal Hoverboard battery: 10s
	#define SPEED_COEFFICIENT   -1
	#define STEER_COEFFICIENT   1
	//#define CHECK_BUTTON		// disable = add '//' if you use a slave board as master
#else

	// LAYOUT_2_X is used in defines.h
	#ifdef GD32F130		// TARGET = 1
		#define LAYOUT 4
		#define LAYOUT_SUB 1	// Layout 2.1.7 exisits as 2.1.7.0 and 2.1.7.1
	#elif GD32F103		// TARGET = 2
		#define LAYOUT 1
	#elif GD32E230		// TARGET = 3
		#define LAYOUT 1
	#elif MM32SPIN05	// TARGET = 4
		#define LAYOUT 1
	#endif

	//#define MASTER		// uncomment for MASTER firmware. Choose USART0_MASTERSLAVE or USART1_MASTERSLAVE in your defines_2-?.h file
	//#define SLAVE			// uncomment for SLAVE firmware. Choose USART0_MASTERSLAVE or USART1_MASTERSLAVE in your defines_2-?.h file
	#define SINGLE			// uncomment if firmware is for single board and no master-slave dual board setup

	#define BAT_CELLS         	6        // battery number of cells. Normal Hoverboard battery: 10s

	#if defined(MASTER) || defined(SINGLE)
		#define MASTER_OR_SINGLE
		
		#define REMOTE_DUMMY
		//#define REMOTE_UART
		//#define REMOTE_UARTBUS	// ESP32 as master and multiple boards as multiple slaves ESP.tx-Hovers.rx and ESP.rx-Hovers.tx
		//#define REMOTE_CRSF		// https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/26
		
		#ifdef REMOTE_UARTBUS
			#define SLAVE_ID	1		// must be unique for all hoverboards connected to the bus
		#endif

		#define TEST_HALL2LED	// led the 3-led panel blink according to the hall sensors
		//#define DEBUG_LED		// uncomment to activate DEBUG_LedSet(bSet,iColor) macro. iCol: 0=green, 1=organge, 2=red

		#define SPEED_COEFFICIENT   -1
		#define STEER_COEFFICIENT   1
		
		#define CHECK_BUTTON		// disable = add '//' if you use a slave board as master
	#endif
#endif

```

RoboDurden, 2024-05-03:
Sure that target is set to GD32F130 ?
Maybe i uploaded keil project set to GD32E230 :-(

pacraf, 2024-05-03:
@RoboDurden target is selected GD32130 
I reloaded project to keil (downloaded fresh from repo) 

now whatever I try to compile, I see error: 

Build started: Project: Hoverboard
*** Using Compiler 'V6.19', folder: 'C:\Keil_v5\ARM\ARMCLANG\Bin'
Build target 'GD32F130'
compiling setup.c...
compiling bldc.c...
compiling main.c...
compiling led.c...
compiling it.c...
compiling remoteUart.c...
compiling remoteCrsf.c...
compiling remoteDummy.c...
compiling remoteUartBus.c...
compiling commsMasterSlave.c...
compiling RemoteAutodetect.c...
compiling gd32f1x0_adc.c...
compiling gd32f1x0_dbg.c...
compiling gd32f1x0_dma.c...
compiling gd32f1x0_fwdgt.c...
compiling gd32f1x0_gpio.c...
compiling gd32f1x0_i2c.c...
RTE/Device/GD32F130C8/gd32f1x0_misc.c(68): error: no member named 'IPR' in 'NVIC_Type'
    NVIC->IPR[nvic_irq] = (uint8_t)temp_priority;
    ~~~~  ^
1 error generated.
compiling gd32f1x0_misc.c...
compiling gd32f1x0_pmu.c...
compiling gd32f1x0_syscfg.c...
compiling gd32f1x0_rcu.c...
compiling gd32f1x0_timer.c...
compiling gd32f1x0_wwdgt.c...
compiling gd32f1x0_usart.c...
compiling system_gd32f1x0.c...
".\Objects\Hoverboard.axf" - 1 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:02

my config is
[image]
[image]

[image]

what is soo strange for me, is that I still can successfully compile old project from  (lets say september... ) 
I do not see any differences in options. but this error - maybe it says something obvious for you?

RoboDurden, 2024-05-03:
Yes you need to doubleclick on the error and replace `->IPR` with `->IP`.
or update your Keil packs..
Old packs had -`>IP` , new pack has `->IPR`

pacraf, 2024-05-03:
packs update resolved problem - as you wrote.
thank you.
anyway I still can not understand how it was possible that today morning I
downloaded repo and I was able to compile for 2.1.8, with 2.1.1 had
problems (motor chirps). It was for sure compiling without error,  because
I also programmed from keil (f7, f8)
then afternoon after your comment, downloaded repo again after your post,
and could not compile current version of repo, but still old was compiling
ok.
whatever.
now compiles , and 2.1.1 rotates - dummy firmware works.

Freshwater123, 2024-05-12:
@pacraf Did you get that remote controlled mower to work with 2.1.1?

pacraf, 2024-05-12:
My mower is only a remote controlled 4wheeler. Not a mower… but a platform
with connected 4 drives and wheels with ppm receiver . This works.

Freshwater123, 2024-05-15:
Gotten stuck. 

_- Flashed Master Uart + Slave with Autoflash, thus did no mods to the code. 

- Flashed the correct ESP32 arduino with the "ESP32_PPM" (again did no mods) and check in Arduino that it reads the input from the RC receiver. Soldered the correct ard pins to the "Remote" on the Master side._

Turn it on and get no lights that turns on except on the RC receiver. No throttle.

Anyone got any suggestions ?

RoboDurden, 2024-05-15:
Always first flash the dummy binary to see if motor is spinning.
Then flash remoteUart or RemoteUartBus and testspeed.ino with the uartBus define uncommented or not.
Finally try the ppm ino.

MathMyne, 2024-05-16:
Hello, we are currently working on a project using hoverboard hardware, and we encountered some difficulties following your tutorial.
We flashed the 2.1.1 hoverboard card with “hoverboard 2.1.1 slave.bin” and hoverboard “2.1.1 master Dummy.bin” for the two cards and it works well.
We want to use an ESP32 WROOM to handle the hoverboard card. We flashed the master with “hoverboard 2.1.1 master Uart.bin” and connected the ESP32 with RX and TX according to the schematics and loaded the testSpeed.ino code.
It shows this error and we are blocked:
[image]
Pictures of the wiring: 
[image]
[image]
Do you know what to try ? 
Thanks in advance for your help.

Freshwater123, 2024-05-20:
Been banging my head against my hoverboard for awhile trying to figure this out. Now found something odd, when i run a simple arduino code it reads PPM signal fine. But when i use RoboDurdens "ESP32_PPM" i get "dead RClink" in the monitor, see pictures

Is this why i can't get my boards to run when it's all connected correctly?

[image]
[image]
[image]
[image]

Freshwater123, 2024-05-20:
I didn't ask you for help, this is a open forum. Anyone who wants to can answer or noone

pacraf, 2024-05-20:
> Don't ask me for help on PPM. I do not care for these old remote controls.

As I remember, and if you analyze code a bit, you should see that dead Rc
link is a message when channels intended to control board are out of range,
and in your case they are (all zeros except one) in case of this code, it
is expected that 3 and 4 channel are used. If you have other, you have to
modify it to your needs.

Anyway, just looking on your screen shots, o see that you have only one
channel, and this is strange as ppm idea is to send more channels. For one
pwm is good enough .

pacraf, 2024-05-20:
that’s what I think too, but I don’t know why I should not reply…