RoboDurden/Hoverboard-Firmware-Hack-Gen2.x #77: A GD32Gen1 board with gd32f103rct6

hiliving, 2024-04-05:
This is a new board of Gen1, yet the pins are somewhat different.I have marked all the Pins, but I don't know how to modify the Gen1 source code to compile the firmware. Can you give me some guidance? Which codes need to be modified? 
Thx ,and Extremely grateful .
[image]

RoboDurden, 2024-04-05:
Go to EFeru GitHub repo and start an issue there. See the defines.h in his firmware.
 This here is Gen2.

hiliving, 2024-04-06:
Sorry, maybe I should change a way of thinking. I wonder if it is possible to modify the source code of Gen2 to compile the firmware for this board to use.

RoboDurden, 2024-04-06:
EFeru firmware has FOC motor control.
And two motors.
And gd32f103 is compatible with stm32f103.

AILIFE4798, 2024-04-06:
gen 2 can support f103 also ,so it should be able to drive one of the motor at least, the motor pin is also the standard one

AILIFE4798, 2024-04-06:
it is not possible to modify the eferu firmware for different layout,because all the pinmodeAf things is hard coded,and the adc channel is also same,not to say this board uses uart2,uart3,uart4,which s different from the generic gen1

AILIFE4798, 2024-04-06:
This would be the place oop would shine, adding a second motor is very easy, just initialize another object

RoboDurden, 2024-04-06:
No, i already had that hope for the Gen2 SimpleFOC C++ platformIO project here on my github account:

https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC

The 6 mosfets need the Advanced Timer0. And the next 6 mosfet of the second motor need another advanced Timer. 
So you can not simply create new BldcMotor objects based on a generic class..
The class would need to aquire the hardware it needs.
With this neccessary hardware knowledge, you can as well initalize the two motors with a define.h 

I am no longer very hopeful to continue with the Gen2_SimpleFOC. Even if it is already running with I2C control. @hiliving could use it for his Gen1 board to drive one motor.. The SimpleFOC class library is so complex (and i even added more classes like Hoverboard and even more classes to come like Remote), that i think i will give up on that.

I rather hope that you @AILIFE4798 will port your Gen2_MM32 to GD32, add FOC and i can happily delete this repo here (= hand it over to you)

AILIFE4798, 2024-04-06:
The future is mm32, maybe lks32, so we should move on to support a gen1 with mm32 and not need to go backwards

AILIFE4798, 2024-04-06:
Some of the mm32 gen1 board still have foc capabilities, but more and more is like the$2 gen2 mm32 boards (and the one you have is also not foc capable), no longer have foc capabilities

closed 2024-04-08 by hiliving